AccelerationConsortium/ac-dev-lab #11: Pi Zero 2 W camera for equipment monitoring

sgbaird, 2024-08-29:
[image]

Need to order tripod, pi zero case, and tripod nut.

Add documentation to devices dir

sgbaird, 2024-09-05:
Also, setting up the tripod and case.

[image]

[image]

There's a second case type, with no tripod attachment

[image]

sgbaird, 2024-09-16:
I ordered ball bearing attachments for the prime cables clamp tripods. 

[image]

The additional four should be set up in a similar manner. There's one extra in the cabinet (might move).

[image]

[image]

sgbaird, 2024-09-20:
For the setup, also important to have a micro USB to female USB-A cable (comes in the pi zero 2 W kits), and a keyboard/mouse combo (however in this case maybe you need two USB ports or a USB splitter)

sgbaird, 2024-09-30:
@Jonathan-Woo any updates?

Jonathan-Woo, 2024-10-03:
[image]
[image]
[image]

https://github.com/user-attachments/assets/75b5787e-c508-45b6-beb6-4ccbc051d9f6

I have the hardware setup but I think we need a smaller nut to connect to the tripod or PrimeCables mount.

On the software side of things, I believe we'll building using [picamera2](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). What sort of MWE/system would you like me to build?

sgbaird, 2024-10-03:
Nice! Ok, can you identify a nut (maybe on Amazon) that you think will work OK? For the software, it would be great if you could try to get a YouTube livestream set up with it. I'll send credentials for an "AC hardware stream" account, or you can use a personal account for testing.

Before that, following a tutorial to get the video feed displaying on the desktop of the Pi Zero 2W would be a good start.

Jonathan-Woo, 2024-10-04:
Sounds good, I'll look into that. Regarding the video feed displaying on the desktop of the Pi Zero 2W, I was wondering if there was something specific you wanted differently than the attached video above. Thanks!

sgbaird, 2024-10-04:
Oh, sorry! Somehow I missed that. That looks great. I'm also seeing that the PDF you linked to is very comprehensive. Anything in particular you'd want to note about following that guide? E.g., certain install commands you used. It would be great if you could also include the code you ran to produce the preview.

Past that, I'd say you're good to move onto setting up a livestream as well some of the other camera modules (the latter of which I imagine will be straightforward).

Thanks for working on this.

sgbaird, 2024-10-04:
Sent you a stream key and the stream URL attached to https://www.youtube.com/@ac-hardware-streams channel.

sgbaird, 2024-10-04:
How is https://www.amazon.ca/Stainless-Small-Pattern-Machine-Finish/dp/B009EEZYE0/ ?

sgbaird, 2024-10-08:
The item arrived from amazon

sgbaird, 2024-10-10:
Accidentally got something that was 1/4" outside dimensions.. planning to order a new part.

Jonathan-Woo, 2024-10-10:
I've recently tested various resolutions, bitrates, and framerates and the best I've been able to do is 720p @30fps with a bitrate of 4 Mb/s. Anything more demanding and we get dropped frames because the Pi Zero can't encode fast enough and we get dropped frames.

This is in line with [youtube's recommendations](https://support.google.com/youtube/answer/2853702?hl=en).

Something to note is that sometimes immediately restarting the stream fails. I'm not too sure why. I suspect it has to do with something on youtube's side where they have a cooldown period between streams.

I've been using the built in command line tool with the below command. I might be overdoing it with the thread_queue_size.

```bash
rpicam-vid -t 0 -o - --width 1280 --height 720 --vflip --codec libav --libav-format mpegts --bitrate 4000000 --framerate 30 | ffmpeg -re -fs16le -thread_queue_size 4096 -i /dev/zero -thread_queue_size 4096 -i - -vcodec copy -framerate 30 -f flv rtmp://a.rtmp.youtube.com/live2/STREAM_KEY
```

Regarding next steps, do you think we should pack this command into something friendlier? I can also get started on assembling the other camera modules if they're ready.

sgbaird, 2024-10-10:
Ok, nice! Not a huge deal if the fps is lower (e.g., 10-20), and overall that seems good. Thanks for trying out different things to get it working. If frames drop, how does that affect the user? Weird lag / skip, or something more dramatic? Weird about the immediate restart. If you have it in a bash script, it would be easier to break over multiple lines and add comments, but it's up to you.

sgbaird, 2024-10-10:
It looks like I may be able to create a shared YouTube channel, in which case I could add you and others as managers. 

This might replace the account attached to https://www.youtube.com/live/g6IhiD4XEMU?si=rAw4ZNUmQB0fNuR6

Jonathan-Woo, 2024-10-15:
@sgbaird Any update on the account?

sgbaird, 2024-10-15:
Sent the credentials for an account with editor access to the account via slack. Turns out invites can be made for regular YouTube accounts.

sgbaird, 2024-10-16:
I'm not sure if I brought this up already, but it would be great to also see an example of the YouTube video embedded into a hugging face space.

sgbaird, 2024-10-17:
Added you to HF. Could you create both a standalone example of embedding a YT video as well as incorporating the same camera stream into the MyCobot app (as a second step)?

sgbaird, 2024-10-24:
@Jonathan-Woo the nuts should be in tomorrow. Ended up ordering from McMaster, since Amazon randomly cancelled and refunded the order.

sgbaird, 2024-10-25:
@Jonathan-Woo it says the hex nuts arrived. They should be in a McMaster package if not already opened. Feel free to open it.

Jonathan-Woo, 2024-10-25:
Updates on the Streamlit HF side of things, I had trouble earlier with pushing but that has been resolved now.

I've pushed changes to embed the youtube livestream in [the cobot space](https://huggingface.co/spaces/AccelerationConsortium/hardware-streams) and a [new space](https://huggingface.co/spaces/AccelerationConsortium/hardware-streams) I intend to contain all the streams.

Jonathan-Woo, 2024-10-25:
I've also modified the `/etc/xdg/lxsession/LXDE-pi/autostart` file according to this [tutorial](https://raspberrypi.stackexchange.com/questions/8734/execute-script-on-start-up/45089#45089) enabling the autostart of GUI apps which rpicam-vid requires to have a preview show up on the desktop.

sgbaird, 2024-10-29:
@Jonathan-Woo great to see that the livestream is up: https://www.youtube.com/live/fF4zEp6LSkg

Yesterday, I think you ran into some issues. Do you mind summarizing here and what you did to resolve? 

As next steps (separate gh issue perhaps), maybe we can look at the programmatic access of the stream data. E.g., some kind of python API for bringing in the video data.

Jonathan-Woo, 2024-10-31:
The command I used originally displayed the feed full screen on the rpi desktop. I had to update the stream key but wasn't able to because the Pi Zero was struggling to display the feed while streaming to youtube. 

So I had to modify the launch script to make it no longer display fullscreen on the desktop. The issue was that since the script auto-started and the Pi Zero was already under maximum load I couldn't stop the full screen feed. So disconnected the camera to cause the script to crash when auto starting so I could modify it before restarting the rpi.

sgbaird, 2024-10-31:
Thanks! Also, any ideas why the HF space embedding is weird?

A workaround might be to use HTML embed code directly (i.e., what's copy pasted from the YouTube share - embed option). Maybe this would be HTML within a markdown block. Maybe some issue with the video block, which honestly might just be a really shallow wrapper.

sgbaird, 2024-11-07:
Also, could you add a Sprite of the camera that's currently set up? Maybe with the camera and the mount. Instructions at: https://ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html#embedding-into-gather-town. You could embed the YouTube video directly if you want (Gather Town supports this I think) or embed the HF space.

sgbaird, 2024-11-07:
Also, the livestream seems to have cut out 8 days ago:

[image]

Looks like the duration was 36:40:08.

I went ahead and created a new livestream with "reuse settings" with the previous one (which I believe preserves the stream key as the same).

[image]

Doesn't seem to be getting data back.

Jonathan-Woo, 2024-11-09:
Hi Sterling,

So I've been working on a significant overhaul of the device code in [src/ac-training-lab/picam](https://github.com/AccelerationConsortium/ac-training-lab/blob/jwoo-camera/src/ac_training_lab/picam/cam.py). The motivation is that the current system relies on a single command line call which is susceptible to lots of errors such as:

* starting too early before internet is connected
* difficult to modify stream keys
* difficult to sync code versions across devices

So to address these issues I've switched to using a parent python script to handle:

1. Downloading the latest version of the script on device reboot
2. Executing the downloaded script
3. Checking for internet connection before streaming
4. Improved preview efficiency by using [QtGL](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)
5. Cleaner way to update stream keys

My current direction is to have the code autostart the stream using a cached stream key. This means that to update the stream key the user would have to interrupt the start procedure, update the cached stream key, and then reboot. To make this smoother I considered setting up a system where rpis could query for stream keys from a DB keyed by the device serial number so we could centralize which device would pertain to which stream key instead of going in and updating the stream keys for each device. However, I thought this might overcomplicating things. I considered putting the stream keys in a `json` and having the rpis wget the serial number to stream key json automatically but that would expose all the stream keys which would be less than ideal haha.

For this undertaking I've had to implement all the preview and streaming functionality using [picamera2](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) so there are still some issues I'm ironing out especially with ffmpeg.

This should be a much more robust system and I'm hoping to get the streams back up ASAP.

Will address the HF spaces and gathertown issues ASAP.

Also, any ideas for where to setup the other cameras? I was thinking one for the microscope and one for the OT-2?

sgbaird, 2024-11-09:
Thanks for digging into this so deeply. I think it will be worth it to spend the time for a robust and replicable solution. One for MyCobot, one for microscope, and one for OT-2 sounds good. There are a number of other devices with motion that would probably go in the same area with an overhead cam. If there's a 4th, this can go into the room with the 3D printer, but it doesn't have to point to anything in particular to start with, but still positioned in a way to avoid catching faces by accident.

For the stream key storage, that does seem tough. One MongoDB user per camera with permissions only to the corresponding stream key is possible, but i think since YT allows you to reuse stream keys (i.e., "reuse settings" option), then maybe it would be OK to have the cached version and document the switchover instructions. Some alternatives: maybe there's a way to add a file to the SD card directly that could be read by the OS. I don't know if this is feasible. I'm not sure if there would be a way to auto start that would allow you to connect a monitor and keyboard and stop the auto-running script to then update the stream key.

Sounds like you're on a good path though.

Jonathan-Woo, 2024-12-19:
Finally finished the port to using pure Python and the Picamera2 library.

https://github.com/AccelerationConsortium/ac-training-lab/blob/jwoo-camera/src/ac_training_lab/picam/stream.py

I was 95% done for a while and everything on the device side looked good but no matter what Youtube wouldn't pick up the stream (initially I thought the Picamera2 library was flawed because it's still in beta, so I debugged by rewriting some parts of the library, no luck). Finally I realized that, despite not having a device to record audio, Youtube required audio to also be streamed otherwise it appeared like it didn't see anything coming through the stream. 

So now in the Ffmpeg command we send blank audio and Youtube displays the stream now.

When I left the lab yesterday I almost finished the autostart but just ran out of time. I set up a camera for the OT-2 but will need one more work session after the break to finish.

While working on this I had to switch to a pi 4 for development and debugging because the zero 2 w was just way too slow to iterate on.

Regarding the stream keys, I couldn't think of a better solution than to manually update the stream key so I've left it at that. Though I've implemented auto updating the script by wgetting the latest version. I designed it to be as plug and play as possible.

One thing I've seen from other projects is that they restart the stream every 8 hours or so because Youtube won't save a recording of a stream beyond that duration so people restart the stream so that they can save logs. Not sure if that's something we want.

sgbaird, 2025-01-06:
Great! Thanks for the update. In our case, at least for these initial ones, if it's easy to auto-restart the stream every 8 hours, great - otherwise not a huge priority at the moment. Glad you were able to switch to Pi 4. Lmk if you need a Pi 5 instead. Once the configuration is finalized, do you think it will still be OK to have a Pi Zero 2W running it?

Manually updating the stream keys sounds good, along with wget for the script.

https://github.com/AccelerationConsortium/ac-training-lab/blob/jwoo-camera/src/ac_training_lab/picam/stream.py

sgbaird, 2025-01-21:
> One for MyCobot, one for microscope, and one for OT-2 sounds good

Let's ignore the MyCobot and microscope for now, focusing instead on OT-2 (#80 ) and A1 mini 3D printer (#161).

Jonathan-Woo, 2025-02-07:
For the OT-2, we've been trying to keep the stream running for as long as possible but, during the week, Yanghuang has had to restart it after a few hours. On the weekend, however, the stream is much more stable which may indicate that the issue has to do with network bandwidth (we are using the Nokia).

sgbaird, 2025-02-14:
Just noticed your comment on this issue. Here is my current leaning:

https://github.com/AccelerationConsortium/ac-training-lab/issues/80#issuecomment-2660441147

However, plenty could go wrong, so it's worth trying it out to verify early. In particular, I'm not sure if livestreams from the channel ID live URL can be embedded (as opposed to the specific video page). See https://www.livecontrol.io/blog-post/how-to-link-directly-to-your-youtube-live-stream

sgbaird, 2025-02-28:
Important update at https://github.com/AccelerationConsortium/ac-training-lab/issues/80#issuecomment-2689533508 related to uptime, whether to use per-equipment channels, and perhaps most importantly a clearer path for using the YouTube Live Streaming API (which might simplify other issues).

sgbaird, 2025-03-11:
As a random aside, I've been thinking about how with security cameras, often it will only record when there's been a noticeable change to the camera feed (i.e., only when there's movement). There would probably be a lot of heavy-duty post-processing and unofficial channels required to make this happen with YouTube livestreams that are already up (i.e., difficult to download, and full downloads would be large).

sgbaird, 2025-03-14:
A bunch more zero 2w's and camera module 3's, along with another prime cable mount (I also have an extra one in my office I think I'll be using for recording overhead webcam tutorials)

[image]

sgbaird, 2025-03-14:
Also noting that the main heatsink of the Pi Zero 2W can get hot to the touch (for me, uncomfortable to touch it for more than a second or two). Just check with `vcgencmd measure_temp` (https://chatgpt.com/share/67d444dd-034c-8006-af9d-65981d1d21bd) which shows 49.4 deg C. I don't think this has been actively running anything for a fair bit of time. There's plenty of clearance to add a taller heatsink in the current configuration if we need to later.

sgbaird, 2025-03-17:
Forgot if I mentioned this somewhere, but since we've been leaning towards a 3D printed design, we should adjust the design to leave a bit more clearance to be able to remove the SD card without disassembling the camera mount. (Non-urgent)

sgbaird, 2025-03-25:
@Neil-YL do we need additional nuts/bolts (i.e., what normally comes with the Pi Zero mount kit, but we would still need with the 3D printed version)?

I.e., do you know what size these are (the bolt and the nut hidden behind the plate) and if we have more available in the office?

[image]

sgbaird, 2025-03-25:
> A bunch more zero 2w's and camera module 3's, along with another prime cable mount (I also have an extra one in my office I think I'll be using for recording overhead webcam tutorials)

Sounds like these didn't have SD cards. Frankly, I don't remember which order this box corresponds to. @Neil-YL found one from the Pi 4B's / 5's in my office.

sgbaird, 2025-03-25:
Just switched to ultra low latency (from normal) on [A1 Mini stream](https://github.com/AccelerationConsortium/ac-training-lab/issues/161). The 20 second delay was a bit annoying, so hopefully less now. At some point we might want to consider adding a running timestamp to the video, though not sure if that will cause issues with the Pi Zero 2W in terms of processing power.

[image]

sgbaird, 2025-03-26:
When it got the broken pipe error, systemd treated this as a clean exit, so it didn't restart. Fixing so that restart is always

sgbaird, 2025-03-27:
Something a bit strange, kind of this subtle and minimal zoom-in/zoom-out behavior, possibly related to the latency change? Took a screen recording while viewing the YouTube page to illustrate:

https://github.com/user-attachments/assets/583c57be-3e62-46cf-8ce5-ebcbf4150bca

Neil-YL, 2025-03-27:
Is it the autofocus of the Picam? I’ve noticed something similar when the motion of the OT-2 causes the desk to vibrate.

sgbaird, 2025-03-30:
Both OT-2 and A1 mini streams seem to be down.

Neil-YL, 2025-03-31:
> > Both OT-2 and A1 mini streams seem to be down.
> > I restarted them on Youtube studio

[image]

Both of them received no data again

sgbaird, 2025-03-31:
Can you check the systemd logs? (instructions at https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/README.md, `sudo systemctl status picam.service`)

Neil-YL, 2025-03-31:
> Can you check the systemd logs? (instructions at https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/README.md, `sudo systemctl status picam.service`)

> ac@rpi-zero2w-a1mini-stream:~ $ sudo systemctl status picam.service
> ● picam.service - Start picam device.py script
>      Loaded: loaded (/etc/systemd/system/picam.service; enabled; preset: enabled)
>      Active: active (running) since Mon 2025-03-31 13:41:11 EDT; 43min ago
>    Main PID: 551 (python3)
>       Tasks: 22 (limit: 178)
>         CPU: 12min 30.562s
>      CGroup: /system.slice/picam.service
>              ├─551 /home/ac/ac-training-lab/src/ac_training_lab/picam/venv/bin/python3 device.py
>              ├─596 libcamera-vid --inline --nopreview -t 0 --mode 1536:864 --width 854 --height 480 --framerate 15 --codec h264 --bitrat>
>              └─597 ffmpeg -f lavfi -i anullsrc=channel_layout=stereo:sample_rate=44100 -thread_queue_size 1024 -use_wallclock_as_timesta>
> 
> Mar 31 13:49:58 rpi-zero2w-a1mini-stream python3[597]: [48.0K blob data]
> Mar 31 13:54:11 rpi-zero2w-a1mini-stream python3[597]: [48.0K blob data]
> Mar 31 13:58:25 rpi-zero2w-a1mini-stream python3[597]: [47.9K blob data]
> Mar 31 14:02:38 rpi-zero2w-a1mini-stream python3[597]: [48.0K blob data]
> Mar 31 14:05:42 rpi-zero2w-a1mini-stream python3[597]: [34.9K blob data]
> Mar 31 14:09:55 rpi-zero2w-a1mini-stream python3[597]: [48.0K blob data]
> Mar 31 14:14:08 rpi-zero2w-a1mini-stream python3[597]: [48.0K blob data]
> Mar 31 14:17:41 rpi-zero2w-a1mini-stream python3[597]: [40.6K blob data]
> Mar 31 14:17:41 rpi-zero2w-a1mini-stream python3[597]: [flv @ 0x5588769400] Non-monotonous DTS in output stream 0:0; previous: 2182573, >
> Mar 31 14:21:55 rpi-zero2w-a1mini-stream python3[597]: [48.0K blob data]

I have rebooted it so I think we lost the log. I will check next time.

sgbaird, 2025-04-01:
Thanks! I'd like to figure out why the restart mechanism doesn't seem to be working correctly..

sgbaird, 2025-04-01:
https://github.com/AccelerationConsortium/ac-training-lab/issues/72

sgbaird, 2025-04-04:
Oddly, nothing was connected to UofT when I went to check.. Everything is back on UofT, so that might help out. But worth checking in the future, especially since it seems to default to the Nokia if it can't connect to UofT (which is nice in one sense, but difficult if we don't realize it's because of the Nokia / bandwidth).

Neil-YL, 2025-04-07:
> Thanks! When I did `sudo reboot`, it seems like it stayed connected to UofT. However, I think setting this explicitly makes sense (and can at least be mentioned in the READMEs).

add commands above

sgbaird, 2025-05-01:
All streams went down - @Neil-YL investigating.

sgbaird, 2025-05-22:
Aside: @yakavetsiv using camera monitoring for deriving sensor data 

[image]

Maybe worth opening a separate issue

sgbaird, 2025-07-03:
Trying out a longer cable setup, courtesy of @kelvinchow23 

https://github.com/user-attachments/assets/391668aa-19e0-47cd-83ed-6432dabc55d8

Also trying with it plugged into a standard USB port instead of an outlet

This is `mfu2`

kelvinchow23, 2025-09-17:
@pranshu-ui was running into issues with setting up a new picam for equipment monitoring and I tried to do it as well. We both ran into the same issues. when manually starting the script, it returns a 200 error: 

[image]

is there something with the aws/youtube connection that changed? i can see it created a playlist and is trying to connect, but the stream doesnt start. is this a hint at what may be the issue? 

[image]

sgbaird, 2025-09-25:
I set up a fresh device with @ZeqingBao and we ran into the following error at the end. I saw the broadcast on YT but the stream hadn't started. Will need to debug later

```
python3[7075]:   Stream #1:0: Subtitle: text

python3[7075]: Codec AVOption preset (Encoding preset) specified for output file #0 (rtmp://a.rtmp.youtube.com/live2/XXXX-XXXX-XXXX-XXXX-XXXX) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some encoder which was not actually used for any stream.

python3[7075]: Stream mapping:
python3[7075]:   Stream #0:0 -> #0:0 (pcm_u8 (native) -> aac (native))

python3[7075]: Output #0, flv, to 'rtmp://a.rtmp.youtube.com/live2/XXXX-XXXX-XXXX-XXXX-XXXX':
python3[7075]:   Metadata:
python3[7075]:     encoder         : Lavf59.27.100
python3[7075]:   Stream #0:0: Audio: aac (LC), 44100 Hz, stereo, fltp, 128 kb/s
python3[7075]:     Metadata:
python3[7075]:       encoder         : Lavc59.37.100 aac

python3[7075]: [flv @ 0x558d3fd140] Failed to update header with correct duration.
python3[7075]: [flv @ 0x558d3fd140] Failed to update header with correct filesize.
python3[7075]: Error writing trailer of rtmp://a.rtmp.youtube.com/live2/XXXX-XXXX-XXXX-XXXX-XXXX: Broken pipe
python3[7075]: size=      10kB time=00:00:10.54 bitrate=   8.1kbits/s speed=13.6x
python3[7075]: video:0kB audio:3kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 286.143280%
python3[7075]: Error closing file rtmp://a.rtmp.youtube.com/live2/XXXX-XXXX-XXXX-XXXX-XXXX: Broken pipe
python3[7075]: [aac @ 0x558d3fd9c0] Qavg: 65536.000
python3[7075]: Conversion failed!
```