AccelerationConsortium/ac-dev-lab #13: Temperature / humidity / pressure / gas sensor

sgbaird, 2024-08-29:
[image]

Default, IoLT setup instructions - except in streaming mode, similar to the magnetometer setup.

sgbaird, 2024-09-30:
@Luthiraa could you open the PR that you submitted on self-driving-lab-demo to this repo instead?

Luthiraa, 2024-09-30:
> @Luthiraa could you open the PR that you submitted on self-driving-lab-demo to this repo instead?
Sure

Luthiraa, 2024-10-26:
[]()https://github.com/user-attachments/assets/3e110270-9fc1-4b87-a275-a5012453f42a

Demo video of temperature sensor

Luthiraa, 2024-10-26:
Remove unnecessary dependencies and unused code, simpler/streamlined code for adafruit bme680 temperature sensor

- previous version of the code utilized a lot fo dependencies for a simple reading. to address this issue I used code driver code from the [https://github.com/robert-hh/BME680-Micropython](url) repo. 

You can view the commit details [here](https://github.com/AccelerationConsortium/ac-training-lab/commit/8f9dd0fef5cfe0e5d82f83fcdd43004eb536dae8).

sgbaird, 2024-10-30:
Have you been able to start working on the MQTT integration?

Luthiraa, 2024-11-28:
Finished Integration with Streamlit for a temperature sensor with real-time dashboard

[Luthiraa/temperatureSensor](https://github.com/Luthiraa/temperatureSensor) - For the Streamlit App 

[image]

Luthiraa, 2024-11-28:
Here is the link to the latest commit in the repository: [Latest Commit](https://github.com/Luthiraa/ac-training-lab/commit/1d1b41b6269bab2e86763690332cbdcf9ba63114)

sgbaird, 2025-01-10:
Streamlit web app should go to HuggingFace (lmk if you need write access), and it can be plugged into one of the outlets in the training lab office space for permanent access. Lmk if you need the WiFi credentials

sgbaird, 2025-02-15:
Have you gotten this onto hugging face yet?

Luthiraa, 2025-03-01:
Hi Sterling. The whole setup is finished, including the hugging face. When should I drop it off?

sgbaird, 2025-03-01:
This one I take it: https://huggingface.co/spaces/AccelerationConsortium/temperature-sensor

Can you also address the feedback from the PRs https://github.com/AccelerationConsortium/ac-training-lab/pull/127#pullrequestreview-2652577163 and https://github.com/AccelerationConsortium/ac-training-lab/pull/63#issuecomment-2584978270. When you address the feedback, please also update the files on the microcontroller and test it against hugging face again with a short recorded video.

Luthiraa, 2025-03-01:
Will do, thanks.

sgbaird, 2025-05-22:
Some other options for sensors being used in SDL6

[image]

SHT31 being used in SDL6 by @yakavetsiv