AccelerationConsortium/ac-dev-lab #15: Design and print automated turntable vial rack

sgbaird, 2024-08-30:
[image]

https://www.tindie.com/products/fluxgarage/turntable-for-stepper-motor-kit/

For automated storage of [Pioreactor vials](https://pioreactor.com/en-ca/collections/accessories-and-parts/products/20ml-glass-vial-closed-cap-and-stir-bar?variant=42836211761208), hopefully some way to make it easier to robotically insert and remove the vials. Maybe some kind of flexible mechanism?

Probably ~8 vials in a single rack. Ideally a Fusion 360 design with parametric modeling.

Something that can be fixtured to the top plate with minimal hardware

Perhaps upload to BambuLab maker world (?)

Aside: vial rack should perhaps have a spot on the side and/or on top for an AprilTag, and numbering of the vial positions along the side (1 through 8).

sgbaird, 2024-10-01:
@MiceeNS thanks for your initial work on this.

sgbaird, 2024-10-10:
[image]

MiceeNS, 2024-10-10:
Do you have dimensions spec sheet because I cant see find any in the link? Do you already have the automated turntable that is linked above? Are you trying to print only the white storage thing? 

Will update the drawing as fast as possible

sgbaird, 2024-10-10:
Ah got it, I think the design files might be around somewhere. Yes, I do have a few of these units. One is set up fully.

sgbaird, 2024-10-10:
I would just be printing whatever attaches to the black top plate.

sgbaird, 2024-10-10:
@cyrilcaoyang has also been looking into something similar.

sgbaird, 2024-10-15:
This should have dimensions and design files for the base system: https://www.instructables.com/Automated-Turntable-With-Steppermotor/

MiceeNS, 2024-10-15:
Thank you, will take a look

MiceeNS, 2024-10-20:
[image]

I think its best to order longer screws for the upper plate and attach the 3d printed vial rack on those screws. Do you want to change something on the vial rack? I suggest printing only one piece of the holder so we are sure fit tight and works well before printing the whole rack.

[image]

When I get more time, I will improve some things on the rack

sgbaird, 2024-10-24:
Overall nice! Thank you. Do you mind starting one of the prints on the Bambu Lab printer?

Everything including and below the yellow plate is something that comes in the kit (except for stepper motor, which we buy separately). The only thing that is 3d printed is the actual rack. Originally, the intention was to have the vial rack attach via the holes on the side.

[image]

(Self-note / aside - file is https://a360.co/3UpCiPU)

MiceeNS, 2024-10-24:
Hey Sterling,

I will adjust so you can attach it with bolts on those holes. Will also specify which ones. For the Bambu Lab, I have the app but the credentials for log in is asking for email. I cant log in with the credentials you sent me. I made my own account but I am not connected to any device printer. 

Could you please help me out with this.

sgbaird, 2024-10-24:
That was my bad - just messaged you about the credentials.

MiceeNS, 2024-10-24:
Thank you, now I am in. I will print it asap. Do you prefer bolts or plastic studs in those holes. Maybe plastic studs would be much rational than bolting the rack?

sgbaird, 2024-10-24:
> Thank you, now I am in. I will print it asap. Do you prefer bolts or plastic studs in those holes. Maybe plastic studs would be much rational than bolting the rack?

Plastic studs sound fine! Could you include a link to a corresponding mcmaster part, if possible?

MiceeNS, 2024-10-24:
The plastic studs would be printed with the rack, cylindrical pins that would go in the holes of the turntable. That way you don't need to bolt it, but it can be lifted up.

sgbaird, 2024-10-24:
Ah, got it! From a 3d printing standpoint, this might change what supports are needed. Considering the small size of the holes, the studs might break off (especially if they're long). Printing in the CF-imbued filament would help a bit with this. I'm not decided one way or the other. Thoughts?

MiceeNS, 2024-10-24:
Yes absolutely, I was thinking the same It would need to create a lot of supports because those studs are going to lift the rack from the ground. Lets go with bolt and nut. For future we might change and create more generative design.

sgbaird, 2024-10-24:
Sounds good!

MiceeNS, 2024-10-24:
Please take a look : https://a360.co/3UpCiPU

Here are the materials you need from McMaster Carr:
https://www.mcmaster.com/91292a113/  x4 
https://www.mcmaster.com/93625A100/ x4 

You can also get any M3 bolt 10mm length and a M3 Nut with compatible thread if you want to go cheaper. I had to make the clearance for the hole close fit since you already have the holes on the Upper plate.

I will send it for print now if you are okey with this.

[image]

sgbaird, 2024-10-24:
Ok, great! Yes, please do. Just have a look at the print chamber to make sure nothing is in it first

sgbaird, 2024-10-24:
Oh, actually you might not be able to print today. Looks like someone has an overnight print they're doing.

MiceeNS, 2024-10-24:
I see printing going at the moment, will take a look later. If not, is it okey for tomorrow?

sgbaird, 2024-10-24:
👍

sgbaird, 2024-10-24:
I don't think the print will be removed tonight, so best to wait until tomorrow when it's been taken off. Maybe send the Bambu 3mf file over? (Not sure if there's a way to store the print in a queue)

MiceeNS, 2024-10-24:
[image]

It is higher than the printer can do probably, do you want me to shorten it?

sgbaird, 2024-10-24:
Yes that sounds good. Maybe adjust for a 20 mL vial?

Neil-YL, 2024-10-25:
Hi @MiceeNS I was using the 3D printer extensively yesterday. Apologies for any inconvenience.

I opened the 3mf file you sent and the printer generated the supports like this:

[image]

Should it be printed this way, or should I adjust the support settings?

MiceeNS, 2024-10-25:
> Hi @MiceeNS I was using the 3D printer extensively yesterday. Apologies for any inconvenience.
> 
> I opened the 3mf file you sent and the printer generated the supports like this:
> 
> <img alt="image" width="475" src="https://private-user-images.githubusercontent.com/179746567/380196378-bcdbfa9a-bf23-477e-9bd2-01532e163e23.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4NjU1ODUsIm5iZiI6MTcyOTg2NTI4NSwicGF0aCI6Ii8xNzk3NDY1NjcvMzgwMTk2Mzc4LWJjZGJmYTlhLWJmMjMtNDc3ZS05YmQyLTAxNTMyZTE2M2UyMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyNVQxNDA4MDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYjYxNzZhNzA3YWEwOTM1N2VjNzQxMGI4ODRhN2E2MTE2ZGQyZWYyYWExNmFlMmNiNmVhZGYxZjBhODY5NTNjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.dgOQllK_ABl4EdIs3mN5qL1pNpx1JHJJNy3Ny1memyw">
> Should it be printed this way, or should I adjust the support settings?

I changed to the normal supports settings, if you think this might cause any issues please adjust. Thank you

Neil-YL, 2024-10-25:
> I changed to the normal supports settings, if you think this might cause any issues please adjust. Thank you

I’m not familiar with the purpose of this part, so I’m just double-checking that these supports won’t interfere with its use. 
I’ll proceed with the print as you’ve set it.

MiceeNS, 2024-10-25:
Are you placing the print in que or you have to wait for the printing to finish?

sgbaird, 2024-10-25:
> Hi @MiceeNS I was using the 3D printer extensively yesterday. Apologies for any inconvenience.
> 
> I opened the 3mf file you sent and the printer generated the supports like this:
> 
> [image]
> 
> Should it be printed this way, or should I adjust the support settings?
> 

Let's hold off on you doing the print. Didn't realize the supports would be quite like this. @Neil-YL can you make sure the print bed is clear of parts at the end of the day? That way @MiceeNS can tune the design and supports as needed and send the print remotely.

Sorry for the back and forth and confusion.

Neil-YL, 2024-10-25:
> @Neil-YL Are you placing the print in que or you have to wait for the printing to finish?

I have not put it in queue but the printer is available now if you want to access

MiceeNS, 2024-10-25:
Supports are 30% of the time, the only way I can make this better is changing the actual design of the model. Or make this holders to be printed separate and attached to the column. What do you think @sgbaird ?
[image]

sgbaird, 2024-10-25:
Ooh, good point about the holders being separate. From a print standpoint, that would make things a lot easier, and a lot better and that it could accommodate a range of vial sizes by inserting different sized holders as needed, and would be relatively straightforward to switch from one vial size to another vial size completely.

If using a mechanism where it clips in to the rack, then as long as there's access from the top for someone's hand to fit through, I think the clip mechanism could be undone.

An alternative (that I don't like as much) is splitting the vial rack in half, printing separately, and attaching the two halves later either with hardware or a "clip together" mechanism.

MiceeNS, 2024-10-25:
No need for clips (they can be hard to design and easy to break), I would make like a turbine blade root which can be attached to the column. Do you know the diameter and approximate height on 20ml vial? Different vials manufacturer have different diameter on google.
[image]

I am having a fever at the moment, but will try to make it by tonight and print it.

sgbaird, 2024-10-25:
Ok, turbine sounds good (makes sense about clips). Take some time to feel better - this one isn't so urgent.

MiceeNS, 2024-10-26:
Share file link - https://a360.co/3YpRoWC
The triangles are to reduce material and not create supports. The model now is without supports. The holders have same "adapter" so you could just draw another model tomorrow and just attach it. some of the important dimensions you could check them on the drawing. 

Same bolts and nuts like in the previous design.

Let me know what you think and if OK do you want me to print it?
[image]
[image]
[image]

sgbaird, 2024-10-26:
Way cool! 😎 Yes, please do!

MiceeNS, 2024-10-27:
[image]

[image]

 
There is a problem with the first layer sticking on the plate. I will stop this print, if somebody has a chance to clean the print bed. 

I check on the internet it is some problem with adhesion with the printing bed, it needs to be cleaned with 70% Isopropyl Alcohol. Do you have any suggestions?

sgbaird, 2024-10-27:
@Neil-YL ?

cyrilcaoyang, 2024-10-27:
> Screenshot.2024-10-26.202914.png (view on web)
> Screenshot.2024-10-26.203216.png (view on web)

Usually alcohols can help remove the grease possibly from skin contact. If glue was used in the previous print, it is best to wash thoroughly with water and wipe dry with some isopropanol.We recently found in SDL2 that the PEI bed can wear out over extended use and become less sticky to PLA. Hope this is not the case :)Yang

Neil-YL, 2024-10-27:
> @Neil-YL, have you dealt with this before? (We can address next week)

Based on my experience, I would suggest enabling the Raft layer function when printing parts with very small details on the first layer or when the geometry is not perpendicular to the X/Y axis of the printer.

MiceeNS, 2024-10-27:
> > @Neil-YL, have you dealt with this before? (We can address next week)
> 
> Based on my experience, I would suggest enabling the Raft layer function when printing parts with very small details on the first layer or when the geometry is not perpendicular to the X/Y axis of the printer.
> 
> <img alt="image" width="410" src="https://private-user-images.githubusercontent.com/179746567/380461269-e6150af3-e345-4609-ba3e-ec463facffbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5OTIwMDAsIm5iZiI6MTcyOTk5MTcwMCwicGF0aCI6Ii8xNzk3NDY1NjcvMzgwNDYxMjY5LWU2MTUwYWYzLWUzNDUtNDYwOS1iYTNlLWVjNDYzZmFjZmZiYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyN1QwMTE1MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ODg0MDE4ZmRhYmEzMzc5NWNlZGY0YmZhZjY2MGE2YmY2ZTk1OGRkY2UyMjIxNTJlNDEzYmNiNzYyNmIzNGE2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ZrcJ0BIg-hNkHP4VZIsxgOlRFbPRR6td6cjBzv7cw6w">

Thank you for sharing your knowledge! Will do that in the next print.

Neil-YL, 2024-10-28:
I have cleaned up the print bed

MiceeNS, 2024-10-28:
> I have cleaned up the print bed 

Thank you, when I get home I will give it another try.

MiceeNS, 2024-10-29:
I had the same problem today with the print. When you get a chance could you please clean the printing bed @Neil-YL?

Also, is it possible for me to send you the file so you can try to print it? I tried multiple times on different places on the printing bed.  This is only the raft being printed.
[image]

sgbaird, 2024-10-29:
@Neil-YL that would be great if you could try to get the file printed. Not a super urgent task - sometime this week or next would be fine.

Neil-YL, 2024-10-29:
> I had the same problem today with the print. When you get a chance could you please clean the printing bed @Neil-YL?
> 
> Also, is it possible for me to send you the file so you can try to print it? I tried multiple times on different places on the printing bed. This is only the raft being printed. ![Screenshot 2024-10-28 211342](https://private-user-images.githubusercontent.com/76826784/380936578-1161ce10-3112-4aca-b10b-bc98382e7df7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyMDg4NzgsIm5iZiI6MTczMDIwODU3OCwicGF0aCI6Ii83NjgyNjc4NC8zODA5MzY1NzgtMTE2MWNlMTAtMzExMi00YWNhLWIxMGItYmM5ODM4MmU3ZGY3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI5VDEzMjkzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzYTJmMGRhMmMyNzEzMTYxOWJlZTdkYmQ1ZTBiNDNhZTE3MGQyYzY4ZTJkMGQxMDI0ZDIzOWQ4MWRhNDNhNDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.t0iB-5ct_2Gx6rRKuh-elf3kJYKqM6MJIwFMiHSQ8HM)

Of course. Please send the model file to Yanghuang.liu92@Gmail.com

sgbaird, 2024-10-29:
@MiceeNS you can also attach directly here (not sure if you'd need to zip it based on filetype). If .zip or the file type isn't supported, email is fine too

MiceeNS, 2024-10-29:
Sent him the file through gmail, luckily I remembered to send it to myself this morning.

Neil-YL, 2024-10-29:
This is a photo of a failed first layer:
[image]

The failure mainly located close to the edge of the bed, I am suspecting an uneven temperature distribution of the printing bed. 

After several trails and errors, I seem to have a promising first layer print:

[image]

What I change to have this print:

1. Initial layer height:  The default setting for Banbu is 50% of the nozzle diameter, which is 0.2 in our case. I changed it to 0.35mm (close to the diameter of the nozzle) to have a better adhesive performance of the raft layer (since we print with raft layer the first layer thickness may effect the quality of the printed parts)
2. Bed temperature: the previous setting is 55 Celsius,  I increase to 65 for the first layer (actually I didn't change it for the whole printing work).
3. I also rearranged the parts to have a more regular placement on the bed and remove the largest part to be print in separate work.

If this print still fails, I would reduce the amount of the parts and place them in the middle of the bed.

Reference:
https://overture3d.com/pages/help-center?hcUrl=%2Fen-US%2Fnot-sticking-to-the-bed--bed-adhesion-issue-765548

Neil-YL, 2024-10-29:
[image]
Could you please help me confirm the number of these two parts, now I have eight for each.

MiceeNS, 2024-10-29:
8 of each, i dont care about the surface inside the legs , sry I am not responding fast 

Thank you

Neil-YL, 2024-10-29:
[image]
[image]
The parts would be ready to use by tomorrow

MiceeNS, 2024-10-29:
Thank you @Neil-YL , you are the hero!

Neil-YL, 2024-10-30:
[image]

MiceeNS, 2024-10-30:
> [image]
> 
> 

How does the connections fit? Are they tight? I would not want them the move and also easy to put them in / take them out. 

What are your impressions?

Thank you again for what you are doing.

sgbaird, 2024-10-30:
@Neil-YL maybe a quick video of insertion and removal?

sgbaird, 2024-10-30:
The grip on the vial when inserted is pretty good. It could be a bit looser.

@Neil-YL maybe tomorrow you could also upload a video of inserting and removing a vial?

MiceeNS, 2024-10-31:
> The grip on the vial when inserted is pretty good. It could be a bit looser.
> 
> @Neil-YL maybe tomorrow you could also upload a video of inserting and removing a vial?

we can adjust that no problem, I had no idea of the diameter of the vials and also it depends on the plastic flexibility

[49 more replies]