AccelerationConsortium/ac-dev-lab #17: Automated turntable electronics

sgbaird, 2024-08-30:
Identify a stepper motor that doesn't assume/require the extra stand-offs, and can be easily used with the Tic 500

Design some kind of housing or attachment for the Tic 500 + Pico W + terminal picowbell + power connection, etc. along with shopping list / BOM and instructions

sgbaird, 2025-01-14:
Chatted with @kelvinchow23 about using a stepper motor with encoder.

Search:
https://www.google.com/search?q=stepper+motor+with+encoder  --> https://www.omc-stepperonline.com/stepper-motor-w-encoder  --> https://www.omc-stepperonline.com/nema-17-closed-loop-stepper-motor-14ncm-19-83oz-in-with-encoder-1000ppr-4000cpr-17hs08-1004d-e1k

More expensive, but probably worth it (at least to try out for one). The positioning is a big deal for this application.

kelvinchow23, 2025-01-16:
found a [servo based turntable from servocity](https://www.servocity.com/servo-driven-base-pan-kit/) similar to the [stepper based turntable from tindie](https://www.tindie.com/products/fluxgarage/turntable-for-stepper-motor-kit/). The servo has some built in position feedback and there recommended servo has a range from 300 degrees.

sgbaird, 2025-01-16:
Awesome find! Just ordered `1 x Servo-Driven Base Pan Kit (Dual Mode Servo)`. Shipping was pricey, but it will be good to verify to see if this works well for us.

kelvinchow23, 2025-01-16:
great, thanks! i had a pretty long conversation with chatgpt about choosing the right motor for a turntable.  For light duty applications, i think a servo is the simplest approach. That servo turntable doesn't seem to have a bearing element, but I'm sure there's something that can be added. Regardless, as you scale up and want larger turntables, a heavy duty servo will get more costly, which may not be as practical. 
A closed loop stepper driver doesn't seem that[ bad price wise ](https://www.omc-stepperonline.com/1-axis-closed-loop-stepper-cnc-kit-42ncm-59-48oz-in-nema-17-motor-driver-1-cl42t-s04-v41). 
Besides that, thinking about an open loop stepper motor with some homing routine. Some aid that may help in this application would be adding some positioning mechanisms such as a [geneva drive](https://www.youtube.com/watch?v=bjkSqzLdjZM) as sterling mentioned or [rotary detents](https://www.youtube.com/watch?v=QnqAtSeqRPo). Also  thinking about the addition of some positioning feedback with hall effect sensors or limit switches.

kelvinchow23, 2025-01-17:
im probably spending too much time thinking about into this, but i'm probably going to order this and see how it compares to the tindie turntable 
https://www.omc-stepperonline.com/60mm-hollow-rotary-actuator-platform-gear-ratio-5-1-backlash-3arc-min-for-nema-17-stepper-motor-hrp60ke-g5-d5

sgbaird, 2025-01-23:
One I ordered should be in tomorrow.

kelvinchow23, 2025-01-27:
Received the part, it was really easy to put together, maybe 10 minutes or so. the rotating platform isnt screwed into the servo which i was surprised by, relying on gravity to keep it in place. 
The servo that it came with also doesn't have positioning feedback, I am probably going to swap the servo out with one with feedback. 

[image]

kelvinchow23, 2025-01-29:
i swapped out the continuous servo the kit came with one with Ilya had given me (https://www.amazon.ca/Miuzei-Torque-Digital-Waterproof-Control/dp/B07MDM1C3M) that was 0-270deg, 20kgcm.  
Thinking about it a bit more, as long as what's on top of the turntable is built so that the centre of mass is at the centre of axis of rotation, the servo doesn't really need to be a high torque motor. 
I did a quick test to mark 4 positions on the bottom plate (0,90,180, and 270) and drew one line on the top plate to see how repeatable it was to get it to these positions.  I think this is a promising approach with very little effort. i put some weight on it and manually added resistance and feel like this is doable.  If the turntable gets larger and bigger, it will need some roller supports around the permitter to support the extra weight.  Will continue progressing in this direction.

[image]