AccelerationConsortium/ac-dev-lab #18: Documentation / links / instructions for MyCobot 280 demos by Yang Cao

sgbaird, 2024-08-30:
https://www.youtube.com/playlist?list=PLKFxDV1Aoxg7gdBsbIH7mB6FjYu3F57uj

Perhaps as Accelerated Discovery post and add a device page.

sgbaird, 2024-09-05:
This primarily is geared towards module 4 and 5 from https://ac-microcourses.readthedocs.io/en/latest/courses/robotics/overview.html#modules

gursi26, 2024-09-08:
Hey @sgbaird , just a few questions.
1. Any example Accelerated Discovery Posts I can look at to see what a post for the cobot should entail?
2. Seems like the device page should contain instructions for how to setup the MyCobot 280. Is that accurate?

sgbaird, 2024-09-09:
Some examples:

1. https://accelerated-discovery.org/t/building-the-high-resolution-motorized-openflexure-microscope-v7-using-the-rodeostat-kit/231/16
2. https://accelerated-discovery.org/t/improvements-to-the-digital-pipette/236/8
3. https://accelerated-discovery.org/t/science-jubilee-adapter-for-the-digital-pipette/255/3
4. https://accelerated-discovery.org/t/automated-microfluidics-programming-an-electronic-pressure-regulator/252/5
5. https://accelerated-discovery.org/t/building-the-open-flexure-delta-stage-reflection-and-transmission-microscopes/254

I'm still working out a few details on what device pages should contain, but it should have some kind of brief intro to what the device is, links to the original websites and documentation (e.g., like the links above), a short description of why this is relevant to scientific lab automation, a link to the corresponding accelerated discovery post, and links to related code within the AC Training Lab repo.

The intent is to not replicate existing documentation, but link out to it where possible, and clarify any differences or specifics to the current demo.

I think Yang is going to make an accelerated discovery post to start out since he was working on it a bit before, and that will be used as a sort of hand-off to you.

Adding some of this to https://ac-training-lab.readthedocs.io/en/latest/contributing.html

gursi26, 2024-10-16:
I have made the post, should this issue be closed (unless you also have something else in mind)?

sgbaird, 2024-10-16:
Just saw - nice! https://accelerated-discovery.org/t/making-the-mycobot-280-accessible-over-mqtt-in-a-hugging-face-space/430

Any thoughts on what a "device page" might look like? This idea isn't fully fleshed out. Curious to hear your opinions (the idea being a focus on replicability)

https://ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html#docs-page-and-tutorial-bom-setup-etc

sgbaird, 2024-10-26:
Since the LEGO connectors can be a pain and a bit flimsy, I flipped the base and used a 3 mm Allen key to screw the embedded screws into the cobot (4 internal screws). Much more stable now, without having to put lots of Lego connectors and make it really difficult to get off.

[image]

Not sure what we'll want to do when we're thinking about the MyAGV, but shouldn't matter for now.

sgbaird, 2024-10-26:
Unfortunately it's neither particularly stable nor very well centered (i.e., the benefits of extra clearance as a "pedestal" are not exactly realized). Not sure if it's worth using these. May come in handy for another demo.

[image]

gursi26, 2024-10-31:
For a device page, would a README in `src/ac_training_lab/cobot280pi/` be the appropriate place to put it? I have a set of setup instructions I wrote down while I was setting up the second cobot.

sgbaird, 2024-10-31:
The original intention was to have these in https://github.com/AccelerationConsortium/ac-training-lab/tree/main/docs/devices and have a "devices" drop-down on the left sidebar or similar. I hadn't considered putting it in `src` as much. Thoughts on the two?

gursi26, 2024-10-31:
Yep, either is fine, Ill add the markdown file to that directory in my PR.

gursi26, 2025-03-07:
Hey @sgbaird, i was working on fixing this along with addressing the slight API change for the cobot. Currently the cobot is using the free tier on my mqtt account, but do we want to switch to something more permanent? Not sure if the AC has a common email address or how this should be handled.

sgbaird, 2025-03-10:
I'll send you some credentials for a training lab broker. What is the MQTT schema you use? (i.e., is there a common stem across all topics?) That way I can send some credentials that are specific to the Cobot.