AccelerationConsortium/ac-dev-lab #26: OT2 arrived, requires setup to point of remote access color matching

sgbaird, 2024-09-09:
[image]

sgbaird, 2024-09-23:
Starting point: https://support.opentrons.com/s/ot-2/get-started-guide

Neil-YL, 2024-09-23:
OT2 device is now connected to the AC-IoT network.
Pipette is required to run the calibration.

sgbaird, 2024-09-23:
P300 to be picked up tomorrow from organic chemistry lab

Neil-YL, 2024-09-24:
[image]
P300 Single channel pipette was mounted on the right mount and calibrations were carried out for the deck, the pipette offset and the tip length with 300 µL Tip Rack. Attached is the calibration data JSON file.

[opentrons-OT2CEP20240218R04-calibration.json](https://github.com/user-attachments/files/17118123/opentrons-OT2CEP20240218R04-calibration.json)

Neil-YL, 2024-09-25:
SSH access for OT-2 (initial SSH set up for new keygen)
https://support.opentrons.com/s/article/Setting-up-SSH-access-to-your-OT-2
Note: To generate RSA2048 key as shown in the support article: 

 `ssh-keygen -t rsa -b 2048 -f ot2_ssh_key`

USB connection is required to install the key pair to OT-2, I installed my key pair on the device and the key pair file on the PC side can be shared to other computers to establish the SSH connection in the future.

Connecting OT-2 with SSH
https://support.opentrons.com/s/article/Connecting-to-your-OT-2-with-SSH

OT-2 have a built-in camera(resolution of 640x480px for still images and 320x240px for video):
https://support.opentrons.com/s/article/Using-the-OT-2-s-camera

sgbaird, 2024-09-25:
Awesome! Like we talked about, it would be nice to know what can be run directly on the OT-2 vs. what needs to happen through the GUI / PC side. Ideally, we'd be able to have an arbitrary Python file running on the OT-2 (i.e., the one that exposes remote access).

Neil-YL, 2024-09-25:
I have written a protocol file matching our lab-ware and setup and successfully have two trial runs via the PC app and via the Jupyter Notebook on the web bowser. (syntax is a little bit different)

To run it directly on OT-2, I think this documentation may be helpful, 
https://support.opentrons.com/s/article/Integrating-the-OT-2-with-other-lab-equipment
I would look it up and have some trial lately.

sgbaird, 2024-09-25:
Awesome! When you get a chance, do you mind taking a short video?

Neil-YL, 2024-09-25:
Here is a video of a trial run via Jupyter Notebook (Pick up tip, move liquid from A1 to B1, drop tip)

https://github.com/user-attachments/assets/b1995e20-23bb-4595-b11d-8bc507aa9187

Neil-YL, 2024-09-26:
I have successfully run a python code on the OT-2 via SSH with:
`opentrons_execute /path/protocol.py`

but the robot ran without loading the calibration file, perhaps some initialising codes should be added prior to the run() code.
[image]

Though the documentation recommend install PyVisa to establish connection between devices, I think we can retain our practice on other devices (MQTT) without PyVisa since OT-2 is based on Raspberry Pi:

>The core of the OT-2 is a Raspberry Pi 3+ running [a custom Linux distribution](https://github.com/Opentrons/buildroot).

Neil-YL, 2024-09-30:
Have some ideas for the color sensor setup:
- Design a custom housing for the sensor and control board, matching the size of Opentrons labware. The control board will be positioned underneath the support/housing. Additionally, integrate custom labware into the OT-2 system to avoid collisions with moving tips. (This can likely be done by setting up a block with a specific height.)
- The color sensor can be picked up and moved to different slots (1, 2, 3) if it’s out of range.（the range of the color sensor needs to be tested first)
- The sensor labware can be easily relocated if the labware arrangement changes in the future, as the housing is designed to match the size of standard labware.
Some sketching help to understand
[image]
[image]

- Power could be supplied via a battery (either integrated with the same labware or standalone) or through other potential wireless charging solutions. One quick idea is to charge the battery when the tip moves to the idle position (top-right corner).

We can discuss together later to see if any new ideas come up.

sgbaird, 2024-10-01:
Nice! Looking forward to chatting. I'm somewhat leaning towards a 6-point kinematic connection similar to what's done for Science Jubilee (except all 3D printed) for the pickup.

Neil-YL, 2024-10-02:
[image]

sgbaird, 2024-10-16:
For the most part, we've moved our discussion to the other issue about the color sensor tool.

Simultaneous to this, Jackie from Alan's group plans to try out some things with Prefect and the OT-2.

Neil-YL, 2024-10-24:
A demo run of 1 color measurement and 1dispense cycle: 

https://github.com/user-attachments/assets/65ea91c4-4bda-48e4-9894-d84f507308ca

Neil-YL, 2024-10-25:
With RYB colors water:

https://github.com/user-attachments/assets/92a1e66f-6047-4ec4-8e00-3254414f7d0d

Noticing remaining liquid in tip after dispensing and droplets on the OT-2 deck, a blow-out process for the tip may be required during the procedure
[image]

Neil-YL, 2024-10-25:
Added a blow-out step to a waste vial before returning the tip, with 25% speed control when moving to the waste vial.

https://github.com/user-attachments/assets/5496f83b-298b-46e8-9eb2-f117069cfed5

https://github.com/user-attachments/assets/c3e9a3e0-6de0-4824-8d7d-e7363f83deca

sgbaird, 2025-02-25:
I think we're all good here! Ok to close?

closed 2025-02-26 by sgbaird

Neil-YL, 2025-03-31:
# Installing and Auto-starting Tailscale on OT-2 (Opentrons)

**Note:** OT-2 runs a minimal [Buildroot-based system](https://github.com/Opentrons/buildroot) with a read-only root filesystem and limited shell and network utilities. This guide works within those constraints.  
  
 You'll use `vi` as the default editor because simpler editors like `nano` are typically not available on Buildroot systems.

---

## Download Static Tailscale Binary

On your computer, download the[ static ARM binary](https://pkgs.tailscale.com/stable/#static) for Tailscale:

For current version of OT2, you will use [Tailscale_1.82.0_arm.tgz](https://pkgs.tailscale.com/stable/tailscale_1.82.0_arm.tgz)

Send the file to the data directory on OT2:

```bash
scp -i <your_ssh_key> tailscale_1.82.0_arm.tgz root@<ot2_ip>:/data/
```

SSH into OT-2 and run:

```bash
cd /data
tar -xzf tailscale_1.82.0_arm.tgz
```

This will extract two files: `tailscaled` and `tailscale`.

---

## Manually Start and Authenticate

With SSH into OT-2, run:

```bash
cd /data/tailscale_1.82.0_arm
nohup ./tailscaled --tun=userspace-networking > /tmp/tailscaled.log 2>&1 &
sleep 5
./tailscale up --ssh
```
Nohup to keep tailscaled alive.  
Since OT-2 does not have the required network driver for tailsclae, you need to use [userspace-networking mode.
](https://tailscale.com/kb/1112/userspace-networking)

You will have an authentication link after run the code. Open the given URL in a browser to authenticate the device to your Tailscale account.

This command`./tailscale up --ssh `also enables SSH access over Tailscale.

Log in to the Tailscale admin account. You should be able to see a new OT2 shown in machines list. Assign the ACL tag `tailscale-SSH `for the new device.

You may also test the SSH before starting up the auto start script.

---

##  Create Tailscale Startup Script

Create a script to start Tailscale.

```bash
vi /data/start_tailscale.sh
```

Press "i" to insert mode in vi.

Paste:

```sh
#!/bin/sh

if ! ps | grep '[t]ailscaled' > /dev/null; then
    echo "Starting tailscaled..." >> /tmp/tailscale-watch.log
    nohup /data/tailscale_1.82.0_arm/tailscaled --tun=userspace-networking > /tmp/tailscaled.log 2>&1 &
    sleep 5
    /data/tailscale_1.82.0_arm/tailscale up --ssh >> /tmp/tailscale-up.log 2>&1
else
    echo "tailscaled already running" >> /tmp/tailscale-watch.log
fi
```

Press "ESC" then ":wq" + Enter to save your edit.

Make it executable:

```bash
chmod +x /data/start_tailscale.sh
```

---

## Set Up Systemd Auto-Start

To make tailscaled start automatically at boot (even without SSH login), we will use a systemd service. Since `/etc/` is normally read-only on OT-2, we must temporarily remount it as writable.

SSH into your OT-2 and remount `/` as writable:

```bash
mount -o remount,rw /
```
> ⚠️ **Note on remounting `/` as writable:**
>
> Temporarily remounting `/` with `mount -o remount,rw /` is necessary to create the systemd service.  
> However, keep in mind that modifying a normally read-only root filesystem may increase the risk of accidental system changes.  
> Make sure to only modify what’s necessary, and the system will automatically return to read-only after a reboot.

Create the systemd service file:

```bash
vi /etc/systemd/system/tailscale-autostart.service
```

Paste the following content:

```ini
[Unit]
Description=Start tailscaled on boot
Wants=network-online.target
After=network-online.target

[Service]
Type=oneshot
ExecStart=/data/start_tailscale.sh
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

Enable the service and reload systemd:

```bash
systemctl daemon-reload
systemctl enable tailscale-autostart.service
```

Note that rebooting the OT-2 will automatically restore the default read-only state.

You can also manually start the service to test:

```bash
systemctl start tailscale-autostart.service
```

To verify:

```bash
/data/tailscale_1.82.0_arm/tailscale status
```

Now, you can reboot OT-2 and see if the device on the admin page of Tailscale will become offline and then `connected` again

---

## Summary of File Locations

| File | Purpose |
|------|---------|
| `/data/tailscale_1.82.0_arm/tailscaled` | Tailscale daemon |
| `/data/tailscale_1.82.0_arm/tailscale` | Tailscale CLI |
| `/data/start_tailscale.sh` | Startup script |
| `/etc/systemd/system/tailscale-autostart.service` | Systemd autostart service definition |

---

sgbaird, 2025-03-31:
Amazing! Sounds like the path to get it up took a lot of turns. Thanks for documenting all this and getting it up and running!

cyrilcaoyang, 2025-06-25:
Confirm it works on OT-2 with the Tailscale_1.84.0_arm.tgz version.
Will try to install the Tailscale_1.84.0_arm64.tgz version on OT-flex.

> # Installing and Auto-starting Tailscale on OT-2 (Opentrons)
> **Note:** OT-2 runs a minimal [Buildroot-based system](https://github.com/Opentrons/buildroot) with a read-only root filesystem and limited network utilities. This guide works within those constraints.
> 
> You'll use `vi` as the default editor because simpler editors like `nano` are typically not available on Buildroot systems.
> 
> ## Download Static Tailscale Binary
> On your computer, download the [static ARM binary](https://pkgs.tailscale.com/stable/#static) for Tailscale:
> 
> For current version of OT-2, you will use [Tailscale_1.82.0_arm.tgz](https://pkgs.tailscale.com/stable/tailscale_1.82.0_arm.tgz)
>

sgbaird, 2025-10-28:
Been having some trouble with opentrons and tailscale, keeping the devices connected.

Recently on a flex with Zeqing:

[image]

I need to double check if the issue is the same on OT-2. I think we might have had an issue where we didn't press `i` and instead just copy-pasted, such that the first `i` triggered the input. Another perhaps compounded issue related to line endings. Will try to reproduce and check the logs.