AccelerationConsortium/ac-dev-lab #35: Investigate the use of Sparkplug-B

sgbaird, 2024-09-13:
- https://github.com/sciotaio/micropython-sparkplugb

http://www.steves-internet-guide.com/introduction-to-mqtt-sparkplug-for-iiot/
http://www.steves-internet-guide.com/sparkplug-payloads-and-messages/
http://www.steves-internet-guide.com/checking-active-mqtt-client-connections/

Related to the idea of being able to check statuses in a consistent way

sgbaird, 2024-09-21:
Aside: Watchdog timer may help with reliability

https://github.com/AccelerationConsortium/ac-training-lab/issues/40#issuecomment-2365047988

sgbaird, 2024-10-17:
Last will and testament (MQTT concept) is also a related topic.

@SissiFeng thoughts on whether Sparkplug would be suitable for this kind of status tracking ("is device A still connected to the broker")?

sgbaird, 2024-10-17:
https://stackoverflow.com/a/49549288/13697228

Looks like retained message + LWT combo is a compelling pattern.

sgbaird, 2024-10-17:
Heavily leaning towards this as a first implementation: https://www.hivemq.com/blog/managing-iot-device-state-within-mqtt/ (at least from a MWE standpoint).

SissiFeng, 2024-10-18:
Agreed, especially for MWE, this approach is more concise and effective.

Let me update those HF Space, with LWT configuration + streamlit Interface Integration.

sgbaird, 2025-03-20:
@AccelerationConsortium/ac-training-lab just a note/fyi that using retained messages with last will and testament (LWT) mechanism is a pretty clean and slick way to handle device status: https://www.hivemq.com/blog/managing-iot-device-state-within-mqtt/