AccelerationConsortium/ac-dev-lab #36: Add MWE for MyCobot 280 python control to `src` directory, connect to MQTT, and set up streamlit

sgbaird, 2024-09-14:
We can wait on the accelerated discovery post until we hear back from Yang.

gursi26, 2024-09-14:
just a few questions before i start working on this;

1. How should the camera snapshots be streamed back? Periodically or each time the user sends a command (similar to the Pico W onboard temperature in the color matching SDL)?
2. Should there be any restrictions on what angles/coordinates are valid to prevent damage to the cobot? I guess this would depend on where the cobot is placed and maybe implemented later down the line, but just making sure.
3. Are there any example streamlit apps I can look at for reference?

sgbaird, 2024-09-14:
1. For now, just point and shoot (streaming or similar could come later). We will likely have a camera pointing at the arm and livestreaming to YouTube or similar.
2. For now, I would say just restrict it so it doesn't go below the plane of the bottom of the base, with some safety factor. This is an error/warning you can send back to the user if they request something out of bounds. This can be done either on the orchestrator (web app) or device (Pi) side. Since this is a bit of a safety restriction (keep the robot from tipping itself over/breaking), perhaps best on the Pi side.
3. Glad you asked. I suggest starting with a hugging face spaces starter tutorial, then exploring the code (see "files" tab) in the following: https://huggingface.co/collections/AccelerationConsortium/hardware-control-66a4506f9876e9781c8a0808. The web app you write will live next to these in this same list.

sgbaird, 2024-09-25:
Just reached out to Elephant Robotics about a few issues @gursi26 mentioned to me. See the following message:

> We are using a MyCobot 280 Pi, but we have two problems that are stopping us from using it in our teaching course on autonomous science.
> 
> Problem #1: The robot stops working every time we turn it off and then turn it back on.
> 
> Problem #2: Many times, the robot does not respond to the commands we send to it.

Related:
1. https://github.com/elephantrobotics/myCobot/issues/48

Unfortunately, the GitHub repository doesn't appear to be monitored very regularly. I'm also realizing that the firmware is proprietary and isn't open source (it simply downloads proprietary binaries). The Python API is still open source, but it creates a dependency on ER to address bugs like these ones.

Aside: I came across https://github.com/zlj-zz/minimycobotflasher

Hopefully they respond soon.

gursi26, 2024-10-04:
Had a slight issue with some messages being dropped, likely because streamlit reruns the script everytime and it's possible the client was not ready in time to recieve the message.

Fixed by adding client, response queue and credentials to the session state. This allows for the client to only be refreshed when the credentials change and for the queue to persist in the entire session.

sgbaird, 2024-10-08:
I think the stands arrived

gursi26, 2024-10-08:
Perfect, I can come in on Thursday to set it up

sgbaird, 2024-10-10:
The stands as well as a bunch of cable management pieces arrived.

[image]

[image]

gursi26, 2024-10-28:
just push some code with multiple user authentication. ill now be using the client api to update the streamlit app (will migrate everything to gradio soon).

sgbaird, 2024-10-30:
Maybe mentioned in another post, but the stands - not so great. Doesn't really function as a steady pedestal to give that extra access.

sgbaird, 2024-10-30:
@gursi26 thoughts on status of this issue?

gursi26, 2024-10-31:
Not sure about the pedestal stuff, since I havent tried it out yet.

As for the code, it is mostly complete bar a few small bug fixes and refactoring for cleanliness. Ill be updating the PR soon and will let you know when it is good to merge (probably after i finish migrating the HF space to gradio and testing everything together). Ill also add an explanation to the PR for how user authentication works the way I've currently implemented it.

closed 2024-11-09 by sgbaird