AccelerationConsortium/ac-dev-lab #50: Chi.Bio peristaltic pumps and reactor

sgbaird, 2024-09-20:
[image]

It might be nice to figure out the control of the standalone peristaltic pump board, since it seems like this would be the easiest to attach a pico wireless too, assuming I can get the power working all right.

Https://chi.bio

sgbaird, 2024-10-24:
@Jonathan-Woo, looks like there's a forum as well: https://chi.bio/forums/

Is this what you were following when you ran into trouble? https://chi.bio/operation/

Jonathan-Woo, 2024-10-24:
I had trouble connecting to the BeagleBone Black (control computer for the system) via USB cable. (https://chi.bio/forums/topic/cannot-establish-communication-durting-setup/)[This forum post] suggests that this might be due to my laptop (Mac) not having the required drivers. I tried following these instructions to setup the drivers and connect:

1. https://www.fis.gatech.edu/mac-driver-installation-for-usb-tethering/
2. https://www.fis.gatech.edu/how-to-connect-to-a-bbb/
3. https://ftdichip.com/drivers/vcp-drivers/

But it doesn't seem like they have a driver version for my mac version. I think the next best step would be trying to connect using a Windows laptop which is what they recommend.