AccelerationConsortium/ac-dev-lab #51: Pioreactor MQTT and streamlit

sgbaird, 2024-09-20:
Pioreactor uses MQTT by default, and it's possible to swap out the broker quite easily. However it's not designed to have a completely separate orchestrator and device. So, the most straightforward way may be to write a small MQTT interface into a Python script that runs automatically (and this MQTT interface could be using the same broker, as long as it isn't interfering with the regular operational messages). See the pioreactor forum for my discussions with the main creator related to this.

From there, it can be set up as a yeast growth demo per pioreactor docs. It will require some careful thought about how this could be a long-term demo.

Perhaps growing the yeast in ethanol with tightly sealed containers could be a way to do this, since YPD broth would likely need to be changed out semi-regularly (though this could be a human-in-the-loop Prefect triggered task).

sgbaird, 2024-09-25:
Here is the topic link on the pioreactor forum: https://forum.pioreactor.com/t/use-with-a-different-mqtt-broker/228.

linx5o, 2024-09-26:
> Here is the topic link on the pioreactor forum: https://forum.pioreactor.com/t/use-with-a-different-mqtt-broker/228.

We want to use a remote MQTT broker (HiveMQ) with Pioreactor for secure cloud-based access, but the default setup assumes a local broker, and TLS over websockets isn't supported, limiting the UI’s functionality.

Pioreactor introduced a solution that allows custom broker configurations and successful remote communication. However, the lack of TLS support for websockets means the user can only control Pioreactor programmatically. To fully resolve this, TLS for websockets needs improvement, and a custom API could offer better remote control options without relying on the UI.

linx5o, 2024-09-26:
We aim to solve the issue by implementing a communication layer ontop of the existing infrastructure to allow for our web connectivity. We will likely communicate via the API, however we will lose the ability to use the web interface remotely.

linx5o, 2024-10-01:
How should we manage experiments on the PioReactor? Should the Streamlit app be able to manage experiments (create, delete, assign pioreactor)? Only allow for Streamlit app to manage the current experiment (other changes would have to be made on the UI of Pio)? 

Streamlit app design, what should be included, what needs to be shown, and how much functionality?

sgbaird, 2024-10-01:
> How should we manage experiments on the PioReactor? Should the Streamlit app be able to manage experiments (create, delete, assign pioreactor)? Only allow for Streamlit app to manage the current experiment (other changes would have to be made on the UI of Pio)?

I forgot about the concept of experiments within pioreactor. The highest priority is access to the low-level commands, and less so on more orchestration-like functionality. At least, I think we could start there and see how things go for a bit.

I think long-term, being able to trigger each of [these activities](https://docs.pioreactor.com/user-guide/activities) and the underlying parameters (within reason) would be a good stopping point. In the simpler cases, a simple "send RPM command, receive RPM data" or "send temperature command, receive temperature data" may be a good starting point.

Worth some thought at what level in the Pioreactor stack we want to "hijack" for our custom MQTT driver (e.g., [automations](https://docs.pioreactor.com/user-guide/intro-to-automations), [CLI](https://docs.pioreactor.com/user-guide/cli), [Python API](https://docs.pioreactor.com/user-guide/intro-python-scripting))

Maybe it would be worth setting up the following two experimental workflows to help guide the architecture decisions:
1. [Yeast growth](https://docs.pioreactor.com/experiments/yeast-growth-by-temperature) (though just with a single pioreactor)
2. Making jello (assume that jello powder is already present in the vial, and OK that we can only reach ~40 C temperature, and can't refrigerate yet) - https://youtu.be/3lxP1ihNV0U (the video was very relaxing 😌)

> Streamlit app design, what should be included, what needs to be shown, and how much functionality?

For now, assume the same credentials block as in the other AC hardware apps. The focus should still be on "send command, receive feedback" approach, with just a subset of the original React UI is fine. A lot of the commands are one-offs and a lot of the readings are continuous feeds (i.e., similar to [the fan control demo](https://huggingface.co/spaces/AccelerationConsortium/fan-control)). Based on the comments above, any ideas? There's potential for having multiple pages in streamlit, though I'm not sure if that's necessary here.

## Additional Resources
- https://docs.pioreactor.com/user-guide/ngrok-remote-access
- https://docs.pioreactor.com/user-guide/tailscale-remote-access
- https://docs.pioreactor.com/user-guide/intro-python-scripting
- https://docs.pioreactor.com/user-guide/available-plugins (spectrometer, temperature expansion, co2 reading, pioreactor-relay-plugin ([for pinch valves](https://docs.pioreactor.com/user-guide/pinch-valves)), and air bubbler being the more relevant ones)

sgbaird, 2024-10-01:
@linx5o btw, upgrade kits came (in room with 3D printer)

[image]

linx5o, 2024-10-01:
> @linx5o btw, upgrade kits came (in room with 3D printer)
> 
> ![PXL_20241001_201027898.jpg](https://private-user-images.githubusercontent.com/45469701/372635165-2618e55a-2040-4f74-891f-791460379a14.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc4MTQ2NDcsIm5iZiI6MTcyNzgxNDM0NywicGF0aCI6Ii80NTQ2OTcwMS8zNzI2MzUxNjUtMjYxOGU1NWEtMjA0MC00Zjc0LTg5MWYtNzkxNDYwMzc5YTE0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAxVDIwMjU0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNkYzE1NGU0ODAwMzI5ZjJkMjQzZTQwMjMxNTg0NzU1ZmI4OTYyMzUzMjQ1NjIwYzk4MmE1ZTYyN2YwZDY3MWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0w9EuRnodzxiSg9CShPS6kNNzEdPBzjEV91NYQBx7Lw)

awesome, i will pick it up and install it

sgbaird, 2024-10-02:
When you start writing some code for it, do you mind opening up a draft PR on the training lab repo? You have write access to anything other than the main branch, so you don't technically need to fork (though you're welcome to if that's more straightforward for you). I think this will make it easier to have tight feedback cycles.

sgbaird, 2024-10-04:
All I found (during the 1-minute search) related to the high voltage warning: https://forum.pioreactor.com/t/self-test-failure-ir-ref/338/3 and https://github.com/Pioreactor/pioreactor/issues/452. Thanks for being willing to post on the forum about this issue as well as the mechanical clearance issue with the AS7341 sensor and one of the caps.

linx5o, 2024-10-04:
> All I found (during the 1-minute search) related to the high voltage warning: https://forum.pioreactor.com/t/self-test-failure-ir-ref/338/3 and [Pioreactor/pioreactor#452](https://github.com/Pioreactor/pioreactor/issues/452). Thanks for being willing to post on the forum about this issue as well as the mechanical clearance issue with the AS7341 sensor and one of the caps.

High voltage was solved by reducing the Ir intensity in the config, a final value of 25% passed the test.

Thread for the sensor and cap. [https://forum.pioreactor.com/t/new-1-1-pioreactor-has-bad-fit-for-spectrometer/483](url)

linx5o, 2024-10-10:
**Problems** 
The way the messaging is currently set up without a client id for each connected client could lead to potential issues regarding commands and the state of the streamlit app. In particular the choices for the amount of data to be shown in graphs is most likely to be wrong if two clients have different choices.
A signal should also be implemented to update clients regarding changes to the pioreactor, such as the start/stop of an automation.
Cannot resolve the types of automation that is currently running. We can see if there is a temperature automation, but can't tell if it is thermostat or only_record_temp. I have created a post asking for more API details. [https://forum.pioreactor.com/t/documentation-regarding-the-api-endpoints/486](url)
Display of data in the graphs does not account for gaps in the data, however the timestamps are correct. (timezone of New York has been hard coded into the graphs)

**Todo**
Implement led features
Implement dosing features
Allow for the creation of new experiments and switching to new experiments

**Done**
Allow for custom cred 
A subset of features for controlling the pioreactor (on huggingface using streamlit)
An orchestrator on the pioreactor to allow for these controls

**Notes**
The orchestrator uses HTTP API calls as this requires the least amount of effort to get up and running, incorporating some python scripts which directly interface with the pioreactor may allow for more control and data to be exposed. 
The payloads for these calls are not shown on the API docs, try looking at packets in the pioreactor UI to get a sense of the payload.

linx5o, 2024-10-10:
Credentials needs to be removed from form. Currently. having it there is easier for development.

linx5o, 2024-10-12:
Additional data regarding the stats of the temp automation can be assessed through mqtt. I have been able to retrieve this data. However, there is no documentation on it, it is a trial and error process.
Extensive digging into the workings of the PioReactor code is likely needed to retrieve all the relevant data.
LED info is updated frequently on 
`pioreactor/{reactor}/{experiment}/leds/intensity`

linx5o, 2024-10-15:
[PioReactor API Handling](https://github.com/Pioreactor/pioreactorui/blob/master/pioreactorui/api.py)
The usage of the API in regards to the payloads can be reverse engineered here.

sgbaird, 2024-10-15:
> Credentials needs to be removed from form. Currently. having it there is easier for development.

Ah, yes, definitely. https://huggingface.co/spaces/AccelerationConsortium/PioReactor/blob/acb53e49a48a866f3ba91e619eedd364c13a103f/app.py#L62

I would also suggest not reusing the `9bb6...` host for anything sensitive.

See https://huggingface.co/spaces/AccelerationConsortium/openflexure-microscope/blob/ef0e367d95e46e4fb568bdf535b4444bf85f309d/key_request.py#L10 and https://discuss.huggingface.co/t/how-to-manage-user-secrets-and-api-keys/67948/4.

sgbaird, 2024-11-07:
Also, could you add a Sprite of the camera that's currently set up? Maybe with the camera and the mount. Instructions at: [ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html#embedding-into-gather-town](https://ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html#embedding-into-gather-town). You could embed the YouTube video directly if you want (Gather Town supports this I think) or embed the HF space.

sgbaird, 2024-11-07:
Oops, sorry meant pioreactor

closed 2024-11-08 by sgbaird