AccelerationConsortium/ac-dev-lab #53: Electrocatalyst experimental equipment

sgbaird, 2024-09-23:
Suppliers:

sigma
vwr
fisher
McMaster-Carr
grainger

programlich, 2024-10-02:
# Prio 1

- [ ] 1 [Pack of 2 hole stoppers size 6.5, 470104-471](https://www.avantorsciences.com/ca/en/product/8886291/2-hole-natural-rubber-stoppers?isCatNumSearch=true)
- [ ] 4 [Erlenmeyer wide neck flasks](https://www.sigmaaldrich.com/DE/en/product/aldrich/z233048)
- [ ] 2 [Lab stands](https://www.sigmaaldrich.com/DE/en/search/lab-stand?facet=facet_shipping%3ADE%2FShips%20Today&focus=products&page=1&perpage=30&sort=relevance&term=lab%20stand&type=product)
- [ ] 7 [Boss heads](https://www.avantorsciences.com/ca/en/product/10368661/boss-head-right-angle)
- [ ] 4 [Support clamps powder coated](https://www.avantorsciences.com/ca/en/product/43875353/eisco-3-prong-support-clamps-pvc-coated)
- [ ] 1 [Plastic Tray](https://www.sigmaaldrich.com/DE/en/product/aldrich/br156646)
- [ ] 500g [Potassium sodium tartrate tetrahydrate](https://www.sigmaaldrich.com/DE/en/product/sial/s2377)
- [ ] 2kg [Potassium Hydroxide](https://www.sigmaaldrich.com/DE/en/product/mm/105033)

# Prio 2

# Not to be ordered yet (waiting for other parts)

- [ ] 1 x [gas inlet tube](https://de.vwr.com/store/product/576686/tubes-air-steam-inlet-90-bend-quickfit)
- [ ] 1 x [condensor](https://www.sigmaaldrich.com/DE/en/product/aldrich/z303267)
- [ ] 1 x [condensor adapter](https://www.avantorsciences.com/ca/en/product/13350042/synthware-vacuuminert-gas-adapters-90-kemtech-america)

sgbaird, 2024-10-03:
https://www.amazon.ca/s?k=label+printer+bluetooth&crid=2R1JGTUI5XO0D&sprefix=label+printer+bluetooth%2Caps%2C99&ref=nb_sb_noss_1

sgbaird, 2024-10-09:
@programlich would these work?

1. https://www.amazon.ca/gp/product/B08K7HQMPT/ref=ox_sc_act_title_1?smid=A3CIRZRRC63UIE&psc=1
2. https://www.amazon.ca/gp/product/B08SLYBRPB/ref=ox_sc_act_title_2?smid=A3CIRZRRC63UIE&psc=1
3. https://www.amazon.ca/gp/product/B07R9KGCXX/ref=ox_sc_act_title_3?smid=A3CIRZRRC63UIE&psc=1 (see picture of it being used)
4. https://www.amazon.ca/gp/product/B07XG77YVG/ref=ox_sc_act_title_4?smid=A3CIRZRRC63UIE&psc=1

sgbaird, 2024-10-09:
@programlich do you still need the [three-way stopcock](https://www.sigmaaldrich.com/CA/en/substance/nalgenestopcock1234598765) and [hole stoppers](https://www.avantorsciences.com/ca/en/product/8886291/2-hole-natural-rubber-stoppers?isCatNumSearch=true)?

sgbaird, 2024-10-11:
@programlich next time you're at AC offices it would be great if you could take the PPE and such to NRC (currently on desk with microscopes)

[image]

programlich, 2024-10-11:
sure, will do that

sgbaird, 2024-10-17:
The cell arrived!

[image]

I think I sent it to UoT so we could take it to the chemstore and be able to check things directly. In retrospect, not sure if that was the best plan.

Still need to get the chemicals.

programlich, 2024-10-18:
Great! So as soon as the chemicals are here the electrochemical part can start.

And yeah, if we had known how fast we (probably) could 3D print our own setup we could have avoided ordering the cell at all. I think that´s most likely a good lesson learned for the future participants of the training labs

sgbaird, 2024-10-18:
[image]

[image]

[image]

[image]

[image]

[image]

![PXL_20241018_150619883.jpg](https://github.[image]

c-ab75-74b7cbe9ad99)

[image]

[image]

[image]

[image]

sgbaird, 2024-10-18:
Cubes printed 

[image]

sgbaird, 2024-10-25:
McMaster parts should have arrived. There are some hex nuts in there that are for a separate project. You can leave those in the room with the 3D printer on top of the opentrons cardboard box. You can open the McMaster package if it's unopened.

programlich, 2024-10-26:
Yep, I got the parts yesterday. The wires and the o rings in different sizes have arrived safely. But I am wondering, if the largest size o rings are really of the correct size. They seem to be rather small to me. Are they meant to be heavily stretched to fit into the lid of the cell?

sgbaird, 2024-10-26:
Could you raise this point and tag Bojan in the other issue? I think that an o ring for the lid may not have been factored into the design.

sgbaird, 2024-10-29:
Eventually let's consolidate a shopping list / bill of materials / tools list for the full setup, along with setup instructions.

sgbaird, 2024-10-31:
Got a notification that the McMaster o rings were delivered (two of the large kind and an extra of each of the small kind).

programlich, 2024-11-04:
# Bill of materials for setting up a 3 electrode setup experiment

- Custom print cell, tap and sample holder
- Sealings: [o ring for sample holder](https://www.mcmaster.com/9319K155/), [washer for sample holder](?), [o ring for tap](https://www.mcmaster.com/9319K262/), [o ring for thermocouple](https://www.mcmaster.com/products/o-rings/o-rings-1~/core-material~silicone-rubber/dash-number~011/), [o ring for counter electrode](https://www.mcmaster.com/products/o-rings/o-rings-1~/core-material~silicone-rubber/dash-number~014/)
- Electrical contact: [stainless steel wire](https://www.mcmaster.com/8860K71/), better: Platinum wire
- Thermocouple connected to Potentiostat: Cell designed for [Bola Temperature Probes K](https://www.fishersci.de/shop/products/bola-temperature-probes-k-smp-connector-2/17197564?change_lang=true)
- Reference electrode: [Gaskatel RHE](https://gaskatel.de/en/shop/hydrogen-reference-electrode-hydroflex/)
- Counter electrode: [Platinum wire](https://gaskatel.de/en/shop/hydrogen-reference-electrode-hydroflex/)
- Sample: Practically anything cut down to 3.4 x 1 cm. E.g. Ni mesh

To be continued...

programlich, 2024-11-04:
# Bill of materials for setting up a 3 electrode setup experiment

- Custom print cell, tap and sample holder
- Sealings: [o ring for sample holder](https://www.mcmaster.com/9319K155/), [washer for sample holder](?), [o ring for tap](https://www.mcmaster.com/9319K262/), [o ring for thermocouple](https://www.mcmaster.com/products/o-rings/o-rings-1~/core-material~silicone-rubber/dash-number~011/), [o ring for counter electrode](https://www.mcmaster.com/products/o-rings/o-rings-1~/core-material~silicone-rubber/dash-number~014/)
- Electrical contact: [stainless steel wire](https://www.mcmaster.com/8860K71/), better: Platinum wire
- Thermocouple connected to Potentiostat: Cell designed for [Bola Temperature Probes K](https://www.fishersci.de/shop/products/bola-temperature-probes-k-smp-connector-2/17197564?change_lang=true)
- Reference electrode: [Gaskatel RHE](https://gaskatel.de/en/shop/hydrogen-reference-electrode-hydroflex/)
- Counter electrode: [Platinum wire](https://gaskatel.de/en/shop/hydrogen-reference-electrode-hydroflex/)
- Sample: Practically anything cut down to 3.4 x 1 cm. E.g. Ni mesh
- Electrolyte: For alkaline electrolysis [KOH](https://www.fishersci.ca/shop/products/potassium-hydroxide-pellets-certified-acs-fisher-chemical-5/p-33011#?keyword=) solution. Concentration can be 1M (best for educational purposes) or 7.6M (concentration of real world electrolysers but extremely basic, so it needs to be handled VERY carefully)
- Temperature control: Some kind of thermostat which pumps heats a liquid to a cre

To be continued...