AccelerationConsortium/ac-dev-lab #56: Sorting out issues with MyCobot reliability

sgbaird, 2024-09-25:
Just reached out to Elephant Robotics about a few issues @gursi26 mentioned to me. See the following message:

> We are using a MyCobot 280 Pi, but we have two problems that are stopping us from using it in our teaching course on autonomous science.
> 
> Problem #1: The robot stops working every time we turn it off and then turn it back on.
> 
> Problem #2: Many times, the robot does not respond to the commands we send to it.

Related:
1. https://github.com/elephantrobotics/myCobot/issues/48

Unfortunately, the GitHub repository doesn't appear to be monitored very regularly. I'm also realizing that the firmware is proprietary and isn't open source (it simply downloads proprietary binaries). The Python API is still open source, but it creates a dependency on ER to address bugs like these ones.

Aside: I came across https://github.com/zlj-zz/minimycobotflasher

Hopefully they respond soon.

sgbaird, 2024-09-25:
@gursi26 I imagine you've already looked through the following:

1. https://docs.elephantrobotics.com/docs/mecharm-pi-en/14-IssueFAQ/14.3-hardware.html

gursi26, 2024-09-26:
didnt see this before, will try out the fixes mentioned

gursi26, 2024-10-02:
All wifi issues have been resolved and tested after rebooting the cobot. The issue was just to do with WPA2/WPA3 not working, which is why it worked on home wifi but not the office wifi.

Side note: It was also very difficult to remove the cobot from it's base to access the SD card slot. The lego technic connectors are impossible to pull off.

sgbaird, 2024-10-02:
@gursi26 thanks! For the WiFi, was it having trouble with the Nokia? I think the Nokia is a WPA2-personal network (though of course UoT and eduroam are WPA2-enterprise, and come with their own set of requirements).

Bracing ourselves as we pulled the cobot apart was quite a sight I'm sure 😅 too many Lego connectors at once == really hard to remove (how it came by default).

SissiFeng, 2024-10-02:
> @SissiFeng mind providing a quick status update about what the company said?

Chinese are enjoying their national holiday for the whole week. They will be back to office next Monday.

SissiFeng, 2024-10-02:
The issue with the cobot, I thing, worth to try by downgrading the firmware to a lower version (6.5) and connecting it to the windows laptop via a wired connection (to maintain signal stability). Last time, I used my phone's hotspot to connect, which allowed the cobot to start, but it was unstable (this could be related to the Type-C connection on the Mac). So be care of the joint losing signal suddenlly.

gursi26, 2024-10-02:
@sgbaird Yep, the wifi had options for both and the default selection of the cobot did not work. I just had to change the network configuration to use WPA2 instead of WPA3.

sgbaird, 2024-10-02:
@gursi26 nice! That makes sense. Could you include instructions or a link for how to change the network configuration to WPA2?

gursi26, 2024-10-03:
Sure, 
Start by ensuring that NetworkManager is running with
```
systemctl is-active NetworkManager
```
Which should output "active". If not, enable it with 
```
sudo systemctl start NetworkManager
sudo systemctl enable NetworkManager
```

Once that is done, open the connection editor with 
```
nm-connection-editor
```
In the connection editor, do the following
1. Click on the network you want to connect to. Then click the gear icon at the bottom of the editor window.
2. Navigate to the tab titled "Wi-Fi Security" in the connection editor
3. Here, for the Security field, select "WPA & WPA2 Personal". Enter the password for the network in the text field below.
4. Click "Save" at the bottom of the window.

These steps should permanently set the correct network configuration and persist on reboot.

gursi26, 2024-10-03:
Also, @SissiFeng found some documentation on the send_coords function for the cobot.

> The six parameters of the send_coords function typically represent the position and attitude of the robot’s end-effector (e.g., the end of a robotic arm or the robot’s tool point) in three-dimensional space.  
> The six parameters are typically listed in the following order:  
> X: X-axis coordinates (forward and backward)  
> Y: Y-axis coordinates (left-right direction)  
> Z: Z-axis coordinates (up and down direction)  
> 4. Roll: angle of rotation around the X-axis  
> Pitch: Angle of rotation around the Y-axis  
> 6. Yaw: Angle of rotation around the Z-axis  
> This representation is called 6DOF (6 Degrees of Freedom) representation. There are several points to consider when specifically setting these parameters:  
> Coordinate System: Make sure you are using a coordinate system that is consistent with the robot control system. Usually a right-handed coordinate system is used.  
> Units:  
> For positions (X, Y, Z), usually use meters (m) or millimeters (mm) as units.  
> For angles (Roll, Pitch, Yaw), radians or degrees are usually used.  
> Range:  
> The range of the position parameter depends on the robot’s workspace.  
> Angle parameters typically range from -180 to 180 degrees (or -π to π radians).  
> Accuracy: Set the appropriate accuracy based on the task requirements and robot capabilities.  
> Safety considerations: Ensure that the coordinates set are within the safe operating range of the robot.  
> (Example settings:  
> def send_coords(x, y, z, roll, pitch, yaw):  
> pass  
> send_coords(0.5, 0.3, 0.2, 0, 1.57, 0)  
> In this example:  
> The robot end-effector is instructed to move to points in space (0.5m, 0.3m, 0.2m)  
> Keep horizontal (roll = 0)  
> Tilt down 90 degrees (pitch = 1.57 radians ≈ 90 degrees)  
> No yaw rotation (yaw = 0)  
> In practice, these values need to be carefully calculated and adjusted according to the specific task requirements, robot model and working environment

sgbaird, 2024-11-21:
@aclasky, thanks for reaching out about the cobot. As an update, @gursi26 was able to get the MyCobot 280 Pi up and running. He set up a second one in the office as well (#74). He should have some instructions soon about any special steps he needed to take, but it seemed pretty manageable.

aclasky, 2024-11-21:
@sgbaird thanks for connecting me, that's great to hear! I can provide a bit more info on my current situation:

Model: myCobot 320 Pi (+ gripper)

After being unable to flash any of the software via direct USB connection, I opted to connect the cobot to my monitor and interact with it directly. This approach helped me to flash picomain and _read_ the angles of the different joints but still couldn't control them (still not sure how to flash the Atom software).

I decided to restart the cobot but now no information shows up on screen (monitor recognizes there is an input but shows nothing), so I'm a bit stuck on debugging it. My thought/worry is that the raspberry pi isn't booting properly (the atom LEDs are no longer lighting up) but I'm a little hesitant to break it open and track down the SD card (see @gursi26 's reply about it being very difficult to access).

Any help would be appreciated!

aclasky, 2024-11-25:
Posting here as a record in case anyone comes across the same issue. Big thanks to @gursi26 for his help and to @sgbaird for connecting us.

**Takeaway: power transfer from Elephant Robotics chip to Raspberry Pi was compromised.** The Raspberry Pi could not boot properly and that's why there was no signal on the monitor when the cobot was turned on.

Explanation/debugging process: We reset the SD card using myCobot 320 Pi - Ubuntu 20.04 software from [https://www.elephantrobotics.com/download/](url) but the system still did not boot properly. This step essentially ruled out any software issues. Then, we connected directly to the Raspberry Pi using a certified power adapter and, under these conditions, it was able to boot and display properly on the monitor. We reasoned that the most likely issue is insufficient power transfer from the Elephant Robotics chip to the Raspberry Pi.

Going to reach out to try and get a replacement chip and plan to update once we actually get the arm set up. However, if this ends up being the main problem, it's not a great look for an arm we've only had for a couple weeks.

sgbaird, 2024-12-06:
@gursi26 @SissiFeng ok to close?

sgbaird, 2025-01-07:
@aclasky any update on the MyCobot 320 and how things have gone for you?

sgbaird, 2025-03-25:
@aclasky finding myself wondering about alternatives to Elephant Robotics, especially considering the firmware and hardware issues that have been at play.