AccelerationConsortium/ac-dev-lab #60: Wireless charging for microcontroller

sgbaird, 2024-09-27:
A possibility for powering the microcontroller for e.g., the color sensor:

https://www.canadarobotix.com/products/2761?srsltid=AfmBOooxDHrNvKuctEIGHWXeIHusmvp-t7ae6e-OptdjdnoMbGhULSo7

https://www.adafruit.com/category/331

Power requirements would be pretty low. 

Thoughts?

sgbaird, 2024-10-01:
I think I'll get some of these items ordered today

sgbaird, 2024-10-01:
I think you can assume in advance that Pico W microcontroller, PiCowbell, and LiPo Shim are already soldered together (i.e., if you have the design files on hand, great, otherwise you could treat it as a box with maybe little extrusions for some of the ports)

Neil-YL, 2024-10-01:
3D CAD files update:

 - PiCowbell: (original stp file is x10 scale, scaled down to actual size)
 [image]

 - Pico W: (details are slightly different)
 [image]

 - AS7341:
[image]

I have not found a detailed LiPo Shim board CAD model. It has a geometry of 21 x 21 x 7 mm (with the connector), may create one later with detailed connector geometry when the item is arrived.

I also made an assembled block model for the microcontroller with the PiCowbell and LiPo Shim(since it stays at the middle I do not need the detail size of it), with the largest connector on Y and X direction.

[image]

sgbaird, 2024-10-01:
We'll also need a charger for the parking spot, and any other connections that would need to be made to attach to the transmitter.

Neil-YL, 2024-10-02:
Added a CAD model on Fusion for the assembled block with P5905
[image]

sgbaird, 2024-10-02:
What I'm planning to get: https://www.digikey.com/short/877rfv28 (and a 5V / 2A charger)

sgbaird, 2024-10-02:
A bunch of battery options at https://www.digikey.com/short/p9vn9dhv, but all are through Digikey.com (none available through Digikey.ca). Something 40 mAh, 150 mAh, 400 mAh, or 500 mAh would probably be enough. Even the 400 mAh isn't more than ~1.5" in the largest dimension, and the wireless receiver is ~2" in the largest dimension I think (i.e., the limiting factor).

sgbaird, 2024-10-02:
@Neil-YL thoughts on https://www.digikey.ca/en/products/detail/pimoroni-ltd/PIM578/15851381 vs. [the shim](https://www.digikey.ca/en/products/detail/pimoroni-ltd/PIM557/15851380)?

Neil-YL, 2024-10-10:
Wireless charging transmitter:
https://www.digikey.ca/en/products/detail/adafruit-industries-llc/2162/6827104
[image]
(power supply from micro-usb port, red arrow)

Receiver: 
https://www.digikey.ca/en/products/detail/adafruit-industries-llc/1901/5629432
[image]

Both C1 and C2 give 5V while tested with multimeter
[image]
[image]

sgbaird, 2024-10-10:
We can use https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_lipo_shim/battery_pico.py?rgh-link-date=2024-10-02T16%3A50%3A05Z for validation

Neil-YL, 2024-10-11:
The battery cannot power PicoW (Power LED not on) and the Pico shim cannot charge the battery (Charging indicator LED not on, noticing smoke and heat when plugged into the USB charger)

https://forums.adafruit.com/viewtopic.php?t=95370

> You might be seeing a threshold issue. The MCP738xx series of chargers wait for the battery voltage to fall below a given level before they start a new charging cycle. The LiPo connected above is higher than the nominal 3.7v level.
> Have you tried connecting a discharged battery.. one whose output voltage is around 3.3v?

> The MCP73871 does automatic load sharing. When an external power source is connected, that power goes to the load first. Only current beyond what the load needs will go to charging the LiPo.

However, I do not think these discussion related to our situation (worth knowing though)

This KDA3 on our Pico Shim is bulging and I think this is where the heat and smoke come out.

[image]
[image]

I have tried to look for it on the schematic: https://cdn.shopify.com/s/files/1/0174/1800/files/lipo_shim_for_pico_schematic.pdf?v=1618573454

It could be one of these capacitors for the Lipo charger connector since the power trigger works normally when the battery is not connected, and battery is not powering the Pico nor getting charged when connected. 
[image]

We may have to purchase another Pico Shim for further test.

sgbaird, 2024-10-11:
I think I'll purchase a few more shims and at least one other type of LiPo charger (probably an adafruit one). I'm a bit worried about the batteries as well, in particular the reading of 0 volts for at least one.

Neil-YL, 2024-10-11:
> I think I'll purchase a few more shims and at least one other type of LiPo charger (probably an adafruit one). I'm a bit worried about the batteries as well, in particular the reading of 0 volts for at least one.

I think so, though the smoke came out when I plugged in the one with a 3.7V readout, which is supposed to function normally.

sgbaird, 2024-10-11:
> > I think I'll purchase a few more shims and at least one other type of LiPo charger (probably an adafruit one). I'm a bit worried about the batteries as well, in particular the reading of 0 volts for at least one.
> 
> I think so, though the smoke came out when I plugged in the one with a 3.8V readout, which is supposed to function normally.
> 
> Edit: I measured all four batteries; only one has a 0V readout, while the other three are around 3.87V.

OK, good to know. Maybe the one with 0V reading is from when it started getting hot during some of your early tests. 

Some parts I'll probably buy (lmk if you think there are others I should look at):

- https://www.adafruit.com/product/4410
- https://www.adafruit.com/product/1904
  - Interesting note here..
    > For use with Adafruit LiPoly/LiIon batteries only! Other batteries may have different voltage, chemistry, polarity or pinout. [Lipo/Lipoly/LiIon Batteries](http://www.adafruit.com/products/258) not included, but we've got tons in the shop.
- https://www.adafruit.com/product/1304

Maybe just skip the battery piece until we get some more parts in, and focus on the wireless transmission (i.e., powering the Pico via VSYS directly from the wireless receiver)

Neil-YL, 2024-10-11:
[image]
The polarity of the battery connector are indeed opposite to those of the shim

sgbaird, 2024-10-18:
I can't seem to find batteries with both a JST connector and the correct polarity that can be shipped within or to Canada 🥲. Digikey.ca, Digikey.com, and Adafruit all have tight restrictions - so no shipping options. https://www.canadianbatteries.com/ has batteries, but they're 

I also ordered https://www.amazon.ca/gp/product/B083NWXLTK, but these are coming late Oct/early Nov.

Neil-YL, 2024-10-18:
The new Pico Shim has arrived. I soldered the shim with PicoW and conducted following tests with one battery (reversed its original connection): 

- Power supply from battery: The battery successfully powered the PicoW, and the power button on the shim functioned correctly, allowing me to switch the power on and off.
- USB charging: The red charging indicator LED on the shim lit up after connecting to the USB supply.

Then I connect the PicoW with the Picowbell and the wireless charging receiver:
- Wireless charging: The blue charging indicator LED on the transmitter turned on, but the red charging indicator LED on the shim did not light up when the receiver was placed on the transmitter.”

[image]

https://github.com/user-attachments/assets/55cfca82-5f49-48c0-a5f6-7ec8c5b1210b

https://github.com/user-attachments/assets/100ecbad-92ac-45e4-8b76-cbe54acba02d

Neil-YL, 2024-10-21:
I tried to test the wireless charging part with a short script to get the battery volume and storage it in a log on RB Pi.

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_lipo_shim/battery_pico.py
```
from machine import ADC
import time

vsys = ADC(29)                      # reads the system input voltage
conversion_factor = 3 * 3.3 / 65535

full_battery = 3.7                # reference voltage for a full battery, in volts
empty_battery = 0               # reference voltage for an empty battery

log_file = "battery_log.txt"         # file to store battery data
interval = 180                      # 3 minutes interval (180 seconds)
count = 1                           # counter for the number of readings

# Function to write data to a file
def log_battery_data(voltage, percentage):
    current_time = time.localtime()
    timestamp = "{}-{:02d}-{:02d} {:02d}:{:02d}:{:02d}".format(current_time[0], current_time[1], current_time[2],
                                                               current_time[3], current_time[4], current_time[5])
    
    with open(log_file, "a") as file:
        file.write("Reading {}: Time: {}, Voltage: {:.2f}V, Percentage: {:.2f}%\n".format(count, timestamp, voltage, percentage))

while True:
    voltage = vsys.read_u16() * conversion_factor
    percentage = 100 * ((voltage - empty_battery) / (full_battery - empty_battery))
    if percentage > 100:
        percentage = 100.00
    elif percentage < 0:
        percentage = 0.00

    log_battery_data(voltage, percentage)
    
    count += 1  # increment the reading counter
    time.sleep(interval)
```

These following values should be determined if we want to have a correct percentage readout, now I am testing if you have a power gain.
```
full_battery = 3.7                # reference voltage for a full battery, in volts
empty_battery = 0               # reference voltage for an empty battery
```

Here is what I have now: 

> Reading 1: Time: 2024-10-21 12:47:59, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.89%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.89%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.96%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 2: Time: 2021-01-01 00:03:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 3: Time: 2021-01-01 00:06:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.89%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.76%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.89%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.83%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.89%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.96%
> Reading 1: Time: 2021-01-01 00:00:01, Voltage: 0.07V, Percentage: 1.89%

The first line of log is measured with connecting to the USB port on my computer, so the Time is correct. The rest of the log are written with the battery/wireless charging power supply so the time is initialised for every measurement. If the measurement is correct then the battery is not charged, but I believe this code is little buggy because I also got similar random readout when using USB charging (with Red indicator LED on).

sgbaird, 2024-10-21:
👍 maybe also worth briefly reaching out to Pimoroni to see if they can comment.

sgbaird, 2024-10-22:
Maybe try powering via VBUS instead of VSYS. Like you mentioned, it might be related to the USB power. I think VBUS is connected directly to USB power. Normally, powering through VSYS is safer because you're less likely to send an electrical spike back up the USB cable, potentially damaging the USB electronics. However, in this case VBUS might be what's required for the Pico Shim to work.

Do you mind giving this a try?

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_lipo_shim/battery_pico.py

Neil-YL, 2024-10-22:
> Maybe try powering via VBUS instead of VSYS. Like you mentioned, it might be related to the USB power. I think VBUS is connected directly to USB power. Normally, powering through VSYS is safer [...]

I will try this out later.

I drained the battery overnight. Now I am trying to see if I can get it charged with the transmitter.

Neil-YL, 2024-10-22:
https://forums.pimoroni.com/t/pico-lipo-shim-used-pins/21995

> 40 - VBUS ,used to charge the battery

Using VBUS can light up the charging indicator:

[image]

Neil-YL, 2024-10-22:
Code to measure VSYS with PicoShim and PicoW:
https://www.reddit.com/r/raspberrypipico/comments/xalach/measuring_vsys_on_pico_w/

Note: The wlan should be disabled in order to get a normal readout, tested with this code and here is what I have:
Voltage: 4.769855

the readout seems to be at a reasonable range but we still need the reference value of the battery to calculate the percentage.

Neil-YL, 2024-10-22:
Test of the charging port and the (semi)assembled sensor package:

https://github.com/user-attachments/assets/5e4756c6-b193-4693-a8a2-a2e4f58f586e

With light color 3d printing material, the indicators could be visible from outside:
[image]
[image]

Neil-YL, 2024-10-30:
Things to purchased:

- Qi wireless charging pairs https://www.adafruit.com/product/2162 https://www.adafruit.com/product/1901
- Short micro-USB to USB 2.0 cable (I will use the short angled one you bought before, seems quite good to fit in the port design)
- PiCowBell P5905 https://www.adafruit.com/product/5905

sgbaird, 2024-10-30:
Have we been using a "regular" 5V / 1A power supply, or are you using a 5V / 2A one? Should I get another?

Neil-YL, 2024-10-30:
I am using the USB port on the power stripe next to the Nokia so I think it is 5V/1A. I have a power supply with an output of 5V/2A though(not sure if the angled USB cable supports 2A or not.)

sgbaird, 2024-10-30:
Cables we have: [[1](https://www.amazon.ca/gp/product/B00ENZDFQ4/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&th=1)] [[2](https://www.amazon.ca/gp/product/B073PQWY2B/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&th=1)]

Neil-YL, 2024-10-30:
[[1](https://www.amazon.ca/AAOTOKK-Nylon-Braided-Charging-Compatible-Smartphones/dp/B09C1H231J/ref=sr_1_11?crid=19DTNJBTS49KR&dib=eyJ2IjoiMSJ9.J1MArIW4UCSHTrOiQkNyeyhpcPRgdllLTurJO3hNBDFJNsvQlu-CFx6M-TEcasq4Hq2kM3f3WCJUF5HyLMxdCEnYffybvRZLFEHFesYj5tS1KgHGvQwi1Y-40QaN3MOIOBYxzsocxqmYIJb7JTHvFJowAF5yY1ubG3HPUsNVaryLV253fxB-LfvEVE7570vZUeVk7ZzzWsHPv3_4l9tQwk0Q3-_ryTkeQ-z48KMmDT4-lBTGqW2BeuGJ0WsLi-pjuYdUeCT36-uYBWYfaILJvogyTtmEl451s50f5YH_7q8.HybXkJDPKMEcs_o5BYURB9klWrPHmRpRio5JuaSFEDY&dib_tag=se&keywords=angled%2Bmicro%2Busb%2B5V%2B2A&qid=1730311155&s=industrial&sprefix=angled%2Bmicro%2Busb%2B5v%2B2a%2Cindustrial%2C138&sr=1-11&th=1)] This one supports 5V/2.4A if we find the efficiency is an issue

sgbaird, 2024-11-01:
This is my current shopping cart: https://www.digikey.ca/short/fjrh0vbq

Neil-YL, 2024-11-01:
> DigiKey

[image]
It seems Digikey only sales the same configuration with one meter or two meters length.

sgbaird, 2024-11-01:
Ordered https://www.amazon.ca/gp/product/B073PQWY2B/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 and https://airtable.com/appTUMy18WlaGHrTJ/tblxxujYNzi0ebwui/viw46XHCvnOKA3jM1/rece1L52DosUhBEbJ/fldZn7KulHwIoWokF?copyLinkToCellOrRecordOrigin=gridView (also has a solenoid and some electromagnets, unrelated to this).

Neil-YL, 2024-11-04:
The rated 5V/1A USB cable can power both receivers for battery charging, though it’s unclear how this setup affects efficiency

[image]

Neil-YL, 2024-11-04:
[image]

The dual-charging design is finished, here is the test video:

https://github.com/user-attachments/assets/7caff239-0f2c-4b0a-bf39-425b64f7bc08

I finished most of the write-up for the sensor package assembly last Friday. I plan to complete the section for the charging port tomorrow and start taking photos/videos after that. 
We could find some time to review the draft (with the parts in hand) to see if there’s anything to add or adjust for the design and assembly documentation.

sgbaird, 2025-01-17:
Random note that I couldn't think of where to put, i.e., please ignore ;)

An interesting board, around 15 CAD from Pishop, with a power switch, lithium-ion battery management, all three motor drivers, and a place for Pi Pico to sit on. It has a NeoPixel, but no stemma-qt port. 
https://wiki.elecfreaks.com/en/pico/extension-module/wukong2040-pico/