AccelerationConsortium/ac-dev-lab #62: Custom electrochemical corrosion cell for electrocatalysis experiments

sgbaird, 2024-09-27:
Perhaps something similar to the following:
1. https://www.basinc.com/products/IP-U-1L-CCWJ
2. https://www.meinsberg.de/en/produkte/kmz.html#

I've had prior experience and success printing a custom electrochemical cell using Formlabs resin (in my case [clear resin](https://formlabs.com/store/materials/clear-resin-v4/)), which has relatively [good chemical compatibility](https://support.formlabs.com/s/article/Solvent-Compatibility?language=en_US). I believe I was using NaOH, and I ran some of my own "print a cube, let it sit in solution for X amount of time" tests and saw negligible mass differences.

The intention is to use this with KOH at a relatively high concentration for extended periods of time.

Aside: Here are some examples of custom ones that I came across
1. https://www.researchgate.net/figure/a-CAD-model-assembly-for-the-electrochemical-cell-b-Complete-electrochemical-design_fig4_356968869
2. https://www.scielo.org.mx/scielo.php?script=sci_arttext&pid=S0035-001X2022000400007#f1
3. https://www.researchgate.net/figure/a-CAD-SolidWorks-model-assembly-for-the-electrochemical-cell-b-Complete_fig1_361162584
4. https://www.ncbi.nlm.nih.gov/pmc/articles/PMC8283263/ (nice table / review, see table 2)
5. https://www.sciencedirect.com/science/article/pii/S2215016122000115 (molten salt)
6. https://www.sciencedirect.com/science/article/pii/S0925400522004920

sgbaird, 2024-09-28:
@MiceeNS, there is a water jacket that is used for keeping the internal container at a constant temperature. Five ports are required on the lid. There needs to be some kind of gasket between the lid and the cell.

In terms of parametrization, the size of each whole and the overall dimensions of the cell (i.e., internal volume) are the main ones.

sgbaird, 2024-09-28:
Internal volume likely somewhere between 200 mL and 1L

programlich, 2024-10-04:
> Internal volume likely somewhere between 200 mL and 1L

1l, please, if thats fine to build

programlich, 2024-10-04:
Some thoughts I had on the cell:

- Heating: Depending on the heat resistance of the polymer, we could go with a single wall cell in the first iteraion and heat the liquid inside through placing the cell on a hotplate. That might be the easiest setup for now. However the polymer should be resistant to at least 100°C. Is that possible?
- Regarding the tap: Is it possible to construct it in a way that allows the whole cell to be gas tight? Maybe a sealing with an O-ring and a srewable tap ore a bayonet lock might be an option

MiceeNS, 2024-10-04:
@programlich just curious, have you seen the design? I will get back to you regarding your comments.

programlich, 2024-10-04:
> @programlich just curious, have you seen the design? I will get back to you regarding your comments.

Yep, Sterling showed it to me yesterday. I was quite thrilled about how fast you got that done. Would have taken me ages

programlich, 2024-10-04:
Another question that came to me regarding the sample holder (not included in the designs, that Sterling posted above). Is it generally possible to interrupt the printing process at a given time to place some wires inside the structure and continue the build process afterwards? Atm we get the wire into our sample holders by embedding it into epoxy but this is quite a painful process.
Current design (with lots of room for improvement):
[image]

What we need in the end is a sample holder which can be electrically connected to some external periphery and two electrical contacts for the connection to the sample. All other parts of the wire must be well protected to not get in contact with the surrounding media, which would ruin the whole measurement.

MiceeNS, 2024-10-04:
I was supposed to upload the renderings here, will do as soon as possible. The design already has an O-Ring sealing. In the files I have an excel file for the O-Ring size design since you can get only get standard dimensions. However, the backside of that is getting the O-Ring when you print the design.

This type of sealing can withstand up to certain pressure depending of the O-Ring squeeze%. 

For the heating, do you require a constant temperature or fluctuating temperature? I will come up with something since the polymer has low conductivity.

programlich, 2024-10-04:
Ah nice, glad to hear, the sealing is already planned. Perfect!

We will normally run one experiment at a given temperature. Each experiment takes roughly about 24h (can also be up to a week in some cases). The temperature will be at 70 to 80°C normally but the hotplate will get warmer than that.

MiceeNS, 2024-10-04:
Stopping the printer and continuing is possible, at least that what quick search says.

If thats not possible maybe you could create a conduit inside the print enter the wires and seal it with 3D printed plugs/adapter.That is how explosion proof wiring is done.

I will have this in mind and come back to you.

programlich, 2024-10-04:
I made a quick overview of the top view of the tap. Hope that helps to get an idea about the required inlets.

[image]

programlich, 2024-10-04:
> Stopping the printer and continuing is possible, at least that what quick search says.
>
> If thats not possible maybe you could create a conduit inside the print enter the wires and seal it with 3D [...]

Alright, sounds good. Once we start thinking about the sample holder in detail, I think it would make sense to discuss that in a meeting. In my opinion the sample holder should be completely redesigned now that we have this possibility. The current design has a lot of drawbacks.

sgbaird, 2024-10-04:
Let's assume we have access to a circulating heating bath (i.e., still have the outer jacket), with 1/4" barbed tubing connectors. Typically, one barb should be very low, and the other barb very high (I forget which one is supposed to be inlet vs. outlet, but shouldn't matter from the design perspective.

programlich, 2024-10-04:
Talking about jacks: It would be great to have the gas inlet designed as a jack with a 90° angle relative to the tap. On the inside it would be perfect, if the inlet would proceed as a pipe ranging to a depth of approx 4cm above the ground of the cell when closed. The pipe could have an inner diameter of 3-5mm.

MiceeNS, 2024-10-06:
Regarding the Haber Luggin Capillary, we could print something like this. I can adopt it regarding your additional request.
It has NS Joint which can be placed easily on the cell lid. 

[image]

[image]

I assume the electrode would be a piece of metal rod that needs to inserted inside the electrochemical cell. For example, we could print the Haber Luggin Capillary with this kind of adapter( referencing the picture below )
[image]
[image]

MiceeNS, 2024-10-06:
> > Internal volume likely somewhere between 200 mL and 1L
> 
> 1l, please, if thats fine to build

Absolutely, just trying to get to fusion 360 shared folder.

> I made a quick overview of the top view of the tap. Hope that helps to get an idea about the required inlets. [...]

Could we please chat more about the type of  joints/connections on this inlets or you want to build everything from scratch? 

> Let's assume we have access to a circulating heating bath (i.e., still have the outer jacket), with 1/4" barbed tubing connectors on the outer jacket. Typically, one barb should be very low, and the other barb very high (I forget which one is supposed to be inlet vs. outlet, but shouldn't matter from the design perspective.
> 
> EDIT: in terms of temperature, there are a few different resin options. We should check chemical compatibility on our end with some simple tests. Clear resin is the preference, but we need to check temperature limit. High-T resin is also an option, but we need to check chemical resistance.

You would want the inlet to be the bottom one since heat rises. Even though the resin would have low conductivity, with the waterjacket the only heat loss would be from the top of the tap.

programlich, 2024-10-06:
If we could print a Luggin capillar like the one you shared, that would be perfect. For the sake of simplitcity and also to assure, that the capillary is always placed at exactly the same location, I suggest, we print it directly connected to the tap. The one we use in our lab looks like this (practically the same):

[image]

The only important thin is, that our reference electrode can be comfortably placed inside. This means that the tip of the electrode should be at the height where the curvature of the capillary starts. Between the electrode and the inner wall of the capillary there should be at least 2mm of free space. This is our [reference electrode](https://gaskatel.de/en/shop/hydrogen-reference-electrode-hydroflex/) (in practice without the black safety cap at the bottom):

[image]

You are quite right, that the counter electrode is a piece of metal. Specifically it is a platinum wire, which is rolled into a spiral inside the cell. So we basically need an inlet for just a small wire. So far we do that by passing the wire through a rubber plug. Do you have any better idea about how to attach it to the tap? This wire should be removable for replacement/cleaning.

Regarding the gas inlet we would have to design something like this, just with a jack instead of the ball connector. Inside the cell it can be just a straight pipe or have a slight curvature at the bottom, whatever is easier to design. This part could also be directly printed with the tap. No need to remove or replace it.

[image]

programlich, 2024-10-06:
Maybe also helpful: A photo of the corrosion cell in a different design with a flat tap. I guess, we'll end up with a design that's a mix of the one in the photo and the one of basinc.

 
[image]

sgbaird, 2024-10-08:
> Regarding the gas inlet we would have to design something like this, just with a jack instead of the ball connector. Inside the cell it can be just a straight pipe or have a slight curvature at the [...]

There's a pretty standard "sparging rod with frit" that we could probably just have an adapter for. There is also the possibility of making a "false floor" with a series of holes and the gas inlet going directly into the "false floor with holes".

programlich, 2024-10-08:
[image]

programlich, 2024-10-08:
This is the [thermocouple](https://www.fishersci.de/shop/products/bola-temperature-probes-k-smp-connector-2/17197564?change_lang=true), we are using. It would have to be inserted into the electrolyte while being held tightly by the lid. Perhaps with an O-Ring. 
[image]

[image]

programlich, 2024-10-08:
Regarding the gas outlet, we now have the [19/22 condensor](https://uoftchemstore.com/Condenser-coil-200-mm-19-22-p414950395), that will be attached to the lid. The part of it connected to the lid is a 19/22 joint.
[image]

programlich, 2024-10-08:
One more thing, that we did not discuss explicitly yet. If we plan to leave the lid attached to the cell and only switch reference and working electrode between the experiments because we can do the cleaning automatically, we need to be able to take the sampleholder out of the setup seperately. Once the experiment is done the order of steps would be:
1. Take out sample and store in safely in water
2. Remove reference electrode
3. Remove electrolyte
4. Run the cleaning protocol
5. Insert fresh electrolyte
6. Insert new reference electrode
7. Insert new sample

In case you wander about the swapping of the reference electrodes: One electrode can only be used for one experiment, afterwards it needs to "recover" for at least one day before it can be used again.

MiceeNS, 2024-10-12:
[image]

Working Electrode Hydroflex

MiceeNS, 2024-10-12:
[image]

I could upload this only as .gif here. This is my first idea for fully automation on the counter electrode. I mean there would be a standing rack and the adapter holding the wire. The adapter has to have enough weight so it can hold the wire, depending on the material flexibility (section picture below). Respectively, the Jubilee tool would have the same design and would hold the wire by the other side of the adapter while transferring to the electrochemical cell. The cell would have opening big enough so the wire spiral can go through it. The opening would have the same design as the standing rack and now again the bottom part of the adapter would hold the wire, hopefully with enough weight/pressure. 

[image]

[image]

I hope you got my point, let me know what do you think. Is it something you think could be potentially a solution? Maybe a rubber gourmet will be a better solution. I just thought this would have less automation problems in future.

sgbaird, 2024-10-12:
The GIF is really informative! I will give this some more thought, but overall a good impression about automation with the removal of the counter electrode. @programlich feel free to weigh in from a workflow perspective (did the counter electrode need to be removed, or is it just the working electrode with sample holder?). Either way, the same concept may apply.

sgbaird, 2024-10-12:
Maybe we could be thinking about this a different way.. we keep thinking "how can I keep the cell sealed?" Perhaps instead, we decide - let the cell be unsealed, and have the actuator motions happen with a "lidless" design, and put a small nitrogen chamber around the actuators. One problem is the evaporation of solvent and how that affects the experiment.

Or maybe the lid could be broken into three parts, and pushed together laterally (kind of like the lidless design). Maybe even at an angle (e.g. 45 degrees) downward vertically. With enough o-ring type material, this could be a good solution. Maybe only two actuators needed. In fact, maybe only one if we are able to fix one side of the cell in place, and focus on opening up the vertical area for the sample holder only.

The sample holder could be this portion of lid with its own sealing mechanism. Gantry gripper brings sample holder into position.. linear actuator presses down. Having the linear actuator at an angle, and not actually attached to the sample holder (just applies pressure and releases), allows clearance for the gantry gripper.

sgbaird, 2024-10-12:
Rough side view sketch:

[image]

programlich, 2024-10-12:
thanks a lot @MiceeNS  for the good work. The gif is very helpful for the understanding. I really like, that you already have further automation steps in mind.
A few thoughts on this:

- The switching of the counter electrode is something which would be nice to automate but does not have the highest priority. Most time can be saved with automatic changing of electrolyte / rinsing as it allows to only switch the sample while leaving the rest of the cell untouched. 
- If it's easy to implement, I don't see why we should not go with your design. I'd leave that choice up to you mainly in the context of development time. The faster we get a prototype printed for the first tests, the better.
- Just to clarify one detail: The coiled wire of the counter electrode is actually in the same plane to the rest of the wire, not perpendicular to it
[image]
- Regarding the question of sealing, maybe you could add an o-ring to the inlet where the counter electrode will be placed in so that with pressing the electrode a little into the o-ring the tightness does not only rely on weight. Does that make sense?

programlich, 2024-10-12:
Also, thinking about actuators, which will work around the cell, I rearranged the position of the several components so that we can have one side from which the actuators can act freely:
[image]

programlich, 2024-10-12:
@sgbaird I find your ideas really interesting. It's a great idea to overthink some processes completely.
If we say that the sample holder should be the only part which is removed between two experiments, I wonder if it would not be simplest to have int inserted right from the top, perhaps with an actuator pressing it down to get the cell tight, instead of an actuator coming from an angle. If we insert everything from the top, we could go with a standard o-ring between tap and sample holder and would not have to worry about special geometries for now.
If we come from the top, we could have one actuator removing the sample holder after the experiment and placing it into a station aside with a water reservoir to not expose the sample to air. After manually changing the sample, the sample holder could be grabbed up and placed back into the cell from the top.

programlich, 2024-10-12:
My feeling is, it would be best to have a simpler prototype ready soon to start collecting real world experience and then iterate to the next version with more automation steps.

programlich, 2024-10-12:
Regarding the electrolyte exchange and rinsing, perhaps we could design the in and outlets so that they enter the cell with an angle. This way the water would rotate within the cell and create some more turbulences which would improve the cleaning efficiency. I made a rough 2D plot. The inlet of both electrolyte and water jacket would be at the bottom of the cell, the outlet on top.
[image]

sgbaird, 2024-10-12:
Clever idea with the angled inlet. The gantry could perhaps be left in place (constantly applying pressure), during the duration of each experiment. Not sure if this will cause long term issues with the gantry, though probably depends on the amount of pressure. We lose the ability to do anything else with the gantry during the experiment, but maybe not a huge deal.

sgbaird, 2024-10-12:
*super* random aside: for the electrolyte supply, there's some potential for customizing the electrolyte composition on a per experiment basis as well, using pumps and solid dispensers.

MiceeNS, 2024-10-12:
I am making a new update with the design, the one we already had like a jar (but all the elements to be  printed from below ) so we can print it out and as Matthias said to start collecting some data. It would be hard to automate it right away, we will do that in second step, after troubleshooting.

Regarding the purging process, everything depends on what we connect the cleaning water line. Maybe a public water line which is already pressurized? We could have solenoid valves with relays for future automating this.

Give me some time and will send you the updated drawing. I have gone through all your comments, I just want to finish this design and we can continue from there.

sgbaird, 2024-10-12:
I'll need to double check about a water line, but yes - I think we will have an "unlimited" supply.

Could you watch this? https://youtube.com/shorts/9ojPpwQMfrM?si=-TakAUyOfbfJxTnN

In particular, you may want to start with checking the current design for printing compatibility using PreForm.

MiceeNS, 2024-10-12:
> I'll need to double check about a water line, but yes - I think we will have an "unlimited" supply.
> 
> Could you watch this? https://youtube.com/shorts/9ojPpwQMfrM?si=-TakAUyOfbfJxTnN
> 
> In particular, you may want to start with checking the current design for printing compatibility using PreForm.

Yes, I just want to finish with the design, then you guys to check it for any small changes. Then I will check it with PreFrom, maybe adapt it and I will try to print it so next week if Matthias can check the design and test it.

programlich, 2024-10-12:
We definitely have "unlimited" water supply inside the fume hood, pressurized to a certain level. We'll need to put a T-piece somewhere in the water supply to get water both for the cooling of the condensor and the purging, but that's not really a problem.
Thinking about it, it might be best, to have the water input and output more or less on the same side of the cell to simplify the arrangement of the water tubes.

programlich, 2024-10-12:
> Yes, I just want to finish with the design, then you guys to check it for any small changes. Then I will check it with PreFrom, maybe adapt it and I will try to print it so next week if Matthias can check the design and test it.

Sounds great to me. Curious to see the design.

programlich, 2024-10-12:
> _super_ random aside: for the electrolyte supply, there's some potential for customizing the electrolyte composition on a per experiment basis as well, using pumps and solid dispensers.

No idea, how that is going to work, but if it works it would be absolutely beautiful! Let's discuss that in detail, please.

MiceeNS, 2024-10-15:
Hello guys, 

Happy Thanksgiving, I hope you are with the ones you love.

I have many things to share at this point. I will just post a bunch of pictures and screenshots now, I will send detailed explanation by tomorrow.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

programlich, 2024-10-15:
This is absolutely amazing! Fantastic work @MiceeNS! I really love, how you got the purpose of the cell and could create such a design so fast. I´m curious to hear about the explanation but the pictures are already great. Thank you so much.

Happy thanksgiving to the two of you, too.

MiceeNS, 2024-10-16:
Good evening,

My first picture is from the cell lid, this will be 3d printed as a whole. The picture includes the lid, the Haber Luggin Capillary, gas inlet with barbed fitting for a 1/4" tube and the working electrode positioner/holder.
[image]

The reference electrode would be inserted from top into the Haber Luggin Capillary. The lid and the Hydroflex electrode have the same body shape like \ /. I just need confirmation for the Hydroflex exact dimension so we can be sure it would fit properly.
[image]

For the counter electrode, I thought of 3d printing the simple small adapter with a smaller diameter hole in the middle than the wire. The adapter would be printed in 2 parts and will be joined and held together by an O-Ring. The O-Ring would eventually work as holder for the adapter on the lid.
[image]
[image]

Condenser has its own NS joint 19/22 on the lid while the thermocouple will be held with O-Ring. 
[image]

Regarding the working electrode, it is constructed of 3 parts. The positioner that will be 3d printed with the lid, 3d printed cap that hopefully will keep the platinum wire from electrolyte and the working sample itself. 2 cm2 exposure.
[image]
The point of contact between the platinum wire and the sample is shown in the next picture. The wire inside of the 3d printed holder and it is extruded in such a way that comes out to the surface where the working sample is. Hoping the 3d printed cap will hold tight the contact between the two. I also created a revision hole on the sample holder if the platinum wire gets stuck somewhere.
[image]

The whole assembly of the working electrode would be something like presented on this next gif

[image]
[image]

For better understanding of the cell, I will post few pictures rather then explaining
[image]
[image]
[image]
[image]

Please let me know what do you think, any small changes I could do? I know there is a lot where can be improved but I also believe we have to start from somewhere.

In the next days will revise the model, check the compatibility with PreForm and will print it if you all agree to.

programlich, 2024-10-16:
I keep being impressed. The overall design is extremely well thought and includes a lot of details I haven´t even thought about so far. Thanks again.
But now about the design itself:
- **Specs of the reference electrode**: Do you want me to contact the supplier and ask for the exact geometry? Would that be helpful?
- I didn´t quite get the job of the tank washing nozzle. Could you explain that, please?
- **Sample holder:** The basic idea looks very good to me. Would it be possible to remove the small hooks of the little "arms" into which the sample is inserted? I fear that these hooks could scratch off the mechanically instable coating of some of our samples
- **Sample holder exposure area**: Did you design it, so the real exposure area is 2cm ²or is this only roughly the case? I did not think about this previously but one of the drawbacks of a round design is that it doesn´t have a perfect rectangle for the exposed area. For us it is crucial to have exactly 2cm² exposed, so perhaps a slight modification would be necessary.

[image]

- **Sample holder wire:** For testing purposes I suggest we work with a stainless steel wire. Should be fine for now until we are sure about the design. No need to waste Platinum.
- **Capillary:** Would it be possible to move the whole capillary a bit closer to the sample holder while reducing the length of the nozzle? The electrochemically relevant distance is the distance between the bottom of the reference electrode and the sample. The nozzle of the capillary is basically just a workaround because its not possible to get the reference electrode as close to the sample as we would wish.
- Could you think of any (simple) design, which allows the removal of the sample holder while leaving the tap in place? This would simplify the sample exchange, which needs to be done before the rinsing procedure to not harm the sample. If it´s a more complex thing, we´ll leave that for version 2
- **Counter electrode**: Nice idea with the 2 printed parts an the O ring.
- **Gas inlet:** Is it possible to design the inlet with multiple small holes instead of one big hole at the bottom? It´s probably better for the solution of the gas into the electrolyte having more smaller gas bubbles. See my sketch, for explanation. 
[image]

- **Wall thickness:** Is this a parameter that can be easily modified? If so, for the beginning I suggest we start with a bit extra thickness for safety purposes until we are sure about the chemical resistance of the resin against the highly aggressive KOH.

MiceeNS, 2024-10-16:
Specs of the reference electrode: This is my dimensions on the Hydroflex, I was wondering if the cone shape has the correct slope. Even if its a bit off it will fit, so lets not worry about that.
[image]
Tank Washing Nozzle: It just have 360 degree opening when you inject water it will spray in 360 degrees, but I believe this things go with a bearing so they can rotate and spray 360. Here is a picture of the Nozzle
[image]
[image]
Sample Holder: Yes, I would remove them and leave only like a slots opening for the working sample. Do you know the thickness of this working sample?

Exposure Area: I did a mistake here, its 2.159 BUT in mm2. I would change it to cm2 and make it exact 2cm2.

Capillary: I would redesign them closer, now I believe is 20mm distance between the nozzle of the capillary and the working sample. Also I would rotate the working sample for 360 degrees so the capillary its not in the way of placing the cap on the working sample.

Could you think of any (simple) design, which allows the removal of the sample holder while leaving the tap in place? This would simplify the sample exchange, which needs to be done before the rinsing procedure to not harm the sample. If it´s a more complex thing, we´ll leave that for version 2
--- I can make the slots on the cap, this was my next idea for improvement. So you take the sample out with the cap?
 Gas inlet: Yes, I would create holes.

Wall thickness: It can be modified, will not take long. The Cell thickness is 4mm, some of the electrode tubes are 2-3 mm and the thinnest is the Capillary with something around 1 mm. Do you think making the electrode holder and the capillary 4mm thick and the cell around 6 mm would be okey?

Will adjust this as soon as possible, let me know what you think.

programlich, 2024-10-17:
- Tank washing nozzle: Great idea. Love it
- Reference electrode: Looks good for now. On monday I´m back in the lab and can measure the exact dimensions
- sample holder: My samples I brought to Toronto have a thickness of ca. 0.6mm. However in our labs have a variety of samples being tested with different thicknesses. Probably a range from 0.2 to 1.2mm.
- exposure area: perfect. Thanks for the redesign
- Capillary: Very good. A distance between sample and tip of the capillary of about 5mm would be nice. Do you know the diameter of the opening of the capillary?
- Sample holder removal: I´m not sure, if I really get your suggestion. Could you explain it further, please?
- Wall thickness: Lets go with 6mm wall thickness for the cell then. The thicknesses of the tubes don´t concern me as they are not that much safety relevant.
- O-rings: Could you send a list of the planned o-rings, so we can order them asap, please?

sgbaird, 2024-10-17:
Maybe the capillary should stay thin (at least within the path between the counter electrode and working electrode) in order to minimize the interference of the electric field between the two electrodes, no? Also, I think the capillary would still need to be "pointed" at the "front" of the working electrode, right? At least this is how I see it in most applications. I don't know if that's a necessity or a convention.

MiceeNS, 2024-10-17:
> Maybe the capillary should stay thin (at least within the path between the counter electrode and working electrode) in order to minimize the interference of the electric field between the two [...]

The interference makes logic to me, maybe we can make it atleast 2mm. 1 mm is very thin. The capillary would still point at the working electrode. I would just make it closer and rotate the working electrode for 180, so the capillary would be placed from "behind". Since it is a circle it shouldnt matter for the test

programlich, 2024-10-17:
Good point @sgbaird. I agree, we should design the tip of the capillary being perpendicular to the sample. Probably electrochemically better and it reduces the risk of a gas bubble entering the capillary.
Also, the thinner the better. 2mm seems fine to me. Most important is to get the tip of the capillary close to the sample. The idea of rotating the sample holder seems good to me. As long as the "order" of the three electrodes remains working electrode - referenece electrode - counter electrode, everything is fine

sgbaird, 2024-10-17:
The resin printers are set back up again and ready.

MiceeNS, 2024-10-18:
> * Tank washing nozzle: Great idea. Love it
> * Reference electrode: Looks good for now. On monday I´m back in the lab and can measure the exact dimensions
> * sample holder: My samples I brought to [...]

The capillary has 10 mm diameter opening
Sample holder removal: I will share the pictures in a moment, just finished with the design
Wall thickness: Done
O-rings:  as568 standard O-ring (Check picture)
[image]
[image]

MiceeNS, 2024-10-18:
[image]
[image]

MiceeNS, 2024-10-18:
Currently, checking with PreForm and trying to log in on the 3d printers. For the Lid it gives me 7hours print time :) (without adjustments)  because of the odd structure it has to print a lot of supports. That barbed fitting for the gas inlet is in the way for placing the lid flat on the printing table

sgbaird, 2024-10-18:
Just sent you access to the Formlabs printers. See if you can print it on the Form 3+, but otherwise using the Form 3L is fine. Jobs can be sent remotely. The Form 3+ has the "Form Auto" attachment, so if the object is short enough, it can perform an auto-remove, making it ready immediately for the next print.

Could you print 10 test cubes with the Form 3+ with auto-removal? (1 cm x 1 cm x 1 cm). Currently, clear resin v4 is loaded.

@programlich we will probably want to test the cubes at both room temperature and the target temperature, with a few repeats.

If clear resin v4 isn't compatible enough, we can also try rigid 10k.

If none of the resins on hand are compatible, we can consider some kind of internal coating or container.

programlich, 2024-10-18:
After having printed the test cubes from the v4.1 resin, I will put them into KOH on monday at different temperatures and hope, that they will resist the harsh environment.
If that goes well, I suggest, we print the first prototype on Tuesday 22nd. Would that be suitable for you, too @MiceeNS?

programlich, 2024-10-18:
Could you please give us specific links for the o-rings, so we can be absolutely sure to order the right ones, @MiceeNS? That would be of great help

sgbaird, 2024-10-18:
Maybe McMaster?

MiceeNS, 2024-10-18:
Just to get home, I will try to print the cubes and also search for supplier for the O-Rings. Test the cubes on Monday and if the Resin is okey we will print the cell for Tuesday. I will also check if McMaster Carr has this dimensions. Will update you tonight.

programlich, 2024-10-18:
Actually, we already printed the cubes, so no need for that :) I will run the chemical compatibility test on Monday/Tuesday and report. Thanks for your help.

[180 more replies]

closed 2024-11-07 by sgbaird