AccelerationConsortium/ac-dev-lab #64: Tool design for locating the color sensor package to the charging port on OT-2 deck

Neil-YL, 2024-10-01:
To ensure proper location alignment of the color sensor package when it’s placed back in the charging port, certain guiding tools should be introduced between the charging port and the sensor package.

Some design concepts:

 - 6-point kinematic coupling parts (used on the Science Jubilee)
[image] 

 - several cone locator pins 
 
[image]

Neil-YL, 2024-10-01:
The 3d-printed finishing surface could be a challenge to have a good fit between the parts. We could have several tests before add it to the package design.

sgbaird, 2024-10-02:
> The 3d-printed finishing surface could be a challenge to have a good fit between the parts. We could have several tests before add it to the package design.

Maybe necessary to use "fine" setting on 3D print

Neil-YL, 2024-10-02:
Drawing for yesterday and today discussion:
I will post all design related contents to this issue from now.
[image]

[image]

sgbaird, 2024-10-03:
[image]

Neil-YL, 2024-10-04:
Preliminary design for the sensor package, the top end geometry has been tested with current pipette on OT-2 for pick-up.
[image]
[image]
Details to add and consider:  battery and receiver, simulate 3D printing supports
Next step will be designing the installation of the microcontroller board side

sgbaird, 2024-10-04:
Awesome! Thank you!

Random comment: maybe round out this edge? 
[image]

Maybe the "fake tip" could be made a bit shorter, so that it's less likely to break off (or maybe cut an internal hole).

Neil-YL, 2024-10-04:
> Awesome! Thank you!
> 
> Random comment: maybe round out this edge to reduce stress build up? ![image](https://private-user-images.githubusercontent.com/45469701/373708420-3122fc8a-a0ec-4476-98e2-e84fb561fdce.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwNTcyMTUsIm5iZiI6MTcyODA1NjkxNSwicGF0aCI6Ii80NTQ2OTcwMS8zNzM3MDg0MjAtMzEyMmZjOGEtYTBlYy00NDc2LTk4ZTItZTg0ZmI1NjFmZGNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA0VDE1NDgzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmMjhmMzc1YTZmOGViNDg2OWY2YjkyMGJlM2ExYzg4ZWMwMTE1NTJjOGI3N2MwN2VkZDgwODVlYzgzZTg5ZDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bZT-1OHheNRW6VDTWyY7XAKAwW5L7qm46YZKUBx82gQ)
> 
> Maybe the "fake tip" could be made a bit shorter, so that it's less likely to break off (or maybe cut an internal hole).

Sure. I will add more round feature when finalising the design (like at the fitting edge of the bottom), in case of any dependency errors if the design is changed significantly.

I have adjusted the length of the tip and also thickened the top face to reduce deformation when pick up.

sgbaird, 2024-10-05:
Assume this battery I think

[image]

Neil-YL, 2024-10-08:
Have the first 3D print parts assembly in hands. First printed part has some failed surface on one side so I changed the printing direction for this one, but resulting in hard-to-remove support within the fake-tip.(Actually I accidentally cracked it while removing the supports)
[image]
[image]
Everything looks good except for the location of the micro-USB connector. I’ll add an additional hole for it in the new design. The next step will be designing support to locate the sensor and the battery.

Neil-YL, 2024-10-08:
[image]
Photo for the failed surface for the first one. This printing direction was suggested by Bambu Studio to reduce supports. I left it printing last night before leaving the office, so I’m not sure if the issue was caused by leaving the printed part in the printer overnight. I will try printing the new design parts today with the same direction again to see if the problem persists.

sgbaird, 2024-10-08:
I would lean towards likely not due to being left overnight.

Neil-YL, 2024-10-08:
I think so. I am printing the new designed parts with the successfully printed direction but using my manually-controlled supports, reducing the support within the fake-tip. Hope it works.

sgbaird, 2024-10-08:
Aside: the batteries arrived from amazon. I ordered 4

Neil-YL, 2024-10-09:
The manual control support works, and now I have a well-assembled block ready for use.

https://github.com/user-attachments/assets/b0c6d32e-cfc8-463a-905d-8b075933b708

https://github.com/user-attachments/assets/a2506936-d6d1-4bd2-a951-300f9c395c4c

Neil-YL, 2024-10-09:
Battery arrived:
[image]
[image]

Neil-YL, 2024-10-09:
Create an assembly model for the sensor package, for future use in the design of charging port
[image]
[image]

Neil-YL, 2024-10-16:
Currently working on the charging port design. 
It would be much easier for the board positioning and cable fitting if we can use this kind angled MicroUSB cable
[image]
[image]

sgbaird, 2024-10-16:
@Neil-YL could you suggest an Amazon item that you think would work?

Neil-YL, 2024-10-16:
> @Neil-YL could you suggest an Amazon item that you think would work?

This one should work for us: https://www.amazon.ca/Startech-Com-Micro-Cable-Black-USBAUB2MD/dp/B00EANUQ5M/

Neil-YL, 2024-10-17:
[image]
[image]

Have a charging port printed and assembled. Next step is to create the json file for OT-2 as a customized labware and tip, and test the control (picking up and returning) on the machine.

Neil-YL, 2024-10-17:
A video of mounting the charging port on the deck

https://github.com/user-attachments/assets/0f355706-7666-4a93-b45b-d30a3c137873

Neil-YL, 2024-10-17:
I have a successful pick up, but OT-2 protocol analyser requires a height of the labware + height of the tip less than the highest bound for a certain pipette to run the protocol analysis (then, the labware calibration and the execution of the protocol). In our case, now the total height for the charging port and the sensor tip is 244mm where Opentron P300 has a highest bound of 202.77+-5mm. 

[image]
[image]

I will redesign the sensor package and the charging port to meet the requirement.

Neil-YL, 2024-10-21:
Now the pipette can pick up the sensor package from the charging port, move to the well plate and put back to the original slot on the charging port, see video:

https://github.com/user-attachments/assets/cd0377a7-1565-4b8c-99ac-6366c5193e97

I also run a *fake* obstacle test:
[image]
 The charging port was placed on the top left corn (Slot 10), with a tip diameter of 66mm. In the first protocol, the tip rack (on Slot4) should block the access to the A1 well on the well plate(on Slot5). However, the robot performed an identical movement with two protocol (so I didn't record another video). Hence, the program will not simulate the obstacle on the route due to the size of the tip (seems quite fair if they don't include it as a parameter when analysing movement).

I may see how to adjust the accessibility of the well/labware in the program later or we have to sacrifice nearby slots in actual operation.

sgbaird, 2024-10-21:
Great to see this!

> Hence, the program will not simulate the obstacle due to the size of the tip on the route (seems quite fair if they don't include it as a parameter when analysing movement)

Also, nice find! Thank you for verifying this. So, from what I'm gathering here, we'll need to account for "obstacle avoidance" on our side. Might also be a good thing to provide feedback to Opentrons about.

sgbaird, 2024-10-21:
Couple Q's:
1. The "dropoff" seems to have a somewhat tall height. Can this be adjusted? (I realize you may have been "stress-testing" to see what it can handle)
2. If obstacle avoidance is up to us, then this also means we can put the tip along the edge or in the corner of the top face and use edge and corner slots, right?

Neil-YL, 2024-10-21:
> Might also be a good thing to provide feedback to Opentrons about.
I will reach out to Opentrons’ tech support to discuss our application and concerns. They may have advanced or undocumented features that aren’t included in their general documentation. Since our goal is to eventually automate the run, it would be very beneficial if we could make some adjustments to the accessibility and test with their simulator code before the robot starts the run.

> Couple Q's:
> 
> 1. The "dropoff" seems to have a somewhat tall height. Can this be adjusted? (I realize you may have been "stress-testing" to see what it can handle)
I believe there is a advance setting of Z offset for the returning, I could try to implement on later trial.

> 2. If obstacle avoidance is up to us, then this also means we can put the tip along the edge or in the corner of the top face and use edge and corner slots, right?

I think so.

Neil-YL, 2024-10-21:
I think we could design an adapter for the labware that requires the sensor tip to access. By making it higher than the other labware, the accessibility would not be an issue.

Neil-YL, 2025-03-04:
Replaced with new design:

[image]

sgbaird, 2025-05-28:
> Replaced with new design:

The pipette has been struggling to pick this up sometimes. I think there's been a failure rate of ~2-5% (mostly during the initial pickup and not during the transfer, I think?). This was reported during one of the external collaboration/science demos (which went really well btw!), and it happened once recently while doing a remote demo.

I wonder if making it "too thick" means it doesn't bend as much to allow the pipette to go deeper and get that "snug" fit.

Would it make sense to change the offset slightly, so the pipette tries pressing harder? (this would at least be a software change and not require a redesign/reprint).

Neil-YL, 2025-05-29:
> The pipette has been struggling to pick this up sometimes. I think there's been a failure rate of ~2-5% (mostly during the initial pickup and not during the transfer, I think?). This was reported [...]

It could be an issue caused by running for a long time without calibration though.  
Actually I haven’t had a chance to calibrate the new design (the thicker one) since I replaced it. It’s still using the calibration data from the previous version.  
Might be worth performing a labware calibration to see if the issue persists.

sgbaird, 2025-05-29:
Good point! Sounds good to me

Neil-YL, 2025-05-29:
> Good point! Sounds good to me

Do you need me guide someone to calibrate it or wait until I am back in person?

Neil-YL, 2025-06-04:
Recalibrated everything.

Testing the pick up and drop off around 20 times. I tend to believe the issue is due to wear on the 3D-printed material over time, which reduces the friction to the point where it can no longer support the weight of the package. I have tried to force the pipette goes deeper/lower but still had serval times of pick up failure. I think the pipette already reaches its limit (on the hardware safety design side) so we better work on the fake tip instead.

sgbaird, 2025-06-04:
We could also add a check that takes an additional background sensor measurement while the tool is parked. If the sensor readings between when it was parked and when it was out are nearly identical, then it's pretty likely it was a failed pickup.

I guess, if it fell off at some point into the chamber, then we'd see this behavior of before and after being identical during the next experiment.

sgbaird, 2025-06-04:
Could you also describe the calibration procedure a bit? Wondering if it's complex enough to require a video / HiTL step or if a bit of minor markdown documentation is sufficient.

Neil-YL, 2025-06-04:
If it fall into the charging port then a different check is enough, but I noticed it also fall to the deck.

Neil-YL, 2025-06-04:
> Could you also describe the calibration procedure a bit? Wondering if it's complex enough to require a video linked to on a Prefect HiTL step/slack message or if a bit of minor markdown documentation somewhere is sufficient.

It is basically following the guide on the Opentrons APP for recalibration and download the calibration file, rename it as "deck_calibration.json" and scp send to /data/ (the protocol  `opentrons.execute.get_protocol_api` will load /data/deck_calibration.json by default)