AccelerationConsortium/ac-dev-lab #72: Launch code upon startup for RPi SBCs

sgbaird, 2024-10-04:
@gursi26 @linx5o @Jonathan-Woo this applies to you. Ideally, we would have the RPis start running the device code by default on startup. @kenzo-aspuru-takata used crontab for the microscope I think. Out of [five options mentioned here](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/), crontab seems like the best. The OS doesn't load if an error is thrown in `etc/rc.local`. `.bashrc` runs every time a new terminal is opened. `etc/init.d` seems to run both at shutdown and startup (i.e., if a while loop, then it wouldn't actually shut down?). Finally, `systemd` is available only from the Jessie versions of Raspbian OS.

This would also need to be run in the background (and if it throws an error, make sure it doesn't brick the whole system).

gursi26, 2024-10-04:
@sgbaird Done setting this up on the cobot. I followed the instructions on the website but had to make a few changes. After running `crontab -e` to edit the crontab file, I added the following line.

```
@reboot sleep 30; python <path to file>
```
I found that the `sleep 30` call was necessary since reboot is relative to when cron starts, which could be before other services like the internet. Can probably be reduced to a smaller amount of sleep time. 

Also had to drop `sudo` to get it to work.

sgbaird, 2024-10-11:
UPDATE: The blog post clarifies that `etc/rc.local` will not prevent the OS from starting if you use an ampersand, but [systemd is recommended over `rc.local`](https://raspberrypi.stackexchange.com/questions/99270/rc-local-open-new-terminal-window).

I was looking back at this, because it seems to be a bit difficult to see the output from a terminal, motivated by debugging purposes. Within the crontab file, it's easy enough to log to a text file on the desktop for example: 

```crontab
@reboot /usr/bin/python3 /home/pi/Desktop/my_file.py > /home/pi/Desktop/log.txt
```

However, in order to see the active log, you would need to run a command to watch the file for updates. This would either require manually running something in a terminal, or perhaps having a systemd implementation on top of crontab.

sgbaird, 2024-10-11:
UPDATE: The blog post clarifies that `etc/rc.local` will not prevent the OS from starting if you use an ampersand, but [systemd is recommended over `rc.local`](https://raspberrypi.stackexchange.com/questions/99270/rc-local-open-new-terminal-window).

I was looking back at this, because it seems to be a bit difficult to see the output from a terminal, motivated by debugging purposes. Within the crontab file, it's easy enough to log to a text file on the desktop for example: 

```crontab
@reboot /usr/bin/python3 /home/pi/Desktop/my_file.py > /home/pi/Desktop/log.txt
```

However, in order to see the active log, you would need to run a command to watch the file for updates. This would either require manually running something in a terminal, or perhaps having a systemd implementation on top of crontab.

EDIT: not sure if rc.local is actually not recommended in general, and perhaps systemd *is* supported on Ubuntu.

Then again, here is a recommendation that still uses crontab:

- https://askubuntu.com/questions/735935/running-command-at-startup-on-crontab
- https://www.reddit.com/r/linuxquestions/comments/110766f/is_there_a_way_to_make_cron_run_a_script_in_a/
- https://unix.stackexchange.com/a/564491/432417

```crontab
gnome-terminal -- /bin/sh -c '/path/to/script; exec bash'
``

sgbaird, 2024-10-11:
The previous command, with or without @reboot prepended, does not seem to work. Just a normal desktop appears.

From what I understand, these two tasks should run independently.

EDIT: I think I'm supposed to run `crontab` to apply latest changes (?). I checked the crontab logs 

- https://askubuntu.com/questions/56683/where-is-the-cron-crontab-log
- https://superuser.com/questions/247803/cron-ignoring-an-update-to-crontab

Overall confused.

sgbaird, 2024-10-30:
@Jonathan-Woo could you comment on this? (Or link to your comment elsewhere if you already added it)

gursi26, 2024-10-31:
Are you still facing issues like you mentioned in the previous comment? Editing with `crontab -e` auto writes to the file when you close it. Its also possible to check if the desired process is running with `ps aux | grep <process_name>`.

Jonathan-Woo, 2024-10-31:
> @Jonathan-Woo could you comment on this? (Or link to your comment elsewhere if you already added it)

I modified an `autostart` file which appears to be rpi specific.

sgbaird, 2025-03-13:
@Jonathan-Woo does the `autostart` file seem to work OK?

sgbaird, 2025-03-13:
https://forums.raspberrypi.com/viewtopic.php?t=238670

Jonathan-Woo, 2025-03-13:
> [@Jonathan-Woo](https://github.com/Jonathan-Woo) does the `autostart` file seem to work OK?

Yes. I used it for the stream before but I'm not sure if it's still on there. The unreliability of the stream made it so that we resorted to ssh'ing in repeatedly to restart it instead of restarting the PI.

sgbaird, 2025-03-14:
@Jonathan-Woo could you provide an update on how things went with the autostart, and anything else that's relevant? Also, we'll need to figure out WiFi

Jonathan-Woo, 2025-03-14:
I used crontab for the auto reboot. I'll add the command here in a bit. I will likely add a bash script to auto execute on startup to activate the python venv and to run the script. Otherwise we can forego the venv. 

Yes wifi is still an issue.

sgbaird, 2025-03-14:
Thanks!

I figured out WiFi - https://github.com/AccelerationConsortium/ac-training-lab/discussions/183#discussioncomment-12502531

I'm conflicted on whether to use the default python or use a venv.. I only imagine us ever having a single Python installation for the one device.

Jonathan-Woo, 2025-03-14:
I've added the following to crontab for daily reboots at 2 am and execution of the device script on reboot using the venv.

```
0 2 * * * /sbin/shutdown -r now

@reboot sleep 30; /home/ac/ac-training-lab/src/ac_training_lab/A1-cam/.venv/bin/python /home/ac/ac-training-lab/src/ac_training_lab/A1-cam/device.py
```

sgbaird, 2025-03-19:
Transcript around reboot topic: https://chatgpt.com/share/67da116e-184c-8006-99b3-a49fc08eb1bb

I'm somewhat leaning towards `systemd` *and* cron [[updated transcript](https://chatgpt.com/share/67da116e-184c-8006-99b3-a49fc08eb1bb)]

Run:
```
nano /etc/systemd/system/a1-cam.service
```

Copy the following code into the file:

```yaml
[Unit]
Description=Start a1-cam device.py script
After=network-online.target
Wants=network-online.target

[Service]
# Launch the device script (adjust the path as needed)
ExecStart=/usr/bin/python3 /home/ac/ac-training-lab/src/ac_training_lab/a1_cam/device.py
# Restart on unexpected failure – if the script exits with an error, systemd will restart it
Restart=on-failure
RestartSec=10

# Limit restart attempts to avoid a rapid infinite loop:
StartLimitIntervalSec=12h # (i.e., up to max 9 times per day, assuming a StartLimitBurst of 3)
StartLimitBurst=3

# Allow up to 60 seconds for the script to start properly
TimeoutStartSec=60

[Install]
WantedBy=multi-user.target
```

Run:
```
sudo systemctl daemon-reload
sudo systemctl enable ac-device.service
sudo systemctl start ac-device.service
```

Run:

```bash
crontab -e
```

Add the following at the end of the crontab file:

```bash
# 
# Restart at 2 am, local time (set up during flashing, or specified manually via e.g., `sudo timedatectl set-timezone America/New_York`)
0 2 * * * /sbin/shutdown -r now
```