AccelerationConsortium/ac-dev-lab #73: "Use custom broker" checkbox and form within hf streamlit apps

sgbaird, 2024-10-04:
Ideally, I'd like for people to be able to swap out the broker with their own. For non-public AC brokers (currently applies to the microscope, but will also apply to the cobot), we'll want to make sure the broker information isn't exposed.

[image]

Thoughts on a checkbox or dropdown, that if a user clicks it opens the following form?

[image]

@gursi26 @linx5o

gursi26, 2024-10-04:
@sgbaird Is this a custom broker or custom credentials for HiveMQ? do all brokers follow the same credential format (username, password, host url, port)?

sgbaird, 2024-10-04:
> @sgbaird Is this a custom broker or custom credentials for HiveMQ? do all brokers follow the same credential format (username, password, host url, port)?

I meant custom HiveMQ broker + credentials (i.e., new instance on HiveMQ), but it's also a good point - generally if you have those four things you can swap out the broker completely - e.g., move to a mosquitto broker, as long as certificates and security are set up in a compatible way

gursi26, 2024-10-04:
alright, implemented this in my recent push https://huggingface.co/spaces/AccelerationConsortium/cobot280pi/commit/90958100bcd0281f68f55dfe275bd84b448c4103

linx5o, 2024-10-04:
ok, i will implement

sgbaird, 2024-10-16:
@linx5o see https://huggingface.co/spaces/AccelerationConsortium/cobot280pi/blob/6341f4f3454dd00b0352a02e666b8935bd8c580c/app.py#L52 for an example of loading the default credentials from an env var.

sgbaird, 2024-10-16:
> > @linx5o see https://huggingface.co/spaces/AccelerationConsortium/cobot280pi/blob/6341f4f3454dd00b0352a02e666b8935bd8c580c/app.py#L52 for an example of loading the default credentials from an env var.
> 
> The default creds are implemented using env vars in Hugging face. However having it in the form is easier for running it locally.

Oh, got it. Often, I'll use env vars in a VS Code debug launch configuration, but I'm not sure what a convenient way would be with streamlit since it's run from the command line.

EDIT: looks like a terminal profile would do the trick: https://code.visualstudio.com/docs/terminal/profiles.

Forgive the perhaps overly cautious sentiment 😅

sgbaird, 2024-10-17:
@linx5o @gursi26 feel free to reopen if needed.

closed 2024-10-17 by sgbaird