AccelerationConsortium/ac-dev-lab #74: Set up second MyCobot 280

sgbaird, 2024-10-09:
[image]

One will be more for prototyping, one more as a dedicated / standalone remote access demo

gursi26, 2024-10-09:
quick question: for the standalone remote access demo, are we leaving it on 24/7? If so, would it be better to use a non-desktop environment OS like Raspberry Pi OS lite, or should we stick with Ubuntu MATE?

sgbaird, 2024-10-09:
@gursi26 that's a great point. I hadn't really considered that - could make a lot of sense as long as there aren't incompatibilities.

sgbaird, 2025-01-20:
> All that remains is to set a place in the office where i can permanently setup the arm along with maybe a few demo objects that users can try to pick and place.

@gursi26 suggestions on where we can put this? I'm half-considering using that giant opentrons box. Or, clearing space and putting it in the corner of the printer office.

For now, you can set it up wherever you think is best and we can tape some paper over the camera while we figure out a good spot that won't capture people's faces.

sgbaird, 2025-02-15:
I've been checking on a few projects. @gursi26 what is the status on this one? You can use my office while I'm gone if you want. Piece of tape over the camera is still an option.

gursi26, 2025-02-15:
Hey @sgbaird, I paused work on this for a while since I got busy with the project at SDL0. I can setup the permanent demo in your office when I'm back in Toronto after reading week.

sgbaird, 2025-02-15:
Sounds good! Thanks for the update. Is that project with SDL0 something you could summarize/forecast a bit in a separate issue?

gursi26, 2025-03-07:
@sgbaird just made a new issue at #181. Sorry that took a while, was a little busy with midterms and assignments.

sgbaird, 2025-03-23:
Could you get the permanent demo set up sometime next week?

gursi26, 2025-03-23:
Sure, I can come by on Tuesday to set up the permanent demo. where in the office should i do it?

sgbaird, 2025-03-24:
Thanks! You can set it up in my office, that way we won't need to worry much about the camera for now (which I may still just end up putting some tape over in the near-term).