AccelerationConsortium/ac-dev-lab #76: Safety Concerns: Granting Access to PioReactor Feature on Hugging Face Space

linx5o, 2024-10-10:
In particular, having access to the dosing feature could result in the vial being over filled and spilling liquids onto the electronics below. This is as there are no sensors to detect the amount of liquid in the vials. 

Ideas for how this should be handled as the Hugging Face Space is supposed to be open and accessible.

sgbaird, 2024-10-11:
Great point, and definitely worth thinking carefully about. I think that in terms of safeguards for a demo, one of the safer options is to not allow more than the amount of liquid that gets to say, 90% full across the whole system (i.e., not very much liquid at all), and to only use water. This would amount to filling a separate vial 90% full, and transferring that water back and forth between the primary vial and an additional "waste" vial (which also has a peristaltic pump). NOTE: *maybe this is the best to start with.*

This would affect what we want to show from a demo perspective. For example, yeast growth or making Jello becomes unfeasible for a number of reasons - not purely water, and to run multiple iterations would require a larger stock of material.

Another safeguard would be to make sure that the system is tightly sealed, and to make sure from a pressure standpoint, if the vial starts to get overfilled the path of least resistance is into a waste container.

From a software perspective, we could implement logic that controls the integrated dosing time, with a generous safety threshold away from the inferred "full".

From a hardware perspective, we could add some kind of liquid detection sensor to the vial through the cap.

linx5o, 2024-10-24:
From a software standpoint, we could have a file on the PioReactor which stores the current amount of liquid in the reactor. Then using this file we could enforce a limit through software. 

Potential issue of how to reset the file - run a large number on the exit pump, could be a bit tedious.

sgbaird, 2025-01-07:
@linx5o any follow up thoughts on how you want to approach this?

linx5o, 2025-01-10:
**Current**
Currently, I have both waste and media pumps connected to the two vials so they form a circuit. Between them, there is one vial worth of liquid (water). This current solution guarantees that the vials cannot overflow. However, we lose the sense of a waste and media vial concept. 

**Physical Sensor**
We could try to add a physical sensor to either the cap or printing a custom housing which has a sensor hole. This would require development of the base Pioreactor code. 
[Level Sensor](https://forum.pioreactor.com/t/interest-in-a-liquid-level-sensor-plugin/144)
The above discussion is a basis for the possibility of adding a sensor to the Pioreactor.
There are two approaches outlined. Both involve developing Pioreactor code.

**Software Limit**
There does not currently seem to be any limits on liquid amount. The reactor will happily add 100ml of liquid. This approach also would likely require altering Pioreactor base code. There is also the issue of inconsistencies between the digital and physical amounts. Long-running implementations would likely result in big discrepancies leading to undesired results. 

**Water Tight Vial**
We create a cap which is water tight, additional liquids are pushed out through tubes to waste bin. Simplest solution. Superglue?

I believe the current solution of having just one vial of liquid in the system is sufficient for our current demos.

sgbaird, 2025-01-10:
Nice! Current solution seems fine to me too. I've used IV-style drip bags before, which is another option for keeping the system from being over pressured, but not much different from the last option with the waste bin.

Thanks for the update!