AccelerationConsortium/ac-dev-lab #77: API access for demo equipment

sgbaird, 2024-10-12:
While the temporary access key approach is working well with HuggingFace + streamlit + HiveMQ Starter plan + MongoDB, [streamlit doesn't support exposing a remotely accessible API directly as of now](https://discuss.streamlit.io/t/how-to-have-api-call-on-streamlit/32248), and while gradio has this, gradio isn't so much of a dashboard style app framework like streamlit. This is why all the hardware control examples are written with streamlit.

Aside: Accelerated Discovery forum has a topic about temporary access credentials. 

Maybe worth considering flask, https://chatgpt.com/share/670a6b28-abc8-8006-ab20-b7b283db7590, though I'd prefer streamlit.

EDIT: looks like gradio can do plot output now: https://huggingface.co/docs/hub/en/spaces-sdks-gradio

sgbaird, 2024-10-12:
The space should then maybe also link to a prefilled colab notebook that accesses the HF API and shows a simple workflow example

sgbaird, 2024-10-17:
Looks like YouTube video embedding may not be an issue if we switch to gradio: https://github.com/gradio-app/gradio/issues/2894#issuecomment-1366277045.

sgbaird, 2024-10-18:
Documentation for token: https://huggingface.co/docs/hub/en/security-tokens (maybe good enough to just share the link with a brief description of why)

sgbaird, 2024-10-21:
Thanks @SissiFeng for trying out Gradio with the microscope: https://huggingface.co/spaces/SissiFeng/microscope.

I really like the way multiple endpoints are exposed in the "use with API" at the bottom, and how it has an "API recorder".

[image]

(more endpoints below)
[image]

It looks like most everything can be accomplished in gradio, same as streamlit. How was the conversion experience moving from one to the other?

(Note for others: the Gradio app linked above is not connected to the microscope)

SissiFeng, 2024-10-22:
I summarized some key points to consider when converting from streamlit to gradio:

1. Streamlit uses a script-like sequential structure;
    Gradio uses a functional API, requiring defined input/output interfaces.

2. Streamlit uses session_state for managing state;
    Gradio primarily passes state through function parameters.

3. Gradio requires explicit definition of events and callback functions.

4. Consider using Gradio's caching mechanisms.

sgbaird, 2024-11-01:
Thoughts on using `Interface` vs. `Blocks`? @gursi26 @SissiFeng @Jonathan-Woo 

States (i.e., whether to share something between all users, within a user session, etc.). Probably best to have one MQTT client instantiated (and reused) within a single user session.
- https://www.gradio.app/guides/interface-state
- https://www.gradio.app/guides/state-in-blocks

SissiFeng, 2024-11-01:
I prefer “Blocks” for its flexibility, fine-grained UI design and complex workflow control, similar to Flask, with coordinated components and data flow across modules. 
State persists throughout each session,  maybe the accuracy in callback updates is essential to avoid unintended overwrites or inconsistencies. Also, "Blocks" may lead to increased concurrent connections of multiple calls from one user.

sgbaird, 2024-11-01:
The more I'm looking, I'm leaning towards Blocks a bit too, but it's also more verbose and I've been noticing issues with the number of threads (not sure if Interface would have a similar issue).

SissiFeng, 2024-11-01:
That's right. With a multi-module, multi-step setup like Blocks, where each session maintains its own state all the time, the accumulated number of concurrent sessions is what will cause the thread count to increase (this should be properly optimized) 
Interface won't be as bad unless it's handling multiple user requests at the same time.

SissiFeng, 2024-11-01:
I made a quick change to my previous gradio deployment and this interface is definitely better!
[image]

SissiFeng, 2024-11-07:
Made a few tweaks of "ping-pong" —details are on:
https://github.com/SissiFeng/ping-pong

sgbaird, 2024-11-08:
Temporary access and user queues have been a consideration for a number of demos. I came across [gradio interface queues](https://www.gradio.app/docs/gradio/interface#interface-queue) and [gradio blocks queues](https://www.gradio.app/docs/gradio/blocks#blocks-queue).

> By enabling the queue you can control when users know their position in the queue, and set a limit on maximum number of events allowed.

cc @gursi26

gursi26, 2024-11-08:
yea, i had a look at it and it doesnt have a way to allocate a specific amount of time per user, since most of these are built for demos of ML models, so users can interleave requests.

However I was able to get a custom queue working in the code itself. Since gradio doesnt rerun the entire script for each user, its possible to maintain a queue in the code itself.

https://huggingface.co/spaces/AccelerationConsortium/cobot280pi-gradio/blob/main/app.py

SissiFeng, 2024-11-11:
https://huggingface.co/spaces/SissiFeng/ping-pong

Run well, have a try!

sgbaird, 2025-02-25:
I think we have gradio mostly figured out here. Closing. Thanks all!