AccelerationConsortium/ac-dev-lab #79: Prefect flow based on GitHub file for OT2

sgbaird, 2024-10-16:
The prefect flow can be linked to a specific file within a repository, which allows for a sort of "over the air" (OTA) update to the device code.

sgbaird, 2024-10-16:
@yunhzou

sgbaird, 2024-10-24:
@yunhzou what are your thoughts/plans for tomorrow for the OT-2 color matching demo? #26

sgbaird, 2024-10-25:
Sounds like you've been having trouble with installing Prefect on the Raspberry Pi (a 3 I think) embedded in the Opentrons, which it sounds like is running Linux.

Do you mind including the steps/commands you ran and the full command line output for the pip install?

Related:

- https://linen.prefect.io/t/47786/hey-everyone-does-anyone-know-what-are-the-resource-requirem
- https://github.com/PrefectHQ/prefect/issues/1722 (docker arm images are a thing for prefect)
- https://linen.prefect.io/t/2467983/hi-everyone-i-d-like-to-share-with-you-a-nice-project-orches (seems to be possible), also https://unix.stackexchange.com/questions/777264/raspbery-ubuntu-install-running-server-ui-server-how-to-access

When you share the full command output I should be able to help debug. We can jump on a screen share / remote access if needed since it's more urgent.

yunhzou, 2024-10-25:
Works now.
Video:
https://youtu.be/4Eu4eYGAP9w
Code:
https://github.com/yunhzou/OT2Demo/issues/1