AccelerationConsortium/ac-dev-lab #80: OT2 camera feed streaming to YouTube

sgbaird, 2024-10-16:
@Neil-YL could you link the docs you found related to accessing the built-in camera?

Neil-YL, 2024-10-16:
Note: it only has a resolution 320x240px for video

> Using the OT-2 Camera as a live feed webcam 
The OT-2 camera is plugged into the Raspberry Pi through a USB cable. In order to establish a live feed camera for your OT-2 you will need to locate and upload the USB cable from the Raspberry Pi and you will need a USB extender for your computer. The cable for the camera only reaches the USB port on the Raspberry Pi which means you will not be able to directly plug the camera into your computer. 
Here is an example USB extender:
[image]
Next you will need to access the rear electronics by following our support article: https://support.opentrons.com/s/article/Accessing-the-OT-2-s-rear-electronics

> Once you have gained access to the rear electronics you will then see cable for the camera plugged into the Raspberry Pi USB port:
[image]

> You will now plug this camera USB cable into your USB extender and then plug the USB extender into your computer. This will act as a normal Webcam for your computer which you will be able to use with webcam software available on your computer. 

Link: https://support.opentrons.com/s/article/Using-the-OT-2-s-camera

sgbaird, 2024-10-16:
Agreed. Adding a second RPi and maybe even just using a separate camera might be better

Neil-YL, 2024-10-16:
If it only support 320x240px for video(not sure the limitation is on controller end or camera end), using a separate camera might be a better option.

sgbaird, 2025-01-21:
Let's go with the usual Pi Zero 2W camera setup.

Neil-YL, 2025-02-03:
I have run a speed test on the network of the Nokia router. 

Live on: 
[image]

Live down: 
[image]

According to YouTube’s streaming network recommendations:
- 480p (30fps): 1.5 Mbps
- **720p (30fps): 3 Mbps**(our current streaming setting)
- 720p (60fps): 4.5 Mbps
- 1080p (30fps): 6 Mbps
- 1080p (60fps): 7.5 Mbps

We may need a network with a higher upload bandwidth to support our device’s streaming. The ping could be a problem as well.

Neil-YL, 2025-02-06:
> [0:02:51.888436087] [1816] ^[[1;32m INFO ^[[1;37mCamera ^[[1;34mcamera_manager.cpp:325 ^[[0mlibcamera v0.3.2+99-1230f78d
[0:02:52.002349869] [1819] ^[[1;33m WARN ^[[1;37mRPiSdn ^[[1;34msdn.cpp:40 ^[[0mUsing legacy SDN tuning - please consider moving SD>
[0:02:52.007813690] [1819] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:447 ^[[0mRegistered camera /base/soc/i2c0mux/i2c@1/imx708@1a t>
[0:02:52.007969317] [1819] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mpipeline_base.cpp:1120 ^[[0mUsing configuration file '/usr/share/libca>
[0:02:52.032179634] [1816] ^[[1;32m INFO ^[[1;37mCamera ^[[1;34mcamera.cpp:1197 ^[[0mconfiguring streams: (0) 1280x720-XBGR8888 (1)>
[0:02:52.032912404] [1819] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:622 ^[[0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected >
Guessed Channel Layout for Input Stream #0.0 : stereo
av_interleaved_write_frame(): Broken pipe
    Last message repeated 1 times
[flv @ 0x55cac6f680] Failed to update header with correct duration.
[flv @ 0x55cac6f680] Failed to update header with correct filesize.
Error writing trailer of rtmp://a.rtmp.youtube.com/live2/rr8p-pbf1-uzp1-tdju-80ag: Broken pipe
Error closing file rtmp://a.rtmp.youtube.com/live2/rr8p-pbf1-uzp1-tdju-80ag: Broken pipe

sgbaird, 2025-02-12:
You can login with eduroam / UoT credentials on the Pi Zero 2W instead of using the Nokia. Pi Zero 2W supports eduroam, etc. Trouble is we probably need to get a login that's not tied to a personal account, but a personal one might be ok in the meantime.

In addition to logging into eduroam, I suggest setting it to 480p or lower instead of 720p, and let's see how long it can go for. High uptime is more important than resolution right now (since right now we just want to see it moving).

sgbaird, 2025-02-12:
Oh, and I think if you click "reuse settings" and choose the dead stream, you might be able to keep the link the same so you don't have to update it everytime it goes down. Or maybe that only kept the stream key the same, not the URL..

sgbaird, 2025-02-14:
It looks like the stream made it to about 60 hours and then cut out again, but you started a new one already.

A couple comments/options: 

We can utilize the backup stream URL, possibly with a second device or cloud service.

Lengthy conversation here: https://chatgpt.com/share/67afc522-62fc-8006-89d0-6b0fa409a08a

Castr might work.

To avoid having a new URL, we can create a new channel on YouTube per camera and then use a certain URL that points to that channel. https://www.reddit.com/r/k12sysadmin/comments/lgv591/a_way_for_my_livestream_to_avoid_ending_in_case/

For example, https://m.youtube.com/@ac-hardware-streams/live

sgbaird, 2025-02-14:
@Jonathan-Woo is this something you can keep supporting on?

I lean towards creating one channel per camera, using the channel ID link, (maybe also setting the stream key to something we specify rather than autogenerated so it's easier to keep it fixed), and automatically stopping the stream and starting a new one every 8 hours via the API:
https://developers.google.com/youtube/v3/live/getting-started

This would provide us a single persistent link we could use per camera and would have the benefit of keeping archived recordings that YouTube stores for us.

Neil-YL, 2025-02-18:
> To avoid having a new URL, we can create a new channel on YouTube per camera and then use a certain URL that points to that channel. https://www.reddit.com/r/k12sysadmin/comments/lgv591/a_way_for_my_livestream_to_avoid_ending_in_case/

I turned on the Auto-stop on the YTB studio to see how it works. 

> 
> For example, https://m.youtube.com/@ac-hardware-streams/live

Nice to know. I will see how to used it as embedded url.

sgbaird, 2025-02-21:
Maybe if we get the video ID programmatically: https://stackoverflow.com/a/76541269

(Would convert the Java to Python first so we could use directly in HF)

sgbaird, 2025-02-28:
Two related notes: per-equipment channels and YouTube Live Streaming API. The latter may provide a lot of flexibility in terms of where/how we include the videos (i.e., the workaround of trying to have one channel per piece of equipment to get the livestream might not be necessary - but it's worth thinking about whether there are other reasons to separate into multiple channels or keep it in one for now). I suppose the Pi Zero 2W would be the one using the Live Streaming API, starting a new stream every 8 hours or so (or if the stream goes offline for some reason). On HF side, we can also use the Live Streaming API to figure out what the latest video we should pull is (based on title/date/etc.). We can also create playlists and add videos to it programmatically. It doesn't seem like we'd run into limitations on YouTube's end in terms of number of videos, but we could move to a multi-channel approach if that becomes an issue.

## One channel per piece of equipment

To create new brand channels: https://www.youtube.com/channel_switcher, then click "create a channel". On the channel itself, scroll on left sidebar to **Settings**, 

[image]
<br>
[image]

When you invite, you get the following dialogue box:

[image]

One important consideration is that new channels/accounts have certain limitations:

[image]

[image]

Verifying a phone number isn't too big of a deal, and with a valid government ID (snapping a picture), the restrictions are pretty minimal.

## YouTube Live Streaming API

After a long rabbit hole looking into [the Live Streaming API](https://developers.google.com/youtube/v3/live/getting-started), I came upon **https://developers.google.com/youtube/v3/code_samples/code_snippets** 🔥🔥

[image]

[image]

Part of the difficulty of finding this is that I think the YouTube Live Streaming API is technically encapsulated by the YouTube Data API. If we run into issues, https://stackoverflow.com/questions/tagged/youtube-livestreaming-api is probably a good place to ask.

<details><summary>My rabbit hole</summary>
<p>

Aside: From the [Live Streaming API page](https://developers.google.com/youtube/v3/live/getting-started), I happened upon a link to GitHub examples of the YouTube API usage, which even has examples specific to Python (e.g., [create a broadcast and stream](https://github.com/youtube/api-samples/tree/master/python#create-a-broadcast-and-stream)). Some others:
- [`Retrieve a channel's live video streams`](https://github.com/youtube/api-samples/blob/master/python/list_streams.py)
- [`Retrieve a channel's broadcasts`](https://github.com/youtube/api-samples/blob/master/python/list_broadcasts.py)
- [`Create a playlist`](https://github.com/youtube/api-samples/blob/master/python/playlist_updates.py)
- [`Upload a video`](https://github.com/youtube/api-samples/blob/master/python/upload_video.py) (one for uploading video thumbnails, too)

The repo hasn't been updated in ~7 years, and [based on the issues](https://github.com/youtube/api-samples/issues?q=is%3Aissue%20state%3Aopen%20python), many aren't working / are meant for Python 2. However, they might be a good reference point still. 

Hopefully someone puts an update in https://stackoverflow.com/questions/78170151/live-video-using-youtube-live-streaming-api-python.

In terms of general examples, maybe the following: https://developers.google.com/youtube/v3/quickstart/python. At first glance, it looked like there was a more official Python client for Livestream API listed at https://developers.google.com/youtube/v3/live/libraries, but it turned out to go to the [generic top-level API](https://github.com/googleapis/google-api-python-client). However, I stumbled upon [`samples/youtube`](https://github.com/googleapis/google-api-python-client/tree/main/samples/youtube) --> https://developers.google.com/youtube/v3/code_samples#apps-script --> **https://developers.google.com/youtube/v3/code_samples/code_snippets** 🔥🔥

This last one is gold:

[image]

[image]

</p>
</details>

sgbaird, 2025-03-11:
Live Streaming API with an auto-start script on the Pi, with new streams every eight hours, and automatically finding of the latest stream from the orchestrator side (i.e., HF) is probably the way to go. The stream is relatively fragile: https://github.com/AccelerationConsortium/ac-training-lab/issues/161#issuecomment-2714124461

This probably allows us to circumvent needing to stream to the backup URL, and we should be able to have it go to a single YT account.