AccelerationConsortium/ac-dev-lab #81: OpenFlexure Microscope communication reliability (unresponsive on Hugging Face)

sgbaird, 2024-10-17:
I decided to run the script from a terminal instead, and added a print statement to show the receipt of a command. Interestingly, the device is still receiving the command and not throwing errors, but the HF space doesn't receive the message back. If I refresh the space and use the same credentials, then it seems to work OK.

Maybe HF just isn't receiving the message.

sgbaird, 2024-10-17:
@SissiFeng maybe you could take a look at the hugging face spaces code and see where the issue related to messages might be happening?

SissiFeng, 2024-10-17:
I suggest adding several code blocks for future debugging:

1. Add logging at key points to track the sending and receiving of commands.
2. Implement a periodic status check mechanism to ensure the connection between the device and the HF space remains active.
3. Implement a retry mechanism for failed commands:
  import tenacity
  @tenacity.retry(stop=tenacity.stop_after_attempt(3), wait=tenacity.wait_fixed(2))
  def send_command_with_retry(command):
      pass

Additionally, I've encountered this before - consider using websockets for real-time bidirectional communication, which can be more stable.

sgbaird, 2024-10-18:
Sounds good! There could also be some modification needed on the device side. I'm not sure what QoS is set there.

The light mixing demo app has had a fairly high uptime - recently, the issue was mainly on the device side, due to a certificate expiring.

Do you see any major differences in how the MQTT is implemented? https://huggingface.co/spaces/AccelerationConsortium/light-mixing

Could you share a link to your code?

SissiFeng, 2024-10-19:
https://github.com/SissiFeng/openflexure-microscope

I will check what you mentioned above today.

SissiFeng, 2024-10-19:
https://github.com/SissiFeng/openflexure-microscope

My MQTT setup follows the structure from the link above: specifying the QoS level and automatically handling TLS settings and so on. Currently, I'm receiving the following message:
[image]

Maybe my HiveMQ configuration isn't stable enough. Could you please debug it with another account?

sgbaird, 2024-10-19:
Thanks for looking into this! HiveMQ has been reliable in other cases, so I'm not sure it's purely a HiveMQ issue. Maybe we can chat about this on Monday.

https://github.com/SissiFeng/openflexure-microscope

2024-10-19.08.03.04.png (view on web)<https://github.com/user-attachments/assets/a219d345-32f3-4415-be69-288a2b8c299c>

SissiFeng, 2024-10-20:
https://huggingface.co/spaces/SissiFeng/microscope

ac-hardware, 2024-10-26:
SGB:

At least part of the issue is probably the device code that's running on the microscope. Restarting the script got things responsive again on HF. 

It's also running on Python 3.7.3. I think @kenzo-aspuru-takata had trouble getting it installed on a different Python version (something more specific to the openflexure OS perhaps), but maybe that could be affecting things.

Tried again, and observed the behavior:
device:
```
Received command {'command': 'get_pos'}. Executing
Done.
```
No response on HF.

EDIT: Checked the hf logs and it seems that maybe on the hf side the code is being called many times. I'm seeing about 10 repeats of printouts for each button click, which would obviously cause some issues. Perhaps the separate file structure and having multiple imports is leading to this. Maybe worth condensing to a self-contained `app.py` implementation to check.

Here's the device code since I've had trouble finding time to get it uploaded.

```python

#maybe implement security for z axis focus and move

import paho.mqtt.client as mqtt
import json
import time
from queue import Queue
import openflexure_microscope_client

import PIL
from PIL import Image
import os
import math
import shutil
import base64
import io
import random

from my_secrets import HOST, USERNAME, PASSWORD, PORT, MICROSCOPE

time.sleep(30)

m = openflexure_microscope_client.MicroscopeClient("localhost")
rangetopx=20000
rangebottomx=-20000
rangetopy = 20000
rangebottomy =-20000

steps_per_full_camera_movemement_x=3600 #estimate this then slightly undershoot if you are having problems with stitching, these values should work though
steps_per_full_camera_movemement_y=2700

client=mqtt.Client()
client.tls_set()
client.username_pw_set(USERNAME, PASSWORD)

commandq = Queue()

def on_message(client, userdata, message):
    received = json.loads(message.payload.decode("utf-8"))
    if commandq.empty():
        commandq.put(received)

client.on_message=on_message  

client.connect(HOST, port=PORT, keepalive=600, bind_address="")

def image_to_base64_with_metadata(image_path):
    # Open the image and get metadata
    with Image.open(image_path) as img:
        # Create a BytesIO buffer
        with io.BytesIO() as buffer:
[... 134 lines cut ...]
        elif command["command"] == "focus":
            try:
                focus(amount=command["amount"])
            except KeyError as e:
                print(str(e))
                client.publish(microscope+"/return", payload=json.dumps({"command":"focus","complete":False,"error":str(e)}), qos=2, retain=False)
            except Exception as e:
                print(str(e))
                client.publish(microscope+"/return", payload=json.dumps({"command":"focus","complete":False,"error":str(e)}), qos=2, retain=False)
        elif command["command"] == "take_image":
            take_image()
        elif command["command"] == "scan":
            try:
                scan(c1=command["c1"], c2=command["c2"], ov=command["ov"], foc=command["foc"])
            except KeyError as e:
                print(str(e))
                client.publish(microscope+"/return", payload=json.dumps({"command":"scan","complete":False,"error":str(e)}), qos=2, retain=False)
            except Exception as e:
                print(str(e))
                client.publish(microscope+"/return", payload=json.dumps({"command":"scan","complete":False,"error":str(e)}), qos=2, retain=False)
        else:
            print("Command invalid")
            client.publish(microscope+"/return", payload=json.dumps({"command":"INVALID","complete":False}), qos=2, retain=False)
        print("Done.")
```

sgbaird, 2024-10-26:
Probably both need to be reduced to the basics with:

https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.4-hardware-software-communication.html

And 

https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.4.1-onboard-led-temp.html

While using the public test broker:

```
HIVEMQ_USERNAME = "sgbaird"
HIVEMQ_PASSWORD = "D.Pq5gYtejYbU#L"
HIVEMQ_HOST = "248cc294c37642359297f75b7b023374.s2.eu.hivemq.cloud"
```

sgbaird, 2024-12-06:
@SissiFeng could you try setting up tailscale on the RPi 4B from the microscope to make it easier to debug things remotely? See https://docs.pioreactor.com/user-guide/tailscale-remote-access for instructions

SissiFeng, 2024-12-11:
[image]
installed the tailscale in a pi 4b. 

The next issue to fix is that tailscale always inexplicably logs out on the pi, even with sudo privileges on. 

Working on figuring out.

sgbaird, 2024-12-12:
Thanks for the update! Was this directly on the Openflexure Pi 4B? If so, do you think it's worth troubleshooting on a separate Pi with a standard RPi OS (e.g., Ubuntu)?

Have you found any resources online that mention tailscale + RPi?

sgbaird, 2025-02-15:
@kenzo-aspuru-takata could you try consolidating some of the files into app.py at https://huggingface.co/spaces/AccelerationConsortium/microscope_gradio/tree/main?

sgbaird-alt, 2025-03-11:
> The next issue to fix is that tailscale always inexplicably logs out on the pi, even with sudo privileges on.

...

> * for Openflexure Pi self: sometimes break out the connection when start, maybe the conflict between vpn and processing.
> * a separated RPi 4b: works ok, but will log out after a while of waiting.

@linx5o have you run into this issue before while using Tailscale (e.g., with Pioreactor)?

linx5o, 2025-03-11:
> linx5o have you run into this issue before while using Tailscale (e.g., with Pioreactor)? Also, have you set up Tailscale on any other devices? cc @gursi26 (having trouble remembering if you've used tailscale

No I have not. I generally install tailscale manually and then login with a access token (instead of using the script). 

```
sudo systemctl enable --now tailscaled
```

Also use this command to have tailscale run on boot. (Command may be wrong, can't remember off the top of my head, should be something similar)

sgbaird, 2026-09-18:
Note for auditors: the HiveMQ/mosquitto credentials visible in this thread are intentionally public test credentials (see [SECURITY.md → "Intentionally public credentials"](https://github.com/AccelerationConsortium/ac-dev-lab/blob/main/SECURITY.md#intentionally-public-credentials) and the [microcourse explanation](https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.4-hardware-software-communication.html#onboard-led-using-mqtt)).