AccelerationConsortium/ac-dev-lab #84: Chemical compatibility test: Formlab resins vs. KOHaq

programlich, 2024-10-21:
| Sample  | Solution| Temperature (°C) | $m_{t_0}$ (g) | $m_{t_1}$ (g) |
| ------------- | ------------- | --- | --- | --- |
| 1 | KOH 29.9 w% | ~20 | 1.206 | |
| 2 | KOH 29.9 w% | 70 | 1.2112| |

programlich, 2024-10-24:
1. Lets to the 10k rigid cubes with the same dimension, as these fit perfectly into the round neck flasks -> 1cm^3
2. Numbering is perfect
3. Please print directly on the platform -> results in more homogeneous shape
4. Sorry, I don´t get the question
5. Yes, exactly

sgbaird, 2024-10-24:
4. sorry, were samples 1 and 2 printed directly on the platform, or were they ones with a base?

sgbaird, 2024-10-24:
[image]

programlich, 2024-10-24:
Ah, I think they were both not printed on the base, if I remember correctly

sgbaird, 2024-10-24:
All good - probably shouldn't matter, just wanted to double check.

programlich, 2024-10-24:
Chemical compatibility after 51 hours still seems to be reasonable good. No clear degradation could be observed. However, I think it will be a lot more interesting, investigating at 70 or 80°C. Haven't done that so far, because the hotplate did not get the water bath warm enough. Will do that next week for a couple of hours with a flask directly placed on the hotplate.

sgbaird, 2024-10-24:
Sounds good! If it stabilizes, that would be ideal.

programlich, 2024-10-24:
Definitely.
I just figured out that the safety temperature of these hotplates controlles the temperature of the hotplate itself, not the one measured by the thermocouple (it works that way with the ones I use back home). This is why I did not get to the desired 70°C in the water bath. Will repeat the high temperature experiments next week with both, v4 and rigid 10k resin.

programlich, 2024-11-06:
I added a last value of the compatibility test at room temperature. As you can see, over time the mass is slightly increasing, which might either be due to not perfectly washed cubes with remaining KOH on the surface or to some kind of swelling mechanism. Either way, the mass remains more or less constant over time at room temperature. The more interesting tests will start when I do electrochemical tests at elevated temperatures. For now, I think we are good to proceed this way

sgbaird, 2025-01-08:
from https://github.com/AccelerationConsortium/ac-training-lab/issues/104#issuecomment-2548576202

> I inspected the cell carefully today after it was now exposed to KOH at 70°C for several days. I noticed a structure at some parts of the interior of the cell, which I believe has formed during the electrochemical experiments:
> 
> [image]
> 
> These parallel scratches do not look like I introduced them during grinding with fine sandpaper and are only present at the inside of the cell. Surprisingly I can´t find them anywhere at the capillary and gas inlet tube or sample holder which are also exposed to hot KOH. I think I´ll need to keep an eye on this as it might be a sign of degradation. If that´s the case, we might consider adding a protective coating layer to the inside of the cell, like Novolac Epoxy.

sgbaird, 2025-01-08:
We should probably try with [high-temp resin](https://formlabs.com/store/materials/high-temp-resin/), too. @kelvinchow23 have you printed anything with high-T resin yet?

sgbaird, 2025-01-08:
[Chemical Resistant Epoxy Novolac Coating](https://www.epoxy.com/633.aspx): Sodium Hydroxide @ 50% = R-Recommended for continuous service. Might be worth trying to apply the coating prior to curing the resin to increase the adhesion.

see also https://chatgpt.com/share/677eb6e9-bdd0-8006-b804-642a59f8b47f

kelvinchow23, 2025-01-08:
> We should probably try with [high-temp resin](https://formlabs.com/store/materials/high-temp-resin/), too. @kelvinchow23 have you printed anything with high-T resin yet?

not yet, no.

programlich, 2025-01-09:
I talked to the formlabs support and the answer was basically, that I need to find a suitable material myself because they don't have experience with these chemical conditions. Applying a Novolac coating might be our best shot for a quick solution. I'll try to order some and play around with it. If it works fine, we might need to adjust the interior of the cell slightly to allow for easier application of the coating in the future.

programlich, 2025-01-27:
Today I coated the cell with a layer of epoxy. It will now take about 24h of hardening time and then we will see, if I was able to create a consistant layer.
A few things I want to note from the learnings today:
- Removing air bubbles from the epoxy-hardener mix using an ultrasonic bath works well
- Increasing the surface roughness of the cell with some sandpaper would be a good idea for the next try
- As the epoxy is very liquid while not hardened, it would make sense to give it approx 5min of hardening time before applying it to the cell as it would decrease the time where one needs to constantly rotate the cell to prevent accumulation of all epoxy at the bottom side

[image]

[image]

programlich, 2025-01-29:
The epoxy is now completely hardened and has formed a continuous layer inside the cell. So far the test was successfull. You cant't really see that on the photo so I guess, you'll have to believe me.
Next, I'll check it's behaviour when the cell is filled with hot KOH. However, due to some internal processes I probably can't do that within the next 1.5 weeks :/. Sorry about the delay.

[image]

[image]

sgbaird, 2025-02-14:
Any update on the hot KOH test?

programlich, 2025-02-14:
As commented in the other issue, I was on sick leave for quite some time. As I am finally recovering I hope to run tests in the coming week