AccelerationConsortium/ac-dev-lab #86: assignement 1.5 creation error

programlich, 2024-10-21:
Container for codespaces seems to have a problem. Here is the log:

```
2024-10-21 21:14:42.013Z: Host information
2024-10-21 21:14:42.013Z: ----------------
2024-10-21 21:14:42.013Z: OS: Ubuntu 22.04.5 LTS (stable release)
2024-10-21 21:14:42.014Z: Image details: https://github.com/github/codespaces-host-images/blob/main/README.md
2024-10-21 21:14:42.014Z: ----------------

=================================================================================
2024-10-21 21:14:42.014Z: Configuration starting...
2024-10-21 21:14:42.064Z: Cloning...

=================================================================================
2024-10-21 21:14:53.618Z: Creating container...
2024-10-21 21:14:53.670Z: $ devcontainer up --id-label Type=codespaces --workspace-folder /var/lib/docker/codespacemount/workspace/5-data-logging-programlich --mount type=bind,source=/.codespaces/agent/mount/cache,target=/vscode --user-data-folder /var/lib/docker/codespacemount/.persistedshare --container-data-folder .vscode-remote/data/Machine --container-system-data-folder /var/vscode-remote --log-level trace --log-format json --update-remote-user-uid-default never --mount-workspace-git-root false --omit-config-remote-env-from-metadata --skip-non-blocking-commands --skip-post-create --config "/var/lib/docker/codespacemount/workspace/5-data-logging-programlich/.devcontainer/devcontainer.json" --override-config /root/.codespaces/shared/merged_devcontainer.json --default-user-env-probe loginInteractiveShell --container-session-data-folder /workspaces/.codespaces/.persistedshare/devcontainers-cli/cache --secrets-file /root/.codespaces/shared/user-secrets-envs.json
2024-10-21 21:14:53.940Z: @devcontainers/cli 0.68.0. Node.js v18.20.4. linux 6.5.0-1025-azure x64.
2024-10-21 21:14:54.464Z: $ docker buildx build --load --build-arg BUILDKIT_INLINE_CACHE=1 -f /tmp/devcontainercli-root/container-features/0.68.0-1729545294458/Dockerfile-with-features -t vsc-5-data-logging-********-1b9190e5706a626c69b91b7bc80dd9f4008f4877c6cdd2388f9a08af6cdc6247 --target dev_containers_target_stage --build-arg VARIANT=3.10-bullseye --build-arg NODE_VERSION=none --build-arg _DEV_CONTAINERS_BASE_IMAGE=dev_container_auto_added_stage_label /var/lib/docker/codespacemount/workspace/5-data-logging-********
2024-10-21 21:14:58.174Z: ERROR: failed to receive status: rpc error: code = Canceled desc = context canceled
2024-10-21 21:14:58.181Z: {"outcome":"error","message":"Command failed: docker buildx build --load --build-arg BUILDKIT_INLINE_CACHE=1 -f /tmp/devcontainercli-root/container-features/0.68.0-1729545294458/Dockerfile-with-features -t vsc-5-data-logging-programlich-1b9190e5706a626c69b91b7bc80dd9f4008f4877c6cdd2388f9a08af6cdc6247 --target dev_containers_target_stage --build-arg VARIANT=3.10-bullseye --build-arg NODE_VERSION=none --build-arg _DEV_CONTAINERS_BASE_IMAGE=dev_container_auto_added_stage_label /var/lib/docker/codespacemount/workspace/5-data-logging-programlich","description":"An error occurred building the image."}
2024-10-21 21:14:58.181Z: Stop: Run: docker buildx build --load --build-arg BUILDKIT_INLINE_CACHE=1 -f /tmp/devcontainercli-root/container-features/0.68.0-1729545294458/Dockerfile-with-features -t vsc-5-data-logging-********-1b9190e5706a626c69b91b7bc80dd9f4008f4877c6cdd2388f9a08af6cdc6247 --target dev_containers_target_stage --build-arg VARIANT=3.10-bullseye --build-arg NODE_VERSION=none --build-arg _DEV_CONTAINERS_BASE_IMAGE=dev_container_auto_added_stage_label /var/lib/docker/codespacemount/workspace/5-data-logging-********
2024-10-21 21:14:58.183Z: Error: Command failed: docker buildx build --load --build-arg BUILDKIT_INLINE_CACHE=1 -f /tmp/devcontainercli-root/container-features/0.68.0-1729545294458/Dockerfile-with-features -t vsc-5-data-logging-programlich-1b9190e5706a626c69b91b7bc80dd9f4008f4877c6cdd2388f9a08af6cdc6247 --target dev_containers_target_stage --build-arg VARIANT=3.10-bullseye --build-arg NODE_VERSION=none --build-arg _DEV_CONTAINERS_BASE_IMAGE=dev_container_auto_added_stage_label /var/lib/docker/codespacemount/workspace/5-data-logging-programlich
2024-10-21 21:14:58.183Z:     at FtA (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:466:1933)
2024-10-21 21:14:58.183Z:     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
2024-10-21 21:14:58.183Z:     at async Pm (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:465:1896)
2024-10-21 21:14:58.183Z:     at async NH (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:465:610)
2024-10-21 21:14:58.184Z:     at async YtA (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:482:3771)
2024-10-21 21:14:58.184Z:     at async eB (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:482:4886)
2024-10-21 21:14:58.184Z:     at async prA (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:663:200)
2024-10-21 21:14:58.185Z:     at async drA (/.codespaces/agent/bin/node_modules/@devcontainers/cli/dist/spec-node/devContainersSpecCLI.js:662:14706)
2024-10-21 21:14:58.196Z: devcontainer process exited with exit code 1

====================================== ERROR ====================================
2024-10-21 21:14:58.202Z: Failed to create container.
=================================================================================
2024-10-21 21:14:58.202Z: Error: Command failed: docker buildx build --load --build-arg BUILDKIT_INLINE_CACHE=1 -f /tmp/devcontainercli-root/container-features/0.68.0-1729545294458/Dockerfile-with-features -t vsc-5-data-logging-programlich-1b9190e5706a626c69b91b7bc80dd9f4008f4877c6cdd2388f9a08af6cdc6247 --target dev_containers_target_stage --build-arg VARIANT=3.10-bullseye --build-arg NODE_VERSION=none --build-arg _DEV_CONTAINERS_BASE_IMAGE=dev_container_auto_added_stage_label /var/lib/docker/codespacemount/workspace/5-data-logging-programlich
2024-10-21 21:14:58.205Z: Error code: 1302 (UnifiedContainersErrorFatalCreatingContainer)

====================================== ERROR ====================================
2024-10-21 21:14:58.221Z: Container creation failed.
=================================================================================
2024-10-21 21:14:58.241Z: 

===================================== WARNING ===================================
2024-10-21 21:14:58.241Z: Creating recovery container.
=================================================================================

=================================================================================
2024-10-21 21:16:48.009Z: Creating container...
2024-10-21 21:16:48.032Z: $ devcontainer up --id-label Type=codespaces --workspace-folder /var/lib/docker/codespacemount/workspace/5-data-logging-programlich --mount type=bind,source=/.codespaces/agent/mount/cache,target=/vscode --user-data-folder /var/lib/docker/codespacemount/.persistedshare --container-data-folder .vscode-remote/data/Machine --container-system-data-folder /var/vscode-remote --log-level trace --log-format json --update-remote-user-uid-default never --mount-workspace-git-root false --omit-config-remote-env-from-metadata --skip-non-blocking-commands --skip-post-create --config "/var/lib/docker/codespacemount/workspace/5-data-logging-programlich/.devcontainer/devcontainer.json" --override-config /root/.codespaces/shared/merged_devcontainer.json --default-user-env-probe loginInteractiveShell --container-session-data-folder /workspaces/.codespaces/.persistedshare/devcontainers-cli/cache --secrets-file /root/.codespaces/shared/user-secrets-envs.json
2024-10-21 21:16:48.213Z: @devcontainers/cli 0.68.0. Node.js v18.20.4. linux 6.5.0-1025-azure x64.
2024-10-21 21:16:48.587Z: $alpine -c echo Container started
2024-10-21 21:16:48.620Z: Unable to find image 'mcr.microsoft.com/devcontainers/base:alpine' locally
2024-10-21 21:16:48.750Z: alpine: 2024-10-21 21:16:48.751Z: Pulling from devcontainers/base
2024-10-21 21:16:48.793Z: 
[1A2024-10-21 21:16:48.794Z: [2K
43c4264eed91: Pulling fs layer 
[1B
[1A[2K
f40462439694: Pulling fs layer 
[1B
[1A[2K
78f82a04246d: 2024-10-21 21:16:48.798Z: Pulling fs layer 
[1B
[1A[2K
ab30627c2e5e: Pulling fs layer 
[1B
[1A[2K
1c6e677dc7f6: Pulling fs layer 
[1B
[... 256 lines cut ...]
43d7c7b4c7f6: Extracting  41.29MB/43.39MB
[... cut ...]
2024-10-21 21:17:14.410Z: @devcontainers/cli 0.68.0. Node.js v18.20.4. linux 6.5.0-1025-azure x64.
2024-10-21 21:17:14.633Z: Outcome: success User: vscode WorkspaceFolder: /workspaces/5-data-logging-programlich
2024-10-21 21:17:14.643Z: devcontainer process exited with exit code 0

=================================================================================
2024-10-21 21:17:14.650Z: Configuring codespace...

=================================================================================
2024-10-21 21:17:14.714Z: Finished configuring codespace.

```

programlich, 2024-10-21:
[image]

SissiFeng, 2024-10-21:
If the assignment is done directly in codespace, then the devcontainer.json file needs to be modified by adding information such as your aws api. Indeed, operating directly in codespace may lead to configuration issues sometimes.

A simpler solution: you can try to clone the job locally, edit and test the code locally, which avoids the container error problem. When you're done, you can push the modified code back to GitHub and run or check it out in codespace.

programlich, 2024-10-22:
I did develop the code locally now and to me it seemed to work all fine. Communication between the Pi and the mongodb, my laptop and the mongodb respectively worked out fine via the AWS Gateway and the Lampda function. However, I could not run the tests in codespace, as pytest itself seems to fail.

sgbaird, 2024-10-22:
There have been some issues with the devcontainers. We'll need to check it out on our side. Thanks for the logs!