AccelerationConsortium/ac-dev-lab #87: Test out AS7341 sensor in liquid color matching

Neil-YL, 2024-10-23:
Currently, we have the charging port and sensor package assembled for the liquid color matching demo. It is necessary to test the performance of the AS7341 color sensor for detecting liquid dye in a well plate within the packaging.

I will start by using the code from the micro-course LED color sensor demo for testing.

Neil-YL, 2024-10-23:
Schematic of Picowbell:
https://learn.adafruit.com/adafruit-proto-under-plate-picowbell/downloads

Raspberry Pi Pico W Pinout:
[image]

SCL - GPIO5 - Pin(7)
SDA - GPIO4 - Pin(6)

By charging the i2c to scl=Pin(7),sda=Pin(6), it should be able to set up the connection of AS7341.

I used a script to see if I can obtain data from the sensor (along with the as7341.py and as7341_sensor.py from the micro-course demo)

```
from machine import I2C, Pin
from as7341_sensor import Sensor  

def test_as7341():
    try:
        
        i2c = I2C(1, scl=Pin(7), sda=Pin(6))  # Use GPIO 5 for SCL and GPIO 4 for SDA
        
        # Pass the i2c object to the Sensor class
        sensor = Sensor(atime=100, astep=999, gain=128, i2c=i2c)

        # Retrieve and print all channel data
        channel_data = sensor.all_channels
        print("Spectral Data (F1 to F8):", channel_data)

        # Retrieve and print all channels along with CLR and NIR data
        channel_clr_nir_data = sensor.all_channels_clr_nir
        print("Spectral Data (F1 to F8, CLR, NIR):", channel_clr_nir_data)

        # Optionally disable the sensor at the end of the test
        sensor.disable()
        print("Sensor disabled.")

    except Exception as e:
        print(f"An error occurred: {e}")

# Run the test
test_as7341()
```

However it returned:

> Detected devices at I2C-addresses: 
> I2C read_byte at 0xA9, error [Errno 5] EIO
> I2C write_byte at 0xA9, error [Errno 5] EIO
> I2C write_byte at 0x70, error [Errno 5] EIO
> I2C read_byte at 0xA9, error [Errno 5] EIO
> I2C write_byte at 0xA9, error [Errno 5] EIO
> I2C write_byte at 0x80, error [Errno 5] EIO
> I2C write_byte at 0x80, error [Errno 5] EIO
> I2C read_byte at 0x92, error [Errno 5] EIO
> Failed to contact AS7341 at I2C address 0x39
> An error occurred: Failed to contact AS7341, terminating

It seems the controller is not able to communicate with the AS7341 sensor over the I2C setting.

I run the same test code with my micro-course demo(changing the pin back to 27 and 26) it did set up the connection and obtained data from the sensor.

sgbaird, 2024-10-23:
Are you using the stemma qt port on the picowbell?

Neil-YL, 2024-10-23:
`i2c = I2C(0, scl=Pin(5), sda=Pin(4))  # Use GPIO 5 for SCL and GPIO 4 for SDA`

This is the correct i2c to use the stemma qt port on the Picowbell:

> MPY: soft reboot
Detected devices at I2C-addresses: 0x39
Spectral Data (F1 to F8): [238, 655, 943, 1148, 1953, 2038, 2559, 1464]
Spectral Data (F1 to F8, CLR, NIR): [238, 654, 941, 1145, 1962, 2049, 2571, 1474, 4143, 371]
Sensor disabled.

Neil-YL, 2024-11-29:
I have modified the wrapped class for AS7341 (AS7341_sensor.py) to add control for the LED:

```
    @property
    def LED(self):
        """Get the current state of the onboard LED."""
        return self._led_state

    @LED.setter
    def LED(self, state):
        """Set the LED state."""
        if state:
            self.sensor.set_led_current(10)  # Use set_led_current method in AS7341.py
            self._led_state = True
        else:
            self.sensor.set_led_current(0)
            self._led_state = False
```

Video:

https://github.com/user-attachments/assets/3890448a-31f3-4ff5-9b1f-8d799147370b

Neil-YL, 2024-12-05:
Updates for this week:

I have a short script to test the sensor: have one measurement with command "read" and "yes or no" to turn on or off LED light on the sensor during the measurement. 

<details>
<summary> Click here for sensor test script</summary>

```
import time
from machine import I2C, Pin
from as7341_sensor import Sensor

# Initialize the AS7341 sensor
sensor = Sensor(i2c=I2C(0, scl=Pin(5), sda=Pin(4)))

def read_sensor_data():
    """Read dictionary of sensor data"""
    # Get all channel data from the sensor
    channel_data = sensor.all_channels

    CHANNEL_NAMES = [
        "ch410",
        "ch440",
        "ch470",
        "ch510",
        "ch550",
        "ch583",
        "ch620",
        "ch670",
    ]

    # Return a dictionary that maps channel names to sensor data
    return dict(zip(CHANNEL_NAMES, channel_data))

print("Testing sensor...")
sensor_data = read_sensor_data()
print(sensor_data)

def test_sensor():
    """Interactive testing interface for the AS7341 sensor."""
    print("AS7341 Sensor Test")
    print("==================")
    print("Type 'read' to sample data or 'exit' to quit.")

    while True:
        # Wait for a command from the user
        command = input("Command: ").strip().lower()

        if command == "read":
            # Ask whether to turn on the LED
            led_choice = input("Turn on LED for this reading? (yes/no): ").strip().lower()
            if led_choice == "yes":
                sensor.LED = True  # Turn on LED
                print("LED is ON.")

            # Trigger sensor data reading
            sensor_data = read_sensor_data()
            if sensor_data:
                print("Sensor Data:")
                for wavelength, value in sensor_data.items():
                    #print(f"{wavelength}: {value}")
                    print(f"{value}")
            else:
                print("Failed to read sensor data.")

            # Turn off the LED if it was turned on
            if led_choice == "yes":
                sensor.LED = False  # Turn off LED
                print("LED is OFF.")

        elif command == "exit":
            print("Exiting the program.")
            break
        else:
            print("Invalid command. Type 'read' to sample or 'exit' to quit.")

# Directly call the test_sensor function
test_sensor()

```
</details>

I prepared some colors mixture for several trial test, each well slot has a volume of 280μL.

[image]

A1:  70% Red, 10% Yellow, 20% Blue
A2:  20% Red, 10% Yellow, 70% Blue
A3: 35% Red, 5% Yellow, 10% Blue, 50% water (50% dilution of A1)
A4: 10% Red, 5% Yellow, 35% Blue, 50% water (50% dilution of A2)
A5: 100% Yellow
A6: 100% Blue
A7: 100% Red

Followings are findings after several measurements: 

- LED on, readings at a range of 10k-20k,  except for wavelength 410, which is around 2k.
- LED off, readings at a range of 100-400, expect for wavelength 410 around 20 and 440 around 40-60. 
- LED on or off, the readings are very stable and repeatable for each color (Errors: With LED off, most of channels are 0, ±2 for maximum channel, and ±10 with LED on).
- LED off, A1 and A2 are distinguishable with 5% difference, and A3 and A4 are not very distinguishable (most of channels less than 1% difference, less than repeat measurement errors )
- LED on,  A1 A2 A3 A4 are not distinguishable with difference less than repeat measurement errors 
- LED on, YBR are not very distinguishable

You may refer to this [excel file](https://github.com/user-attachments/files/18028922/color.test.comparison.xlsx) for detail results.

I think we will not use the LED for the measurement, so may not need the LED control for the test code (since we need to update the sensor wrapped class as well)

For Prefect on OT-2, I think we do not need to install Prefect or modify any codes on OT-2. Just simply include SSH tasks in the Prefect workflow to connect to the OT-2 and then to execute Opentrons protocol code on the robot.

I also found some codes from Jackie's demo codes, have valuables regarding to Neopixels, which is not necessary in liquid matching. Although some of them could be necessary valuables but with previous code valuables naming. Do you think I should put some efforts to modify them or just leave them as before (I think if we choose to use the same naming we may not need to modify the code for the MQTT from hello world course?)

Since my Prefect courses are to held in Dec 17-18 so I think I will finish the documentation for the rest of the software (sensor, PicoW MQTT, OT-2 setup reference, etc) before that days and try to finish the Prefect part after the courses.

sgbaird, 2025-02-25:
Anya update required here you think? Or should we close? I know we moved to acrylic paint. Not sure if any of that should go here.

Neil-YL, 2025-02-26:
> Anya update required here you think? Or should we close? I know we moved to acrylic paint. Not sure if any of that should go here.

No sure if you want to add the background measurement and white calibration to the liquid color matching (for long term deployment), but we could move to the other issue for discussion so we can close this one.

sgbaird, 2025-02-26:
I think it depends on how things are going with the optimization. If students are able to obtain meaningful results during the optimization campaign, adding the background measurement can wait (but I think should still be a priority for later). If the noise is so great that the students feel like they're not really optimizing anything, then I think adding the background would be good. Maybe in-between assignments we add this? Workflow wise it's "move sensor to the well prior to dispensing, take spectrum, put sensor back, <rest of workflow>".

sgbaird, 2025-02-26:
Porting over some of the discussion around light sensor measurements of the liquids from https://github.com/AccelerationConsortium/ac-training-lab/issues/152#issuecomment-2599087735 to here (specifically about the switch from food coloring to dilute acrylic paint).

> > * let's use washable, non-toxic paint instead of food coloring. Can you find something that looks like it would work from Amazon and include the link here?
> 
> [RYBG paint](https://www.amazon.ca/Melissa-Doug-Poster-Paint-Crafts-Supplies/dp/B072QFSBRC/ref=sxin_15_pa_sp_search_thematic_sspa?content-id=amzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24%3Aamzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24&cv_ct_cx=washable%2C+non-toxic+paint&keywords=washable%2C+non-toxic+paint&pd_rd_i=B072QFSBRC&pd_rd_r=45adc370-c22c-4000-9ce7-fa3e3724bdd2&pd_rd_w=kf8Ge&pd_rd_wg=wvsbb&pf_rd_p=46621be6-fabe-4126-8501-d32c96c42a24&pf_rd_r=BBAT6MT9SE4PB8GPTNBN&qid=1737141917&sbo=RZvfv%2F%2FHxDF%2BO5021pAnSA%3D%3D&sr=1-4-acb80629-ce74-4cc5-9423-11e8801573fb-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9zZWFyY2hfdGhlbWF0aWM&psc=1): no sure what kind of paint is it
> 
> [Tempera Paint](https://www.amazon.ca/Marte-Tempera-Paint-Bottles-Washable/dp/B0DBVH2LNB/ref=sr_1_32?dib=eyJ2IjoiMSJ9.-kpuh5nDMeAa0FUfIRJlxQj2-iH4vYSO8JIwP5ZtjltzS-I9YBHUZ4SH6-cl-0jMPDqWEVw_3Hc2zhC1ksF0k5S1i7O7PbChfHxWs_TgQbWnR7iFXxQi914ld0KYcGRzerrfKj80BkRN_8-UQlKQCIKHoGTqV-HxoMpvmds906xskgKGg3Gpdj0xRcrxzTYeKP1dTS63p1J78OiETFbGJfTR11mJDsXNNMCzw7iB2E8junCroaWdE94fPRS1y8bibHrK_sg5Qg1_JISSyxRRfDgseH8mQmTSjJavP_W3oy1gpzbscVABzBzxHZF6VfaIBXD_SzlLiNXMSbC32iLN0YvOVVNwm9jCqzYjEN7MogE_nJwA4BbFnMUqh113zYv2XCEZEx7Y0lCiACgBwOOYem7wlMKu1CSe2WukKlDekvW4F3YyfPzrthZKW8mLBii2.Sf_jRWkxrHdV9IG_TCEq8rZDUTFnegJrhL3douIlPm4&dib_tag=se&keywords=Prang&qid=1737143030&sr=8-32): Tempera is water-based paint and the thickness can be adjusted (just like food coloring), but could become solid leftover in pipette tip if fully dried out.
> 
> There are options of tempera RYB color in single bottle as well: [red](https://www.amazon.ca/Prang-Washable-Tempera-16-Ounce-10701/dp/B000QE2WDS/ref=sr_1_26?dib=eyJ2IjoiMSJ9.-kpuh5nDMeAa0FUfIRJlxQj2-iH4vYSO8JIwP5ZtjltzS-I9YBHUZ4SH6-cl-0jMPDqWEVw_3Hc2zhC1ksF0k5S1i7O7PbChfHxWs_TgQbWnR7iFXxQi914ld0KYcGRzerrfKj80BkRN_8-UQlKQCPOV8pcXwEnXjaCxzxb-Ic0tauYFrS9fBQShIsVAr7FrKP1dTS63p1J78OiETFbGJfTR11mJDsXNNMCzw7iB2E8junCroaWdE94fPRS1y8bibHrK_sg5Qg1_JISSyxRRfDgseH8mQmTSjJavP_W3oy1gpzbscVABzBzxHZF6VfaIBXD_SzlLiNXMSbC32iLN0YvOVVNwm9jCqzYjEN7MogF8XlFbPAxMk9jGGW7CoBiZ7TqA-GgbDwDtwmL4-eN45SudMdGt4UyV3KL8kNYbHSi4F3YyfPzrthZKW8mLBii2.-OKC67K1SG25yPy3wzFTVNlGiwrntFR6oY0ZcAMHOJk&dib_tag=se&keywords=Prang&qid=1737143397&sr=8-26), [yellow](https://www.amazon.ca/Prang-Washable-Tempera-16-Ounce-10703/dp/B0017DBCKG/ref=sr_1_17?dib=eyJ2IjoiMSJ9.-kpuh5nDMeAa0FUfIRJlxQj2-iH4vYSO8JIwP5ZtjltzS-I9YBHUZ4SH6-cl-0jMPDqWEVw_3Hc2zhC1ksF0k5S1i7O7PbChfHxWs_TgQbWT7H6A2yIw9wmwcJIXHoMcJBH4BSomMPSt7bTsj1mtK_OV8pcXwEnXjaCxzxb-Ic0tauYFrS9fBQShIsVAr7FrKP1dTS63p1J78OiETFbGJfTR11mJDsXNNMCzw7iB2E8junCroaWdE94fPRS1y8bibHrK_sg5Qg1_JISSyxRRfDgseH8mQmTSjJavP_W3oy1gpzbscVABzBzxHZF6VfaIBXD_SzlLiNXMSbC32iLN0YvOVVNwm9jCqzYjEN7MogF8XlFbPAxMk9jGGW7CoBiZ7TqA-GgbDwDtwmL4-eN45SudMdGt4UyV3KL8kNYbHSi4F3YyfPzrthZKW8mLBii2.FMxZXu8fjhNY-A1yKC26rqBpILPwnxT6XD-Rpri_5HE&dib_tag=se&keywords=Prang&qid=1737143257&sr=8-17) [blue](https://www.amazon.ca/Prang-Washable-32-Ounce-Turquoise-10912/dp/B000QSEKZM/ref=pd_ci_mcx_di_int_sccai_cn_d_sccl_3_11/140-9582717-8730642?pd_rd_w=Dyk9J&content-id=amzn1.sym.d6674fdf-bd00-4d07-8317-6dfd6c498cdf&pf_rd_p=d6674fdf-bd00-4d07-8317-6dfd6c498cdf&pf_rd_r=YX9Z727XJDXQST965Y4C&pd_rd_wg=mBxuX&pd_rd_r=571f6d71-8e80-4a99-8c74-c5841f2d242d&pd_rd_i=B000QSEKZM&psc=1) (cannot find a 16oz blue on amazon, this is a 32oz one)

---

> We've had good luck with acrylic paint thinned in water, probably a 10:1 water:paint ratio, if not weaker. Tempura paint is probably equivalent - I think we just have whatever was on the shelf at a craft store. We've generally used opaque colored liquids to mix. You can also use food coloring + milk in a pinch, but then you need to carefully wash everything or it starts to smell.

---

> I went ahead and ordered some acrylic and some washable. Thanks, both!
> 
> Acrylic: https://a.co/d/5aOw6DW Washable: https://a.co/d/35i7Ost
> 
> Should arrive Monday.

---

> > I think we can just stick with this for now, and we'll worry about having a per-student budget/restriction a bit later. I need to give this some more thought.
> 
> Currently I have a MongoDB to log-in the data of well-plate status (empty/used) so I think I can also mark down the student_id for the budget consumption purpose, but we may need students sign up for their ID first.
> 
> Update for the colors:
> 
> The Crayola washable paint is a little bit too viscous to use in our current set up (given the size of pipette tip and well volume), resulting in very visible layering of colors after mixing. I turned to acrylic paint and it mixed better. <img alt="Image" width="953" src="https://private-user-images.githubusercontent.com/179746567/404989395-999e25d6-01e0-4358-8c12-01fb8c1e1cb3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDA1OTY0NDEsIm5iZiI6MTc0MDU5NjE0MSwicGF0aCI6Ii8xNzk3NDY1NjcvNDA0OTg5Mzk1LTk5OWUyNWQ2LTAxZTAtNDM1OC04YzEyLTAxZmI4YzFlMWNiMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjI2JTJGdXMtZWFz [... cut ...]
> The A-row data were eliminated because it has a large variation compared to other rows. I marked all A-row wells as "used" on MongoDB so Hugging face will skip A row for the students practical.

---

> With the light mixing demos, I typically take a background with all the LEDs off. I suppose the equivalent in this case is to have a step at the beginning of the workflow that captures the spectrum of the empty well before dispensing liquid into the well (i.e., as if you put 0,0,0 for the RYB liquid amounts).