AccelerationConsortium/ac-dev-lab #98: ML for electrolysis life cycle testing

sgbaird, 2024-10-30:
Some initial discussions

[image]

sgbaird, 2024-10-30:
@programlich

sgbaird, 2024-10-31:
Maybe something interesting here onc it processes:

https://hasanyone.com/?id=53ee7bc7

sgbaird, 2024-10-31:
Likewise:

https://hasanyone.com/?id=33ee89fc

programlich, 2024-11-01:
Pretty cool! Haven't heared about this service before.

I think the first point is interesting, because it leaves a lot of room for filling this research gap.
Regarding the second question, there are probably quite a few AST protocols out there (however, most of them for experiments on cell level, rather then in basic 3 electrode setup, if I'm informed correctly). However in our case, we might be able to create an AST protocol, which not only accelerates the aging of the electrode but also gives insight into the kinds of stress parameters, which are particularly bad for our specific material. This will hopefully provide insight into the degradation mechanism(s) occuring in Raney-Ni coated Ni meshes.

programlich, 2024-11-01:
# Parameters
- Current density $0<j<2 A/cm²$
- Cycle time $1s < t_c < 12h$

Note: A current densitiy of 0 A/cm² is equivalent of OCP. The potentiostat will not actively set a current of 0A 

# Objectives
The arrow indicates if the parameter should be maximized or minimized within this experiment
- Overpotential $\eta \uparrow$
- Tafel slope $b \uparrow$
- Exchange current density $j_{0} \downarrow$
- Double layer capacity $C_{dl} \downarrow$

sgbaird, 2024-11-01:
Part II (a) https://youtu.be/HyBDDhy_0nQ

Part II (b) https://youtu.be/vhXEI5bqvus

[image]

[image]

sgbaird, 2025-01-06:
Do you want to start doing this with your existing setup, or would you want to wait until we have things a bit more locked in with the 3D printed cell? 

Do you have the code?

programlich, 2025-01-09:
With this, I'll need to wait until we have a more automated setup because currently I don't have the ressources to run a large number of experiments with manual setup.

I still have the code we developed together (however unmodified since our last chat). Do you want me to paste it somewhere?

sgbaird, 2025-02-14:
Could you put the code here? (And links to corresponding ChatGPT conversations if applicable). Sounds good about waiting until it's more automated

sgbaird, 2025-02-14:
@programlich would you also say that in the setup you listed above, EIS isn't required as suggested in https://chatgpt.com/share/67afadf9-151c-8006-9937-bb3b94829b56?

(I'm guessing the key here is the use of CV instead of EIS per that article that you mentioned)

sgbaird, 2025-04-04:
> Do you want me to paste it somewhere?

Could you paste this here? (if you have the LLM transcript still, that would be great if you have the link - no worries if not)

programlich, 2025-04-07:
I'll paste the code, that we developed during our online meeting using honegumi. However, I have not touched it since then, so it surely needs some fine tuning. I'd appreciate to talk to you once again about the code to get started with it in the coming weeks/months.

```import numpy as np
from ax.service.ax_client import AxClient, ObjectiveProperties
from ax.modelbridge.factory import Models
from ax.modelbridge.generation_strategy import GenerationStep, GenerationStrategy
import pandas as pd

obj1_name = "eta"
obj2_name = "b"
obj3_name = "j0"
obj4_name = "Cdl"

def branin3_moo(w1, h1, w2, h2, w3, h3, w4, h4, w5, h5):
    # Custom objective functions using all 10 variables
    y = float(
        (h2 - 5.1 / (4 * np.pi**2) * w1**2 + 5.0 / np.pi * w1 - 6.0) ** 2
        + 10 * (1 - 1.0 / (8 * np.pi)) * np.cos(w1)
        + 10
    )
    y = y * (1 + 0.1 * w1 * h1 * w2 * h2 * w3 * h3)

    y2 = float(
        (w2 - 5.1 / (4 * np.pi**2) * h1**2 + 5.0 / np.pi * h1 - 6.0) ** 2
        + 10 * (1 - 1.0 / (8 * np.pi)) * np.cos(h2)
        + 10
    )
    y2 = y2 * (1 - 0.1 * w4 * h4 * w5 * h5)

    return {obj1_name: y, obj2_name: y2, obj3_name: y2, obj4_name: y}

# Define total for the compositional constraint
total = 12.0

# Define the training data
# Add a row per run experiment here
X_train = pd.DataFrame(
    [
        {"w1": 1.0, "h1": 2.0, "w2": 1.5, "h2": 2.5, "w3": 3.0, "h3": 1.0, "w4": 2.0, "h4": 0.5, "w5": 0.5, "h5": 1.5},
        {"w1": 0.5, "h1": 1.5, "w2": 2.0, "h2": 3.0, "w3": 1.0, "h3": 2.0, "w4": 4.0, "h4": 1.5, "w5": 2.5, "h5": 1.0},
        {"w1": 2.0, "h1": 3.5, "w2": 0.5, "h2": 1.5, "w3": 1.5, "h3": 2.5, "w4": 2.5, "h4": 0.5, "w5": 3.0, "h5": 1.5},
        {"w1": 1.5, "h1": 0.5, "w2": 3.0, "h2": 1.0, "w3": 2.0, "h3": 3.0, "w4": 1.0, "h4": 2.5, "w5": 0.5, "h5": 1.0},
        {"w1": 0.0, "h1": 2.0, "w2": 4.0, "h2": 1.5, "w3": 2.5, "h3": 0.5, "w4": 3.5, "h4": 1.0, "w5": 1.0, "h5": 2.5},
    ]
)

# Calculate y_train values
y_train = [branin3_moo(**row) for _, row in X_train.iterrows()]

n_train = len(X_train)

# Generation Strategy
gs = GenerationStrategy(
    steps=[
        GenerationStep(
            model=Models.SOBOL,
            num_trials=6,
            min_trials_observed=3,
            max_parallelism=5,
            model_kwargs={"seed": 999},
            model_gen_kwargs={},
        ),
        GenerationStep(
            model=Models.FULLYBAYESIANMOO,
            num_trials=-1,
            max_parallelism=3,
            model_kwargs={},
        ),
    ]
)

# Initialize AxClient
ax_client = AxClient(generation_strategy=gs)

min_current_density = -2.0

# Create experiment
ax_client.create_experiment(
    parameters=[
        {"name": "w1", "type": "range", "bounds": [0.0, total]},
        {"name": "h1", "type": "range", "bounds": [min_current_density, 0.0]},
        {"name": "w2", "type": "range", "bounds": [0.0, total]},
        {"name": "h2", "type": "range", "bounds": [min_current_density, 0.0]},
        {"name": "w3", "type": "range", "bounds": [0.0, total]},
        {"name": "h3", "type": "range", "bounds": [min_current_density, 0.0]},
        {"name": "w4", "type": "range", "bounds": [0.0, total]},
        {"name": "h4", "type": "range", "bounds": [min_current_density, 0.0]},
        {"name": "w5", "type": "range", "bounds": [0.0, total]},
        {"name": "h5", "type": "range", "bounds": [min_current_density, 0.0]},
    ],
    objectives={
        obj1_name: ObjectiveProperties(minimize=False),
        obj2_name: ObjectiveProperties(minimize=False),
        obj3_name: ObjectiveProperties(minimize=True),
        obj4_name: ObjectiveProperties(minimize=True),

    },
    parameter_constraints=[
        f"w1 + w2 + w3 + w4 <= {total}",  # Updated compositional constraint
    ],
)

# Add existing data to AxClient
for i in range(n_train):
    parameterization = X_train.iloc[i].to_dict()

    # Remove parameters that aren't part of the search space
    parameterization.pop("w5")

    ax_client.attach_trial(parameterization)
    ax_client.complete_trial(trial_index=i, raw_data=y_train[i])

# Run optimization trials
#for _ in range(21):

parameterization, trial_index = ax_client.get_next_trial()

# Extract parameters
w1 = parameterization["w1"]
w2 = parameterization["w2"]
w3 = parameterization["w3"]
w4 = parameterization["w4"]
# Derive w5 and h5 based on the composition constraint
w5 = total - (w1 + w2 + w3 + w4)

print(parameterization, w5)

# Run my experiment here
#results = branin3_moo(w1, parameterization["h1"], w2, parameterization["h2"], w3, parameterization["h3"], w4, parameterization["h4"], w5, parameterization["h5"])

#ax_client.complete_trial(trial_index=trial_index, raw_data=results)

# Get Pareto-optimal results
#pareto_results = ax_client.get_pareto_optimal_parameters()
```

sgbaird, 2025-05-14:
I guess we can leave this issue in ac-training-lab for now, since the echem-cell repo might be more specific to the design rather than the usage for AST? Happy to move it over if you prefer

programlich, 2025-05-15:
I'll accept whatever works best in your opinion. I guess, the issue is fine in any of the three repos