AccelerationConsortium/ac-dev-lab #99: Front left corner of Bambu lab printer leading to print failures

sgbaird, 2024-10-30:
Maybe related to temperature or leveling.

Thanks for raising this @Neil-YL

Neil-YL, 2024-10-30:
[image]

sgbaird, 2024-10-30:
Maybe post on or search Bambu lab forums at some point?

Neil-YL, 2024-10-31:
> Maybe post on or search Bambu lab forums at some point?

I was told Bambu‘s got an Ai chatbot for troubleshooting, may also give it a try tomorrow

Neil-YL, 2024-10-31:
I asked the AI chatbot about the heat bed adhesion troubleshooting and followed [this wiki](https://wiki.bambulab.com/en/x1/manual/manual-bed-leveling) for the manual reset of the bed leveling.
The process required running a G-code command provided in the wikipage and manually adjusting three screws beneath the bedplate. After that, I ran the calibration for both the micro lidar and bed leveling from the Bambu studio.
[image]

It seemed that the micro lidar’s calibration board on the right needed cleaning, so I wiped it with a disinfecting wipe. [Guide of cleaning lidar and calibration board ](https://www.youtube.com/watch?v=gtLa2dNL03Q&ab_channel=TryNotToBreakAnything)
This is the micro lidar calibration board:
[image]

I tested with two previously failed print jobs, and the adhesion performance has noticeably improved on the left/top edge:

[image]
[image]

Kevin will have some models printed in the coming days, so we’ll see if the issue is fully resolved.

Neil-YL, 2024-11-04:
Kevin said he had several parts printed without any issues last week. If the printer in SLD2 is experiencing similar adhesion issues (according to @cyrilcaoyang ), it may be worth trying this [manual bed levelling process](https://wiki.bambulab.com/en/x1/manual/manual-bed-leveling) as well.