AccelerationConsortium/ac-dev-lab #104: Custom electrochemical cell (v2) - enhancements and automation

sgbaird, 2024-11-07:
@MiceeNS @programlich, way to go on the achievements in #62. Very excited to continue this with version 2. Perhaps to help orient ourselves, @programlich, could you summarize the current state of the cell, as well as anticipated improvements for v2? Then, in response, @MiceeNS perhaps you could respond with what is already planned and we can get the discussion started from there.

programlich, 2024-11-07:
The current status I would describe as follows: All general requirements are fulfilled, the cell is ready for real world testing. A bunch of possible minor improvements have already been detected and can be tackled withing v2. Before getting into it, I recommend, we wait for my first high temperature test, in case in brings any unexpected results. 

Here is a list of issues and ideas, we can work on in the future:

# Necessary
- Adjust sample holder/position of it so it does not hit the bottom of the cell

# Repairability
- Make the washing nozzle screwable into the cell wall from the inside -> replacement possible
- Screw the barb for the gas inlet into the tap
- Screw the gas inlet tube into the tap? -> Replacement possible

# Electrochemistry
- Turn the inner part of the gas inlet tube by 90° to have the bubbles coming out away from the reference and working electrode. We want as little interference between the electrodes and the inert gas as possible
- Adjust the channel of the wire inside the sample holder, so that it is less curved in the region of contact to the sample. flatter curve = larger contact area = less ohmic resistance
- Add more space between top of the sample and the sample holder, to avoid interaction between gas bubbles and sample
- Make the inner opening of the capillary a bit smaller
- Reduce distance between sample and capillary (if possible)
 
# Automation
- Reposition sample holder and capillary: Right now, the sample holder cannot be inserted with a pure movement in the z axis, because it would hit the capillary. For automation, a straight z movement would be preferrable

# Other improvements
- give the barb for the gas inlet a 90° angle -> more natural connection to the gas tube
- Redesign the holder for the counter electrode or just replace it with a standard rubber stopper
- Think about other kind of connections for the tubes to insert/remove KOH then barbs. Perhaps there are more secure ways to transport KOH
- Add mechanism to separate removed KOH from removed water for washing -> avoid adding lots of water to the KOH waste
- Print a solid block to close the hole of the sample holder during washing protocol when sample holder is outside for the replacement of the sample
- Increase the wall thickness again, to have a more reliable print with less risk of blowouts
- Position the barb for the thermal oil inlet a little bit higher to add space for the tube around the barb
- Adjust the inlet for the reference electrode (RHE), so it is positioned a few mm higher than now, to give easier access to the electrical connection in the RHE

sgbaird, 2024-11-08:
> Adjust sample holder/position of it so it does not hit the bottom of the cell  

maybe best to move the capillary and the counter electrode "up" as well, i.e., keeping the relative positions of the counter electrode and luggin capillary the same.

[image]

> Turn the inner part of the gas inlet tube by 90° to have the bubbles coming out away from the reference and working electrode. We want as little interference between the electrodes and the inert gas as possible

if we make the gas inlet screwable, we need to keep the orientation of the gas inlet tube in mind if we keep the curved design.

Likewise, I wonder if the gas inlet should be embedded within the cell itself rather than into the lid.

[image]

> Adjust the channel of the wire inside the sample holder, so that it is less curved in the region of contact to the sample. Flatter curve = larger contact area = less ohmic resistance  

using a flat, thin, flexible strip of 316 SS and adjusting the design could also make it easy to have a large area of contact. 

> Add more space between top of the sample and the sample holder, to avoid interaction between gas bubbles and sample  

e.g., via a cutout, but we need to keep in mind that if we use a round o-ring, then we'd need to make the entire sample holder radius larger. If we move to a rectangular shape (not sure what we'd do about the seal), then we could make a larger cutout without issue

[image]

(btw, 

> Make the inner opening of the capillary a bit smaller  

Half the diameter? Smaller? What is the internal diameter of the capillary that you use? I just sent an email checking on my side with the other unit.

> Reduce distance between sample and capillary (if possible)  

Do you also want thinner walls for the capillary? (keeping in mind blowout and breakage concerns of course)

> Think about other kinds of connections for the tubes to insert/remove KOH than barbs. Perhaps there are more secure ways to transport KOH  

We also want to keep in mind that any automated handling of concentrated KOH is safe and robust, given how corrosive it is to skin/eyes/etc., especially keeping in mind overpressure situations and tubes or mechanisms wearing out. As you mentioned, it takes one drop of 30% KOH in the eye for complete and unrecoverable blindness (with a window of ~5-10s to wash it out to reduce damage).

Likewise, it would be great if we could have the full system (including automation bits) represented within the Fusion 360 file. For example, specific pumps, with tube sizes / adapters, etc. I think this would help streamline getting everything connected.

> Add mechanism to separate removed KOH from removed water for washing -> avoid adding lots of water to the KOH waste  

a custom insert with two barbs instead of one would work for this.

> Print a solid block to close the hole of the sample holder during washing protocol when sample holder is outside for the replacement of the sample  

The hole below? What's the intention here?

[image]

(aside: @MiceeNS is the hole in the picture below necessary?)

> Increase the wall thickness again, to have a more reliable print with less risk of blowouts  

It looks like current thickness is 4 mm:

[image]

What was the original thickness? Seems we need either the original thickness or somewhere in-between. Even if we're using ~20% more resin, having the bottom of the cell oriented flat against the build platform and reducing the overall risk of blowout may be worth it.

sgbaird, 2024-11-08:
@MiceeNS as an aside, have you been using rollbacks on the timeline when making some of the edits within Fusion 360?

[image]

MiceeNS, 2024-11-08:
Yes, I did

MiceeNS, 2024-11-09:
Sry, I did not saw that you have both written a lot above. Will respond tomorrow

programlich, 2024-11-09:
Thanks for all the ideas, @sgbaird. I´ll respond to them one by one. Also, is there an option to use a kanban board or something similar in github? I think it would be great to have a separate tool to track all the tasks once we have decided on which ones to work.

> Sry, I did not saw that you have both written a lot above. Will respond tomorrow

No worries, it´s all jut brainstorming for now :)

programlich, 2024-11-09:
> maybe best to move the capillary and the counter electrode "up" as well, i.e., keeping the relative positions of the counter electrode and luggin capillary the same.

Yes, I agree. This would also resolve the task of moving the reference electrode slightly upwards to have better access to it´s electrical connection

> Likewise, I wonder if the gas inlet should be embedded within the cell itself rather than into the lid.

This is a wonderful idea. Actually, why not have the connection to the gas source at the side of the cell and then build another channel (like for the outlet of the electrolyte) to the bottom of the cell, so that the gas can bubble all the way up from the bottom to the top? Would simplify the design and increase gas solubility

> using a flat, thin, flexible strip of 316 SS and adjusting the design could also make it easy to have a large area of contact.

True story. If a flat Platinum wire is not available, we could just press/cold roll it to get it flat

> e.g., via a cutout, but we need to keep in mind that if we use a round o-ring, then we'd need to make the entire sample holder radius larger. If we move to a rectangular shape (not sure what we'd do about the seal), then we could make a larger cutout without issue

I´ll leave that to @MiceeNS to decide. Also this task isn´t a very high priority, more a nice to have for even better performance

> Half the diameter? Smaller? What is the internal diameter of the capillary that you use? I just sent an email checking on my side with the other unit.

Need to discuss this with my colleagues. There was a rule of thumb of the diameter of the opening in relation to the distance from the working electrode, which I can´t remember

> Do you also want thinner walls for the capillary? (keeping in mind blowout and breakage concerns of course)

No, thickness is fine. I prefer it the more robust way.

>The hole below? What's the intention here?

I mean the complete opening in the tap for inserting the sample holder. If the process is "take sample out" -> "electrolyte out" -> "wash" -> "fresh electrolyte in" -> "new sample in" the washing would take place with the opening for the sample holder not being closed by anything and water would exit the cell. That´s why we need something to close it during washing.

Thanks for pointing the security issues out. I think this is something we should always have particularly in mind. It´s always preferable to design a bit of extra safety for the use of these chemicals.

programlich, 2024-11-09:
@sgbaird, could you add your idea about the replacement of the o-ring for the thermocouple, with another kind of seal, please?

sgbaird, 2024-11-09:
I've used silicone grommets before. While decent, this won't give us the same chemical resistance as the FEP encapsulated one, but the FEP is pretty rigid..

https://www.mcmaster.com/products/grommets/grommets-2~~/?s=Silicone+Grommet

Just need to identify the correct ID, OD, and material thickness, modifying the design slightly to accommodate the limited selection. If it turns out there are multiple flange sizes for a desired ID/OD/thickness combo (unlikely), larger is maybe better, as long as it doesn't get in the way of other components.

The main thing will be identifying an ID/OD combo that has "good" tolerances - large enough that it can fit, but small enough that it creates a reasonable seal. We will probably want to get a couple within a certain range and test print a few sizes of holes in a "plate" of identical thickness to the lid.

@MiceeNS could you identify three to five grommets with reasonable dimensions and prepare a CAD model "plate" with various hole diameters and the same thickness as the lid?

EDIT: maybe look into "hollow push-in bumpers" in addition to "grommets". Before when I used silicone grommets, I would usually cut off the flange on one side of the grommet to make insertion and removal easier.

However, looks like there's only one size for a silicone push-in bumper:
https://www.mcmaster.com/products/push-in-bumpers/hollow-push-in-bumpers~~/.

EDIT: also came across https://www.mcmaster.com/products/stopper-plugs/hollow-high-temperature-tapered-plugs/. Looks like maybe a thin layer on the bottom needs to be cut out, but is otherwise hollow.

MiceeNS, 2024-11-09:
> > Adjust sample holder/position of it so it does not hit the bottom of the cell
> 
> maybe best to move the capillary and the counter electrode "up" as well, i.e., keeping the relative positions of the counter electrode and luggin capillary the same.

I will adjust the height for both of them.

> if we make the gas inlet screwable, we need to keep the orientation of the gas inlet tube in mind if we keep the curved design.
>
> Likewise, I wonder if the gas inlet should be embedded within the [...]

I think the screwable part would be only the barbed fitting on the lid. Good point for the gas tube to be connected with the cell. We should think of more what is better for automation. Wherever it would be I am accounting the barbed fitting to be with thread at this moment.

> using a flat, thin, flexible strip of 316 SS and adjusting the design could also make it easy to have a large area of contact.

We cant bend the flat strip in both direction I could try to draw something where the strip is rotated inside the holder so it could be bended in the other direction. 

> e.g., via a cutout, but we need to keep in mind that if we use a round o-ring, then we'd need to make the entire sample holder radius larger. If we move to a rectangular shape (not sure what we'd do [...]

I do not understand what are you guys referring to. If I could get more details

> > Make the inner opening of the capillary a bit smaller
> 
> Half the diameter? Smaller? What is the internal diameter of the capillary that you use? I just sent an email checking on my side with the other unit.

Now the diameter is 3.175mm, will reduce it to 2mm

> > Reduce distance between sample and capillary (if possible)
> 
> Do you also want thinner walls for the capillary? (keeping in mind blowout and breakage concerns of course)

The distance between the ID (12.7mm) of the capillary and the OD (8mm) of the reference electrode is 2.35mm. I will reduce it to 1mm. So the ID of the capillary will be 10mm. Thickness of the capillary is 2/32inch (1.5875mm) I will keep the same thickness unless you specify the thickness.

> We also want to keep in mind that any automated handling of concentrated KOH is safe and robust, given how corrosive it is to skin/eyes/etc., especially keeping in mind overpressure situations and [...]

From top of my head, in terms of safety, maybe a compression connector fitting. One that will always be on the tube side so you would have to unthread to take it off and the other side threaded to the cell. Something similar to this 
[image]

As I see there are already some of this fittings for peristatic pump tubes. We should also think of using more standardized and off the shelf products for the model instead of everything to be custom made.

> > Print a solid block to close the hole of the sample holder during washing protocol when sample holder is outside for the replacement of the sample
> 
> The hole below? What's the intention here?
> 
> <img alt="" width="100" src="https://private-user-images.githubusercontent.com/45469701/384467176-33723ab5-5051-4b36-a0de-07791c6370f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzExNzIzNTgsIm5iZiI6MTczMTE3MjA1OCwicGF0aCI6Ii80NTQ2OTcwMS8zODQ0NjcxNzYtMzM3MjNhYjUtNTA1MS00YjM2LWEwZGUtMDc3OTFjNjM3MGYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA5VDE3MDczOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzMDU0YjViOGJhNWY1ZTQwYTZhMDU5ZjAwZWNkZTY4OGQ1ZDY0YmNlZDg3Yzk0Y2VmYTg3N2M4NTY3MTY0NGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.hJTm7x9kts8lR2FkoyB6AghAnX27qncJ8v_R3s_vzFc">
> (aside: @MiceeNS is the hole in the picture above necessary?)

Because the wire was supposed to go in two different ways just below this, thinking it is just "prototype" I add like revision hole.

> > Increase the wall thickness again, to have a more reliable print with less risk of blowouts
> 
> It looks like current thickness is 4 mm:
> 
> <img alt="" width="200" src="https://private-user-images.githubusercontent.com/45469701/384468484-e9c7947d-9c69-413e-8d57-a827b815983a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzExNzIzNTgsIm5iZiI6MTczMTE3MjA1OCwicGF0aCI6Ii80NTQ2OTcwMS8zODQ0Njg0ODQtZTljNzk0N2QtOWM2OS00MTNlLThkNTctYTgyN2I4MTU5ODNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA5VDE3MDczOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI2NzczMTk0ZDkwYWI3NjU0YzUwZTdhZmQwOTEwNjA3YzM5YjgzNzY1NmRjYzc5MmU4ODEwYTZkNTc4ODk4YmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jmANZDYfaBlLHazBWrFRKF7y_hWT9AAAjMG2lYGwnuw">
> What was the original thickness? Seems we need either the original thickness or somewhere in-between. Even if we're using ~20% more resin, having the bottom of the cell oriented flat against the build platform and reducing the overall risk of blowout may be worth it.

The initial thickness was 6mm, then for the second one i reduced it to 4mm. The water jacket wall thickness was always 4mm

> @MiceeNS as an aside, have you been using rollbacks on the timeline when making some of the edits within Fusion 360?

are there any issues with doing this?
[... cut ...]
> @MiceeNS could you identify three to five grommets with reasonable dimensions and prepare a CAD model "plate" with various hole diameters and the same thickness as the lid?

Sure, let me specify the grommet that we are going to use and then I will print few different holes just to test the tolerances, but usually there are specs provided for the hole size with the grommet. 

Will provided details with what has been changed and modified or ideas by end of this week. 

I also want to completely redraw the design of the tap, instead of changing a different file then inserting that component to the assembly. Something fusion is not meant for. When I get free time...

sgbaird, 2024-11-09:
I'll think about the project flow.

Maybe we can all sit on things and recuperate for a week after the rapid iterations. We can arrange for a time to do a virtual check in and discuss next steps. In particular, I think the cell is at a pretty solid point from an automation perspective. Beginning to consider what's external to the cell will likely get us closer to the automation objectives. Of what we've talked about, vertical insertion of the sample holder is the only cell design aspect restricting autonomy over multiple iterations. Liquid flow control, gas flow control, and automated electrical connection/disconnection are the main things external to the cell.

Eventually, we might consider a rectangular design, since automation modules are often laid out in a grid-like fashion.

We have an Opentrons Flex, and I could imagine a high throughout setup with multiple of these cells (scaled down a bit) could be set up in parallel.

MiceeNS, 2024-11-10:
Are we using Vinyl tubes?

MiceeNS, 2024-11-11:
https://www.softenerparts.com/7209574.htm
[image]
https://www.softenerparts.com/How-to-Use-Quick-Connect-or-John-Guest-type-Fittings--Video_c_329.html

On the other side we can use thread or whatever that could be connected to the cell

The collar in the first link is sold separate from the fitting, since we cant use the plastic this fittings are made of we can print our main body and buy the collar. I still have not figured out how the collar is manufactured but is one smart idea. It has stainless steel teeth that bites the tubing when pulled.

I went to home depot and did a bit of reverse engineering. 

[image]

https://github.com/user-attachments/assets/236d7171-f36f-498b-947d-92bb3c975d66

[image]

MiceeNS, 2024-11-12:
Something similar 1/4" vinyl tube with M22x2.5 Threading. The other end to be threaded into the cell. Hopefully the compression ring will create enough force to hold the tube
https://github.com/user-attachments/assets/653670d7-62da-4375-b361-5939cc3bff01

programlich, 2024-11-12:
> I went to home depot and did a bit of reverse engineering.

Super cool! I think that's the way to go. Let's use whats already out there and modify it to our needs.

> Something similar 1/4" vinyl tube with M22x2.5 Threading. The other end to be threaded into the cell. Hopefully the compression ring will create enough force to hold the tube. I want to test this

Is this all custom built by you or are these standard components? Quite impressive anyway. I know that this kind of connection is being actively used for the transport of KOH. I'll get some details about manufacturers, material, etc. when I'm back in the lab tomorrow.

> We should also think of using more standardized and off the shelf products for the model instead of everything to be custom made.

I absolutely agree with this. No need to invent everything from scratch if low cost solutions are already available.

> Are we using Vinyl tubes?

Let me check.

programlich, 2024-11-12:
> Maybe we can all sit on things and recuperate for a week after the rapid iterations. 

That would be a good idea from my point of view. I'll be busy this week with getting started here again. I'm carefully optimistic, that I can run a whole experiment with our prototype next week, so if we could meet virtually by the end of next week, that might be a good thing.

sgbaird, 2024-11-12:
Nice! Likely silicone tubing, also another brand/type with PTFE. Swagelok is a pretty big name in this space. https://products.swagelok.com/en/c/straights/p/T-400-1-4

programlich, 2024-11-12:
> Liquid flow control, gas flow control, and automated electrical connection/disconnection are the main things external to the cell.

I think the most complex part will be the (dis)connection of the electrical contacts with alligator clamps. No idea how to automate this, but perhaps you guys have some good ideas. The control of liquids and gases can probably be done with a microcontroller and some pumps and valves.

programlich, 2024-11-12:
> Nice! Likely silicone tubing, also another brand/type with PTFE. Swagelok is a pretty big name in this space. https://products.swagelok.com/en/c/straights/p/T-400-1-4

Yep, Swagelok is definitely the de facto standard for reliable connections

programlich, 2024-11-13:
Thus kind of fitting is proven to work with KOH:

[image]

https://www.bola.de/en/HT-Laboratory-Screw-Joints-BOLAHT-Laboratory-Screw-Joints-BOLA

[image]

https://www.bola.de/en/Screw-In-Tube-Fittings-BOLA

programlich, 2024-11-13:
The material for tubes should ideally be PFA (or PTFA) for transport of KOH. For the gases, tubing material is not so important.

sgbaird, 2024-11-13:
I think Tygon tubing might be somewhat more chemically resistant than silicone, while still being compatible with a peristaltic pump (i.e., compressible). Thicker tubing means less likely to wear out as soon, but makes it harder to pump (not a big deal, to a point). We could also use Teflon tubing everywhere except at the actual pump, making it easier to periodically replace or inspect the pump tubing.

The temperature is also going to play a role. Potentially, we could wait for the KOH to cool somewhat before pumping it to waste. Also, inlet KOH will be room temperature.

Overall, a tough problem to deal with in terms of safety, reliability, and cost.

sgbaird, 2024-11-13:
> I think the most complex part will be the (dis)connection of the electrical contacts with alligator clamps.

Likely, we'll ignore the alligator clips completely, and focus on something more like an audio cable hookup. I'm sure there are some best practices and straightforward solutions I'm not aware of.

sgbaird, 2024-11-13:
@programlich do you mind making an Accelerated Discovery forum post asking about suggestions for automated electrical connect/disconnect procedures for e.g., potentiostat working electrode connections? I can review the post if you want.

MiceeNS, 2024-11-16:
Do you think you would want the gas to go all around the cell? Did something really fast for representation. Is there any gas flow rate or volume that need to be injected at certain time? 
[image]

sgbaird, 2024-11-17:
I think it won't be a huge deal one way or another. It's mostly about getting oxygen out of the system, but I'm not sure if the sensitivity of these experiments is enough to require incredibly thorough gas dispersion. I'm open to it (in a prior design, I put holes in a "false" floor at one point), but significant changes to the design could lead to complications with printing and new steps for post-processing.

programlich, 2024-11-17:
If it does not add extra complications to the print process, I'm totally in favor with the round bottom inlet. This way, gas enters the cell uniformly but well away from the electrodes, so it can't influence the measurement. Whatever is easiest.
We have to make sure, that the electrolyte dies not geht sucked out of the cell in case there is a underpressure in the gas supply line. Maybe with a one way valve? Vor is there another simple option to do that?

MiceeNS, 2024-11-17:
> If it does not add extra complications to the print process, I'm totally in favor with the round bottom inlet. This way, gas enters the cell uniformly but well away from the electrodes, so it can't [...]

One way valve is the way to go unless we have some restrictions. But I dont think thats the case in our model, unless vacuum occurs in the gas line.

[image]

For your information:
I have to create like extension from the cell so we could thread the adapter to the cell. And it will again create supports, if we print it from bottom to top I could make gussets where the green area is to remove the supports. I will have to check if it will create supports inside the threads. Another way is to not print threads and create them after with tapping tool. I will brainstorm for another way around

programlich, 2024-11-18:
> One way valve is the way to go unless we have some restrictions. But I dont think thats the case in our model, unless vacuum occurs in the gas line.

True. It would be more like an extra safety feature.

Regarding the stewing of the adapters, I think supports don't really matter as long as they are not inside the thread itself. Cutting a thread manually would be fine, but looking at the use for other groups, avoiding an extra step might be preferable.

MiceeNS, 2024-11-18:
@sgbaird for automation are we planning to use the Opentrons Flex or the Jubilee? Is there a possibility of changing the working head on the Opentrons Flex to our needs?

sgbaird, 2024-11-19:
The plan is to use the Opentrons Flex, and to create a fake tip for the pipette to grip onto.

@Neil-YL has been working on designs related to this. Could you pass on to @MiceeNS info related to your designs? (Your custom docs, f3d file, and the Opentrons height limitations)

MiceeNS, 2024-11-19:
Alright, I was curious about that. I have the f3d files from the shared team. Also, i have seen some of the videos how it works. @Neil-YL do you think we can mount something next to the tip. For example, some push and latch mechanism so it can grab things around? I have seen the design cone where the tip holds onto but fittings off the shelf dont have such features.

fyi: https://www.amazon.ca/Latch%EF%BC%8CTouch-Latch%EF%BC%8CRebound-self-Locking-Device%EF%BC%8COpen-Cupboard/dp/B07SK14KJ1/ref=asc_df_B07SK14KJ1/?tag=googlemobshop-20&linkCode=df0&hvadid=706738175649&hvpos=&hvnetw=g&hvrand=5546603168547897037&hvpone=&hvptwo=&hvqmt=&hvdev=m&hvdvcmdl=&hvlocint=&hvlocphy=1002249&hvtargid=pla-777348070048&psc=1&mcid=006a7d8f89493404a8f42834b0319211&gad_source=1

sgbaird, 2024-11-19:
Could you explain more? What's the purpose?

> do you think we can mount something next to the tip.

Crude 2D drawing?

> For example, some push and latch mechanism so it can grab things around? I have seen the design cone where the tip holds onto but fittings off the shelf dont have such features.

Off the shelf fittings?

The Opentrons pipettes allow you to both grab tips and eject them automatically. To "grab things and move them around", only the fake tip is required. Maybe I misunderstood something.

MiceeNS, 2024-11-19:
> Could you explain more? What's the purpose?
> 
> > do you think we can mount something next to the tip.
> 
> Crude 2D drawing?
> 
> > For example, some push and latch mechanism so it can grab things around? I have seen the design cone where the tip holds onto but fittings off the shelf dont have such features.
> 
> Off the shelf fittings?
> 
> The Opentrons pipettes allow you to both grab tips and eject them automatically. To "grab things and move them around", only the fake tip is required. Maybe I'm missing something though.

I was thinking if we want to connect and disconnect the compression fittings on the cell you wont be able to do it alone with the tip even if its quick connect/disconnect fitting

Neil-YL, 2024-11-19:
> do you think we can mount something next to the tip. For example, some push and latch mechanism so it can grab things around? I have seen the design cone where the tip holds onto but fittings off the shelf dont have such features.

I would lean to mount tools on the pipette rather than external mechanism next to the tip, which could be obstacle of the movement and actual functionality of the pipette.

You may refer to [this issue](https://github.com/AccelerationConsortium/ac-training-lab/issues/64) for our previous  implementation design on OT-2's pipette. One thing I would like to point out is pay attention to the height limit of the working area of Flex (you may see [this reply](https://github.com/AccelerationConsortium/ac-training-lab/issues/64#issuecomment-2420261474), but it is for OT-2) 

Our "fake tip" design basically mimics a regular tip that can be picked by the pipette, here is a cross-section view:

[image]

and a close look of the fitting:

[image]

Feel free to let me know if you need more information or any suggestion from me. Sorry for no reply last night, I did not receive the email notification from Github

Neil-YL, 2024-11-19:
Fusion file: https://a360.co/4hYtOcr

I believe the tip geometry also depends on the size of the pipette. Our current design is for Opentrons P300 single channel pipette.

Neil-YL, 2024-11-19:
I asked one rep from Opentrons website if they have a similar page of height limits of Flex. 

She replied:

> I believe we do but it is not public, our ability to share documents has changed since we released Flex, so I will dig around for this

I left her my email and to see if anything they can provide to us

programlich, 2024-11-19:
Today, I finally got the chance to do a first electrochemical test with our design from v1 in our labs. There are a lot of interesting aspects to share:
- The double wall, fittet its purpose and heating could be accomplished through it. The holes in the outer wall which I had to fix  with epoxy kept tight
- After polishing the surface of the cell with P1200 sandpaper, it was very smooth and the parts fitted into each other nicely
- All the connections (condensor, thermocouple, potentiostat, etc.) worked well
- The sample in the sample holder was a bit loose in the beginning, as I had to cut off on of the 3 lock points to get the sample holder fitting into the cell
- The cell didn´t break when confronted with concentrated KOH at 70°C for several hours

However, the contact between sample (standard Ni mesh) and the wires in the sample holder turned out to be to loose, which resulted in a very high contact resistance. Also, after the measurement, I realised, that the sample holder had unscrewed itself a bit so the sample was only very loosely pressed against the wires. This also explains the very bad values in overpotential, which I measured.

[image]

In this plot you can also see, that we need to plan some extra time for the heating of the electrolyte as the thick polymer walls have a bad heat conductivity. I had started the heating about half an hour in before the measurement.

Next, I will try one measurement with our own sample holder to make sure, the bad values are caused by the contact resistance in our sample hoder. Then we can think about how to improve it. Probably we need to tune the locking mechanism as we didn´t consider the loosing of the parts when being heated up.

sgbaird, 2024-11-20:
@Neil-YL thanks for exploring by contacting Opentrons. At this point, a pretty hard no for the NDA. It's a lengthy process to get NDAs signed at UoT/Acceleration Consortium, and for something as simple as a couple dimensions on a spec sheet that we could just go measure ourselves.. seems asinine to me. Does she know we own a Flex? Not sure if that would make a difference. Alternatively, maybe she could annotate the public version of the OT-2 with updated values relevant to the Flex, since then she wouldn't be sharing the drawings.

sgbaird, 2024-11-20:
@programlich thank you for the update! That's great that nothing was catastrophic 😅 sounds like our priority should really be on the sample holder. Tbh we might just want to use an online service to send you a part that a local supplier prints, since they're small. It's usually a premium per mL of resin (at least the trend), but maybe worth it over the hassle of mailing small parts. We'd send a revamped system including the full cell through mail though.

Neil-YL, 2024-11-20:
> @Neil-YL thanks for exploring by contacting Opentrons. At this point, a pretty hard no for the NDA. It's a lengthy process to get NDAs signed at UoT/Acceleration Consortium, and for something as [...]

I understand it might be difficult to sign the NDA through UofT’s process, so I mentioned that I would need to seek permission on our side first. I provided her with my UofT email and added her to my contacts in case we need more information later.

> Does she know we own a Flex? Not sure if that would make a difference. Alternatively, maybe she could annotate the public version of the OT-2 with updated values relevant to the Flex, since then she wouldn't be sharing the drawings.

Yes, she knew we already have the products (Flex and OT-2, I actually said we have several in different labs, lol). I think the parameters she already provided are quite sufficient to start a design, and we can also have our own measurements if needed. Like she said have our pipette installed and measure to have a subtraction from 250mm clearance. We may also test with the App. It has an analyser and if the custom labware are out of bound it would have an error message (Not very sure about Flex but I know it work like this on OT-2)

sgbaird, 2024-11-20:
(NOTE: See end for my conclusion and a diagram, likely we just use a molex or banana plug/socket pair and find a way to keep the printed and metal parts joined and aligned reasonably well)

@MiceeNS I might still be misunderstanding, but from what @Neil-YL and I have seen, I'm pretty sure we can implement something relatively straightforward that only uses the fake tip mechanism (from both hardware and software perspective). Worth brainstorming more, but I have a few ideas which are hard to explain without drawing.

https://www.google.com/search?q=mcmaster+tapered+steel --> https://www.mcmaster.com/products/steel-taper-pins/ --> https://www.mcmaster.com/products/taper-pins/locating-pins-1~/ --> https://www.mcmaster.com/products/taper-pins/removable-locating-pins-7/ --> e.g., https://www.mcmaster.com/product/2397A51

[image]

This one seems to assume a locking nut, so we'd probably need to play around by hand with one of these to see if hand pressure could be used to create a tight connection and also remove it.

I brought up another idea a while back on the other issue, which assumed use of a linear actuator situated at an angle. When it's retracted, there's enough clearance to allow the sample holder to be inserted vertically, but there's no tight electrical connection yet. When it's extended, it applies constant pressure (between e.g., a battery connector spring and a flat plate) to keep the electrical connection solid. It would still require some level of fixturing of the linear actuator, either relative to the cell, to some kind of cell holder, or to the Flex itself.

However, to @MiceeNS's point, we might want to use something more off the shelf (and specifically meant for removable electrical connections).

NOTE: As I've been going through the thought and search process, I think I have a good solution.

https://www.google.com/search?q=mcmaster+mating+electrical+connection --> https://www.mcmaster.com/products/electrical-connectors/ --> https://www.mcmaster.com/products/electrical-connector-contacts/manufacturer~molex/ --> (https://www.mcmaster.com/product/69295K111 + https://www.mcmaster.com/product/69295K115)

(Related follow up search that also led me to realize the wire gauges mean for crimping, not for the insertion/deinsertion pins, which are selected in pairs): https://www.google.com/search?q=mcmaster+molex+signal+power+connector+mating+single+pin)

I've used custom molex connections before. I think we should go with this. This doesn't address the gas tightness of the sample holder in the cell (the linear actuator idea can both seal an o-ring and create an electrical connection), but there might even be a similar solution for that (maybe simply applying a bit of grease to the mating surfaces of the sample holder and cell would be sufficient, similar to when using ground glass tapered fittings).

We might just need some kind of alignment piece (maybe just printed) that keeps the molex connector of the sample holder fixed vertically and in a repeatable xyz location. Banana plugs come to mind, too, and are pretty standard (https://www.mcmaster.com/products/banana-connectors/test-end-component~banana-plug). Same principle.

To illustrate this idea:

[image]

MiceeNS, 2024-11-21:
For the electrical connections:
I have used binding post (https://www.mcmaster.com/6937K62/) that can be panel mounted so we can just make cutout and mount it on the lid. I have to reconfirm if this exact one can work with banana jack inserted from top. 
We can try different type of jacks and see what works the best for us at the end.
 Presented in this pictures
[image]
[image]

For the connections on the cell:
I found this type of fitting which can be easily connected and disconnected from the cell. Compression fitting for safety on the tube side. This fitting also have shut off feature which when disconnected immediately shuts off ( atleast that what description says, have to confirm which side of the fittings shuts off)  
https://www.mcmaster.com/5012K125/

[image]

For the sample holder: 
I will try to design 3d printed compression spring that will constantly create force on the washer and the electrical connections. Also providing good seal.  
Something similar to:
[image]
The spring will be behind the washer connected to plate same size as the washer that will push constantly the washer forward. Few mm will do the trick and I think its feasible
[image]

sgbaird, 2024-11-21:
Electrical connections: thanks for exploring. I think we'll only implement the automation-friendly electrical connections on the sample holder. Binding post could simplify things, but we also have to think about how to actually get the part in the lid after the lid has been printed (in this case, I'm guessing it's just a simple disassembly step). I'm not sure I follow why there was an upper and a lower part though.

Nice find on the cell fittings!

I'm a bit worried about trying to make a spring out of clear v4, which is very rigid and brittle.

sgbaird, 2024-11-21:
Oh, also as a reminder KOH outlets need to be highly chemically resistant wherever the KOH touches. As long as the ports are high up, I don't think there's a big need for it to be self closing upon disconnect.

programlich, 2024-11-21:
Thank you all for your great ideas about the electrical connection and the fittings! Really nice, following this discussion.

> I found this type of fitting which can be easily connected and disconnected from the cell. Compression fitting for safety on the tube side. 

Pretty nice! From my side, I´m happy with any fitting, which safely connects cell and tube and is resistant to KOH (PP and PTFE definitely fine). Shut off mechanism is nice but not urgent.

> I will try to design 3d printed compression spring that will constantly create force on the washer and the electrical connections. Also providing good seal.

This would probably have to be done with an elastic resin. I agree with @sgbaird that the clear v4 will most likely be too brittle. However, elastic resins are available, but I don´t want to bet on their resistance against KOH. However, we might reconsider using a thread mechanism for locking the sample holder. By now I think we have decided, that mounting the sample into the holder will be done manually and we just go with a set of sample holders to be exchanged one by one. So if printable a thread might be the easiest solution to answer to different thicknesses of samples. What do you think? 

Also I suggest, we move the discussion about the sample holder to a separate issue, as there are lot´s of things to be discussed about the next design.

Electrical contact: I really like your idea, @sgbaird. Perhaps, we could take it one step further and implement a solution, where the potentiostat is permanently connected to an electrical contact in the tap of the cell. When the sample holder is inserted it is pressed against the other end of the same contact. This way, we could avoid touching the cable connected to the potentiostat completely. See my sketchy drawing about this idea:
[image]

programlich, 2024-11-21:
Today I had a long discussion with a very experienced colleague. Regarding the safety aspect of having hydrogen and oxygen mixed in the same compartment he came up with an easy to implement idea for gas separation. We could print a wall attached to the tap which comes down to a height just over the electrodes. As the evolving hydrogen and oxygen rise up directly, they would be separated by the wall, as soon as they get to the height of the wall. Then we could simply implement to gas outlets with two attached condensors and the gases would be mostly seperated. Not 100% seperation but definitely enough to increase security significantly at almost no cost. Could I express the idea or should I draw a sketch?

Also, the idea of having the nitrogen bubbling out from a ring in the bottom of the cell like suggested by @MiceeNS was considered to be very good.

programlich, 2024-11-21:
Also from todays discussion: The reference electrode can remain in the cell for about 1-2 weeks in continuous use without need of changing it. So I´d say there is no need to think about an automated changing of the reference electrode for now

sgbaird, 2024-11-22:
Agreed, I think a threaded design may be best. Pressure applied isn't very repeatable unless there's a hard stop to the threads. Too much pressure could damage a coating, correct? Too little pressure and you get the issue shown before. We can get something working and worry about tuning the pressure later.

@MiceeNS could you open a new issue focused on the sample holder? Just a brief summary of our latest ideas/challenges, image(s) of the latest sample holder design, and a link back to this comment.

Glad you had the conversation with your expert colleague!

I forgot about reference electrode storage. I think I'd actually like to push for having a similar mechanism removing the cap from the storage container and lifting the reference electrode out and and putting it into the storage container (and in reverse order). The electrical connection can stay permanent for that one I think.

I think the counter electrode, being Pt, can stay there somewhat indefinitely (months), correct?

I'm not sure I follow as much on the gas separation. With inert gas flowing most of the time, isn't it going to get whisked away and dispersed largely in inert gas? I think I'd need to see a rough sketch to know what you mean with the wall.

Good point about having a permanent potentiostat WE connection to the cell. Originally, I was thinking about the situation of having multiple cells, but then we'd probably need multiple potentiostats anyway, correct? Curious to hear if you have any ideas on whether it becomes useful to plug the WE / RE / CE into other setups (either other identical cells or completely separate experiments). My impression of these experiments is one cell monopolizes the potentiostat for the entire duration of the experiment.

As an aside, I'm leaning away from the Flex and Science Jubilee for something more suited and lower cost. More details to come.

For mating a metal part and a resin part, we can consider putting them together prior to curing.

programlich, 2024-11-22:
> I forgot about reference electrode storage. I think I'd actually like to push for having a similar mechanism removing the cap from the storage container and lifting the reference electrode out and [...]

I think exchange of the reference electrode (RHE) is something, which can be implemented in the future without too much problems but I´d say it has a rather low priority for now.

> I think the counter electrode, being Pt, can stay there somewhat indefinitely (months), correct?

Exactly

> Originally, I was thinking about the situation of having multiple cells, but then we'd probably need multiple potentiostats anyway, correct?

Practically speaking, yes. There are large potentiostats/potentiostat-booster combinations, which can control 2 or even 3 experiments at a time but then we talk about prices of min 35000$. Not really feasible for now. However, if it´s not necessary to apply industry relevant currents we could use several small potentiostats from palm sense. 

> For mating a metal part and a resin part, we can consider putting them together prior to curing.

This might be especially interesting for adjusting the sample holder. I think we should have another meeting to discuss sample holder related details in person.

programlich, 2024-11-22:
[image]

vicobit, 2024-11-26:
Hi! @programlich invited me to check out your project, and I must say, it is super interesting for our group since we’re working on coatings onto SS and Ni foams provided by IFAM! We’re very interested in implementing at least the sample holder soon. It’s a bit challenging to provide feedback at this stage, as there are so many details to consider. My main question or concern would be: could bubbles accumulate inside the sample holder?

Have you tested similar sample holders (or even this design) in a real EQ cell? Any insights or comments on this aspect would be greatly appreciated.

Thanks again for sharing this exciting project!

MiceeNS, 2024-11-27:
> Hi! @programlich invited me to check out your project, and I must say, it is super interesting for our group since we’re working on coatings onto SS and Ni foams provided by IFAM! We’re very [...]

Hi @vicobit,

If you are referring to gas bubbles, there is a O-Ring and a rubber Washer around the only aperture at the sample holder. With enough pressure I think we will have descend seal. Regarding the testing, @programlich did some testing on this version of the sample holder.

programlich, 2024-11-29:
Hey @vicobit, great to have you looking at this project, too. It´s awesome to have another experienced electrochemist providing her input.
Thanks for your question regarding the sample holder. As @MiceeNS mentioned, we hope to not have any bubbles inside the sample holder by sealing it against the electrolyte with one o ring and one washer. If you refer to the bubbles which might accumulate at the top inside of the sample holder, we haven´t tested it so far, but we actually might improve the current design a bit. It´s a valid point you are asking!

@MiceeNS, would it be possible to add an angle to the sample holder at the place where the gas bubbles coming from the sample hit the inside of the sample holder, so that they can flow off more easily? Atm they just hit the sample holder and it will take some time for them to "disappear". See my sketch for what I mean. Sorry for the terrible drawing, I hope I can explain my thought.

[image]

programlich, 2024-11-29:
> Have you tested similar sample holders (or even this design) in a real EQ cell? Any insights or comments on this aspect would be greatly appreciated.

Not yet, but this is planned for the next iteration. Once we have v3 of the sample holder printed, I will test it in the IFAM labs in the known cell setup.

programlich, 2024-12-04:
Current state of development, all STL files included

[Electrochemicall Cell STLs.zip](https://github.com/user-attachments/files/18005114/Electrochemicall.Cell.STLs.zip)

programlich, 2024-12-17:
Yesterday I had the chance to run another electrochemical experiment in our cell with one of IFAMs sample holders to check the general functionality of the cell. Here are two plots from two experiments where I played around with the current density. The sample was a standard Ni mesh and the electrolyte 30m% KOH. The thermostat was set to 70°C.
[image]

[image]

From these plots imho we can learn two things:
- The general concept of the cell works as expected. The values are in the range of what I would expect. This is good news!
- The thermal conductivity of the v4 resin is very poor (as expected) so in the future we might consider preheating the electrolyte before adding it to the cell and only use the double wall design for keeping the temperature constant. Note the steady increase of temperature over time in the plots above.

Once the next version of our custom sample holder is printed, I will carefully run two precise head-to-head comparisons between our custom cell and the IFAM standard setup.

programlich, 2024-12-17:
An idea from a discussion with an experienced electrochemist: We might consider removing the condensor from our design completely and only have two separate outlets for the gases (anode and cathode side) which are then passed to a gas washing bottle. In his experience the condensor is not absolutely necessary and in his labs he is now moving away from them as they bring very little benefit but add an extra layer of complexity (glass equipment & water supply).
This might be beneficial for us as it would also give us more space at the top of the cell which might simplify the movement of the robot

sgbaird, 2024-12-17:
This is only towards the top of the inside of the cell?

programlich, 2024-12-17:
No, I can find it everywhere on the inside of the body

[8 more replies]