AccelerationConsortium/ac-dev-lab #107: Form Auto seeming to not record auto-release videos anymore

sgbaird, 2024-11-08:
Sometimes, the FormAuto can have trouble releasing off of the build platform. See the following two articles:

- https://support.formlabs.com/s/article/Best-practices-for-printing-with-the-Form-Auto?language=en_US
- https://support.formlabs.com/s/article/Removing-unreleased-printed-parts-from-the-Form-Auto?language=en_US

While there is a camera on the FormAuto that can record part removal videos, if the part removal video doesn't get recorded, then typically I think the only indication might be on the Activity Log for your print from [dashboard.formlabs.com](http://dashboard.formlabs.com/). For example, see the attached activity log. In this case, I suspect it didn't auto-release because the Form 3+ wasn't aligned properly. Note that the orientation should be with the tallest edge of the Form 3+ facing the collection basket (see attached image and [printing prep](https://support.formlabs.com/s/article/Setting-up-your-Form-Auto?language=en_US#printing-prep) instructions).

Could be firmware related.

kelvinchow23, 2024-11-08:
A test print with a small cube was conducted.  Attached is the preform file and a screen recording of the preparation. The part did successfully release from the build plate for this instance.  I don't believe I see a print removal video on the formlabs dashboard of this job. 
See zip file below.
[debug cube.zip](https://github.com/user-attachments/files/17682632/debug.cube.zip)
[image]

sgbaird, 2024-11-08:
@kelvinchow23 any chance you could open a new topic describing this on the [formlabs forum](https://forum.formlabs.com/)?

sgbaird, 2024-11-09:
Thanks @kelvinchow23 ! https://forum.formlabs.com/t/form-auto-print-removal-video-not-available/40560/2

kelvinchow23, 2025-02-03:
i notice this issue has been resolved after reaching out to formlabs support. we are able to view form auto part removal videos again to confirm a successful part removal before sending a new job.

closed 2025-02-21 by sgbaird