AccelerationConsortium/ac-dev-lab #109: Setting up Prefect on the OT-2

sgbaird, 2024-11-08:
@yunhzou can you comment on how you went about to set this up with the custom Linux distribution on the Raspberry Pi 3+ embedded within the Opentrons OT-2?

(i.e., the device-side code corresponding to the https://github.com/yunhzou/OT2Demo shared recently)

sgbaird, 2025-02-15:
@yunzhou bump.

Xref: https://github.com/PrefectHQ/prefect/discussions/16685

sgbaird, 2025-04-08:
@Neil-YL we can probably revive this one soon and implement a human-in-the-loop step for swapping out the 96 well plate when it's full.

I have a yt video related to this. I'll dig up. Also good to consider if we want a reset button that the user presses to signify it being complete.

Neil-YL, 2025-04-09:
Current workflow, with HF as an orchestrator, (and a funny human in the loop)

[image]

(1) HF checks if the student ID has experiment quota and retrieves the next available well location from MongoDB (then adds the task to the queue).  
(1') HF updates the well status and student quota when the task is added to the queue.  
(2) HF sends experiment commands (student ID, RYB values, well location, experiment ID) via MQTT to the OT-2 to perform color mixing.  
(3) After completing color mixing, the OT-2 moves the color sensor to the target well and sends a sensor status ("in place") to HF via MQTT.  
(4) HF sends a read command to the color sensor via MQTT.  
(5) After measurement, the color sensor sends back an 8-channel result to HF via MQTT.  
(6) After receiving the sensor data, HF sends a command  via MQTT to the OT-2 to move the sensor package back to the charging port.  
(7) Once the sensor package is moved to the charging port, the OT-2 sends a sensor status ("charging") to HF via MQTT.  
(8) HF logs the experiment data into MongoDB, reporting experiment result to the output.

sgbaird, 2025-04-09:
I love the arrow arms 😂 

For provenance, here is the related YouTube video: https://youtu.be/MnE5O41lCPw (in the context of https://youtu.be/qcSUsrVRiEU)

Within HF, we need to see if we can get Prefect working #224 

Above HF, to keep things simple for now, we can have our BO loop calling the HF API and running 24/7 on an oracle instance, but we'll need to handle the case where the experiment fails or is waiting on a hitl step.

cc @masonmenges

sgbaird, 2025-04-10:
Maybe something like the following:

Some mix of:
- https://github.com/AccelerationConsortium/ac-training-lab/blob/main/scripts/prefect_scripts/my_gh_sample_transfer_workflow.py
- https://github.com/AccelerationConsortium/ac-training-lab/blob/main/scripts/prefect_scripts/create_sample_transfer_deployment.py
- https://github.com/AccelerationConsortium/ac-training-lab/blob/main/scripts/prefect_scripts/run_my_gh_sample_transfer.py

Create a script that handles the slack interaction similar to `my_gh_sample_transfer_workflow.py`, upload it to GitHub, and run a script locally similar to `create_sample_transfer_deployment.py` (store on GitHub too for provenance/ease), and then use something like https://github.com/AccelerationConsortium/ac-training-lab/blob/main/scripts/prefect_scripts/trigger_flow_programatically_1.py within HF to call that deployment.

Pulling your reply from slack:
> To add the Prefect control HITL on HF, it would be making the `add_to_queue()` a main flow, having a pause for HITL at when  `if not empty_wells: raise ValueError("No available wells")` and keeping most of the other things on HF as it (for our early stage exploration)

Neil-YL, 2025-04-10:
Although yesterday we've discuss that Prefect could use the HF+devices system as a blackbox, now it seems make more senses if Prefect is on the same level of the HF, given the fact that there still could be users on HF level outside of BO loop.

Current plan is trying to add the Prefect control HITL on HF. If any difficulties, create reproducer and seek help from Prefect at the same time, move HITL to higher level.

-------
Update:
- Simply calling Prefect client with Prefect api key would not enable any tracking or visibility on Prefect Cloud UI. Now I turned to using "work-pool" and "worker", seems promising now:

HF logs:

> 
> Discovered type 'process' for work pool 'ot2-pool'.
> Worker 'ProcessWorker d76ad014-f944-45bf-b660-fbed96a42042' started!

Prefect UI:
[image]

I will continue on this testing [HF space](https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM-test) later with the HITL setup.

Neil-YL, 2025-04-11:
Prefect pause on HF:

https://github.com/user-attachments/assets/085003de-daa1-4685-9c75-77a2110cb9be

sgbaird, 2025-04-11:
Few quick notes from our conversation:
1. Since we want to add background spectrum capture to the workflow, and it would complicate things and require a major refactor of the code, it makes sense to try swapping out MQTT with Prefect for the HF <--> OT-2 communication

Neil-YL, 2025-04-14:
Some updates today (on [HF test space](https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM-test))
- Add a `maintenance_flow.py`.
- Add a trigger_maintenance_request to` prefect_utils.py`.
- Add a maintenance status/flag check (avoid multiple maintenance request before HITL complete) and a maintenance log in db to `DB_utils.py`.
- Use `find_unused_wells`, when empty_wells == 1, to trigger the maintenance request.

Issue need addressing:
- Description and input variables name on flow resume. (I am using this [template](https://docs.prefect.io/v3/develop/inputs) but seems a little buggy)

I will post a video later.

Neil-YL, 2025-04-14:
https://github.com/user-attachments/assets/f486dea8-bfb2-4de7-8805-48a3f2f76f02

(My original clip is just serval MB later than 100 MB limits on the Github issue reply so I had to trim the beginning and the ending.)

Neil-YL, 2025-04-15:
[image]
Update the resume interface on Prefect UI

Neil-YL, 2025-04-15:
Put refs here for comments on OT2 system and install python packages on OT2:
https://support.opentrons.com/s/article/Integrating-the-OT-2-with-other-lab-equipment
https://support.opentrons.com/s/article/Using-Python-packages-in-Python-API-protocols

A previous issue mentioned by @MatPopp https://github.com/Opentrons/opentrons/issues/15110 

I also found that the file explorer under the OT-2’s Tailscale connection doesn’t include the file path to /data.(that Opentrons documentation recommends users to storage data)  /data is located at a higher level than the default SSH path.  

> Where on the OT-2's filesystem can I keep my files?
> Most of the OT-2's filesystem is read-only. However, two places you can store files in are:
> 
> The /data/user_storage directory, which is good for general-purpose storage.
> 
> The /var/lib/jupyter/notebooks directory, which is also be accessible through your OT-2's Jupyter Notebook.
> 
> Files placed in either of these directories will persist across reboots and software updates.

[image]

Neil-YL, 2025-04-16:
More updates:
I still cannot resolve the dependencies issues.

According to [this article](https://support.opentrons.com/s/article/Using-Python-packages-in-Python-API-protocols),

> Some of the more sophisticated Python packages, like [NumPy](https://numpy.org/), aren't written purely in Python. Parts of them are written in other languages, like C.
> You won't be able to install those packages on the OT-2. They require a [compiler](https://en.wikipedia.org/wiki/Compiler), which the OT-2 doesn't have.

Since Prefect depends on some packages written in both C and Rust, I tried to use Docker to prebuild the virtual environment for the prefect deployment. However, building and packaging required package pendulum3.0.0 for the OT-2’s architecture (armv7) remains problematic, even with required Maturin and Rust in place.

`PIP_CONFIG_FILE=/dev/null` is to fix the PIP preset path on the OT2 (It automatically adds `/var/user-packages` to your installation paths, causing path conflicts and makes using a virtual environment much more complicated). It will not help with the lack of compliers.

We may need to have another device working as the Prefect agent/worker for the OT2, and perhaps with similar approach of Jackie's SSH wrapper class to use OT2 robot control API.

sgbaird, 2025-04-16:
Thanks for exploring this. Do you think we should pivot back to MQTT, explore more options in Prefect (maybe there's a way to use requests and not install Prefect -- could be a huge pain or rabbithole though), have a dedicated separate device with SSH like you mentioned, or stick with an MQTT implementation?

Likewise, we could pivot and refocus on the YT Livestream restarts and auto download workflow. Maybe leave some documentation or issues with Prefect and/or Opentrons folks to see if there are any alternative recommendations.

Neil-YL, 2025-04-17:
OK. Failed to have the Pi 4 pre-compiled file work, but I think I understand the issue better now.  

It turns out that Prefect requires Pendulum 3.0.0:  
> Collecting pendulum<4,>=3.0.0 (from prefect==3.3.4)

However, Pendulum 3.0.0 has a known bug when building on 32-bit systems, as noted [here](https://github.com/python-pendulum/pendulum/issues/784). Unfortunately, the OT-2 runs on a Pi 3 with 32-bit armv7l, which explains why I always got stuck.  

It looks like the issue was closed via [this commit](https://github.com/python-pendulum/pendulum/commit/b84b97625cdea00f8ab150b8b35aa5ccaaf36948), but it hasn't yet been included in the latest release (the commit is from last November, and the latest release is from 2023).  

This gives me some hope — we might be able to use that specific commit verison of Pendulum to see if we can get Prefect running on the OT-2.

sgbaird, 2025-04-21:
Great! Nice intuition. Could you also include a link to the slack message you mentioned from the Prefect slack?

Neil-YL, 2025-04-21:
Update: successful have the fixed commit Pendulum and the Prefect installed on the RPi4 mimic OT2 environment, but the packed venv file still failed to run on OT2. 

I believe there're still some issues on the Pendulum on OT2, I will try to pre-comply a Pendulum for OT2 and then install Prefect.

sgbaird, 2025-04-23:
Random aside: at some point it would be good to record a short video of exchanging the wellplate, add to YT, and include the link in the slack message. Not urgent

Neil-YL, 2025-04-23:
> be good to record a short video of exchanging the wellplate, add to YT, and include the link in the slack message. Not urgent

As a video guide of the maintenance?

sgbaird, 2025-04-23:
Correct (doesn't have to be complex, just testing out what a good way to have the "standard operating procedure" embedded into the human-in-the-loop step)

Neil-YL, 2025-04-23:
OK, I think it’s time to wrap up the pre-compiled tryout I’ve been working on over this week:  

As I mentioned last Friday, Pendulum has a known issue when installed via pip on 32-bit armv7 devices. There was a commit that fixed it, and I managed to install that specific commit version successfully, along with Prefect, on a mimic environment using an RPi 4.  

Then I moved the virtual environment to the OT-2, placing it under the user storage path as suggested by the official documentation. That’s when I ran into another issue: Prefect was built with Glibc 2.34, but the OT-2 only has 2.25 installed:  

```
ldd --version
ldd (GNU libc) 2.25
$Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$Written by Roland McGrath and Ulrich Drepper. 
```

At that point, I figured I should try matching the Glibc version. I eventually found that I would need a 2018 version of RPi Lite OS to do that — but our RPi 4B doesn’t support something that old.  

So I pivoted to using Docker with a Debian Stretch (2017/2018-ish, if I remember correctly) base image, still using a 32-bit armv7 setup. I went through the install process all over again — more dependencies and requirement issues on this old system like Python 3.10 + SSL compatibility issues. Eventually, I got Prefect installed inside the Docker container, only to find out that the installation still required at least Glibc 2.28 (needed for one of Pendulum’s requirements, which I think required at least G++8).  

So… I think this might be a dead end for installing Prefect natively on the OT-2 before we touch the system-wise no-go-zone.  

There’s one last slim hope: our OT-2 version might be kind of old (around Sept. 2024?), so maybe they’ve upgraded Glibc in a newer release(not very likely). But since Christine will be borrowing the device from this Friday through Monday for the ChemPALS event, I’ll hold off on upgrading until after their event.  

I’ll pivot to the streaming manager tmr.

sgbaird, 2025-04-24:
Thanks for digging in and documenting what you tried! This is great. Agreed that it's probably time to pivot - maybe still worth opening up a discussion on Prefect Discussions and reaching out to Opentrons (either on gh or through their support, not sure which would be better), at least to let them know about it.

Neil-YL, 2025-04-24:
> Thanks for digging in and documenting what you tried! This is great. Agreed that it's probably time to pivot - maybe still worth opening up a discussion on Prefect Discussions and reaching out to [...]

Actually Opentrons has a [PR for python package](https://github.com/Opentrons/opentrons-python-packages) for their robots but it has not been updated for three years. I can try to open an issue to request Prefect and mentioned what we have tried. They also provide a build tool but it said it does not work on arm64 device (like my M2 Mac) so I did not try.

sgbaird, 2025-04-28:
Maybe worth trying to install Prefect's `main` branch while forcing it to avoid the `pendulum` dependency (e.g., through git clone, removing that line from pyproject.toml, and `pip install -e .` for local installation)

Aside: I think this is the light version (I think you mentioned earlier), not sure if this still has the same dependency issue: https://docs.prefect.io/v3/get-started/install#minimal-prefect-installation or is applicable in our case

Related: https://github.com/PrefectHQ/prefect/discussions/17843

Neil-YL, 2025-04-29:
> Aside: I think this is the light version (I think you mentioned earlier), not sure if this still has the same dependency issue: https://docs.prefect.io/v3/get-started/install#minimal-prefect-installation or is applicable in our case

I forgot to mention that Prefect-client also requires Pendulum 3.0.0.

sgbaird, 2025-04-30:
https://support.opentrons.com/s/article/Using-Python-packages-in-Python-API-protocols has a link to https://github.com/Opentrons/buildroot/issues. Might be worth x-posting there as a request (not sure they would support this for Prefect on all systems though, but might be more likely to get a response).

Random aside: just came across the [opentrons simulator](https://support.opentrons.com/s/article/Simulating-OT-2-protocols-on-your-computer)

sgbaird, 2025-05-09:
Got as far as being able to install Prefect without dependencies; however this probably doesn't work anymore because I rebooted the OT-2 and probably lost the files where it was installed. However, most of the history of what I tried documented in https://claude.ai/share/00ca2edb-10ca-457b-9f50-d6d72b88e32f (and decided to stop at end).

sgbaird, 2025-05-09:
We can set up an OT-2 robot simulator per https://github.com/AccelerationConsortium/ac-training-lab/issues/248

Neil-YL, 2025-05-15:
Things to address:

If both OT2-LCM and OT2-LCM-test HF space are up, the well status maintenance request HITL on OT2-LCM will lead to a update on the OT2-LCM-test well status database(in flow resumed stage), but not the OT2-LCM one. 
If I stop the space of OT2-LCM-test, the well status maintenance update works normally for OT2-LCM.

Neil-YL, 2025-05-24:
Yes. There is a wrapper for each command of the OT2 protocol via the ssh client.

sgbaird, 2025-06-23:
Last week, Prefect was successfully installed on an OT-2. I verified serving a flow and viewing the flow run on the AC's prefect cloud workspace: https://github.com/sgbaird/opentrons-python-packages/pull/12#issuecomment-2982441682. I'm in the middle of trying to get copilot to retrace its steps so we can have a lean, self-contained set of installation instructions

Copying that message below:

> @copilot it works. I SSH'd manually and separately, and then logged in with Prefect Cloud, adding the API key permanently. So, you should be able to test one of those quickstart flows (don't go overboard) directly with the Prefect Cloud instance. Make sure you're not exposing any sensitive info. Using the quick start example:
> 
> ```shell
> 02:31:05.506 | INFO    | Flow run 'blazing-crab' - Beginning flow run 'blazing-crab' for flow 'main'
> 02:31:05.602 | INFO    | Flow run 'blazing-crab' - View at https://app.prefect.cloud/account/5b838504-64cf-4297-9b35-b881ac6169b3/workspace/d2718b4c-b49a-43ce-83c2-baf6fb3b9665/runs/flow-run/0685224e-8b4a-72d6-8000-feb7356e87f5
> 02:31:06.175 | INFO    | Task run 'get_customer_ids-eea' - Finished in state Completed()
> 02:31:08.508 | INFO    | Task run 'process_customer-e00' - Finished in state Completed()
> 02:31:08.511 | INFO    | Task run 'process_customer-99d' - Finished in state Completed()
> 02:31:08.623 | INFO    | Task run 'process_customer-e51' - Finished in state Completed()
> 02:31:08.632 | INFO    | Task run 'process_customer-5ca' - Finished in state Completed()
> 02:31:08.714 | INFO    | Task run 'process_customer-c10' - Finished in state Completed()
> 02:31:08.727 | INFO    | Task run 'process_customer-00b' - Finished in state Completed()
> 02:31:08.835 | INFO    | Task run 'process_customer-7fb' - Finished in state Completed()
> 02:31:08.868 | INFO    | Task run 'process_customer-195' - Finished in state Completed()
> 02:31:08.885 | INFO    | Task run 'process_customer-27a' - Finished in state Completed()
> 02:31:08.890 | INFO    | Task run 'process_customer-d61' - Finished in state Completed()
> 02:31:09.393 | INFO    | Flow run 'blazing-crab' - Finished in state Completed('All states completed.')
> ```
> 
> ![image](https://private-user-images.githubusercontent.com/45469701/456289079-bc28e340-3884-4d27-9ad3-ecb6cf9a63cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTA3MTEyODIsIm5iZiI6MTc1MDcxMDk4MiwicGF0aCI6Ii80NTQ2OTcwMS80NTYyODkwNzktYmMyOGUzNDAtMzg4NC00ZDI3LTlhZDMtZWNiNmNmOWE2M2NmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA2MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNjIzVDIwMzYyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3ZTE0MDE5OGE3Mzg1M2Y2MDM2OGVhYmY4MjY0ODlkOGYyYjE4OWEyZTA3ODJhZmVmZmVlMmVkMGQ1ZTcwY2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ve8zBkgnH9SF8qX5nx9LkgoGCS7DNCw2wYdzceZO0pE)