AccelerationConsortium/ac-dev-lab #113: Autotrickler v4 - custom RS232 control with A&D scale

sgbaird, 2024-11-12:
As a follow up to #20, the goal is to get a similar interface and [IolT setup](https://ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html) to that of the US Solid scale. Some related material at https://ac-microcourses.readthedocs.io/en/latest/courses/robotics/3.2-serial-communication.html. In particular:

> ✅ Read [HR Analytical Balance Literature](https://weighing.andonline.com/sites/default/files/documents/HR-A_HR-AZ%20Lit_web.pdf)
> 
> This document provides an overview of the analytical balance we'll be interfacing with. Understanding the capabilities and specifications of the instrument is crucial for effective integration.
> 
> ✅ Skim [HR Analytical Balance Manual](https://weighing.andonline.com/sites/default/files/documents/HR-A_HR-AZ_Manual.pdf). Read Sections 15 and 17 closely.
> 
> Sections 15 and 17 detail the RS232C interface and the commands used to communicate with the balance. This information is essential for programming our microcontroller to interact with the balance correctly.

The connection at the back of the scale is similar. We have two scales (the 1 mg and the 0.1 mg resolution). The 0.1 mg resolution scale is at NRC. Let's focus on the 1 mg at UoT.

evelynnesher, 2024-11-13:
[image]
Iset up the RS232 connection with the scale. As shown below, I needed to unplug the auto trickler as it was plugged into the RS-232C serial interface.
[image]
[image]
[image]
Currently, it prints “ST” when the weight measurement is stable and “US” when the weight is actively changing and unstable.

sgbaird, 2024-11-13:
That was fast! Great. There should be a second pico 2ch rs232 module available (I think in the same room?), so we can have one for the US solid scale and one for the A&D scale in the office.

evelynnesher, 2024-11-13:
Got HiveMQ set up and publishing with no issues, so I’m moving on to the app now. Do you want this one to have the autotrickler feature too? I’ll have to figure out a way to make it work since they both use the same port.

sgbaird, 2024-11-13:
Nice! You'll use the same port, but adjust the MQTT topic structure to differentiate between the two devices. I think a separate app might be better, but I'm open to the idea of integrating it into the existing app.

evelynnesher, 2024-11-13:
[image]

[image]
I think you misunderstood, I meant the physical port on the scale, where both the Raspberry Pi (for the RS232 connection) and the AutoTrickler are connected. But as you can see, only one can connect at a time.

sgbaird, 2024-11-13:
Ah, my bad 😅 The autotrickler can be left disconnected for now.

evelynnesher, 2024-11-13:
[image]
I set up the app in a private space on my personal account to test it out first. It is currently identical to the previous one but ready to be implemented into the AC Hugging Face Spaces.

evelynnesher, 2024-11-13:
[image]
[image]
App is public and scale is embedded into GatherTown.

evelynnesher, 2024-11-13:
Done!

evelynnesher, 2024-11-25:
[image]
I added a connection status bar that is identical to the one for the continuous output scale. I set the connection status refresh rate to 10 seconds, matching the other scale. 
The graph now skips unstable data inputs, keeping it cleaner and only showing weights of properly placed items—not the weight of your hand as you place them.

sgbaird, 2024-11-26:
Thanks! It's not urgent, but I'd rather that the unstable values be included to be more realistic to if someone had it in front of them. Likewise, there may be other uses for what would be considered unstable data.

evelynnesher, 2025-01-10:
Created a more secure connection as mentioned in #130 and made sure that the Hugging Face app still works.
[image]