AccelerationConsortium/ac-dev-lab #117: Set up second pioreactor

sgbaird, 2024-11-14:
@cyrilcaoyang is borrowing one that I think might not be used anymore.

@linx5o when we get a second one, could you set it up for more permanent deployment? This should leave one for a bit of prototyping later if needed

sgbaird, 2025-01-07:
@linx5o might not get a second pioreactor for now, unless you think there's a strong motivation for it. @cyrilcaoyang, no worries about returning the one you have. I understand it's been useful to you which is awesome 😎 we'll get another one if we need to

sgbaird, 2025-01-20:
@linx5o I ended up purchasing a second pioreactor. Could you set this up? Ideally, I'd like to have one fully set up pioreactor sent over to National Research Council this week for permanent deployment (i.e., where all that's required is to switch out the WiFi credentials).

linx5o, 2025-01-22:
I have both reactors setup. They can be accessed independently. They are currently sharing the same HiveMQ cluster.

Don't seem to see a way to change Wifi creds. Unsure about remote setup. May require reinstalling the OS.

Side note: New fan is significantly quieter, no whining compared to old one.

linx5o, 2025-01-22:
New reactor missing:

- Temp probe 
- Pump calibration
- Light sensor
- Tailscale setup
- Vial for demo

sgbaird, 2025-01-22:
That's great. Thank you! Ah, right. Forgot to get a high-T expansion kit. Maybe we leave off. I have some 12V supplies in my office I think. There may be an extra vial around in the printer office.

Could you do a quick search on the forum and open up a new topic if there isn't anything about changing WiFi credentials after initial setup?

sgbaird, 2025-01-22:
12V supply, light sensor, a set of stemma qt connectors (pick one as you see fit). I need to get more vials..

[image]

[image]

@cyrilcaoyang do you know where I can get vials similar to the ones for the pioreactor, or do you think I should just buy more from pioreactor?

linx5o, 2025-01-23:
https://forum.pioreactor.com/t/issues-connecting-to-worker-after-trying-to-switch-wifi/343

For switching wifi

sgbaird, 2025-02-15:
@linx5o did the suggestion from the forum work? If so, could you copy the instructions or line here?

linx5o, 2025-02-16:
We can use the buildin OS config I believe. [raspi-config](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-networking)

This tool can be used to change wifi.

sgbaird, 2025-03-22:
Is this second one set up and connected?

linx5o, 2025-03-22:
Sorry it wasn't. They are both connected now.

sgbaird, 2025-03-22:
All good. Thank you!