AccelerationConsortium/ac-dev-lab #119: Documentation of the wireless color sensor

Neil-YL, 2024-11-18:
I’ve completed the first draft of the documentation for the wireless color sensor, with step-by-step photos.

Here is the [preview link](https://neil-yl.github.io/wireless-color-sensor/)

I will start to work on the video.

Neil-YL, 2024-11-27:
Step-by-step assembly guide videos have been added. I think this wraps up the documentation for the hardware.

My plan for the software:

-  a simple test program run on PicoW to verify that the sensor can successfully read and output data (add it to the assembly part before closing up the sensor package). I am thinking should I use Thonny to maintain the consistency with course1, or VScode since we need to use Python for OT-2 and Prefect for later set up. Personally I prefer to keep using Thonny and micro-python as in course1 for the PicoW and sensor
-  try to obtain repeatable measure with same composition of colors mixture and also see if it can tell the difference between dark colors (say R: 60% Y:10% B:30% and R: 30% Y:10% B: 60%, if not, may consider to dilute base RYB colors)
-  Also try testing with the light source pad on the sensor board
- MQTT set up for PicoW
- Ways to install Prefect on OT-2
- OT-2 workflow (with sensor measurement), also collect some documentations by Opentrons as reference in our doc
- Visualisation of process and matching results

For the later tasks, I don’t have detailed thoughts yet and may add more ideas bit by bit as I proceed. I am welcoming any suggestions or thoughts from your perspective to help refine the plan.

sgbaird, 2024-11-27:
Simple test scripts sound good. You can refer to module 1 tutorial of the hello world microcourse as a prereq if you'd like. Keeping the instructions vague like "run the following MicroPython script using Thonny (or VS Code with the MicroPico extension)" and "here's the expected output/outcome", is fine. You can have the script inline with the documentation for the simple tests or point to an external file.

Neil-YL, 2024-12-05:
Updates on: https://github.com/AccelerationConsortium/ac-training-lab/issues/87#issuecomment-2521312788