AccelerationConsortium/ac-dev-lab #123: Automated vial capper/decapper RS485 Python API implementation

sgbaird, 2024-11-23:
At one point, I was able to verify that I could connected the RS485 to computer via USB, use one of their desktop apps, and control the state of one of the grippers (sort of a dummy check).

For a while, the plan has been to write a custom RS485 protocol that's written in Python or MicroPython using the https://www.waveshare.com/wiki/Pico-2CH-RS485. This seemed favorable over trying to hack the internals of the programmable logic circuit (PLC) or messing around with the touchscreen interface. Instead, just directly send RS485 commands with this module.

https://youtube.com/shorts/jQLtg0luPNc captures the need pretty well.

sgbaird, 2024-12-06:
@evelynnesher could you provide links to the documentation for the various DH Robotics gripper/actuator components (the linear actuator, the rotating parallel gripper, and the regular parallel gripper)? The product numbers and some relevant snippets for the RS485 commands would be great.

Please include a screenshot of the settings page on the capper/decapper. This should give us an idea of what parameters we'll need to consider for capping/decapping workflow and how to implement that with the RS485 commands, especially given we need the rotating parallel gripper and the actuator to work together with each other (or at least be sure that the protocols have started within a short timeframe of each other).

We might consider using a https://www.pishop.ca/product/rs485-can-hat-for-raspberry-pi/ with an RPi Zero 2W, RPi 4B, or RPi 5, since this might be more robust (important for safety considerations since these grippers are relatively strong)

sgbaird, 2024-12-06:
https://github.com/han-xudong/pyDHgripper will also likely be helpful, if not exactly what we want to use. See e.g., https://github.com/han-xudong/pyDHgripper/blob/main/pyDHgripper/AG95/Gripper.py. For this reason, we may want to go with an RPi single board computer (SBC) rather than the Pico W microcontroller.

Once we do get to testing with the physical equipment, I suggest we place things such that the grippers are guaranteed not to crash into each other.

evelynnesher, 2024-12-11:
### Documentation for parts used:

**Slim-Type Electric Parallel Gripper (PGE-100-26-O-S-M1)**
[PGE-Series-Internal-Controller-Short-Manual_DH-Robotics_ENV3.1.pdf](https://github.com/user-attachments/files/18099860/PGE-Series-Internal-Controller-Short-Manual_DH-Robotics_ENV3.1.pdf)
[PGE-Series-External-Controller-Short-Manual-v2_DH-Robotics.pdf](https://github.com/user-attachments/files/18099861/PGE-Series-External-Controller-Short-Manual-v2_DH-Robotics.pdf)

**Infinite Rotating Gripper (RGI-100-30-O-B-M1)**
https://en.dh-robotics.com/wp-content/uploads/2022/12/RGI-Series-Short-Manual_V3.1_DH-Robotics.pdf

**Miniature Electric Table-Type Cylinder (MCE-3WG-02-050-C-O-B)**

### Startup Screen
First screen upon turning on the device.

- **Decap**: Initiates decapping.

- **Cap**: Initiates capping.

- The three buttons (left to right) correspond to ID2, ID3, and ID4.

[image]
**ID2- Rotary Gripper**
This debug screen provides detailed information about the rotary gripper's operational states and control options.

Table featuring Gripping and Rotating States:
Displays the current status of the gripper's gripping and rotating functionalities.

Controls:

- **Open/Close**: Commands to open or close the gripper.

- **Initialize**: Performs initialization to prepare the gripper for operation.

Rotation Directions:

- **Forward Rotation**: Counterclockwise direction.
One forward rotation ends at position/angle: 1800

- **Reverse Rotation**: Clockwise direction

[image]

**ID3 - Parallel Gripper**
This debug screen provides detailed information about the rotary gripper's operational states and control options.

Table featuring Gripping State:
Displays the current status of the parallel gripper's gripping functionalities.

Controls:

- **Open/Close**: Commands to open or close the gripper.

- **Initialize**: Performs initialization to prepare the gripper for operation.

[image]
**ID4- Electric Cylinder**
This debug screen provides detailed information about the Electric cylinder.

Table featuring Gripping State:
Displays the current position and operational state of the cylinder.

Controls:

- **Up/Down**: Moves the cylinder up and down.  
  - While the cylinder is in motion, the state displays as `In Motion`.  
  - When stationary, the state is `In Place`.  
  - The position value starts at `0` and increases up to a maximum of slightly more than `5000` when moved.

- **Initialize**: Performs initialization to prepare the cylinder for operation.

[image]

evelynnesher, 2024-12-11:
From my experience with RS232, I think it’s best to start with the physical connection between the Raspberry Pi Pico, RS485 CAN Hat, and the grippers. Setting up the hardware first helps make sure everything’s stable and the signals are handled correctly, which makes debugging easier. Once the hardware is all set, I’ll move on to coding for communication and control.

https://socialcompare.com/en/comparison/raspberrypi-models-comparison
https://versus.com/en/raspberry-pi-4-model-b-vs-raspberry-pi-zero-2-w
https://versus.com/en/raspberry-pi-4-model-b-vs-raspberry-pi-5
https://versus.com/en/raspberry-pi-5-vs-raspberry-pi-zero-2-w
Based on the above sources, the RPi 4B or RPi 5 might be a better fit than the RPi Zero 2W.

sgbaird, 2024-12-19:
Two RS485 CAN HAT units arrived (we just need one). There is a Pi Zero 2W (should be plenty for what's needed).

@Neil-YL could you solder header pins onto the Pi Zero 2W for @evelynnesher?

[image]

Neil-YL, 2024-12-19:
> Two RS485 CAN HAT units arrived (we just need one). There is a Pi Zero 2W (should be plenty for what's needed).
> 
> @Neil-YL could you solder header pins onto the Pi Zero 2W for @evelynnesher?
> 
> ![PXL_20241219_181708643 MP](https://private-user-images.githubusercontent.com/45469701/397489462-886463fc-2d2f-4a36-adc4-f2be35f04f34.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2MzU0OTIsIm5iZiI6MTczNDYzNTE5MiwicGF0aCI6Ii80NTQ2OTcwMS8zOTc0ODk0NjItODg2NDYzZmMtMmQyZi00YTM2LWFkYzQtZjJiZTM1ZjA0ZjM0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE5VDE5MDYzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmYzFlNmVlMDQ0NmIwYzljNTA0MjZjZjU3YjVkMmIwOTEzNzRhNDk0ZjkwZjEyMTA5OTk1M2EzMzBkMDZmMDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.h33kqtNraeg8E04vi_U062qokWw8fd77F-ConvPqdT8)

No problem

Neil-YL, 2024-12-19:
Done. 

[image]

You can find it on the table of the vial rack, in the anti-static bag. @evelynnesher 

[image]

evelynnesher, 2024-12-19:
@sgbaird You mentioned screenshotting a settings page for the capper/decapper. Both catalogs for the RGI and PGE series list detailed settings for various commands, such as rotation speed and other parameters. Could you clarify which specific settings you had in mind?

I've attached a screenshot of section 2 from the RGI series catalog to show the various types of settings I'm referring to.
[image]

sgbaird, 2025-01-07:
There's a settings page on the touch monitor that it would be good to have a screenshot / transcribed table of. I don't remember exactly how to get to it, but I think if you poke around you'll find it.

Also, feel free to use a Pi 5 while prototyping and debugging. Once things are fairly set, we can consider moving back to the Zero 2 W, but someone brought up a good point that it slows the development cycle. These already have headers on them, and I think the HAT is compatible.

Could you also include a video of the vial capper/decapper here? May help the discussion about reimplementing the workflow from scratch.

sgbaird, 2025-01-17:
I left an extra SD card by the vial capper.

sgbaird, 2025-01-20:
@evelynnesher could you attach some camera images of the settings pages on the touchscreen?

evelynnesher, 2025-01-20:
Will do this week!

evelynnesher, 2025-01-23:
Adding extra settings
[image]
[image]
[image]
[image]
[image]

sgbaird, 2025-01-23:
Nice! The 2nd image is what I was looking for. Could you share images of the settings for the parallel gripper and the electric cylinder?

evelynnesher, 2025-01-23:
I included both of those in the comment I posted on December 11, 2024. They are quite detailed unless you’re looking for something different.

sgbaird, 2025-01-23:
I see that you attached the manuals for the individual parts and camera snapshots of the debug interfaces in https://github.com/AccelerationConsortium/ac-training-lab/issues/123#issuecomment-2536721343. The settings that are exposed via the touchscreen helps with figuring out the workflow involved which is embedded in the logic of the programmable logic circuit within the capping/decapping system. The manuals are specific to the individual components, not for the capping/decapping system as a whole (which has virtually no documentation). Those three setting pages work together to control the allowable dimensions of the vial being capped and the torque involved.

The ask is for images of each of the two tabs to the right of the one shown in:

[image]

evelynnesher, 2025-02-12:
Parallel gripper parameters
[image]
Electric cylinder parameters
[image]

evelynnesher, 2025-02-25:
I set up the Raspberry Pi to work with both an external monitor and a virtual system on a laptop to give users flexibility if they don’t have external accessories. Personally, I prefer using a monitor, mouse, and keyboard because the setup is more straightforward.

Right now, I’m debugging the RS485 connection. No other updates—I was on vacation.

sgbaird, 2025-02-25:
Virtual system meaning SSH? Something to keep in mind is that only the developers will really be using the monitor or SSH'ing into the machine. Once set up, ideally the only interaction would be via MQTT on e.g., Hugging Face.

If you could keep this issue updated as you continue the RS485 connection, that would be great. Agreed on prioritizing getting contact of some kind between the grippers and the Pi.

Hope you had a nice vacation! ☺️

evelynnesher, 2025-03-19:
[image]
Huzzah! I’ve done it! I’m able to speak to the grippers using the raspberry pi 5

[image]
Current setup uses the USB-RS485 adapter

sgbaird, 2025-03-19:
That's great! Nicely done

evelynnesher, 2025-03-23:
On March 13, I submitted the following inquiry via the DH Robotics contact form regarding RS485 communication with the Automated Decapper Demo Kit. At the time, I had not yet established what I believed to be a stable RS485 connection.

[image]

On March 23, I received the following response:

[image]

Along with these PDF files attached.

[DH Electric Cylinder Operation Manual_V2.0_EN.pdf](https://github.com/user-attachments/files/19410860/DH.Electric.Cylinder.Operation.Manual_V2.0_EN.pdf)
[RGI-series Short Manual_v3.1.pdf](https://github.com/user-attachments/files/19410862/RGI-series.Short.Manual_v3.1.pdf)
[PGE-Series-Internal-Controller-Short-Manual_DH-Robotics_ENV3.1 (13).pdf](https://github.com/user-attachments/files/19410863/PGE-Series-Internal-Controller-Short-Manual_DH-Robotics_ENV3.1.13.pdf)

sgbaird, 2025-05-22:
Aside @yakavetsiv from SDL6 using RS485 for low temperature freezer

[image]

[image]