AccelerationConsortium/ac-dev-lab #124: Custom Sample Holder (v3) - design and automation

MiceeNS, 2024-11-24:
@MiceeNS could you open a new issue focused on the sample holder? Just a brief summary of our latest ideas/challenges, image(s) of the latest sample holder design, and a link back to this comment.

MiceeNS, 2024-11-24:
### Sample Holder v1
The **1st version** of the sample was designed from two components:
1. Main Body
2. Closing Cap

Both components are assembled with snap fit joints that are part of the closing cap. 

1. The conductor is routed through the main body "tunnel" and it would only surface at the point of contact. 
2. Closing Cap accommodates the sample. 

Joining both components would create a seal and electrical contact between the sample and the conductor at the point of surface.

The design is 3d printed with resin v4.

_This design had multiple flaws:_ 
1. The snap fit joint broke after 1 snap

- [resin v4 elasticity] 

4. Conductor could not be routed through the main body "tunnel" 

- [tunnel route angles were demanding for the conductor bending] 

**Experiment not conducted**

Reference pictures:
[image]
[image]
[image]
[image]

sgbaird, 2024-11-25:
Amazing, thank you for the clear summary! Slight clarification: 

> 8. Conductor is supposed to be easily inserted and removed from the holder

I think this one is not as critical. For example, I wouldn't mind if the conductor were permanently fixed into the sample holder with additional resin that gets cured after the conductor has been inserted.

##

In the previous discussion, we've been leaning towards some kind of threaded mechanism to allow the user to tighten it to a particular pressure. We've also verified that we can do threaded connections with clear v4 resin. It might be difficult to remove the sample after a long test in 80 deg. C KOH, but I think it would be OK to leave that for later testing, and I think it could be solved with some Teflon tape around the threads.

It may be worth again considering two long, thin, flat strips of a SS sheet going into the holder with the SS slightly raised where the electrical connection with the sample is meant to take place. I think this would provide a larger surface area of contact without leading to damage of the sample (i.e., larger surface area / more uniform pressure) compared with a round wire. I think that the threaded mechanism would provide a reasonable amount of pressure. I've worked with some really thin SS sheets before (can be cut with regular scissors, rather flexible). I'm not sure what sizes are standard for SS strips / "flat wire" (I had some McMaster links before), but I don't think it would be a big deal to assume cutting a strip from a larger sheet manually.

This would also likely require a redesign such that both SS strips are inserted straight into the holder vertically, with only that slight bend near the point of contact. A connection would need to be made between the two sheets (either by using a larger sheet and cutting out a smaller rectangle to make a sort of upside down "U" shape or making the connection between the two with some kind of clamp. If the SS is thin enough, we could also do a bit of origami with an extra long strip of SS ☺️ I lean a bit towards the origami solution.

Lmk and I can draw some diagrams.

programlich, 2024-11-25:
Thanks for the great summary of our work so far @MiceeNS and the cool ideas for the next design @sgbaird.

I agree, that the removability of the electrical conductor is not a high prio. It would be a nice addon for when we start using Pt-wires, as these are a bit costly but not too important for now. I like the idea with the flat stainless steel strips, as we would natively get a broader contact area this way.
Also, we dont really have the get the wire/strip back into the sample holder like in the current design. Having them come out of the sample holder at one point and being in contact with the sample is enough. This way we only would have to seal 1 hole per side against KOH. Talking to our experienced colleagues I found out, that sealing can be quite tricky on the long term, because KOH really gets into everything after many weeks of exposure. So this is a details which we might give some extra attention. Curing of the resin after the insertion of the wire seems a very promising approach to me.

What's your opinion on the closing mechanism @MiceeNS? Would you also give threads a try or do you prefer improving the current design?

MiceeNS, 2024-11-25:
Yes I will do now a design with threads(because I do not want to waste additional time figuring out if something works or not), and I am thinking of a design with the stainless strips. However, after we print the design with the threads and hopefully everything works and there is good electrical contact I can think of some additional mechanism. Also will update this issue with the v2 today

MiceeNS, 2024-11-26:
Check this design out and let me know, next design I will do with flat wire. Threads are "custom".

Also, I just noticed some important fillets are missing.

[image]
[image]

sgbaird, 2024-11-26:
Oh, interesting! I was imagining the whole mating piece being the threaded piece (rotating the main piece is the locking mechanism), rather than having multiple screws on the outside. @programlich what are your thoughts? What are you picturing when you're thinking of using a threaded design?

In the current design, do you think we'd need longer screws and a custom nut, too? (The nut just for an extra bit of security).

If we take this approach, maybe worth considering using Teflon screws and nuts: https://www.mcmaster.com/products/screws/material~ptfe/.

I think this design could work well without a lot of iterations required. Maybe worth having four screws in a square pattern to give a more even pressure distribution of the sample contacting the SS.

MiceeNS, 2024-11-26:
> Oh, interesting! I was imagining the whole mating piece being the threaded piece (rotating the main piece is the locking mechanism), rather than having multiple screws on the outside. @programlich what are your thoughts? What are you picturing when you're thinking of using a threaded design?

I thought of that but you might not always get the sample straight or it might fall off.
> 
> 
> In the current design, do you think we'd need longer screws and a custom nut, too? (The nut just for an extra bit of

Good point, can add tomorrow.
> 
> 
> If we take this approach, maybe worth considering using Teflon screws and nuts: https://www.mcmaster.com/products/screws/material~ptfe/.

I think we wont need teflon unless tolerances are way off. I think teflon is more for a leaking problem in threaded connections.

> 
> 
> I think this design could work well without a lot of iterations required. Maybe worth having four screws in a square pattern to give a more even pressure distribution of the sample contacting the SS.

Totally agree.

sgbaird, 2024-11-26:
> I thought of that but you might not always get the sample straight or it might fall off.

Great point. Now that you mention it, that sounds like a headache trying to not rotate the sample during it.

> I think we wont need teflon unless tolerances are way off. I think teflon is more for a leaking problem in threaded connections.

I was thinking more so in terms of standardization. I think Teflon would be less likely to get "stuck" or have some kind of unintentional bonding compared with the clear v4 resin (while also having the best chemical resistance), but we don't know if that's a real problem for the resin.

##

Aside: I found a potential candidate for the SS 316, which I think we would just cut to shape with scissors or similar. I don't think there's a product listed as "flat wire" that is thin enough and wide enough. See https://www.mcmaster.com/product/3774T55

The next closest thing I found was a strip ~4x thicker, and 1" wide (too wide): https://www.mcmaster.com/product/9090K3-9090K53.

MiceeNS, 2024-11-28:
We can test print one spring, in the name of science :laughing: 

[image]
[image]

Neil-YL, 2024-12-05:
[image]
Is this the printed spring? I also noticed several other printed parts, such as nuts and bolts.

This print has supports that are very challenging to remove.

MiceeNS, 2024-12-05:
Yes those are the parts. Test the bolt to the nuts and try what you can do
with the spring.

Let me know how it goes

sgbaird, 2024-12-06:
@Neil-YL thanks for your help!

@MiceeNS it looks like you sent the print for the springs and the bolts to the Bambu Lab printer, not the Formlabs printer. The former (PLA) isn't chemically compatible, and the clear v4 resin is much more rigid.

@MiceeNS could you identify a suitable combination of "[Extreme-Temperature Chemical-Resistant PTFE Pan Head Slotted Screws](https://www.mcmaster.com/products/screws/material~ptfe/rounded-head-screws~/rounded-head-style~pan/)" and [PTFE Hex Nuts](https://www.mcmaster.com/products/ptfe-nuts/material~ptfe/?s=ptfe-nuts) based on the model, i.e., whatever is close in size and length to what you're hoping for with.

[image]

If you could add it to the 3D model (and include some snapshots) that would be great so you make sure the size and length look OK. I suggest having the through-hole in the sample holder straight without threads so we can avoid using a tap and just use the nut.

Could you also identify [a corresponding taper chamfer tap](https://www.mcmaster.com/products/taps/taps-3~/economy-carbon-steel-taps/chamfer-type~taper/) that would allow us to create or clean up threads? Then, print a couple different sizes of pucks with holes (with and/or without threads) to help with seeing what would work best with the tap and the teflon screws. This will provide some flexibility for us to pivot to a nut-free design if we wish.

sgbaird, 2024-12-06:
For context, I'd like to get a set of five screws, five nuts (1 extra in case of breakage), and a tap ordered sometime soon. I'm also planning to get [the 316 SS linked above](https://www.mcmaster.com/3774T55/).

MiceeNS, 2024-12-06:
> @Neil-YL thanks for your help!
> 
> @MiceeNS it looks like you sent the print for the springs and the bolts to the Bambu Lab printer, not the Formlabs printer. The former (PLA) isn't chemically compatible, and the clear v4 resin is much more rigid.
> 
> @MiceeNS could you identify a suitable combination of "[Extreme-Temperature Chemical-Resistant PTFE Pan Head Slotted Screws](https://www.mcmaster.com/products/screws/material~ptfe/rounded-head-screws~/rounded-head-style~pan/)" and [PTFE Hex Nuts](https://www.mcmaster.com/products/ptfe-nuts/material~ptfe/?s=ptfe-nuts) based on the model, i.e., whatever is close in size and length to what you're hoping for with.
> 
> <img alt="" width="200" src="https://private-user-images.githubusercontent.com/45469701/393370431-e02bd46c-6802-45eb-bbf4-d7ae26a65435.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1MTIyMjEsIm5iZiI6MTczMzUxMTkyMSwicGF0aCI6Ii80NTQ2OTcwMS8zOTMzNzA0MzEtZTAyYmQ0NmMtNjgwMi00NWViLWJiZjQtZDdhZTI2YTY1NDM1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA2VDE5MDUyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4ZjY2ODE1MmE4Mjk2ZjFmN2I3ZjQ1MGFmOThmMzUwMWNlNTRjZWE3Zjk2Mzc0NDM2NzhjMTczMjQ4NDQ0NzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.d5au7MUFM-hp4NC7xubNxBuwG2Ff75CXJwyxIhTVQZo">
> If you could add it to the 3D model (and include some snapshots) that would be great so you make sure the size and length look OK. I suggest having the through-hole in the sample holder straight without threads so we can avoid using a tap and just use the nut.
> 
> Could you also identify [a corresponding taper chamfer tap](https://www.mcmaster.com/products/taps/taps-3~/economy-carbon-steel-taps/chamfer-type~taper/) that would allow us to create or clean up threads? Then, print a couple different sizes of pucks with holes (with and/or without threads) to help with seeing what would work best with the tap and the teflon screws. This will provide some flexibility for us to pivot to a nut-free design if we wish.

Yes, I send them on the bambulab on purpose to see if the prints would actually work and the thread could fit.  Regarding the nuts such a good idea! Will do and send as soon as possible, buy and ship them to you.

MiceeNS, 2024-12-07:
Update regarding the last changes.  Bolts nuts and tap is ordered, lets figure out the design for flat wire then I will order the sheet metal also.
[image]
[image]
[image]
[image]

MiceeNS, 2024-12-09:
Some renders..

[image]
[image]
[image]
[image]

https://github.com/user-attachments/assets/8a1e6ccc-db58-4202-84ce-4069e996f248

[image]

MiceeNS, 2024-12-11:
For testing the Tap with modified threads sent for print on Bambulab. If you received the components from McMaster Carr this can be tested with the bolts.

[image]
[image]

programlich, 2024-12-11:
short question regarding the electrical contact on top of the sample holder: Do you have a link to the part, which is meant to connect the wire to the contact? Also, did you get a chance to brainstorm the counterpart on the tap of the cell? I´d really like to see, what kind of electrical connection this is

MiceeNS, 2024-12-11:
> short question regarding the electrical contact on top of the sample holder: Do you have a link to the part, which is meant to connect the wire to the contact? Also, did you get a chance to [...]

The part that connects the wire to the "connector" is just a bolt something similar like in the picture below, you insert the wires inside the compress them with the bolt
[image]

The "connector" is like small pin with curved face on the counter part I intend to put compression spring so when you insert the sample holder that counter part can retract a bit and have a constant force for a better electrical contact. Picture of the connector:

[image]

Other idea that I had is with socket connectors, something like in the picture. The female socket on the tap itself
[image]

And the male part on the sample holder 
[image]

If we go with flat wire, we can use this type of socket to connect both wire in the sample holder
[image]

sgbaird, 2024-12-12:
I lean heavily towards using a molex connector or a banana plug. See https://github.com/AccelerationConsortium/ac-training-lab/issues/104#issuecomment-2487458489

One advantage of a Molex connector is that they usually involve crimping rather than soldering, as opposed to banana plugs which typically require soldering (and then we need to figure out a good way to get SS soldered to the banana plug in a robust way).

> NOTE: As I've been going through the thought and search process, I think I have a good solution.
> 
> https://www.google.com/search?q=mcmaster+mating+electrical+connection --> https://www.mcmaster.com/products/electrical-connectors/ --> https://www.mcmaster.com/products/electrical-connector-contacts/manufacturer~molex/ --> (https://www.mcmaster.com/product/69295K111 + https://www.mcmaster.com/product/69295K115)
> 
> (Related follow up search that also led me to realize the wire gauges mean for crimping, not for the insertion/deinsertion pins, which are selected in pairs): https://www.google.com/search?q=mcmaster+molex+signal+power+connector+mating+single+pin)
> 
> I've used custom molex connections before. I think we should go with this. This doesn't address the gas tightness of the sample holder in the cell (the linear actuator idea can both seal an o-ring and create an electrical connection), but there might even be a similar solution for that (maybe simply applying a bit of grease to the mating surfaces of the sample holder and cell would be sufficient, similar to when using ground glass tapered fittings).
> 
> We might just need some kind of alignment piece (maybe just printed) that keeps the molex connector of the sample holder fixed vertically and in a repeatable xyz location. Banana plugs come to mind, too, and are pretty standard (https://www.mcmaster.com/products/banana-connectors/test-end-component~banana-plug). Same principle.
> 
> To illustrate this idea:
>
> [image]

There would also likely be a fake tip slightly offset from the center of the sample holder, to allow for picking up and moving sample holders around.

Off-center for the sample holder fake tip because I think it would be easiest to get the desired electrical connection by having the molex connector or banana plug fully centered on the sample.

Aside: flat wire = thin SS sheet that's been cut (regular scissors are fine in my experience) such that it's like an upside down U (imagine it's unrounded) with a small "line thickness". This also assumes there are two insertion points from the top, which allows for having a direct electrical connection on both sides of the sample  in a straightforward way.

Lmk if a diagram would help.

programlich, 2024-12-12:
I´m not sure if I get your idea completely, @sgbaird. Would your design idea be an electrical connection 

**potentioastat - tap - sample holder** 
or would it rather be 
**potentiostat - sample holder**?

The advantage of the first one is, that we would only have to move the sample holder around and didn´t have to worry about (dis)connecting the electrical connection from the sample holder manually with every sample exchange.

Besides this, I agree with you that banana plug/molex has quite an advantage because once the connection is established, it is always the same, so we could be sure, to have a constant electrical resistance regardless of slight differences in the positioning of the sample holder.

programlich, 2024-12-12:
> The part that connects the wire to the "connector" is just a bolt something similar like in the picture below, you insert the wires inside the compress them with the bolt

This looks completely fine to me. Easy and functional.
 Regarding the connection sample holder - (tap) - potentiostat I am leaning more to a standardized connector. See answer above.

sgbaird, 2024-12-12:
Thanks for the great comments! The desire for being able to move electrodes around is something more specific to my goals/vision. I want to make it easier to reuse the same potentiostat for entirely separate workflows in the same fumehood/lab, without having to manually go in and swap electrode connections each time. The current focus is on the working electrode, but the same principle applies to the counter and reference electrodes.

Also, I remembered that I'm leaning away from a "fake tip" with an OT-2 pipette towards instead using a low-cost robotic arm (specifically MyCobot 280 Pi is the one I would first try with, then move to something a bit more expensive/higher performance if needed). Instead of a fake tip, it would be something easy to grip onto with the corresponding gripper. This could be using a [parallel gripper](https://shop.elephantrobotics.com/en-ca/products/mycobot-gripper-parallel), [adaptive gripper](https://shop.elephantrobotics.com/en-ca/products/adaptive-gripper), or [suction gripper](https://shop.elephantrobotics.com/en-ca/collections/mybuddy/products/suction-pump-2-0) with the MyCobot. @MiceeNS, thoughts? Looping @gursi26 and @SissiFeng to weigh in, since both of you have had experience with this arm and with robotic arms in general. See https://accelerated-discovery.org/t/455 for context.

> I´m not sure if I get your idea completely, @sgbaird. Would your design idea be an electrical connection
> 
> **potentioastat - tap - sample holder** or would it rather be **potentiostat - sample holder**?

It would be potentiostat -- "molex holder" -- sample holder. The workflow would be:

1. Initialize experiment - molex holder (connected to working electrode lead) is sitting in a storage rack. Sample holder with the preloaded sample is sitting in a storage rack. Potentiostat reference electrode and counter electrode leads are connected to the electrochem cell, which is waiting to have a sample loaded.
2. Cobot arm picks up a sample holder from storage and places it into the electrochem cell
3. Cobot arm picks up the molex holder from storage and pushes the molex plug into the sample holder
4. Run electrochemical experiment
5. Cobot arm removes the molex holder from the sample holder and places it back in storage
6. Cobot arm removes the sample holder from the cell and places it back in storage
7. Repeat steps 2-6

Step 5 will likely require some trial-and-error. For example, will the mechanical connection of the molex connector be so strong that it picks up the sample holder along with it? (seems unlikely to me based on weight of the sample holder and my experience with molex connections -- banana plugs would be a different story, I could imagine a banana plug picking up the sample holder, in which case we'd need a workaround to separate the two, which has a precedent in the N9 platform ([figure 1.D](https://www.science.org/doi/10.1126/sciadv.aaz8867)) and [digital pipette v2](https://github.com/ac-rad/digital-pipette) (not yet released) with a syringe and a tip)

> The advantage of the first one is, that we would only have to move the sample holder around and didn´t have to worry about (dis)connecting the electrical connection from the sample holder manually with every sample exchange.

True, as long as we can get the connection/disconnection to be fairly reliable. There's an equivalent option, which is similar to what @MiceeNS has brought up. The "molex holder" would be embedded into the tap (with the connector pointing up instead of down) and the sample holder molex connector pointing down instead of up.

> Besides this, I agree with you that banana plug/molex has quite an advantage because once the connection is established, it is always the same, so we could be sure, to have a constant electrical resistance regardless of slight differences in the positioning of the sample holder.

👍

SissiFeng, 2024-12-13:
I have some ideas. It looks like there are two items that need to be grabbed.
1.sample Holder: fitted with a “fluted holder” that acts like a tray, the bottom of the sample holder can be securely recessed into the fluted area. Guide slots may need to be added for alignment.
2. molex holder: fitted with a “U-shaped gripper” with a flexible slot on the bottom to hold the Molex plug.

Swivel adapter for switching between two gripping heads:
triangular or circular platforms, each facing a gripping head interface, driven by a small stepper motor to control the switching of gripping heads between segmented positions.

Neil-YL, 2024-12-13:
> For testing the Tap with modified threads sent for print on Bambulab. If you received the components from McMaster Carr this can be tested with the bolts.
> 
> ![Screenshot 2024-12-10 223911](https://private-user-images.githubusercontent.com/76826784/394561261-dbee8a21-28fd-41f7-965f-74b30a741d44.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMDA0MjEsIm5iZiI6MTczNDEwMDEyMSwicGF0aCI6Ii83NjgyNjc4NC8zOTQ1NjEyNjEtZGJlZThhMjEtMjhmZC00MWY3LTk2NWYtNzRiMzBhNzQxZDQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDE0Mjg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhMjZhMDhiMzgzNTU4NjEyNmYxNDVmYWExNzFiNWNlYzA0YzBhZTE3MmNkZDU2NzFhYjJmOTMzYWMyZTU4ODkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SDnpYLuwy22xJMPpE2joWoDRjPL6YHnFqz68QmoKqvE) ![Screenshot 2024-12-10 223952](https://private-user-images.githubusercontent.com/76826784/394561257-4493bc1e-7d89-446c-a10c-10717f5ca874.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMDA0MjEsIm5iZiI6MTczNDEwMDEyMSwicGF0aCI6Ii83NjgyNjc4NC8zOTQ1NjEyNTctNDQ5M2JjMWUtN2Q4OS00NDZjLWExMGMtMTA3MTdmNWNhODc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDE0Mjg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmM2IxMmNiMWJkZGZlOTljYmVjZTYwZmNkYmNlNzdlNWU4M2IyY2EyMWZkMGEwMmQ2ZWU1NjY1NzZhZTBlY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.fRbjbsMNtEOqV-kR9azHz5pOSDZNsNSuLDK5nqRv2Fo)

I have picked up this printed part, are there any tests I need to carry out with it?
The threads seem to be very thin not sure if they're within the precision of our 3D printer.

MiceeNS, 2024-12-13:
> I have picked up this printed part, are there any tests I need to carry out with it? The threads seem to be very thin not sure if they're within the precision of our 3D printer.

The threads are for testing standardized NPT ( 4-40 screw at the printed part) and nothing I could change there. We need to see if the 4-40 can fit in the thread. the holes without thread should be threaded with the TAP and see if the bolts can be screwed.

If the bolts do not fit in the printed threads, try to thread those also with the TAP.  Basically what we want to test if its possible to print thread that we can get rid of the nuts in the design.

Bolts tap and the nuts should be -> This can go into the training lab space on the 2nd floor on 2620 Speakman Drive, Mississauga 

Thank you,

MiceeNS, 2024-12-13:
> I lean heavily towards using a molex connector or a banana plug. See [#104 (comment)](https://github.com/AccelerationConsortium/ac-training-lab/issues/104#issuecomment-2487458489)
> 
> One advantage of a Molex connector is that they usually involve crimping rather than soldering, as opposed to banana plugs which typically require soldering (and then we need to figure out a good way to get SS soldered to the banana plug in a robust way).
> 
> > NOTE: As I've been going through the thought and search process, I think I have a good solution.
> > https://www.google.com/search?q=mcmaster+mating+electrical+connection --> https://www.mcmaster.com/products/electrical-connectors/ --> https://www.mcmaster.com/products/electrical-connector-contacts/manufacturer~molex/ --> (https://www.mcmaster.com/product/69295K111 + https://www.mcmaster.com/product/69295K115)
> > (Related follow up search that also led me to realize the wire gauges mean for crimping, not for the insertion/deinsertion pins, which are selected in pairs): https://www.google.com/search?q=mcmaster+molex+signal+power+connector+mating+single+pin)
> > I've used custom molex connections before. I think we should go with this. This doesn't address the gas tightness of the sample holder in the cell (the linear actuator idea can both seal an o-ring and create an electrical connection), but there might even be a similar solution for that (maybe simply applying a bit of grease to the mating surfaces of the sample holder and cell would be sufficient, similar to when using ground glass tapered fittings).
> > We might just need some kind of alignment piece (maybe just printed) that keeps the molex connector of the sample holder fixed vertically and in a repeatable xyz location. Banana plugs come to mind, too, and are pretty standard (https://www.mcmaster.com/products/banana-connectors/test-end-component~banana-plug). Same principle.
> > To illustrate this idea:
> > ![17339914566523172421278705913037](https://private-user-images.githubusercontent.com/45469701/395068630-b56ed242-edac-4acf-afc9-23fa4c42e094.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMDkwNjEsIm5iZiI6MTczNDEwODc2MSwicGF0aCI6Ii80NTQ2OTcwMS8zOTUwNjg2MzAtYjU2ZWQyNDItZWRhYy00YWNmLWFmYzktMjNmYTRjNDJlMDk0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDE2NTI0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3Mzk0Y2FmOWViYTU3ODIyYTIxYzI0ZjkzNzY0MmE0Y2Y1Y2RhNDliNzhmNTA2NDc3ZjUyMGU2Yzg4Njg2OTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-fpitkYNIOsd0CDsGfXq-KYuA8rhloGSSeMcRXiFBRE)
> 
> There would also likely be a fake tip slightly offset from the center of the sample holder, to allow for picking up and moving sample holders around.
> 
> Off-center for the sample holder fake tip because I think it would be easiest to get the desired electrical connection by having the molex connector or banana plug fully centered on the sample.
> 
> Aside: flat wire = thin SS sheet that's been cut (regular scissors are fine in my experience) such that it's like an upside down U (imagine it's unrounded) with a small "line thickness". This also assumes there are two insertion points from the top, which allows for having a direct electrical connection on both sides of the sample in a straightforward way.
> 
> Lmk if a diagram would help.

Neil-YL, 2024-12-13:
> The threads are for testing standardized NPT ( 4-40 screw at the printed part) and nothing I could change there. We need to see if the 4-40 can fit in the thread. the holes without thread should be [...]

My previous experience for small threads on 3D printed part is using [embedment nuts](
https://www.amazon.ca/Hilitchi-Knurled-Threaded-Pressed-Projects/dp/B07VFZWWXY/
) or thread-head inserting into the 3D-printed holes, may need to apply some glue for critical connections.

[image]

MiceeNS, 2024-12-13:
> > > > For testing the Tap with modified threads sent for print on Bambulab. If you received the components from McMaster Carr this can be tested with the bolts.
> > > > ![Screenshot 2024-12-10 223911](https://private-user-images.githubusercontent.com/76826784/394561261-dbee8a21-28fd-41f7-965f-74b30a741d44.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMDA0MjEsIm5iZiI6MTczNDEwMDEyMSwicGF0aCI6Ii83NjgyNjc4NC8zOTQ1NjEyNjEtZGJlZThhMjEtMjhmZC00MWY3LTk2NWYtNzRiMzBhNzQxZDQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDE0Mjg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhMjZhMDhiMzgzNTU4NjEyNmYxNDVmYWExNzFiNWNlYzA0YzBhZTE3MmNkZDU2NzFhYjJmOTMzYWMyZTU4ODkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SDnpYLuwy22xJMPpE2joWoDRjPL6YHnFqz68QmoKqvE) ![Screenshot 2024-12-10 223952](https://private-user-images.githubusercontent.com/76826784/394561257-4493bc1e-7d89-446c-a10c-10717f5ca874.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMDA0MjEsIm5iZiI6MTczNDEwMDEyMSwicGF0aCI6Ii83NjgyNjc4NC8zOTQ1NjEyNTctNDQ5M2JjMWUtN2Q4OS00NDZjLWExMGMtMTA3MTdmNWNhODc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDE0Mjg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmM2IxMmNiMWJkZGZlOTljYmVjZTYwZmNkYmNlNzdlNWU4M2IyY2EyMWZkMGEwMmQ2ZWU1NjY1NzZhZTBlY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.fRbjbsMNtEOqV-kR9azHz5pOSDZNsNSuLDK5nqRv2Fo)
> > > 
> > > 
> > > I have picked up this printed part, are there any tests I need to carry out with it? The threads seem to be very thin not sure if they're within the precision of our 3D printer.
> > 
> > 
> > The threads are for testing standardized NPT ( 4-40 screw at the printed part) and nothing I could change there. We need to see if the 4-40 can fit in the thread. the holes without thread should be threaded with the TAP and see if the bolts can be screwed.
> > If the bolts do not fit in the printed threads, try to thread those also with the TAP. Basically what we want to test if its possible to print thread that we can get rid of the nuts in the design.
> > Bolts tap and the nuts should be -> This can go into the training lab space on the 2nd floor on 2620 Speakman Drive, Mississauga
> > Thank you,
> 
> My previous experience for small threads on 3D printed part is using [embedment nuts](https://www.amazon.ca/Hilitchi-Knurled-Threaded-Pressed-Projects/dp/B07VFZWWXY/) or thread-head inserting into the 3D-printed holes, may need to apply some glue for critical connections.
> 
> <img alt="image" width="1136" src="https://private-user-images.githubusercontent.com/179746567/395644024-170ab351-d054-4070-9fff-132c3aa0dbf8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMTA3NTgsIm5iZiI6MTczNDExMDQ1OCwicGF0aCI6Ii8xNzk3NDY1NjcvMzk1NjQ0MDI0LTE3MGFiMzUxLWQwNTQtNDA3MC05ZmZmLTEzMmMzYWEwZGJmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxM1QxNzIwNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYmQwZjlkNTdjYmVjNTVjNzFjNDIzOTUyMWY1MzM1ZDAxZjc2MDc2OTVhYWNlYzhjY2FmODQ5ZjQ4YWM2MzY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.08C-Dz2xsKjmXrZ2DFRQ6vKTkKW646z04LNaFI9uujI">

We can not use because they are not chemically compatible and metal can interfere with the electro magnetic fields. The real saga

sgbaird, 2024-12-14:
We don't have anyone regularly going to NRC from the AC, so it may take some time before we get the bolts/nuts/tap back to UoT for testing.

programlich, 2024-12-16:
Thank you all for the great ideas. A lot is going on here!

About the electrical contact:
> I want to make it easier to reuse the same potentiostat for entirely separate workflows in the same fumehood/lab, without having to manually go in and swap electrode connections each time. The [...]

Now I understand you better, @sgbaird. I think this is a very valid point, which we need to address even though counter and reference electrode have a lower priority than the working electrode, as they can be used for at least a week. However, I agree, that we should standardize the electrical connection **potentiostad - any electrode** to simplify the future exchange process.
Personally, I still think the most robust way for the first automated design would have the potentiostats cables attached to the tap, to only have one automated movement, that combineds positioning and electrical contact. The honest reason why I prefer this for now is that I would like to use this setup for my PhD and there are only 1,5 years left so I´m in favor with any design that provides a robust and automated workflow in reasonable time.

On the weekend, I discussed this topic with an electrical engineer (thanks @ggramlich26 ) and he suggested to use [spring loaded pogo pins](https://www.ccpcontactprobes.com/application/7-mounting-options-spring-loaded-connectors-pogo-pins) for the electrical contact. The spring allows a reliable contact and these parts are commercially available with cold coating, which would be great in terms of corrosion resistance. The pogo pins might be loaded on a little PCB, which also connects to a banana plug/whatsoever to the potentiostat.
Here´s an idea about how this might work for us:

https://github.com/user-attachments/assets/b1c45e03-79b3-49e8-ab5b-2a25a8ba01c9

[image]

For the future, we might think about building small adapters for the reference and working electrode, so they can also be connected via the pogo pins. What do you think, @sgbaird? Would this be an option?

programlich, 2024-12-16:
@MiceeNS, could you please post the specs of the PTFE bolts and nuts for this sample holder design? I might have the chance to print a sample holder with a formlabs printer here in Germany in the near future and would like to order the bolts & nuts

sgbaird, 2024-12-16:
Line Product Ordered Shipped
1 25995A125 Economy Carbon Steel Tap, Taper Chamfer, 4-40 Thread Size, 9/16" Thread Length 1
Each
1
2 94701A611 PTFE Plastic Screw, Extreme-Temperature, Pan Head Slotted, 4-40 Thread, 1/4" Long 5
Each
5
3 90059A005 High-Temperature Chemical-Resistant PTFE Hex Nut, 4-40 Thread Size, Packs of 1 5
Packs
5

[image]

programlich, 2024-12-17:
The stainless steel wire by now shows quite some signs of corrosion after using it in hot KOH at different potentials.

[image]

For the sake of cost reduction the nicest thing would be to have a stainless steel or copper wire where only the end is coated/massive platinum. That way only Platinum would be exposed to KOH while the wire which is inside the sample holder and thereby not exposed could be of any cheap material.

sgbaird, 2024-12-17:
Looks like the image upload might not have processed all the way. Could you edit the post and reattach?

programlich, 2024-12-17:
AS a  sidenote @sgbaird : would you mind creating a github project for us, so we can keep track of all the minor things we need to address?

sgbaird, 2024-12-17:
> use [spring loaded pogo pins](https://www.ccpcontactprobes.com/application/7-mounting-options-spring-loaded-connectors-pogo-pins) for the electrical contact. The spring allows a reliable contact

Is this the mechanism I should be picturing? https://youtu.be/-oxunE-RMjA. From the video, it seems like it depends on mechanical pressure being constantly applied to maintain the connection. Is that correct? Not sure how that would affect the results (similar to what was mentioned about the sample to wire connection not being robust enough).

sgbaird, 2024-12-17:
> AS a sidenote @sgbaird : would you mind creating a github project for us, so we can keep track of all the minor things we need to address?

Added you and @MiceeNS as admins, and made the project publicly visible. I didn't add much to it - feel free to customize as needed: https://github.com/orgs/AccelerationConsortium/projects/5

MiceeNS, 2024-12-17:
> Is this the mechanism I should be picturing? https://youtu.be/-oxunE-RMjA. From the video, it seems like it depends on mechanical pressure being constantly applied to maintain the connection. Is [...]

Yes those are the pins. This link is to the same ones I used in the design https://www.mill-max.com/products/spring-loaded/pogo-pin-header-strip/827/10-002101

In the description of the pins it says: 
Grams Force @ Mid Stroke = 60g
Initial Force (Pre-Load) = 5g

So I think that will be a problem

Also useful link: https://www.ptr-hartmann.com/know-how/spring-loaded-connectors/

programlich, 2024-12-18:
> So I think that will be a problem

You mean that's a problem because of too much force pressing the sample holder up? Perhaps we can find Pogo pins with less spring force?

programlich, 2024-12-18:
> Added you and @MiceeNS as admins, and made the project publicly visible. I didn't add much to it - feel free to customize as needed: https://github.com/orgs/AccelerationConsortium/projects/5

Done. Please feel free to add and modify anything.

MiceeNS, 2024-12-19:
Update:
[image]

Do you think the banana jack should be from the top or from the side? Like this the banana jack would create a box that would be 40mm height. Another option is to find smaller banana jack or place it sideways or remove it from the PCB. The pins by itself are small but placing a banana jack next to them for the potentiometer is a problem.

sgbaird, 2024-12-19:
Is it banana jack *and* pogo pins? I was under the impression that the pogo pin suggestion was in place of the banana jack. Also, should be from top.

Also, I'm a bit confused why there's a potentiometer.

Is what you showed above an example of three possible designs?

programlich, 2024-12-19:
I think it depends if we want to have the banana jacks removable with a robot arm in the future. If that's the plan, we should have it placed from the top but if we plan to leave it more permanently connected to the cell, it should be from the side to leave more room.

Also, please try to not place the PCB into the tap, but rather on top of it. I am worried about taking material away from a component which will be attacked by hot KOH vapor.

@sgbaird, I think Bojan refers to the Potentiostat as the connection with the current design would be sample holder -> pogo pins -> PCB -> banana jack.

I think it would be best to setup a meeting with the three of us because here's a rather important decision to be taken which is best discussed in person.

programlich, 2024-12-19:
@MiceeNS, one question regarding the ptfe screws for the current design of the sample holder. Would metric M3 screws with 8mm length fit through the designed holes? I am trying to order the screws and bolts but it is challenging to get ptfe screws in imperial dimensions here.
Same thing for the nuts - would metric nuts fit?

programlich, 2024-12-19:
Printed  spring with Formlabs high temp resin:

https://github.com/user-attachments/assets/6c9d3260-7f0a-426b-8c55-a03db4c1929d

Looks pretty promising to me. Thanks @ggramlich26

MiceeNS, 2024-12-20:
> Is it banana jack _and_ pogo pins? I was under the impression that the pogo pin suggestion was in place of the banana jack. Also, should be from top.

Because the potentiostat has banana pin.

> Also, I'm a bit confused why there's a potentiometer.
I was thinking of potentiostat
> Is what you showed above an example of three possible designs?
No it is only one, with the box, PCB, banana jack, 2 set of screws  and heat inserts.

> @MiceeNS, one question regarding the ptfe screws for the current design of the sample holder. Would metric M3 screws with 8mm length fit through the designed holes? I am trying to order the screws [...]

The screw clearance is okey, however from wherever you order them check the nut size. On Mcmaster the metric is bit smaller and the cutout is a bit big for that nut size. I can adjust that quick. Let me know if you want it changed before you print.
1/4"=6.35mm

[image]
[image]

programlich, 2024-12-20:
If it's something that can be done quickly, it would be nice, if you could adjust the but cutout. No pressure though, as I'm in holidays now :)

MiceeNS, 2024-12-23:
@programlich 
For M3 Screws and nuts, please confirm nut size before ordering or printing..
I also adjusted the distance between the washer so you could potentially have more pressure
Are you going to order the screws from McMaster?? or you will get them local, let me know if you need some additional help
[Sample Holder V3.1 Metric.zip](https://github.com/user-attachments/files/18223801/Sample.Holder.V3.1.Metric.zip)

programlich, 2025-01-09:
@MiceeNS are the specs of the screw and nut from above fine? Also, I need to decide, which length to order for the srew. Is 10mm ok, or should I order longer ones?

Also, for testing I'll order a bunch of PEEK screws, too:
- https://highperformancepolymer.de/products/peek-hex-socket-cylinder-head-cap-screw?variant=44169074802986
- https://highperformancepolymer.de/products/peek-hexagon-nuts?variant=44169069560106

MiceeNS, 2025-01-10:
@programlich  I do not know how I skipped your first message. The links you have sent are M6 and you should be looking for M3 screw and nuts. Screw length of 6mm should be fine but not below that. I would prefer 8mm

Links from Mcmaster Carr for reference:
https://www.mcmaster.com/94701A197/
https://www.mcmaster.com/90059A115/

From your websites:
https://highperformancepolymer.co.uk/en-ca/products/ptfe-hex-socket-cylinder-head-cap-screw?variant=46518645096746
https://highperformancepolymer.co.uk/en-ca/products/ptfe-hexagon-nuts?variant=40858647691459

PEEK:
https://highperformancepolymer.co.uk/en-ca/products/peek-hex-socket-cylinder-head-cap-screw?variant=44169073918250
https://highperformancepolymer.co.uk/en-ca/products/peek-hexagon-nuts?variant=44169069461802

NOTE: This nuts are bit bigger then the ones from McMaster Carr which is on what I initially draw the design.
Take a look at this comment with the picture.

From the High Performance Polymer website the width of the nut is 5.5mm, the width of the Mcmaster Carr nut is 5.32mm. The clearance is 5.62mm. You would be able to insert it but would probably use small force. I could increase the clearance, just let me know whatever works for you best.

> > Is it banana jack _and_ pogo pins? I was under the impression that the pogo pin suggestion was in place of the banana jack. Also, should be from top.
> 
> Because the potentiostat has banana pin.
> 
> > Also, I'm a bit confused why there's a potentiometer.
> > I was thinking of potentiostat
> > Is what you showed above an example of three possible designs?
> > No it is only one, with the box, PCB, banana jack, 2 set of screws  and heat inserts.
> 
> > @MiceeNS, one question regarding the ptfe screws for the current design of the sample holder. Would metric M3 screws with 8mm length fit through the designed holes? I am trying to order the screws and bolts but it is challenging to get ptfe screws in imperial dimensions here.
> > Same thing for the nuts - would metric nuts fit?
> 
> The screw clearance is okey, however from wherever you order them check the nut size. On Mcmaster the metric is bit smaller and the cutout is a bit big for that nut size. I can adjust that quick. Let me know if you want it changed before you print. 1/4"=6.35mm
> 
> ![Screenshot 2024-12-19 203227](https://private-user-images.githubusercontent.com/76826784/397586777-557cfdec-f8ad-485f-ad88-ba176a0a1368.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY0NzA2MTcsIm5iZiI6MTczNjQ3MDMxNywicGF0aCI6Ii83NjgyNjc4NC8zOTc1ODY3NzctNTU3Y2ZkZWMtZjhhZC00ODVmLWFkODgtYmExNzZhMGExMzY4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDAwNTE1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk0Y2U2ODMyYmUwZGUwZWM1ZDliYjZkOTg0Mjk4ZGRlYjlhYjA2OTRhNWZlMTczYzgzZWZjMmFjYmE2MmUyNzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.WahjDzBa-ifL5rnjyQh6HhsUIMCkeybNklsNqZBnx9M) ![Screenshot 2024-12-19 203131](https://private-user-images.githubusercontent.com/76826784/397586781-1900af93-0dca-46b3-a960-21c78d7ac1e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY0NzA2MTcsIm5iZiI6MTczNjQ3MDMxNywicGF0aCI6Ii83NjgyNjc4NC8zOTc1ODY3ODEtMTkwMGFmOTMtMGRjYS00NmIzLWE5NjAtMjFjNzhkN2FjMWU2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDAwNTE1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4NzJlYjNkYjFlOTdhYmZiZWU5NmMxYzUyNjM3MThiNDU0ODQ3M2Q1YWRlZDA1NTE0NTU3OGFhZGYxZTZkYjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xTyGgiqMtUQ-bwn3Cwmb8IYdSIloVsVxMtVyokBkyAc)

programlich, 2025-01-10:
thanks for catching my mistake with the M6 thread. I have now ordered several screws with M3 and a length of 8mm. If you could adapt the design slightly, so that the bolts can be easily entered into the sample, that would be very nice. Once this is done, I´ll print it and test it.

MiceeNS, 2025-01-10:
Sure, will do tonight. Excited to see results

MiceeNS, 2025-01-11:
@programlich  Here is the zip file with the main body and the cap body. I also changed some tolerances for the rubber washer. You have the rubber washer and the O-ring with you right?

I hope everything works out fine. Let me know if you think I should adjust something for you.

[Working Sample Holder Metric V3.1.zip](https://github.com/user-attachments/files/18386800/Working.Sample.Holder.Metric.V3.1.zip)

programlich, 2025-01-13:
Beautiful. Many thanks @MiceeNS. With a bit of luck, I´ll have a version of this sample holder printed by the end of the week from formlabs high temp, so I can test it next week.

MiceeNS, 2025-01-17:
@sgbaird  The wire length (bends included) from the bottom to the top of the tap is 125mm

sgbaird, 2025-01-17:
Thanks!

Based on conversation, priority is to get sample holder with good electrical contact and no corrosion of the electrical connection, and worry a bit less about the automation piece for now. There is some platinum foil that would probably work (https://www.mcmaster.com/product/6224N18), and we can test with SS in similar dimensions to make sure we can get it through. Some worry about clearance if we try to have a crimped connection between SS and platinum near the sample, so having two lengths of platinum foil would do the trick. Width is ~0.8 mm, and compared with the legacy design which had two 1 mm wires contacting the sample on the left and right sides, we should be ok with the one length of foil on either side. Not sure if there needs to be contact on left/right **and** back/front (i.e., 4 points of contact). @programlich, does the legacy design have this, or was it just left/right on back (not front). Resin can be added and cured ad-hoc to prevent KOH from traveling up the sample holder (less of a concern if it's platinum the full way) or as @MiceeNS suggested, there could be a separate insert/plug.

For later automation, I lean away from having a separate PCB and pogo pins - this makes it harder for others to replicate. This is something I want to still keep in mind though, especially for other projects.

Crimping a banana socket onto the foil after a bit of origami will work OK I think (technically can use a single piece of foil, ~250 mm). Securing the banana socket could be straightforward. In total ~$80 USD for the foil for a single sample holder I think.

MiceeNS, 2025-01-17:
Just an idea that came to my mind: we could use electrical paste for a better connection where the KOH is not present.

MiceeNS, 2025-01-22:
I want to propose one design idea for the contact between the foil and the sample

I draw something fast just so I can present the idea

[image]

Basically we use the same design that we had until now for the sample holder with the routing inside the component. We create holes on both ends of the sample holder where the electrical contact should be.

We will simply put the foil over the foil "base" (assuming it will have raised face for the foil like 0.25mm) and hold the platinum foil with O-ring on the "base". The "base" component or the sample holder itself (i need to check what will work best) will have groove inside the sample holder so it holds the "base" in place and provides waterproof for the stainless steel wire inside the route.

We can use foil clamp terminal like this ones: https://www.te.com/en/products/brands/termi-foil.html

Ideally we will have waterproof routing and only the platinum foil will be in contact with the KOH. In this example we do not need the clearance for the clamping since we can do that outside and just fish the stainless steel wire from the bottom to the top of the sample holder.

I tried to create something similar at home just for idea
Imagine the tube is not hollow inside rather it will be a solid body

[image]
[image]
[image]

programlich, 2025-01-22:
I´ve been trying to find flaws in this idea but couldn´t find any. Actually, I think this is an awsome idea, definetely being worth a shot! It might be the closest, we can get to a broad Pt-sample contact with a minimal amount of Pt being used. We just need to ensure tightness. Depending on the thickness of the Pt foil, we could add a small trench for the foil, something like this:

[image]

Fantastic idea, @MiceeNS. What do you think, @sgbaird?

[75 more replies]