AccelerationConsortium/ac-dev-lab #130: Implement long-term/permanent "always connected" setups

sgbaird, 2025-01-06:
@gursi26 @linx5o @evelynnesher @Luthiraa @Jonathan-Woo could you comment on the status of the hardware you're working with? (To try to have a summary in one place). Especially relative to the web app being able to connect to the real equipment that's set up semi-permanently and connected (i.e., could someone go to the virtual lab and interact with the equipment remotely)

@SissiFeng could you comment on the microscope and the AprilTag printer?

@Neil-YL could you comment on the state of the OT-2 / color matching demo?

All, you can link to relevant issues as well.

Neil-YL, 2025-01-06:
Before the holiday, I mentioned that after completing the Prefect course, I felt our previous demo (by Jackie) didn’t fully  utilized Prefect’s potential. Since then, I’ve started restructuring everything. I’ve completed the new Prefect task/flow setup for the OT-2 and am now planning to integrate the sensor and optimization workflow into the orchestrator flow. Additionally, I’ve been testing AWS Lambda to log and read data from MongoDB within orchestrator.

evelynnesher, 2025-01-06:
The Autotrickler uses an F-F adapter (the yellow piece in the picture below) that tends to come loose. If the device is moved, the adapter might disconnect, stopping MQTT from receiving data and preventing graph updates. A quick fix is to secure it with something like tape or replace it with a sturdier adapter.
[image]

sgbaird, 2025-01-06:
@Neil-YL, @evelynnesher thanks for the updates!

@Neil-YL Let's use PyMongo directly instead with the orchestrator (https://github.com/AccelerationConsortium/ac-microcourses/issues/139). See https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.5.1-pymongo.html for the tutorial.

@evelynnesher I think I got some DB9 connectors that let you customize the connections. Could you see if one of those has a tight fit? If so, then maybe we could make the custom cable connections (I think there are only a few that are in use). Let's follow up with each other in one of the other issues.

Neil-YL, 2025-01-06:
> @Neil-YL, @evelynnesher thanks for the updates!
> 
> @Neil-YL Let's use PyMongo directly instead with the orchestrator ([AccelerationConsortium/ac-microcourses#139](https://github.com/AccelerationConsortium/ac-microcourses/issues/139)). See https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.5.1-pymongo.html for the tutorial.

No problem.

linx5o, 2025-01-07:
The Pioreactor is in a semi permanent working state, the Pioreactor and web app are plug and play. 

The UI/UX of the web app could be improved as it is currently a bit unintuitive and requires prior knowledge to use. However, due to the nature of Pioreactor and MQTT, a preferably two way system using a one way communication network this would be difficult. 

Core functionality of the Pioreactor can be accessed through the web app. Continual maintenance may be needed as the Pioreactor devs change the API endpoints.

An overflow sensor should be installed or some other kind of implementation to limit fluid overflow.

gursi26, 2025-01-07:
The cobot is pretty much ready to go, the client-side (which includes the gradio webapp, python api and gradio api endpoints) and server side code is done and is set to auto run when the arm turns on. 

All that remains is to set a place in the office where i can permanently setup the arm along with maybe a few demo objects that users can try to pick and place.

Jonathan-Woo, 2025-01-10:
Cameras should be ready to go. Just need to find a day to come in to fix the autostart then I'll have them running continuously. I have one so far with the cobot and one with the OT. We have at least one though, happy to setup wherever.

Luthiraa, 2025-01-10:
The temperature sensor is pretty much good to go on the client side (aside from a few questions I had - seen on that issue). Just need a spot to set up permanently.