AccelerationConsortium/ac-dev-lab #132: SEM custom open/close with linear actuator

sgbaird, 2025-01-07:
Perhaps an actuonix one with feedback. We'll need to do a bit of testing to get a sense of the force required. Maybe we can have someone from Hitachi High Technologies send a video of pulling it and pushing it back in.

sgbaird, 2025-03-23:
Cc @chasedamagician who may be involved with this later (related #148 #140)

sgbaird, 2025-04-01:
Major question: to try to adhere, mount to SEM externally (e.g., a base that wraps around), or modify the SEM with some mounting holes or similar. Whether to mount it to the side, the top, or have it pull out from ahead of it (i.e., mounted separately, relying on weight of the SEM to keep the SEM in place). Can't mount internal to chamber because of outgassing of plastic/etc.

Will want to do a force test of some kind to get a sense of what's needed to open and close the door. I think one of the SDLs has done something loosely like this (Maybe SDL2?).

I lean a bit towards using a strong adhesive initially, and knowing what's needed to remove/dissolve the adhesive.

sgbaird, 2025-04-01:
Just sent a message to Patrick and Rod requesting a video of the chamber door being opened and closed.

sgbaird, 2025-04-02:
Opening and closing door video below. Seems to not require much force, and I also asked if pressing the evacuate button is something that can be done programmatically or not.

https://github.com/user-attachments/assets/f5afedb4-abdc-4323-b377-9ce2122981bf

sgbaird, 2025-04-02:
Maybe some ideas that can come from the videos at https://www.linkedin.com/posts/sterling-baird_from-a-lab-automation-infrastructure-perspective-activity-7136432975298117632-eXss/.

In particular, from the following where a linear actuator opens a furnace door on a hinge (rather than a door that slides out like above)

https://github.com/user-attachments/assets/9f22fece-1ecf-4d56-ae37-a172a1cfbafd

kelvinchow23, 2025-04-04:
SDL1 will get a [furnace ](https://www.thermofisher.com/order/catalog/product/F6010?ef_id=Cj0KCQjwhr6_BhD4ARIsAH1YdjB8h-8OXuOt8CNpRrdJ7kyatQ-GIpkYZakSGWuv9U0c24Qzj1CCYv8aAnOSEALw_wcB:G:s&s_kwcid=AL!3652!3!706695403320!!!g!!!21493194447!165617053176&cid=lpd_wlp_wps_r02_co_cp0000_pjt0000_lpd00000_0se_gaw_dy_awa_ecommerceDSA&gad_source=1&gclid=Cj0KCQjwhr6_BhD4ARIsAH1YdjB8h-8OXuOt8CNpRrdJ7kyatQ-GIpkYZakSGWuv9U0c24Qzj1CCYv8aAnOSEALw_wcB) soon. It might be the exact same model as the video above, and it will need to automated to open/close the door. we're also thinking about a linear actuator

sgbaird, 2025-04-08:
I have a hard time imagining something more straightforward than the equivalent of a command strip on a mounting panel that can connect to one end of the actuator, and a clamp of some kind attached to the door handle and the other end of the actuator. Also, probably best to use the lower force actuator.

[image]

kelvinchow23, 2025-05-06:
@yakavetsiv inspiration for moving that microscope head

sgbaird, 2025-05-22:
This one?

https://github.com/user-attachments/assets/c06256bb-a92f-40ae-9e28-8d3a32a0dee0

[image]

pranshu-ui, 2025-06-27:
The issue has to do with opening and closing of the SEM door and pressing of its button at NRC.

Based on the desktop setup, it would be best to keep it low profile, electronically actuated and have the base be secured to the machine such that it can be moved later on.

Thinking about this, our major constraints here would be:
- Does not leave residue/marks on the SEM
- Must be secured.

Functional Criteria:
- Open and close the door
- Press and release the button

Non Functional Criteria:
- Small form factor

Based on this, I would design a U-mount that would be secured underneath the table, which holds an arm out that rotates with a med-torque stepper. The stepper motor, when rotated down, would pull on the handle until fully open, and when rotated up, would push the door back shut as the arm would be rigid. 

In terms of the button press, I believe a simple 3d printed mount that would attach to the wall of the SEM, using silicon ends to suction to the wall attached with a small stepper motor would do the job.

kelvinchow23, 2025-07-02:
hey that looks cool. few comments

- is there room to mount that track actuator on a table? i think ideally, the track actuator is secured on both ends and right now the end sticking out might be floating (dependent on where the equipment is placed on the table). 
- anything that is automated should should not lock a manual user from operating the equipment, so something like a button to operate the actuator (and door) would be nice. 
- I guess this piece of equipment is in NRC and not easily accessible. Could we get like a rough estimate on how easy it is to open that door? use something like a [force meter](https://www.amazon.ca/Mxmoonfree-Digital-Portable-Storage-Power-Off/dp/B08YJNGVKM/ref=sxin_16_pa_sp_search_thematic_sspa?content-id=amzn1.sym.8648fd4f-faab-4fd6-b373-b95aa19853ef%3Aamzn1.sym.8648fd4f-faab-4fd6-b373-b95aa19853ef&crid=1N987CNYMG0C4&cv_ct_cx=force%2Bmeter&keywords=force%2Bmeter&pd_rd_i=B08YJNGVKM&pd_rd_r=dffab498-2de0-4d3d-98d2-b878279bb398&pd_rd_w=EnSQC&pd_rd_wg=K7pzb&pf_rd_p=8648fd4f-faab-4fd6-b373-b95aa19853ef&pf_rd_r=D6N3A8DZ3CSDQA1PQKGE&qid=1751469804&sbo=RZvfv%2F%2FHxDF%2BO5021pAnSA%3D%3D&sprefix=force%2Bmeter%2Caps%2C88&sr=1-1-30a71431-0640-4d25-b8df-74e773a218a5-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9zZWFyY2hfdGhlbWF0aWM&th=1) and find a way to hook it to the handle?

sgbaird, 2025-07-02:
> * is there room to mount that track actuator on a table? i think ideally, the track actuator is secured on both ends and right now the end sticking out might be floating (dependent on where the equipment is placed on the table).

Good point. You can kind of see from https://github.com/AccelerationConsortium/ac-training-lab/issues/132#issuecomment-2855850348 that there's not a whole lot of space. At the same time, I don't mind rotating it. It seems like these track actuators have a little bit of give, but might require a more severe accident rather than a bump (i.e., something heavy falling onto it / colliding), to damage it.

> * anything that is automated should should not lock a manual user from operating the equipment, so something like a button to operate the actuator (and door) would be nice.

Agreed. Could you link to the code/designs where you've implemented this before? (I think the automated sash (?)). No worries if you don't have on hand

> * I guess this piece of equipment is in NRC and not easily accessible. Could we get like a rough estimate on how easy it is to open that door? use something like a [force [...]

Force meter would be good. The video from https://github.com/AccelerationConsortium/ac-training-lab/issues/132#issuecomment-2773142239 gives the impression of "requires a bit of manual force" (pulled out with a slight tug of a single finger).

pranshu-ui, 2025-07-02:
> * is there room to mount that track actuator on a table? i think ideally, the track actuator is secured on both ends and right now the end sticking out might be floating (dependent on where the equipment is placed on the table).

Great point. The design was based on hopes that the actuator could fit under the SEM and above the table while being secure. However if it can't fit, I can redesign a piece that will reach over the table and onto the handle with the actuator under the table. This'll keep the arm out of the way and would let the actuator be more secure. Note that it might require a higher gear ratio to support the weight and pull the drawer open.

> * Force meter would be good. The video from https://github.com/AccelerationConsortium/ac-training-lab/issues/132#issuecomment-2773142239 gives the impression of "requires a bit of manual force" (pulled out with a slight tug of a single finger).

The current 22:1 gear ratio planned to be used can push a max of 50N, so if the drawer can be opened with less than that, perfect.

[image]

sgbaird, 2025-07-04:
Could you share a bill of materials (and shopping list) with links for the proposed solution?

  *   Force meter would be good. The video from #132 (comment)<https://github.com/AccelerationConsortium/ac-training-lab/issues/132#issuecomment-2773142239> gives the impression of "requires a bit of manual force" (pulled out with a slight tug of a single finger).

image.png (view on web)<https://github.com/user-attachments/assets/25f6479d-30fa-4bc4-b3ef-e68fc075e675>

sgbaird, 2025-07-04:
@pranshu-ui I sent the fusion 360 invite to your waterloo email. Once you're logged in / accepted, I think you should be able to view the folder SEM folder: https://utoronto63.autodesk360.com/g/projects/20240927811133626/data/dXJuOmFkc2sud2lwcHJvZDpmcy5mb2xkZXI6Y28uLUpvY3JvSlFUemVsbXNoQzJESVNrZw (not entirely sure, lmk if you have trouble accessing). This is where you can put the various designs. Ideally, would be great if you could re-create those designs in Fusion instead of SolidWorks and add them there. Please also use "File --> Share File Link" and enable "Share the latest version with anyone using this link" to True and enable the "Allow item to be downloaded to True". You can leave "Require a password to access the link" disabled.

[image]

(as an example, for a vial rack from #15 )

[image]

This will make it easier to review designs on the fly (e.g., on my phone) and acts a bit as self-documentation. That link will always point to the latest version, which is fine.

sgbaird, 2025-07-07:
cc'ing @pwoo-htc for visibility.

pranshu-ui, 2025-07-08:
> * in sterling's post here (https://github.com/AccelerationConsortium/ac-training-lab/issues/132#issuecomment-2773157636), there's like an extrusion fence around the equipment, and the acutators mounted onto it. Someone at the AC replicated this and am hoping he will share what he did once hes done (should be soon).

This solution could also work. I'll start on a draft of multiple solutions and post them here. The only concern with this edition would be that the actuator would then stick out from the side as the actuators only move the attachment as opposed to a telescopic linear actuator. 

T-16 actuator movement for reference:
https://www.youtube.com/watch?v=y0zkD59bofE&t=67s

> * we also buy a lot of metric optical breadboards (https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=159). Both the sem and actuator can be fixed directly onto the breadboard. You could also make risers around the feet to elevate the sem.

Just an idea, what do you think about having one of these below it as the riser and just connecting the actuator from there?

pranshu-ui, 2025-07-14:
@IlsCode 

I made a model of that sketch on Fusion! However, the telescopic actuator I'm experimenting with doesn't seem to be long enough to open the SEM fully. Could you link which actuator you used for the automated furnace door mechanism? We need the stroke length to be about 1ft or 300mm. 

The one in this model is 200mm.

https://youtu.be/87Y8RjoZWkI

Thanks!

IlsCode, 2025-07-14:
Here's the model I used for the furnace, which has a stroke of about 12 in:
https://www.firgelliauto.ca/products/tubular-linear-actuator?variant=1095573748

Fair warning it is quite bulky so I would recommend looking for a bullet actuator like this:
https://www.firgelliauto.ca/products/bullet-series-23-cal?variant=29196799377513

Or even just using a stepper motor and designing a mount for the handle accordingly:

[image]

pranshu-ui, 2025-07-21:
Improved the design to include the Bullet Series 23 Cal. Linear Actuator from firgelliauto.

The video highlights the mounting from the SEM to the actuator and from the actuator to the handle where there's plenty of room to mount a button presser. 

The mounts are ready to print and I have also update the BOM.

[https://www.youtube.com/watch?v=QZ0NZa7gNZU](url)

sgbaird, 2025-07-24:
@pwoo-htc amazing, thanks Patrick! If I interpreted correctly, the slide distance is ~17 cm or 6.7".

@pranshu-ui this is perfect because it's actually better if the stroke length is slightly less than the maximum travel distance as a built-in safety, and there is an L16-R that's 15 cm (5.9"): https://www.actuonix.com/l16-100-35-12-p

@pranshu-ui could you update the design accordingly?

All, feel free to weigh in.

sgbaird, 2025-08-19:
@pwoo-htc is checking about the ability to programmatically control the evac/vent button. If need be, we can also consider hijacking that relay.

[image]

[image]

https://github.com/user-attachments/assets/15231425-4e4b-451f-ad8c-f006231a608d

kelvinchow23, 2025-08-19:
this looks really promising. im going to assume 2 of those wires are for the led, and 2 are for the contact.

sgbaird, 2025-08-20:
Also would be good for us to document what currently happens software-wise when the evac button is pushed. I think the beam has an auto-shutoff if the chamber gets vented. I don't think there's any sort of software lock-out on the evac/venting getting triggered though - if you push it, it does what you ask. So, if we do anything custom we'll need to be cautious about robustness such that the button isn't getting accidentally triggered.

pranshu-ui, 2025-09-05:
Made changes to support the L16-P Linear Actuator (140mm 35:1). I also ensured all pieces are modular and fit within the printbed. 
Printable pieces are black.
[https://www.youtube.com/watch?v=KBVw0UCgBLI](url)

kelvinchow23, 2025-09-05:
- if your 3d printed part is one body, you might have a difficult time installing the actuator to the 3d printed part. 
- also how does it attach to the handle? does the front part of the handle come off easy, i cant remember.

pranshu-ui, 2025-09-05:
- They are split into smaller printable parts which are attachable by either a rod or mountable screw & threaded holes.
- I didn't really showcase it but the arm has an opening, kind of like a lego hand where it'll just grip onto the handle tight. The only issue with that is I'm not entirely sure what the handle's exact diameter is so I'm going to wait on printing that arm.

kelvinchow23, 2025-09-05:
I meant on the top part where the actuator is mounted. From looking at the model, I'm not sure you can slide it in and install it. I think that square piece is forcing you to slide it in, and there's something blocking you from doing so. I could also be wrong

pranshu-ui, 2025-09-08:
I think I see what you mean, I did a quick redraw to allow the actuator to slide in a bit better. The securing piece is further up so it has more room and a little incline is added to the top to give it room to slide it at an angle without interfering with the rear extrusion. If everyone's okay with this design, I can begin printing the actuator holding piece.

[image]

sgbaird, 2025-09-09:
Cool! Let's give it a try

pranshu-ui, 2025-09-10:
I've picked up the actuator. Everything is here. [image]

pranshu-ui, 2025-09-16:
Almost, just need to reprint some as the gap tolerances were too large and the thread mounts didn't fit. Should be done by today though.

sgbaird, 2025-09-24:
Nice! For the attachment, let's try using a U-bolt with a plate/fastener. Could you provide the following six dimensions?

[image]

Probably you'd still use the following, or at least reuse the bolt and nut that goes through the clevis (hole at the end of the actuator). 

[image]

Since the orientation of the clevis is probably fixed, if we want it to be rotated by 90 degrees, we can probably change the side of the actuator that's mounted up against the side of the SEM.

@pranshu-ui could you also lay out the hardware that the actuonix linear actuator came with and take a picture? This would help to know what we're working with that's already available.

pranshu-ui, 2025-09-25:
**Dimensions:**
- Handle Diameter (left to right): 24mm
- Handle Diameter (top to bottom): 20mm
 _(Handle is elipsoid)_ 
- Right side to first handle length: 120mm (the 3d printed piece itself is 150mm across)
- Distance form front of SEM face to face of clevis on the actuator: ~15mm
 
I will lay out all the disassembled pieces there. 
It does actually come with a threaded rod end piece.

sgbaird, 2025-12-23:
@pwoo-htc great to hear from you! Also noting that some of the code Pranshu has been working on is at https://github.com/AccelerationConsortium/ac-dev-lab/issues/455#issuecomment-3687265719. We're considering incorporating the SEM as a remote access device for a course being taught at UofT.