AccelerationConsortium/ac-dev-lab #133: SEM API demo, jupyter notebook, and MQTT integration

sgbaird, 2025-01-07:
We'll need to get the API documentation from Hitachi High Technologies. There may be a simulator class that we could use as well. We can take some lessons from the OpenFlexure ecosystem as well, especially in terms of stitching and position calibration.

Worth considering if / what to expose to public usage (of course with safeguards). Since this one is more likely to be a part of experimental workflows, it may be better to keep the control private, especially given the cost.

pranshu-ui, 2025-10-20:
The API Covers the following commands based on the user manual:

`EXT.SetHv(OnOff='ON'|'OFF')`
Turn high voltage on/off.

`EXT.RunScan(ScanState=0|1|2)`
0=Run (live scan), 1=Freeze, 2=Forced Freeze. 

`EXT.SetMagnification(Value=<int>)`
Set nominal mag. (Manual covers mag behavior/limits.) 

`EXT.RunAutoAbc(Mode=0|1, Bm=0|1)`
Auto brightness/contrast. Mode: 0 Single, 1 All; Bm toggles backscatter bias option (if applicable). Manual describes brightness/contrast controls. 

`EXT.RunAutoAfc()`
Autofocus. Manual describes AF operation. 

`EXT.RunCapture(Type=0|1)`
Trigger capture.

`EXT.RunStageMove(X=<nm>, Y=<nm>)`
Stage move in nanometers. Keep motions modest and time for settling.

pranshu-ui, 2025-10-20:
Also related:

https://github.com/AccelerationConsortium/ac-dev-lab/issues/455#issuecomment-3423884018