AccelerationConsortium/ac-dev-lab #141: Autonomous battery slurry optimization system

sgbaird, 2025-01-08:
https://hasanyone.com/?id=f2572055

Talking with @MatPopp,

Using Science Jubilee and performing optical microscopy and viscosity measurements. Possibly machining a hole into the Science Jubilee plate to make room to attach the OpenFlexure delta stage microscope (reflection version) underneath for imaging the slurry from the bottom of a wellplate, where the wellplate is moved around via a press fitted 3D printed attachment into one of the wells near the center of the well plate (maybe with some adhesive for robustness)

We also considered a custom attachment to a glass slide to enable pickup, but drop casting into a wellplate and moving it around may be simpler to implement. Attaching the delta stage as a science jubilee tool might be a bit unwieldy in terms of size, but it's not overly heavy. It would probably leave room for one other tool (possibly an ultrasonicator and/or a doctor blade).

Rather than machining a hole into the Science Jubilee build plate, creating a standoff plate to raise the vials and the well plate (or perhaps just making a base for the wellplate and vial stand that's really tall), and putting the microscope and anything else required directly on the build plate is another option. There just needs to be enough XY clearance relative to the cross bar and the tool.

Stacking wellplates and moving them with the fake tip procedure mentioned above is an option.

There was mention on the discord forum about possible difficulty with calibration with too much weight on the build plate. Maybe not a big deal if we don't need high XY precision. Not sure how much center of gravity will affect things either. For sure need to be careful to keep the weight towards the end with the two rods instead of the two corners nearest the sides with the the single rod, since too much pressure on those corners can tip the build plate.

Many of the suggestions and comments in https://chatgpt.com/share/677df81f-13a8-8006-8817-51b40cad70ce seem reasonable to me. Use of a glass slide where the slurry has been spread may prove more informative than the well plates, similarly coating the slurry onto aluminum foil and using the microscope as a science jubilee tool would allow for inspection in the as-coated state.

Heated/vacuum drying may be difficult to implement. An ultrasonic probe would be especially helpful with the  water-based slurries it would seem.

sgbaird, 2025-01-08:
@Yang1Bai feel free to weigh in with any comments here.

MatPopp, 2025-01-09:
Thanks @sgbaird  for summarizing our discussions so nicely! 

I will set this comment up as a TODO list / work plan. 

## High Level Targets
* Contribute to cheap and performant batteries for climate change mitigation
This project only contributes small parts.
* Develop SDL / MAPs methods serving a very broad scope in science.
** Make make machine learning and artificial intelligence usable for (material) science by connecting it to (fully) automated experiments
** Find ways to these methods in the context of recycling / circular economies -> automatically build process models for optimizations in an overarching scope

## Targets: 
(1) get modules for slurry mixing map up and running for the following involved processes:
* (a) mixing of (viscous) liquids 
* (b) measuring viscosity (proxy) values 
* (c) moving material (weelplates, microscopy slides (optional)) around (@Neil-YL)
* (d) (optional) microscopy on slurry in wet or dried state
* (e) (optional) deposit slurry, e.g. on microscopy slides with aluminum / copper foil. 
* (f) (optional) sample drying.
* (g) (very optional) electrical characterization of dried active material (4Point / 4Bar Measurements) 
* (h) (here out of scope) electrochemical characterization of dried active material (e.g. in half-cell geometry, needs electrolyte, therefore out of scope)
* (j) (here out of scope) (temporary) assembly of cells. (ideas like "continuous endless cell": Assembling only Electrodes+Active material and re-using structural parts in between experiments)  
* (k) (here out of scope) performance characterization of (temporary) Laboratory cells

(2) Apply Machine Learning / AI
* (a) use well-established Bayesian Optimization for validation of experimental setup
* (b) find a way to use Bayesian Optimization to deal with varying input quality

## TODOs:

###(1) Experimental preparation
* (a) Slurry mixing
* * get used to working with Science Jubilee to be able to transport samples
* * get used to working with the available pipetting systems
** test if mixing of viscous liquids (i.e. the slurry components) can be accomplished by 
* (b) viscosity proxy
** set up software for LPS22 Pressure sensor 
*** Done: (micropython <-> LPS22)
** design + build air-tight enclosure for LPS22 Pressure sensor with connector to hose
** design + build air-thight connector between syringe (for first tests) and hose 
** acquire first test data with syringe + water
** acquire first test data with syringe + glycerine
** design + build air-thight connector between OT2 pipette and hose
** design + build mechanical attachments to OT2 pipette. (including wiring) 
* (c) moving stuff:
** @Neil-YL design press-fit connector between wellplate and OT2 Pipette 
** @Neil-YL show working example (moving + positioning wellplate) 
** @Neil-YL (?) Design new (plastic) Frame attachment for Jubilee (aluminum) build-Plate:
*** two places for piles of wellplates, raw material containers, ... pipette tips (?) 
*** One large "working yard" where stuff can be positioned freely. The edges of that "yard" might be used to align wellplates rotationally, by shortly touching off
* (d) microscopy
** test with actual battery slurry if it is worth it
** @sgbaird decide, if we can have a cutout into the aluminum plate, make contact to necessary personell/contractor
** attach microscope to bottom of build-plate.

* integration of modules into setup
** wire everything 
** transport it to NRC (@sgbaird a car might be beneficial for that) 

## Phase 2: Utilizing setup + Algorithmic / Software work
* (a) Bayesian Optimization campaign:
** prepare pre-mixed raw-slurries: [water, NMP, CarbonBlack, CMC in water, SBR in water, NMC in water, NMC in NMP, Graphite in water, PVDF in NMC] (?)
** inputs: mixture fractions of raw-slurries (or rather subsets thereof) 
** outputs: pressure-derived viscosity proxy, shear rate dependence proxy (?)
** optimization target: deviation from target viscosity, (optional) shear rate dependence, (optional) microscopy homogeneity measure, (optional) microscopy particle size distribution

* (b) Algorthm enhancement for recycling/Process modelling: 
** optimization campaign that re-uses already characterized slurry mixtures 
*** comment: this would make results (more) transferrable to industry: the properties af the slurry in a large mixing pot could be used to determine what has to be added. 
*** free inputs: inputs of (2)(a) 
*** clamped inputs / parameters: all available measurements on intermediate slurry mixture
*** optimization target: like (2)(a) but with extra cost for the use of (precious) raw material
-> **"Find a way to mix good battery slurry using as few expensive (raw) material as possible. Use cheap (recycled) material yielding well enough quality"**
-> **"Remember what you did to build an data-based understanding of what dominates the behavior of this type of process"**

Neil-YL, 2025-01-10:
> * ** @Neil-YL show working example (moving + positioning wellplate)

The [video](https://github.com/AccelerationConsortium/ac-training-lab/issues/143#issuecomment-2583271322) is posted in this issue.

sgbaird, 2025-01-10:
Wellplate above delta stage

Because there is a lip to the wellplate, not all wells can be accessed while keeping it flat.

https://github.com/user-attachments/assets/11fa2368-034b-460d-8039-c598f3fb6d3a

There would likely need to be something stick or something grippy to make sure the wellplate that is dropped onto the microscope maintains it's orientation relative to the microscope stage.

https://github.com/user-attachments/assets/7ec0327e-8ede-4913-a555-84dbeb577a10

Mounting the microscope as a tool essentially negates any utility of the XYZ stage. SciJub is the XYZ at that point.

There is clearance to put the microscope on the OT-2 base and put a wellplate on top of it.

MatPopp, 2025-01-13:
## First tests of pressure sensor viscosity proxy
### Setup
[image]
* syringe actuated by linear actuator (controlled via pi pico)
* pressure sensor (in white box) connected to syringe, data recorded by pi pico
### Documentation
* n=1: inhale and exhale air:
[image]

* The sensor is exact enough to sense pressure changes due to sucking air into the pipette and pushing it out again. 
* The movement of the syringe is retarded compared to the setpoint (which is expected)

* video: aspirate dish soap:
https://github.com/user-attachments/assets/961f66c5-6803-41f0-95d5-530d268eeef4

* some experiments, aspirating water or soap. The Actuator was moved only as few as possible (stick-slip behavior of the large syringe), then the drop-off of pressure was observed (more stable than when servo is still active):
[image]
[image]
[image]
* rather exponential decay observable

[image]
[image]
[image]
* again, exponential decay observable with larger decay time (30x smaller).
* https://www.michael-smith-engineers.co.uk/resources/useful-info/approximate-viscosities-of-common-liquids-by-type suggests viscosities for shampoo of 3000 cP which is rather a factor of 3000 larger than water. 

* the origin of spikes in measurements is currently unclear. Possibly false packets in i2c conneciton.
possible reasons for the deviaiton:
* the 3D-printed connection-piece between the syringe and the pipette tip is not fully air-thight. (tested by closing the tip with a finger. After sucking + waiting air is pushed out when exhaling fast. 
* non-newtonian behavior (seems a bit unlikely to me)

sgbaird, 2025-01-13:
Nice! Any luck with position feedback from the new linear actuator? Maybe we do a resin print, assemble, apply some additional resin, and cure it to keep everything airtight.

MatPopp, 2025-01-14:
No luck with position feedback so far. A voltage booster might be needed. (https://www.actuonix.com/lac-board-intro?srsltid=AfmBOopJR0bP6b83xY1mzNJnjBf2eja3fOiL6rOgy8MKNan4byd39Wzr)
The housing of the sensor is air-tight, Superglue worked fine. The leakage was at the adapter between the syringe and tip. There is a small kink at the circumference of fdm printed round parts which lets some air through 

Possible solutions:
* sanding the circumference (I tried that, will see tomorrow how it worked)
* resin 3d printing for better accuracy
* including an o-ring or some soft material at the connections.

sgbaird, 2025-01-14:
With enough super glue you'd be able to plug it though, right? I'll open a separate issue for the linear actuator.

MatPopp, 2025-01-16:
@sgbaird played a bit with the jubilee python interface. Do you know if/where the end stop of the pipette is implemented in science jubilee? Could not find a spot i  python. Does it stop the z motion within duet?

sgbaird, 2025-01-16:
> @sgbaird played a bit with the jubilee python interface. Do you know if/where the end stop of the pipette is implemented in science jubilee? Could not find a spot i python. Does it stop the z motion within duet?

@yakavetsiv do you know for this one?

sgbaird, 2025-01-16:
@MatPopp starting around here https://science-jubilee.readthedocs.io/en/latest/building/pipette_tool.html#wiring-connections maybe something that can help.

MatPopp, 2025-01-16:
I finally found it: the science jubilee version was not up-to-date and the "H4" that makes the movement stop when the switch is pressed was missing. 

https://github.com/machineagency/science-jubilee/blob/079c43d02f991f1a2c07129b77c9af8fdadeb5a2/src/science_jubilee/tools/Pipette.py#L663

(line 663)

sgbaird, 2025-01-16:
Ah, wow! Really nice find

yakavetsiv, 2025-01-16:
@MatPopp I can check Jubilee setup configuration at SDL6 tomorrow, if you will have more questions. Please let me know if you will have any questions

MatPopp, 2025-01-21:
First tests with OpenTronsFlex pressure sensing described in https://github.com/AccelerationConsortium/ac-training-lab/issues/150

MatPopp, 2025-01-25:
# First semi-systematic viscosity-proxy measurements (pressure decay time)

## target
create a routine for dispensing liquids with strongly varying viscosities and measure the viscosity of the created mixture.

## documentation

* I programmed routines for the Opentrons Flex that can dispense dish-soap and water from two reservoirs (vials, ca. 10 ml, each)
    * it is still not perfect. As soon as soap came into play it became challenging: 
        * it takes much time to aspirate 
        * a lot of it sticks to the outer walls of the tips, it spreads "everywhere", which makes it difficult to dose 
        * as soon as bubbles form within the tips, pressure readings are hevily distorted. 
In the following plot-series the fraction of soap in a mix with water was varied from 0.6 to 1 (for values <0.6 it seems to be less interesting / so far not measurable with the current setup and routine). Total volumes were 200 ul (but the error-bar on the ammount of soap might be 40%). Left settling after aspiration, right settling after dispensing:

0.6 soap, 0.4 water:

[image]

0.64 soap, 0.36 water: 

[image]

0.68 soap, 0.32 water:

[image]

0.72 soap, 0.28 water:

[image]

0.76 soap, 0.24 water:

[image]

0.80 soap, 0.20 water:

[image]

0.84 soap, 0.16 water:

[image]

0.88 soap, 0.12 water: (distortion probably due to air bubbles / foam)
[image]

0.92 soap, 0.08 water: (right plot: rather large air bubble dispensed, resulting in a kink)

[image]

0.96 soap, 0.04 water:

[image]

1.0 soap, 0.0 water:

[image]

### decay time vs. soap fraction

[image]

## possible improvements

* soap / viscous liquid dispensing
    * use the pressure sensor to get the pressure to zero before leaving the storage vial. This should reduce the tendency of bubble formation and drastically improve accurracy.
    * use pressure sensor decetct liquid surface height, in order to immerse the tip as few as possible -> less slowly moving viscous liquid on outer side of the pipette
 * viscosity measurement
     * the decay time method seems to work OK for rather viscous fluids without bubbles but is inaccurate / not usable for water-like viscosities (I already used a pretty small tip)  -> Other excitation/measurement methods might be preferrable: excite with sawtooth-displacement (this is what e.g. the mixing-api command does) and use some signal-theory on measured pressure to recover the response function of the liquid. Using some fourier-transform and higher harmonics evaluation / non-linear response theory, this could hint on non-Newtonian behavior.  
     * remove bubbles!!! this + inaccurate soap dispensing are currently the main error mechanisms.

sgbaird, 2025-01-25:
Nice! And again, thank you for the thorough description. Likely, slurry won't have this kind of bubbling effect. Is there a replacement you can think of that can act as a proxy for the slurry? I have glycerin at the AC. There should be some vials with it. Maybe on your desk?

How about glycerin + corn starch + activated charcoal powder (health item)? See https://chatgpt.com/share/67948cca-8b08-8006-a50b-a1aa0432952e

sgbaird, 2025-01-25:
Also, could you PR your implementation? Sorry I didn't come up with a good way to have git (maybe check if dnf is installed? -- https://git-scm.com/book/en/v2/Getting-Started-Installing-Git).

MatPopp, 2025-01-29:
# First Bayesian viscosity optimization

* with the mixing / measuring method described above, I ran a Bayesian optimization with the target to optimize the viscosity of a soap-water mixture to a proxy-value of tau = 2s (pressure decay time after aspirating liquid).

the algorithm had some bad luck after 10 experiments whith a fluctuation suggesting an optimum at soap_fraction = 0.75 (which is too low): 

[image]

probably this is why it took ~30 iteration to get a good result: 

[image]

it continued for some further iterations but did not improve further, because it ran out of soap: 

[image]

## lessons learned
* ideas on further improvement of viscosity sensing: 
    * use time-integrated pressure of aspiration - time-integrated pressure of dispensing
        * this will work better for low viscosities: decay times are below 0.1 s then, yielding only few relevant data points. The pressure during the main aspiration part will however still depend on low viscosities. 
        * gravity will then have an influence, but will be cancelled when subtracting dispensing from aspiration. 
        * A monotonically rising dependence on viscosity can be expected. 
        * takes less time for low viscosity liquids: the measurement can be stopped once a threshold pressure is reached.
        * This should be more resilient to kinks by bubbles as in this measurement: 

[image]

## Planned further steps: 
* implement new p(t) integral viscosity proxy (funny: this will have the unit Pa*s), evaluate if it really is an improvement.
* generate calibration curve
* extend algorithm for input qualification measurements

* run experiment with actual battery slurry
* (optional) include imaging of prepared samples (in cooperation with @Neil-YL)

Neil-YL, 2025-01-31:
For the Picam capture script, I think [this ](https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/A1-cam/server.py
) might be helpful:

```
from picamera2 import Picamera2
from libcamera import controls, Transform

# camera setup
picam2 = Picamera2()
picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})
config = picam2.create_still_configuration(transform=Transform(vflip=1))
picam2.configure(config)
picam2.start()
```

MatPopp, 2025-02-04:
@Neil-YL  thanks for the links, these were helpful! 

# comparing some viscosity approximation techniques with pipette + pressure sensor

## description of methods

The measurement protocol was similar for the follwoing methods: 
* pipette is moved as fast as possible (both for aspiration and dispensing)
* pressure signal is recorded (both during movement and settle_time_s afterwards)
* the following graph shows the pressure for aspiration - waiting - dispensing - waiting, here for pure soap:

[image]

### Method 1: pressure-decay-time fit 
* see also previous comments in this issue

* fit of exponential function $f = Ae^{-t/\tau}+c$ to decay part, yielding $\tau$, the decay time.

[image]

### Method 2: pressure integral
* pipette is moved in arbitrary fashion, can moved fast. 
* pressure signal is recorded (as in (1)
* pressures_integral is the time-integral over pressure: $\int_0^{t_{max}} p(t)\text{dt}$ with t being the time.

### Method 3: time-specific proxy with derivative
* signal recording as above
* for each point the time/pressure specific viscosity-proxy is calculted: $prox = -p(t) / \frac{\partial p}{ \partial t}$. This corresponds to a local fit of the decay time (for exponential functions the ratio between value and (negative) derivative is the decay-time.)
* The mean-value can be taken as a viscosity proxy, the slope of the $proxy(p)$ dependence an indicator for the presence of shear-thinning / thickening
* It can be seen in the following graph, that the resulting values are pressure-dependent (for soap) -> shear-thinnning can be seen (albeit not very exact, the shear rate across the pipette tip crossection is not constant, but it might be OK as a proxy.)

[image]
**** graph 3 y-axis title fraction should be inverted****

## calibration and comparison between evaluation methods:
For the follwoing graph, measurements and evaluations were performed for water (1mPas) and viscosity standards with (10,50,200,500,1000) mPas with the 200 ul Opentrons pipettes, 5 consecutive measurements for each standard. Green lines are for aspiration and blue for dispensing:

[image]

The methods show the follwoing properties with regard to optimal (linear) behavior: 

Method 1 decay time (here, second graph) 
* very linear for viscosities above 10 mPas
* slight discrepancy between aspiration and dispensing 
* deviations from linear behavior for low viscosities due to 
    * few points within relevant fit range (sampling rate was ~30 Hz, decay time for water < 0.1s)
    * possibly momentum-dominated flow of water and therefore deviations of physical model

Method 2 integration time (here, first graph)
* deviations from linear both for
    * low viscosities (momentum dominated?) 
    * high viscosities: exponential response signal is cut at finite observation time. Therefore nonlinearity.

Method 3 proxy value from derivatives:
* seems to be an allround-method (though slightly less repeatable than decay times)
* has extra-advantage of non-newtonianness-proxy! (relevant for battery slurry)
* fun fact: proxy values are roughly equal to viscosities in mPas without further ado. (this is a coincidence, except Opentrons have planned their pipette volumes + tip diameters such that this happens)

In the following graph, the same method was repated with a 1000 ul tip with roughly the 2-fold tip-diameter yielding 16-fold flow according to Haagen Poiseuille's law: (not 1/16 proxy values due to increased air volume, but substantially lower values than 200 ul pipette)

[image]

Method 1 decay time (here, second graph) 
* not applicable for water, not enough points for fit

Method 2 integration time (here, first graph)
* nonlinear for low viscosities but not completely fale

Method 3 proxy value from derivatives:
* evaluation difficulities at low viscosities. (probably due to inclusion of pressure-measurement offset values at times when the pressure is already roughly constant and therefore the derivative small. Might be fixed by better edge case handling)

### non-Newtonianness proxy 

plots created with Method 3 for 

* F1000 viscosity standard: 

[image]

* Dish soap:

[image]

--- I would argue that there is a significant difference. The method might still not be perfect -- after all it is a pipetting machine, no rheometer. The point of all the experimentes above is not to build a high fidelity rheometer replacement, but to get a proxy measurement within a pipetting robot without further added instrumentation.

MatPopp, 2025-02-04:
## Chemical compatibility

Today, we tested, if our currently available labware is compatible with the available battery materials (NMC, in NMP solvent, with PVDF binder and Carbon-Black as conductive additive).

It is not, NMP etches its way through the wellplates: 

[image]

[image]

(It feels a bit weird to write this down publically, but it contains one of the more important lessons, in case somone will continue with this)

Label of Wellplate package: 
MICROPLATE, 96 WELL, PS, F-BOTTOM, CLEAR, 10PCS/BAG
PS means Polystyrene 

according to https://www.tom-parker.co.uk/upload/files/literature/CPC_Chemical_Compatibility_Chart_(_261_KB_)_.pdf 
Polypropylene should have done the job. 

I do so far not know how the interior of the Opentrons Pipettes would react to NMP.
Here is a table with chemical compatibility: https://opentrons.com/archives/faq/chemical-resistance-of-opentrons-pipettes?srsltid=AfmBOopbKdK2nxTJI9sZ_jNxGld1eCdLnNeYfwoKNTwWWNV7NkQYzULr 
 -> Polypropylene tips should be fine. 
There is no NMP (N-Methyl-2-pyrrolidone) within the table, a trained chemist should comment on that.

MatPopp, 2025-02-05:
Plan for last experiments: 

Demonstrate "recycling" operation, using https://github.com/AccelerationConsortium/ac-training-lab/issues/164

-> use soap + water just as in example. (corn starch would be available, but let's start simple)
-> use 1000 ul tips, since these work better in transferring viscous soap (and should also work better for battery slurries) due to larger tip diameter. 

-> possibly this method can be used for recycling of active material (e.g. aged batches that shall be recycled).

sgbaird, 2025-03-01:
Hope your time off has gone well :) lmk when you'd want to debrief and do some follow-ups on this. We might be able to have someone continue working on the setup at NRC with three of us keeping it moving forward asynchronously.

sgbaird, 2025-03-03:
Awesome! Quick aside: which code should I look at for the campaigns? The lack of sampling on the left side seems strange

MatPopp, 2025-03-09:
@sgbaird the code for the campaigns above is now in https://github.com/MatPopp/STORIES_MP/blob/main/opentrons/soap_mixing/soap_recycling_Bayes.ipynb

I have just put the corresponding code into this repository: 
https://github.com/MatPopp/STORIES_MP/tree/main 
@sgbaird @Neil-YL @yakavetsiv  You can also find it on the opentrons Flex in your lab.

@sgbaird if you want me to also put parts of this code in one of your repositories please point out a location, I am happy to contribute a pull request. 

## Content
### opentrons 
(this contains the most relevant scripts)
* soap_mixing.ipynb : Code for using an opentrons Flex robot to mix two liquids, measure their viscosities and optimize the mixing ratio with standard Bayesian Optimization for some target viscosity. 
* soap_mixing_1000ul_pipette.ipynb : same as soap_mixing.ipynb but calibration curves recorded with 1000 ul pipettes. These have a larger diameter at their opening an can therefore deal with higher viscosities. 
* soap_recycling_Bayes.ipynb : code modified for
    * (1) model-building phase: automatically builds a process-model including input-characterization, i.e. output prediction depending on measurements on the input (material).

    * (2) production-optimization phase: use the process-model for being able to optimize the same kind of process for fixed (but unknown!) input material with hopefully less iterations.
 
I recommend to download the repository and view it in vscode, to be able to view contained plotly plots: 
[image]

### openflexure
* some code to enhance contrast for camera streams produced with an open-flexure microscope

### CameraPi
(@Neil-YL )
* code to be used on a raspberry-pi with a camera
* uses the picamera2 library for interfacing with the camera and streaming to an http endpoint. -> convenient and quite fast transmission of camera images within a local network.

MatPopp, 2025-03-09:
> Awesome! Quick aside: which code should I look at for the campaigns? The lack of sampling on the left side seems strange. I'll give a more detailed response soon. Thanks for how thorough this is

The lack of sampling on the left side is due to the "custom" sampling algorithm without exploration. I created this in order to get a very close result within 10 iterations, trusting the previously built global process model to suggest a good starting parameter set.

sgbaird, 2025-11-12:
cc @mseifrid