AccelerationConsortium/ac-dev-lab #143: Create press-fit insert or similar with fake tip for picking up wellplates with OT-2 pipette

sgbaird, 2025-01-08:
@MatPopp and I are thinking of a simple cylinder that gets pushed into the wellplate, with enough tension to stay in (or maybe some superglue or other adhesive to keep it in there), with the fake tip above it.

Neil-YL, 2025-01-08:
@MatPopp discussed his experiment with me today afternoon. I will prepare some 3D printed parts tomorrow.

Neil-YL, 2025-01-10:
I have one wellplate with a connector glued on the "D6" slot and I will try to run a pickup/drop on OT-2 to test the mechanism work as we expect.

However, I find something could be problematic, maybe only for OT-2 protocol but I think it still worths mentioning. If I want to have a pick up tip command on a location, I should define it as a tip_rack first, but for the most of the time we want it to act as a well plate to dispense liquid. I think I could try to force the machine to have a pick up on a *wellplate* location if the pickup mechanism works well to move the plate.

Neil-YL, 2025-01-10:
[image]

Here  is a video for the pickup connector.

https://github.com/user-attachments/assets/f7486c3e-2d27-41eb-b481-5b30b029fd0e

OT-2 pipette have a lower limit (around 40mm, it is not documented) so I have to stack up several plates.

sgbaird, 2025-01-10:
Amazing! Could you put the CAD file here, too? I think it was super glued into there, right?

Maybe there's an unload labware command, or maybe it doesn't throw an error if you just load to the same location

https://github.com/AccelerationConsortium/ac-training-lab/blob/3d4df26f76663bddccf9e24b9293377820a9f421/src/ac_training_lab/opentrons/ot-2/_scripts/Oct28%20demo%20loop.py#L11

Neil-YL, 2025-01-10:
STP file: [wellplate_connector v2.step.zip](https://github.com/user-attachments/files/18380940/wellplate_connector.v2.step.zip)

It does have a command to[ move the labware](https://docs.opentrons.com/v2/moving_labware.html#moving-labware) to a "off deck position"(unloaded from the deck) and I think we can use it to offload and have a new load for the normal well plate labware after we move the well-plate.

>  # pause to move the spent tip rack off-deck
    protocol.move_labware(labware=tips1, new_location=protocol_api.OFF_DECK)
    # pause to move the fresh tip rack on-deck
    protocol.move_labware(labware=tips2, new_location=1)
    pipette.pick_up_tip()

sgbaird, 2025-01-10:
Thanks! We might want to verify that this doesn't throw an error (e.g., this might be specific to Flex)