AccelerationConsortium/ac-dev-lab #144: Set up higher quality camera monitoring for bambu lab

sgbaird, 2025-01-10:
For potential incorporation into SDL workflow

AMS can go on side instead of on top, allowing for a top down camera. Maybe one of the higher quality RPi ones. Lid can be removed, but probably best to leave it and live with the tinting.

sgbaird, 2025-01-10:
[image]

Using a Pixel 9

0.5x
[image]

1x
[image]

8x
[image]

Jonathan-Woo, 2025-01-11:
Would this be a live feed thing or just for capturing photos?

sgbaird, 2025-01-11:
Ah, right. I think just photos, as high-res as we can reasonably get. Is there a higher quality RPi camera / lens setup we could get? Alternatively, there is an iPhone which is used for tracking airtags that could be used for a livestream. Maybe asking a bit much, but controllable tilt or some XY travel could be nice.. maybe just whatever is available for now though and we check resolution and swap out as needed. Though we should probably buy a high-res RPi cam soon. If you provide links to what's needed, I can purchase them.

The intended use is to be able to evaluate the quality of print lines from a distance of maybe 6 inches to a foot.

Jonathan-Woo, 2025-01-21:
> @Jonathan-Woo<https://github.com/Jonathan-Woo> would you be able to come in sometime in the next 2 days or so to get high res image capture set up with one of the Pi Zero 2W camera setups?
>
> xref: #158 (comment)<https://github.com/AccelerationConsortium/ac-training-lab/issues/158#issuecomment-2605232138>
>
> Ideally, as high-res as possible for the given RPi camera. If you could just work on getting the image capture set up, I'll keep you posted on the data transfer procedure and put that in a separate post.

Yep I can come in tomorrow. Sorry for the delay, been wrapping up a conference submission.

sgbaird, 2025-01-21:
Awesome. Thank you! Hope the conference submission goes well :)

Aside: Looks like Camera Module 3 has an HDR mode: https://www.raspberrypi.com/products/camera-module-3/. Not sure exactly how to access it, but I think it might be worth using that. We'll also want to use the auto-focus feature. If you get HDR and autofocus working, would be great if you could provide some documentation on how you did it.

sgbaird, 2025-03-04:
Lmk if should be reopened

closed 2025-03-04 by sgbaird