AccelerationConsortium/ac-dev-lab #145: Integrate takasago pinch valve with pioreactor

sgbaird, 2025-01-10:
[image]
[image]

https://docs.pioreactor.com/user-guide/pinch-valves

@linx5o it would be nice to see this (at least to give it a quick try). Thoughts?

linx5o, 2025-01-11:
Of course. I will take a look.

linx5o, 2025-01-13:
After looking at this approach I don't really see a use case where this is useful.

If we connect it up to the Pioreactor we would need to go through software to activate it, as an interrupt of some sort. However, we would still have no knowledge of liquid levels in the vial. Also since we are interrupting the current running we could just stop the pumps maybe?

I think the main issue right now is we don't have knowledge of the liquid level.

sgbaird, 2025-01-13:
It's more as a proof of concept for being able to control gas flow (e.g., if someone is flowing nitrogen into a system). E.g., someone comes into the lab and sees the Pioreactor - we describe and show how it can do heating, stirring, optical measurements, liquid transfer, etc. and also control the flow of gas.

Sorry for the lack of clarity - this is unrelated to the liquid level issue.

linx5o, 2025-01-13:
Understood.

sgbaird, 2025-01-14:
Might be good to know where we can purchase these types connectors to make a custom one with our valves (which I think simply have two wires). We'll also need to double check the orientation (+ vs -) and whether it's normally closed or normally open. I have the specs for what we bought so lmk if I need to dig it up.

[image]

Also, when you make the pioreactor forum post, please include a link to it in this issue

linx5o, 2025-01-14:
[Forum post for connector type](https://forum.pioreactor.com/t/pinch-valve-connector/631)

linx5o, 2025-01-17:
The PS-1615NC pinch valve appears to operate regardless of polarity.

For the Pioreactor after installing the [relay plugin](https://github.com/camdavidsonpilon/pioreactor-relay-plugin) set the `start_on` for the relay in the configurations to 0.  May require restart.
[image]

sgbaird, 2025-01-17:
Nice! At some point a video would be great. Thanks for documenting.

linx5o, 2025-01-17:
[image]

Current setup, two individual connectors plugged into the PWM port. No stand or support for the pinch valve. Need thinner tubing to test. I believe 1.6 inner diameter 3.2 outer diameter tubing. 

Functionality has been added to huggingface.

Possible concern of continuous use of pinch valve leading to high temperatures. (60 degrees Celsius)

sgbaird, 2025-01-17:
Nice! Do you know if the pinch valve is on/off only or if the position can be controlled partially? I think it mentioned PWM as a way to avoid overheating, and good point about the temperature being a concern.

@Neil-YL could you print one of https://www.printables.com/model/356918-dovetail-pinch-valve-holder in black please?

linx5o, 2025-01-17:
[image]
The Pioreactor only has a switch, hardware might support it though.

sgbaird, 2025-01-17:
Ah, got it! Do you mind asking on the forum? They might know. Thanks for digging into this issue. Feel free to close when you're ready.

Neil-YL, 2025-01-17:
[image]
Only need one right?
Do I need to test the fitting?

sgbaird, 2025-01-17:
Just one, correct. If you could just leave it somewhere that @linx5o can get to that would be great. Thanks!

Neil-YL, 2025-01-17:
@linx5o you can find it next the pinch valve on the same desk

sgbaird, 2025-02-25:
Not urgent, but at some point let's get a video of the pinch valve operating and add a link to HF here (was any modification required on the device side?)

linx5o, 2025-02-25:
> Not urgent, but at some point let's get a video of the pinch valve operating and add a link to HF here (was any modification required on the device side?). Then we can close 

No additional edits needed. Just following the guide above and installed one module was all that was needed. I will work on video.

closed 2025-03-01 by sgbaird

sgbaird, 2025-05-22:
Aside: valve being used by Ilya

https://github.com/AccelerationConsortium/ac-training-lab/issues/139#issuecomment-2901831775