AccelerationConsortium/ac-dev-lab #147: Set up remote Python control on A1 mini

sgbaird, 2025-01-11:
https://github.com/greghesp/ha-bambulab likely best

Just ordered AMS version with extra 0.2, 0.6, and 0.8 mm nozzles

https://ca.store.bambulab.com/products/a1-mini

Some possibility of using:
- https://github.com/mchrisgm/bambulabs_api (https://github.com/mchrisgm/bambulabs_api/issues/48)
- https://github.com/mattcar15/bambu-connect (

Neil-YL, 2025-01-15:
I will lean to using this https://github.com/mchrisgm/bambulabs_api, given it has more developer-oriented features and a well-documented [usage guide]( https://mchrisgm.github.io/bambulabs_api/api.html).

The set up seems quite straightforward but I still need to register the printer with a Bambu account to collect all the information for the Python API connection setup:

```
import time
import bambulabs_api as bl

IP = '192.168.1.200'
SERIAL = 'AC12309BH109'
ACCESS_CODE = '12347890'
```

Now I have the IP and SN, but for the access code is now shown as an IP address not the numerical code, according to the community it seems I need to login to the Handy App then the Studio app to enable this access code feature.

https://forum.bambulab.com/t/a1-mini-where-do-i-find-the-access-code/56991/5

Neil-YL, 2025-01-15:
I have set up the connection with the Bambulabs-api. Most of the commands work well:

- Light on/off
- Pause/resume print job
- Get status of printer (Job status, temperature of nozzle/bed, print speed, fan speed)
- Set print speed
- Capture photo from camera

Example photo: 
[image]

Example video (the printer reacts faster than I expect):

https://github.com/user-attachments/assets/4b55c2c4-6874-4d49-9797-a21fce66ab57

However, the set_bed_temperature command doesn’t update the target temperature on the UI or change the actual bed temperature (as checked using the bed_temp command), regardless of whether a print job is in progress.

though I think still possible to adjust the bed temperature (and most print settings) by modifying the G-code file and maybe a better practice comparing to changing the speed and temperature during the print job

@sgbaird @SissiFeng Do you have any idea on to which level / how many parameters the students will need to change in the MO practical?

sgbaird, 2025-01-15:
Thank you! Probably at minimum, nozzle temperature, print speed, and bed temperature. Maybe something related to extrusion speed? Nozzle temperature and print speed would be a good start.

Neil-YL, 2025-01-17:
I believe all of these parameters can be modified directly in the G-code for the print model. If the G-code is too complex for the students to understand, we could provide a script to directly edit these parameters in the G-code and send them to the printer using the Python API. Alternatively, modifying the API to expose these parameters could be another option.

@SissiFeng and I have discussed that to have a relatively complicated pattern to print so the results will be more evaluable than a simple one. One possible choice could be the UofT logo or the tree on the logo. I think Sissi would give more comments on this after we have several trial prints.

[image]

sgbaird, 2025-01-17:
We'll want the students to only interact with the Python function, not with g code, so we should have a wrapper. Up to you to send g code via API vs. modify the package to expose those commands. One way or another the top level Python function will take the parameters as inputs and enforce reasonable bounds for safety. If you modify the package, please fork the repo, make the changes there, include a link to it here, and make a pull request to the main repo. Might be worth opening an issue to ask about recommendations for getting this functionality, but doesn't have to stop you from trying out things.

Oh, nice! I like this idea. Maybe the AC logo / graphic? (Without the text). I have an SVG if you need.

sgbaird, 2025-01-22:
Does the Python package assume the orchestrator is connected to the same network as the A1 mini, or can it be remote accessed? If it can't be remote accessed, then we probably need a dedicated device for it, right? (I suppose we could dual-purpose the Zero 2W on the camera toolhead)

Neil-YL, 2025-01-22:
> Does the Python package assume the orchestrator is connected to the same network as the A1 mini. 

I think so. It is using the local IP address for connection.

sgbaird, 2025-02-14:
https://github.com/AccelerationConsortium/ac-training-lab/issues/168#issuecomment-2654319066 (API suddenly stopped working, more discussion later in that issue)

sgbaird, 2025-02-14:
@Neil-YL maybe try a factory reset? https://www.reddit.com/r/BambuLab/comments/1gizs3s/suddenly_getting_05004003_error_when_printing/

sgbaird, 2025-02-14:
You could also try with the X1E to help with troubleshooting if you want. Up to you how you want to approach it

Neil-YL, 2025-02-14:
> [@Neil-YL](https://github.com/Neil-YL) maybe try a factory reset? https://www.reddit.com/r/BambuLab/comments/1gizs3s/suddenly_getting_05004003_error_when_printing/

Yes I have seen a similar suggestion from the community forum as well. That is the very last thing I have not tried on the list. We are setting up the RBPi for the printer and I will reset the machine later(maybe next Tuesday).

sgbaird, 2025-02-15:
Looks like someone else may have had a very recent issue begin: https://forum.bambulab.com/t/new-firmware-doent-work-how-do-i-downgrade/47771/13

sgbaird, 2025-02-15:
I'll probably be asking a bunch of things you've already tried:

Have you tried:
1. removing the SD card completely and trying without the SD card installed? (Maybe a bad suggestion if our workflow depends on the SD card) https://forum.bambulab.com/t/parsing-errors/3528/7
2. reformatted SD cards using onboard controls https://forum.bambulab.com/t/parsing-errors/3528/10
3. is the GCode being exported from Bambu studio per https://forum.bambulab.com/t/parsing-errors/3528/15? (I.e., using exported .GCode instead of .3mf file)
4. making sure the file is sliced for A1 mini https://forum.bambulab.com/t/parsing-errors/3528/16
5. reinstall Bambu studio
6. trying a different WiFi (e.g., phone hotspot)
7. deleting files from SD card manually via Bambu studio

sgbaird, 2025-02-15:
Are you able to downgrade firmware or are you locked out of doing that?

Neil-YL, 2025-02-18:
> Are you able to downgrade firmware or are you locked out of doing that?

I didn't upgrade the firmware after reset. Current firmware is 01.04.0.0, I believe this is also the version we had the successful print via API. 

I could try roll back to 1.03 tomorrow.

> I'll probably be asking a bunch of things you've already tried:
>
> Have you tried:
>
> 1. removing the SD card completely and trying without the SD card installed? (Maybe a bad suggestion if our workflow [...]

It is always Gcode on my side, but via the API it will be packed as .3mf to save on the printer, according to the API send GCode [example](https://github.com/mchrisgm/bambulabs_api/blob/main/examples/print/print_gcode.py)
  
> 4. making sure the file is sliced for A1 mini https://forum.bambulab.com/t/parsing-errors/3528/16 ✅ 

It was the previous 10x10 square file, should be for A1 mini. I will try sending another part again.

> 5. reinstall Bambu studio  🔜 will try today
> 6. trying a different WiFi (e.g., phone hotspot)  🔜 will try today
> 7. deleting files from SD card manually via Bambu studio ✅

sgbaird, 2025-02-18:
Thanks! Lmk how it goes

sgbaird, 2025-02-19:
Any luck?

sgbaird, 2025-02-19:
Possibly, reformatting needs to happen multiple times: https://forum.bambulab.com/t/prints-not-downloading/17575/45?u=sgbaird

sgbaird, 2025-02-19:
Are we able to send single lines of GCode successfully via https://github.com/mchrisgm/bambulabs_api/blob/0965030b725ed4c1aa859917ea790d630f5fd269/bambulabs_api/mqtt_client.py#L459?

Neil-YL, 2025-02-19:
The BambuStudio always works on my Mac for slicing part, sending print job and starting print. I try to reinstall and generate the Gcode again, still failed to start the print with the same error. My iPhone's internet sharing cannot found by the printer, even though I turn off/on the Maximise compatibility. 

Sissi will try her PC and phone as well as the downgrading firmware.

Sorry for not working too much with A1 today. After the meeting with Jae, I moved to NIR testing in order to know the potentials of his idea.

sgbaird, 2025-02-19:
All good. We talked about Feb 23rd as a go / no-go day for the 3D printing optimization, and I'd like to prioritize this one as much as we can until then. Thanks for your patience through the process.

SissiFeng, 2025-02-19:
I am summarizing the current troubleshooting efforts to keep everything organized and to facilitate further debugging and discussion.

1. BambuStudio Works, But Lacks Automation
BambuStudio functions as expected—we can slice parts, send print jobs, and start prints without any issues. Our gcode file with sliced information is extracted studio.

2. G-code Compatibility
The G-code exported from BambuStudio should theoretically be fully compatible with the A1 mini. There is no indication that the G-code format itself is causing the issue.

3. API Printing Issue: My Setup vs. Yang Huang's Setup
Initially, Yang Huang was able to send print jobs via the API, while my system was never able to start prints via the API, despite trying multiple troubleshooting methods.

To ensure consistency, I asked him to package his working environment into a Docker container. However, starting from that day, his API-based printing also stopped working, exhibiting the same behavior as mine. This suggests that the issue is not specific to individual machine setups but could be related to the API behavior, firmware constraints, or network communication issues.

4. Troubleshooting Efforts
Since then, we have tested available solutions that we could find online, including:
Various API request modifications
Network settings adjustments
System reconvery
Format SD card several times
Despite these efforts, the issue remains unresolved.

This disrupts automation and suggests that the API might require additional authentication to allow fully automated printing.

5. Firmware Downgrade Issue
Currently, A1 Mini firmware version is 1.04, cannot be downgraded to 1.02 through SD card is not possible, as firmware versions below 1.04 do not support offline updates.
Official online updates do not support downgrading, only upgrading.
I have contacted Bambu Lab support to check if they can provide a 1.02 firmware file for manual installation.

6. API Response: M115 returns valid information, confirming that the API can successfully communicate with the printer. However, G-code execution requires additional permissions that seem to be either embedded in BambuStudio or require manual confirmation on the touchscreen. 

Current Status:
✅ The API successfully sends print commands, and the A1mini receives them.
❌ However, the printer does not automatically start printing.
❌ Manual intervention is required—we must press a button on the Bambu touchscreen to start the print job.

sgbaird, 2025-02-20:
Thanks! Tomorrow, could you please record and share a video of the following pieces you mentioned?

> ✅ The API successfully sends print commands, and the A1mini receives them.
> ❌ However, the printer does not automatically start printing.
> ❌ Manual intervention is required—we must press a button on the Bambu touchscreen to start the print job.

SissiFeng, 2025-02-20:
https://github.com/user-attachments/assets/34b78338-84a5-46c3-9f64-97ac64eb537c

Neil-YL, 2025-02-20:
<details>
  <summary>Code</summary>

```python
import time
import bambulabs_api as bl
from io import BytesIO
import zipfile

IP = '192.168.1.124'
SERIAL = '0309CA471800852'
ACCESS_CODE = '14011913'

INPUT_FILE_PATH = '10x10 0.1.gcode'
UPLOAD_FILE_NAME = '10x10 0.1.3mf'

def create_zip_archive_in_memory(
        text_content: str,
        text_file_name: str = 'file.txt') -> BytesIO:
    """
    Create a zip archive in memory

    Args:
        text_content (str): content of the text file
        text_file_name (str, optional): location of the text file.
            Defaults to 'file.txt'.

    Returns:
        io.BytesIO: zip archive in memory
    """
    zip_buffer = BytesIO()
    with zipfile.ZipFile(zip_buffer, 'w', zipfile.ZIP_DEFLATED) as zipf:
        zipf.writestr(text_file_name, text_content)
    zip_buffer.seek(0)
    return zip_buffer

if __name__ == '__main__':
    print('Starting bambulabs_api example')
    print('Connecting to BambuLab 3D printer')
    print(f'IP: {IP}')
    print(f'Serial: {SERIAL}')
    print(f'Access Code: {ACCESS_CODE}')

    # Create a new instance of the API
    printer = bl.Printer(IP, ACCESS_CODE, SERIAL)

    # Connect to the BambuLab 3D printer
    printer.connect()

    time.sleep(2)

    # Get the printer status
    status = printer.get_state()
    print(f'Printer status: {status}')

    bed_temperature = printer.get_bed_temperature()
    nozzle_temperature = printer.get_nozzle_temperature()
    speed = printer.get_print_speed()
    print(
        f'Printer status: {status}, Bed temp: {bed_temperature}, '
        f'Nozzle temp: {nozzle_temperature},'
        f'Print speed: {speed}')

    with open(INPUT_FILE_PATH, "r") as file:
        gcode = file.read()

    gcode_location = "10x10 0.1.gcode"
    io_file = create_zip_archive_in_memory(gcode, gcode_location)
    if file:
        result = printer.upload_file(io_file, UPLOAD_FILE_NAME)
        if "226" not in result:
            print("Error Uploading File to Printer")

        else:
            print("Done Uploading/Sending Start Print Command")
            printer.start_print(UPLOAD_FILE_NAME, 1)
            print("Start Print Command Sent")

```
</details>

Output:
> Starting bambulabs_api example
Connecting to BambuLab 3D printer
IP: 192.168.1.124
Serial: 0309CA471800852
Access Code: 14011913
Starting camera thread.
Printer status: UNKNOWN
Printer status: UNKNOWN, Bed temp: 0.0, Nozzle temp: 0.0,Print speed: 100
Done Uploading/Sending Start Print Command
Start Print Command Sent

and if send single start print command:

[image]

sgbaird, 2025-02-20:
Thanks! Could you also include pictures of the two SD cards that had been tried with it? (The one that came with the printer and the other one that you tried)

SissiFeng, 2025-02-20:
[image]

linx5o, 2025-02-20:
I have created a Huggingface interface and raspberry pi script to get the status, bed temp and nozzle temp. 

[HuggingFace UI](https://huggingface.co/spaces/AccelerationConsortium/A1_Mini)

SissiFeng, 2025-02-21:
> I have created a Huggingface interface and raspberry pi script to get the status, bed temp and nozzle temp.
> 
> [HuggingFace UI](https://huggingface.co/spaces/AccelerationConsortium/A1_Mini)

Thank you for your dedicated work. Based on our last group meeting discussion, it’s important to keep your PI aligned with the overall HF system. Johnathan has already worked on integrating his monitoring PI into the full implementation. It would be great if you could also explore a similar approach, ensuring a seamless integration rather than providing a separate link that requires additional effort from the team.

https://huggingface.co/spaces/AccelerationConsortium/Bambu_A1mini/tree/main

SissiFeng, 2025-02-21:
If this feels a little pain, no worries — I will handle it on my end.

sgbaird, 2025-02-21:
Currently, https://huggingface.co/spaces/AccelerationConsortium/Bambu_A1mini has a runtime error. Have the points mentioned in https://huggingface.co/spaces/AccelerationConsortium/Bambu_A1mini/discussions/1 been addressed yet? Currently @linx5o's implementation follows with the idea of MQTT being at the top-level.

sgbaird, 2025-02-21:
For the hugging face piece, we can port over the conversation to https://github.com/AccelerationConsortium/ac-training-lab/issues/160.

sgbaird, 2025-02-21:
@Neil-YL could you open an issue at https://github.com/bambulab/BambuStudio (search for related issues first)?

Maybe related: https://github.com/bambulab/BambuStudio/issues/1864

Neil-YL, 2025-02-21:
> [@Neil-YL](https://github.com/Neil-YL) could you open an issue at https://github.com/bambulab/BambuStudio (search for related issues first)?
> 
> Maybe related: [bambulab/BambuStudio#1864](https://github.com/bambulab/BambuStudio/issues/1864)

I don't think BambuStudio has any problems, since everything works normally with BS. The error raises when we send print command to the printer **via API**

sgbaird, 2025-02-21:
I was thinking that something might have changed with the firmware or someone might know some troubleshooting steps for address the specific error code we've been running into. Also, it sounded like the Python API version was the same when it was working vs. when it wasn't, which means something else had changed.

SissiFeng, 2025-02-21:
https://github.com/SissiFeng/bambu-printer-control

Pi 4b is responsible for core functions:
- Direct printer control
- Database management and backup
- MQTT communication (interaction with HF Space)
- Location management
- G-code generation and processing

sgbaird, 2025-02-25:
@SissiFeng could you please transfer the ownership to AccelerationConsortium?

sgbaird, 2025-02-28:
@Neil-YL, could you confirm that the bed temperature is still something we can't set via the API yet? (https://github.com/mchrisgm/bambulabs_api/issues/99#issuecomment-2675210319). Could you also confirm which parameters we've been able to successfully control?

Neil-YL, 2025-02-28:
> [@Neil-YL](https://github.com/Neil-YL), could you confirm that the bed temperature is still something we can't set via the API yet? ([mchrisgm/bambulabs_api#99 (comment)](https://github.com/mchrisgm/bambulabs_api/issues/99#issuecomment-2675210319)). Could you also confirm which parameters we've been able to successfully control?

No. The set_bed_temperature command still does not change the bed temperature, but we can use [send Gcode command](https://mchrisgm.github.io/bambulabs_api/api/printer.html#bambulabs_api.Printer.gcode) to set the bed temperature. (See video below)

https://github.com/user-attachments/assets/343ae717-1f16-417b-ad1b-f4e36a85f381
current bed temp: 55
via API send Gcode command "M190 S65" (M190 for bed temp and S65-->set 65 degrees)
bed temp on Bambu: changed to 65 ✅

sgbaird, 2025-02-28:
Oh, great! I thought that even Gcode commands weren't affecting the bed temperature, but maybe I was just remembering the issues with contacting the printer.

Neil-YL, 2025-02-28:
Today Sissi finished the set up on the RB Pi4 and also tested it with sending customised Gcode file and start print via the RB Pi4 with BambuAPI. 

We also successfully changed the nozzle temperature and the bed temperature by pre-set in the G-code file. Followings are the video of starting the print job and printing with 3 (squares) x 2 (different parameters pre-sets) = 6 squares with one Gcode file.

https://github.com/user-attachments/assets/7990856d-b7db-4d73-ad2f-b961717f34cf
https://github.com/user-attachments/assets/178ce05d-d082-4c46-bb4b-3b47cebaa67d

sgbaird, 2025-02-28:
Amazing! Thank you @SissiFeng, @Neil-YL, and @Jonathan-Woo!

Jonathan-Woo, 2025-02-28:
PR here (sorry if this is a double notif, not sure if HF sends notifs)

https://huggingface.co/spaces/AccelerationConsortium/Bambu_A1mini/discussions/2

sgbaird, 2025-03-04:
There was a linked PR that closed this (typical behavior when you link a PR to an issue). Reopening