AccelerationConsortium/ac-dev-lab #148: Position feedback from linear actuator

sgbaird, 2025-01-14:
https://github.com/AccelerationConsortium/ac-training-lab/issues/141#issuecomment-2588638958

https://www.actuonix.com/lac-board-intro?srsltid=AfmBOopJR0bP6b83xY1mzNJnjBf2eja3fOiL6rOgy8MKNan4byd39Wzr

@MatPopp could you describe what you tried? I can ask someone to continue the debugging if needed.

MatPopp, 2025-01-14:
* tried to connect it (the shield) solely via USB (without extra power supply) -> could not establish a connection
* thought about directly connecting the actuator to a pi pico -> not as simple as with the 3 wire actuators: of the 5 wires, 3 are for a potentiometer, 2 for dc at motor, so no rc-type control possible, H-bridge or so would be needed. 

Then I stopped, since the 3 wire actuator was rather easy to control, and I wanted to proceed with pressure measurements.

Suggestion to try next: supply shield with 6 V, then use usb connection.

sgbaird, 2025-01-14:
Sounds good! Re-reading it, it sounds like that is the issue.

MatPopp, 2025-01-15:
Just hooked up the actuator with the power supply from the microscopy booth. 
* Works fine when adhering to the video-explaination in https://www.actuonix.com/lac-board-intro?srsltid=AfmBOor5Nq6NOY6UCE4yBNgUPAPF2YBUuorXLC0PyTyGMoHuzkrWkaK5
* python interface is still TODO. There is a labrary here: https://github.com/DamnedFacts/actuonix-lac but it does not work out of the box on windows; error message: usb.core.NoBackendError: No backend available

sgbaird, 2025-01-15:
Nice! It said it supports PWM, at which point I don't think you'll be using the USB interface (instead you would use a Pico W with a PWM pin connected to the right spot on the LAC)

MatPopp, 2025-01-15:
Got it. I was nat aware of that the lac board also has a pwm output for the current position signal. 

This arduino description should work with the pi pico as well: https://www.actuonix.com/lac-board-arduino

sgbaird, 2025-01-15:
Nice find. I also found https://www.actuonix.com/assets/images/datasheets/ActuonixLACDatasheet.pdf 

[ActuonixLACDatasheet.pdf](https://github.com/user-attachments/files/18418318/ActuonixLACDatasheet.pdf)

And it seems that while we can control position with PWM, speed is either controlled manually (and coarsely) by adjusting a screw or it can be controlled via USB (i.e., what you were describing). Shame it doesn't have a plug and play Python interface.

I'm seeing X2 (position feedback), X6 (PWM control), and P1 (**max** speed set by screw adjustment) from the datasheet as the most important for us right now.

Maybe we can just set the speed to max via the screw and use a while loop with a timer on the Pico W to manually create a lower speed (i.e., not specific to the LAC board).

sgbaird, 2025-01-21:
cc @kelvinchow23

kelvinchow23, 2025-02-19:
Here's a demo of the actuator hooked up to the LAC and Raspberry Pico to send position information via PWM and reading the analog position feedback from the actuator. I'll throw the wiring instructions and script [here](https://github.com/kelvinchow23/actuonix-actuator-feedback-demo): 
https://github.com/user-attachments/assets/9b675ec4-6ea3-40f6-8395-f95342c70da7

sgbaird, 2025-03-27:
Cc @cyrilcaoyang

cyrilcaoyang, 2025-03-30:
A Python package was added to control LAC, which drives L16-P models via mini-USB.

https://github.com/cyrilcaoyang/actuonix_lac

Model L16-R will be easier to control.