AccelerationConsortium/ac-dev-lab #149: Set up A1 mini

sgbaird, 2025-01-14:
@Neil-YL I think this will arrive tomorrow. This is intended to be used for Jae's course.

#147 #144

Neil-YL, 2025-01-14:
I will set it up with the Bambu app to have a trial print and then with the remote Python control later

Neil-YL, 2025-01-14:
[image]
I finished the assembly of the new Bambu A1 mini 3D printer. It is now set up at the cubicle next to the chocolate printer. I will set up the connection with the Bambu App and later with the remote Python control tomorrow.

sgbaird, 2025-01-14:
Great! Was the setup straightforward?

Neil-YL, 2025-01-15:
> Great! Was the setup straightforward?

Yah, just follow the guide on the manual, the most challenging part is entering the wifi password on that small touch screen.

Neil-YL, 2025-01-15:
I attached a yellow PLA filament and had a scraper printed and I think all the mechanical set up are finished this stage. For the python API I will update to [this issue](https://github.com/AccelerationConsortium/ac-training-lab/issues/147)

closed 2025-01-20 by sgbaird