AccelerationConsortium/ac-dev-lab #151: Maximizing compressive strength of a 3D printed resin lattice

chasedamagician, 2025-01-16:
Hello, I am Chance, and I am currently working with Dr. Baird for my AP Research Project which aims to see how the post-processing of resin 3D printed lattices affects the structure's compressive strength. This is to see if it is possible for scientists to mimic the physical properties of high-strength polymers such as Polyetheretherketone (PEEK)  through the use of  accessible 3D printed resin structure
### **TODO: Printing**

- Done: Basic Octet-Truss Lattice Structure has been designed
- Select initial 10 batches for the surrogate model each with 3 structures per batch
- Currently, the plan is to base the initial points on the suggested post-processing values on the Formlabs website for Clear Resin which maximizes compressive and tensile strength. 
- The project will also experiment with alterations to the design of the lattice structures, altering the thickness of the supporting columns and the amount of unit cells used in the structure
- Currently, it is unclear if I will be able to do the post-processing myself because I do not have access to the building so I might have to rely on Mr. Kelvin and others to complete the post-processing
- At the current moment, I am using Clear Resin to print the lattice structure, but a possible variable would be the resin used in the printing (clear, grey, Rigid 10k, etc…)

### **Step 2: Testing**

- A Universal Testing Machine will be used to measure the compressive strength of the lattice structures
- My primary concern at the moment is that I will not be allowed to operate the machine due to safety concerns
- I am currently talking with some of the people using the machine to see if they are willing to help with my project. 

### **Step 3: Analyzing**

- I will use the AxPlatform and Honegumi to design a Bayesian Optimization program for this experiment
- My current configuration of the Honegumi Settings is:

- Multi-objective
- Default Model-type
- Single Task
- Composition constraint
- Custom Threshold
- Single Synchrony
- And Visualize Results
- I will also need to create a formula for the relationship between the variables

- I will need to set the boundaries of the variable values to create my search

sgbaird, 2025-01-22:
[image]

[image]

[image]

https://github.com/user-attachments/assets/a4e37eb8-5620-473b-8584-6a85dfb61321

sgbaird, 2025-01-22:
[image]

~15 mm clearance

sgbaird, 2025-01-22:
[image]

chasedamagician, 2025-01-22:
Hello here is the first batch of data for the compressive strength testing: 

[Chance - Lattice test 1_01222025_150649.csv](https://github.com/user-attachments/files/18512119/Chance.-.Lattice.test.1_01222025_150649.csv)

This data is not usable for analysis because the plat tilted during compression, but it proves the experiment is feasible. For the actual batches of lattice structures, I will reduce the size of the prints and add a 2mm layer to the top.

sgbaird, 2025-01-22:
Nice! Do you also have an image of the single layer after post processing?

chasedamagician, 2025-01-22:
Here is the single-layer

[image]

 lattice structure after the compressive test:

sgbaird, 2025-01-22:
Thanks! We'll also want to look at the load (N) vs. time curve to see what might be required for a DIY setup.

sgbaird, 2025-01-23:
Looks like the maximum is ~701 N (~110 seconds into the workflow) at a maximum position of 8.48 mm. It reaches a load of ~100 N at 0.56 N, ~150 N at 0.67 mm,  ~200 N at 0.82 mm, ~127 N at 5 mm, ~200 N at 6.9 mm, ~600 N at 8.3 mm, ~701 N at 8.48 mm.

For safety context, past ~150 N might be around the threshold for hazard/injury in terms of finger / pinch force. https://pubmed.ncbi.nlm.nih.gov/12820907/

chasedamagician, 2025-01-24:
Here are the plots for the first compression test, keep in mind there was some wobbling in the plate during the compression which affected the data:

Load over Time:
[image]

Stress vs. Strain
[image]

I am currently in the process of smoothing out the displacement over time plot. I have also almost finished the table of the values I want to use for my 10 initial experiments. I still need to redesign the lattice structure before resetting the bounds for the edge-length and strut-diameter values. Thank you

sgbaird, 2025-01-24:
That's great! Might be good for Hongchen or Keith Brown to weigh in on this as well.

sgbaird, 2025-02-14:
@chasedamagician any updates here or things I can help with? (Checking intermittently on certain projects)

chasedamagician, 2025-02-14:
Hello sorry for not updating this channel. So far I have completed the printing of all 11 batches of initial lattice structures that I will use to make my surrogate model. Additionally, I have completed experimentation for 3 of the 11 batches and have analyzed the Young's modulus and compressive strength of each batch. I will include a link to my data sheets for the individual lattices as well as the batches below. 

Currently I am in the process of programming my Bayesian model however currently my issue is figuring out how I will design the "measure properties" function that will calculate the compressive strength based on the input parameters. My current plan is to see if the Axplatform offers.

Batches Datasheet : [https://docs.google.com/spreadsheets/d/1v1shYJLnGDtbV6S0ZlWaOzrl0ChahEquiaUSqRNOxpM/edit?gid=0#gid=0](url)
Lattice Data: [https://drive.google.com/drive/folders/1xzbUMMLOpimWTpVn8GQ4QyIpdvvY6gQ5?usp=drive_link](url)

sgbaird, 2025-02-15:
Have a look at https://youtu.be/4tnaL9ts6CQ?si=_d3owbn76x9KsDxX

sgbaird, 2025-02-15:
Also, the training/predicting for the surrogate model is something that happens under the hood in this case.

sgbaird, 2025-02-20:
Aside: see https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#links for formatting links in markdown (also, you can include the link without any markup like I did here and it will render correctly).

chasedamagician, 2025-02-20:
Hello, the link should work now

chasedamagician, 2025-02-20:
Thank you so much for the suggestions. I will apply these changes to my code tomorrow. I have started taking before and after photos for the 2 latest batches and also have a recording of a crushed lattice structure. Thank you for your time.

sgbaird, 2025-02-20:
Nice! Could you attach a couple of those photos and the video to this issue? You should be able to click "add files" (either on your phone or your computer) when adding a new comment.

chasedamagician, 2025-02-25:
Dear Dr. Baird

The video file is too large to upload here, so I will send it to you via Slack. I was unable to attend the latter half of the experiments, so I do not have photos of those lattice structures. However, I will take additional photos and videos of the structures suggested by the program.

Additionally, I have successfully run the first draft of my Bayesian program, thanks to your suggestions. I will share it in a follow-up comment. I also plan to refine the input and output variable thresholds and boundaries further.

I am considering adding another objective to address cases where the lattice structure did not undergo plastic deformation, which makes the highest recorded stress value an inaccurate measure of compressive strength. To better distinguish high-performing lattices, I am thinking about incorporating Young’s Modulus as a third objective, even though it optimizes stiffness. I would appreciate your thoughts on this approach.

Lastly, I am adjusting the goal of my experiment. Instead of using post-processing to mimic the compressive strength of complex polymers, I am focusing on optimizing the compressive strength of the resin lattice structures. While my initial research objective was compelling, I realized that I lack the necessary background to produce an effective report on that topic.

I will attach my current code in the next comment. Thank you for your time.

chasedamagician, 2025-02-25:
`
import numpy as np
import pandas as pd
from ax.service.ax_client import AxClient, ObjectiveProperties
import matplotlib.pyplot as plt
from mpl_toolkits.mplot3d import Axes3D

obj1_name = "compressive_strength"
obj2_name = "relative_density"
obj3_name = "y_modulus"

def measure_properties(cure_time, cure_temp, wash_time, pre_time, edge_len, strut_dia):
    #Insert matethematical model once the intial experiments are complete.
    y= 0

    #return y

# Define the training data

#training data
X_train = pd.DataFrame(
    [
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
    ]
)

# Define y_train (normally the values would be supplied directly instead of calculating here)
y_train = [
    {obj1_name: 3.01, obj2_name: 0.0245, obj3_name: 1.10},
    {obj1_name: 1.45, obj2_name: 0.0245, obj3_name: 0.59},
    {obj1_name: 1.44, obj2_name: 0.0245, obj3_name: 1.11},
    {obj1_name: 2.00, obj2_name: 0.0245, obj3_name:0.62},
    {obj1_name: 2.30, obj2_name: 0.0245, obj3_name:0.81 },
    {obj1_name: 90.0, obj2_name: 0.089, obj3_name:33.89},
    {obj1_name: 97.3, obj2_name: 0.089, obj3_name: 32.22},
[... 124 lines cut ...]
[... cut ...]
               color="red", s=80, label="Pareto-optimal Points", marker="^")

    # Labels and title
    ax.set_xlabel("Compressive Strength")
    ax.set_ylabel("Relative Density")
    ax.set_zlabel("Young's Modulus")
    ax.set_title("3D Scatter of All Trials and Pareto Front")

    ax.legend()
    plt.show()

    plt.scatter(df_all[obj1_name], df_all[obj2_name], alpha=0.5)
    plt.xlabel("Compressive Strength")
    plt.ylabel("Relative Density")
    plt.title("2D Projection: Compressive Strength vs Relative Density")
    plt.show()
`

chasedamagician, 2025-02-25:
```
obj1_name = "compressive_strength"
obj2_name = "relative_density"
obj3_name = "y_modulus"

def measure_properties(cure_time, cure_temp, wash_time, pre_time, edge_len, strut_dia):
    #Insert matethematical model once the intial experiments are complete.
    y= 0

    #return y

# Define the training data

#training data
X_train = pd.DataFrame(
    [
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 42.0, "cure_temp": 35.0, "wash_time": 10.0, "pre_time": 225.0, "edge_len": 4, "strut_dia": 0.2},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 108.0, "cure_temp": 78.0, "wash_time": 3.0, "pre_time": 69.0, "edge_len": 2.21, "strut_dia": 0.21},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 32.0, "cure_temp": 71.0, "wash_time": 11.0, "pre_time": 29.0, "edge_len": 3.16, "strut_dia": 0.380},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 136.0, "cure_temp": 36.0, "wash_time": 2.0, "pre_time": 22.0, "edge_len": 2.21, "strut_dia": 0.263},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 115.0, "cure_temp": 78.0, "wash_time": 17.0, "pre_time": 39.0, "edge_len": 1.34, "strut_dia": 0.246},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 72.0, "cure_temp": 53.0, "wash_time": 3.0, "pre_time": 25.0, "edge_len": 2.70, "strut_dia": 0.371},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
        {"cure_time": 149.0, "cure_temp": 47.0, "wash_time": 13.0, "pre_time": 21.0, "edge_len": 2.11, "strut_dia": 0.338},
    ]
)

# Define y_train (normally the values would be supplied directly instead of calculating here)
y_train = [
    {obj1_name: 3.01, obj2_name: 0.0245, obj3_name: 1.10},
    {obj1_name: 1.45, obj2_name: 0.0245, obj3_name: 0.59},
    {obj1_name: 1.44, obj2_name: 0.0245, obj3_name: 1.11},
    {obj1_name: 2.00, obj2_name: 0.0245, obj3_name:0.62},
    {obj1_name: 2.30, obj2_name: 0.0245, obj3_name:0.81 },
    {obj1_name: 90.0, obj2_name: 0.089, obj3_name:33.89},
    {obj1_name: 97.3, obj2_name: 0.089, obj3_name: 32.22},
    {obj1_name: 130.6, obj2_name: 0.089, obj3_name: 27.93},
    {obj1_name: 155.54, obj2_name: 0.089, obj3_name: 35.70},
    {obj1_name: 131.8, obj2_name: 0.089, obj3_name: 27.94},
[... cut ...]
               color="red", s=80, label="Pareto-optimal Points", marker="^")

    Labels and title
    ax.set_xlabel("Compressive Strength")
    ax.set_ylabel("Relative Density")
    ax.set_zlabel("Young's Modulus")
    ax.set_title("3D Scatter of All Trials and Pareto Front")

    ax.legend()
    plt.show()

    plt.scatter(df_all[obj1_name], df_all[obj2_name], alpha=0.5)
    plt.xlabel("Compressive Strength")
    plt.ylabel("Relative Density")
    plt.title("2D Projection: Compressive Strength vs Relative Density")
    plt.show()
```

sgbaird, 2025-02-25:
Nice! That's great. I can support on the post processing aspects and reflections in the report. I'd recommend keeping it. For the video, are you able to upload it to YouTube as an unlisted video and share the link here? 

No worries about the photos, it's an optional nice to have. Also don't feel like you have to take videos of every single test if it's too much of a pain.

> Can you look at the BEAR paper to see how they handled their objective (e.g., normalizing by the weight, keeping the top-view area constant, keeping height the same) and report back?

Any update here? This would help me comment on adding other objectives.

chasedamagician, 2025-02-25:
Sorry for the oversight! From what I understand from the paper, the researchers normalized their objectives by setting a maximum force threshold of 4.8 kN to prevent excessive strength and by ensuring consistency through the same material and printing process for each lattice.

I have a meeting with Dr. Brown early this week, and I'll ask him to confirm that I didn’t miss anything.

Thank you

sgbaird, 2025-02-25:
Thanks! And sounds great. Odd, it still says `We're processing this video. Check back later.`: https://www.youtube.com/watch?v=L6YDirzY3qw

chasedamagician, 2025-02-25:
That's odd, I re-uploaded the video so it should work now. 

https://youtube.com/shorts/0_pvi-PM770

sgbaird, 2025-03-04:
Any updates from the meeting with Keith?

sgbaird, 2025-03-12:
Based on our meeting today, let's also take a look at a parity plot between the model's predicted `relative_density` and the actual analytical `relative_density` (since we're technically treating `relative_density` as a black box objective even though we know the the equation for it)