AccelerationConsortium/ac-dev-lab #152: OT-2 liquid color matching for MSE403H1 practical

Neil-YL, 2025-01-16:
I am testing a Hugging face app demo locally for the practical.

Input block:
student id here acts as the session_id in our MQTT setting and there are another experiment_id for each run with the sensor
R and Y will be used to calculate the B (R+Y+B=1)

[image]

If there is a queue for the experiment 
[image]

when your experiment is running 
[image]
 
Result:
[image]

I will finalise the page tomorrow and deploy it on Hugging face to test remote access.

@sgbaird I might need the admin login for the Nokia router to enable Hugging Face to access the OT-2 via SSH.

Another things to do:

- Collab page for the objective calculation
- The sensor may need adjustment to provide more distinguishable readings. (compared with the light demo, the reading is much smaller). I may also try turning on the light within the OT-2 to see if result will be better.
- any suggestions?

sgbaird, 2025-01-16:
Nice! ☺️ 

- leave as RYB and have the values be in uL. Throw an error if someone requests more than the max allowed in a single well plate
- if you deploy to Hugging Face Spaces and use MQTT (or Prefect) then you shouldn't need SSH access, right?
- please try printing the sensor housing in black
- let's use washable, non-toxic paint instead of food coloring. Can you find something that looks like it would work from Amazon and include the link here?
- how did you implement the queue?

Can you share the code on Hugging Face Spaces and share the code that's running on the OT-2? I'll need that to provide better feedback

Thanks for working on this. Getting closer!

sgbaird, 2025-01-16:
I'd like to take a look at how you implemented the queue mechanism.

Aside: @Neil-YL have a look at https://ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html#hugging-face-spaces (specifically https://ac-training-lab.readthedocs.io/en/latest/devices/setup_iolt_devices.html#hugging-face-spaces)

Related to the remote access and Prefect: https://github.com/PrefectHQ/prefect/discussions/16685

Neil-YL, 2025-01-17:
> * let's use washable, non-toxic paint instead of food coloring. Can you find something that looks like it would work from Amazon and include the link here?

[RYBG  paint](https://www.amazon.ca/Melissa-Doug-Poster-Paint-Crafts-Supplies/dp/B072QFSBRC/ref=sxin_15_pa_sp_search_thematic_sspa?content-id=amzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24%3Aamzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24&cv_ct_cx=washable%2C+non-toxic+paint&keywords=washable%2C+non-toxic+paint&pd_rd_i=B072QFSBRC&pd_rd_r=45adc370-c22c-4000-9ce7-fa3e3724bdd2&pd_rd_w=kf8Ge&pd_rd_wg=wvsbb&pf_rd_p=46621be6-fabe-4126-8501-d32c96c42a24&pf_rd_r=BBAT6MT9SE4PB8GPTNBN&qid=1737141917&sbo=RZvfv%2F%2FHxDF%2BO5021pAnSA%3D%3D&sr=1-4-acb80629-ce74-4cc5-9423-11e8801573fb-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9zZWFyY2hfdGhlbWF0aWM&psc=1): no sure what kind of paint is it

[Tempera Paint](https://www.amazon.ca/Marte-Tempera-Paint-Bottles-Washable/dp/B0DBVH2LNB/ref=sr_1_32?dib=eyJ2IjoiMSJ9.-kpuh5nDMeAa0FUfIRJlxQj2-iH4vYSO8JIwP5ZtjltzS-I9YBHUZ4SH6-cl-0jMPDqWEVw_3Hc2zhC1ksF0k5S1i7O7PbChfHxWs_TgQbWnR7iFXxQi914ld0KYcGRzerrfKj80BkRN_8-UQlKQCIKHoGTqV-HxoMpvmds906xskgKGg3Gpdj0xRcrxzTYeKP1dTS63p1J78OiETFbGJfTR11mJDsXNNMCzw7iB2E8junCroaWdE94fPRS1y8bibHrK_sg5Qg1_JISSyxRRfDgseH8mQmTSjJavP_W3oy1gpzbscVABzBzxHZF6VfaIBXD_SzlLiNXMSbC32iLN0YvOVVNwm9jCqzYjEN7MogE_nJwA4BbFnMUqh113zYv2XCEZEx7Y0lCiACgBwOOYem7wlMKu1CSe2WukKlDekvW4F3YyfPzrthZKW8mLBii2.Sf_jRWkxrHdV9IG_TCEq8rZDUTFnegJrhL3douIlPm4&dib_tag=se&keywords=Prang&qid=1737143030&sr=8-32): Tempera is water-based paint and the thickness can be adjusted (just like food coloring), but could become solid leftover in pipette tip if fully dried out.

There are RYB color single bottle option as well: [red](https://www.amazon.ca/Prang-Washable-Tempera-16-Ounce-10701/dp/B000QE2WDS/ref=sr_1_26?dib=eyJ2IjoiMSJ9.-kpuh5nDMeAa0FUfIRJlxQj2-iH4vYSO8JIwP5ZtjltzS-I9YBHUZ4SH6-cl-0jMPDqWEVw_3Hc2zhC1ksF0k5S1i7O7PbChfHxWs_TgQbWnR7iFXxQi914ld0KYcGRzerrfKj80BkRN_8-UQlKQCPOV8pcXwEnXjaCxzxb-Ic0tauYFrS9fBQShIsVAr7FrKP1dTS63p1J78OiETFbGJfTR11mJDsXNNMCzw7iB2E8junCroaWdE94fPRS1y8bibHrK_sg5Qg1_JISSyxRRfDgseH8mQmTSjJavP_W3oy1gpzbscVABzBzxHZF6VfaIBXD_SzlLiNXMSbC32iLN0YvOVVNwm9jCqzYjEN7MogF8XlFbPAxMk9jGGW7CoBiZ7TqA-GgbDwDtwmL4-eN45SudMdGt4UyV3KL8kNYbHSi4F3YyfPzrthZKW8mLBii2.-OKC67K1SG25yPy3wzFTVNlGiwrntFR6oY0ZcAMHOJk&dib_tag=se&keywords=Prang&qid=1737143397&sr=8-26), [yellow](https://www.amazon.ca/Prang-Washable-Tempera-16-Ounce-10703/dp/B0017DBCKG/ref=sr_1_17?dib=eyJ2IjoiMSJ9.-kpuh5nDMeAa0FUfIRJlxQj2-iH4vYSO8JIwP5ZtjltzS-I9YBHUZ4SH6-cl-0jMPDqWEVw_3Hc2zhC1ksF0k5S1i7O7PbChfHxWs_TgQbWT7H6A2yIw9wmwcJIXHoMcJBH4BSomMPSt7bTsj1mtK_OV8pcXwEnXjaCxzxb-Ic0tauYFrS9fBQShIsVAr7FrKP1dTS63p1J78OiETFbGJfTR11mJDsXNNMCzw7iB2E8junCroaWdE94fPRS1y8bibHrK_sg5Qg1_JISSyxRRfDgseH8mQmTSjJavP_W3oy1gpzbscVABzBzxHZF6VfaIBXD_SzlLiNXMSbC32iLN0YvOVVNwm9jCqzYjEN7MogF8XlFbPAxMk9jGGW7CoBiZ7TqA-GgbDwDtwmL4-eN45SudMdGt4UyV3KL8kNYbHSi4F3YyfPzrthZKW8mLBii2.FMxZXu8fjhNY-A1yKC26rqBpILPwnxT6XD-Rpri_5HE&dib_tag=se&keywords=Prang&qid=1737143257&sr=8-17) [blue](https://www.amazon.ca/Prang-Washable-32-Ounce-Turquoise-10912/dp/B000QSEKZM/ref=pd_ci_mcx_di_int_sccai_cn_d_sccl_3_11/140-9582717-8730642?pd_rd_w=Dyk9J&content-id=amzn1.sym.d6674fdf-bd00-4d07-8317-6dfd6c498cdf&pf_rd_p=d6674fdf-bd00-4d07-8317-6dfd6c498cdf&pf_rd_r=YX9Z727XJDXQST965Y4C&pd_rd_wg=mBxuX&pd_rd_r=571f6d71-8e80-4a99-8c74-c5841f2d242d&pd_rd_i=B000QSEKZM&psc=1) (cannot find a 16oz blue on amazon, this is a 32oz one)

brendenpelkie, 2025-01-17:
We've had good luck with acrylic paint thinned in water, probably a 10:1 water:paint ratio, if not weaker. Tempura paint is probably equivalent - I think we just have whatever was on the shelf at a craft store. We've generally used opaque colored liquids to mix. You can also use food coloring + milk in a pinch, but then you need to carefully wash everything or it starts to smell.

Neil-YL, 2025-01-18:
> We've had good luck with acrylic paint thinned in water, probably a 10:1 water:paint ratio, if not weaker. Tempura paint is probably equivalent - I think we just have whatever was on the shelf at a [...]

Thank you for sharing your experience! 

Originally, acrylic was the first common paint that I ruled out because I heard it could be used to paint on clothing and the drawings would be waterproof. However, your successful experience and suggestion of the 10:1 water-to-paint ratio is really helpful and gives me more confidence to give it a try.

sgbaird, 2025-01-18:
I went ahead and ordered some acrylic and some washable. Thanks both!

Neil-YL, 2025-01-20:
I am working on switching the SSH to MQTT on OT-2 and I have some questions about how you think on the queuing and the MQTT topic assigned for the device. 

Currently, the queue for students to submit their parameters is managed on Hugging Face. Would you prefer that I manage this queue through MQTT instead? If so, I believe this functionality should be integrated into the OT-2 MQTT setup.

Another thing is the sensor now only subscribes to one same topic (I am using your "absurd-gazelle" course_id for the topic) but it is getting different student_id (which is session_id) submitted by the students and a randomised experiment_id for each run . Is this the intended way to repeatedly use the same MQTT topic or I should assign different course_ids to different students and let the sensor subscribe to all the course_ids topic? I would lean towards reusing same topic and tagged the in/out data with different student/session_id.

sgbaird, 2025-01-20:
> Currently, the queue for students to submit their parameters is managed on Hugging Face. Would you prefer that I manage this queue through MQTT instead? If so, I believe this functionality should be integrated into the OT-2 MQTT setup.

The queue can be on the app.py side.

This is how I've implemented it before: https://huggingface.co/spaces/AccelerationConsortium/light-mixing/blob/916acb2bcee4d81d81020c3ee5f46808d2572a77/app.py#L78. I think we can just stick with this for now, and we'll worry about having a per-student budget/restriction a bit later. I need to give this some more thought.

Aside: @gursi26 has implemented a concurrency limit set to 1 on gradio, but this might have been more about temporary access (i.e., here's your five minute access pass):
https://huggingface.co/spaces/AccelerationConsortium/cobot280pi-gradio/blob/39dd3bf0b5dca8e0aebe286594fd8e65d20df40c/app.py#L359

Neil-YL, 2025-01-20:
> I think we can just stick with this for now, and we'll worry about having a per-student budget/restriction a bit later. I need to give this some more thought.

Currently I have a MongoDB to log-in the data of well-plate status (empty/used) so I think I can also mark down the student_id for the budget consumption purpose, but we may need students sign up for their ID first.

Update for the colors:

The Crayola washable paint is a little bit too viscous to use in our current set up (given the size of pipette tip and well volume), resulting in very visible layering of colors after mixing. I turned to acrylic paint and it mixed better. 
[image]
The results are quite good with the OT-2 light on.:

[image]

C4 appears slightly bluer than C5, and the results for CH440 and CH470 show the most noticeable differences between the two measurements.

Thank you again for @brendenpelkie sharing your successful experience.

Basic colors I have now: 
[image]

sgbaird, 2025-01-20:
Nice! Aside: Can the crayola colors be diluted?

Neil-YL, 2025-01-20:
> Nice! Aside: Can the crayola colors be diluted?

I diluted it quite a bit (at least 1:15-20) but it still has three layering colors and very sticky bubble after the blow-out process of the tip. 

[image]

sgbaird, 2025-01-20:
Acrylic it is, then! :D Thanks for checking.

sgbaird, 2025-01-21:
@Neil-YL thoughts on printing the color sensor tool in black PLA?

Neil-YL, 2025-01-21:
> [@Neil-YL](https://github.com/Neil-YL) thoughts on printing the color sensor tool in black PLA and then trying the onboard LED again? [#87](https://github.com/AccelerationConsortium/ac-training-lab/issues/87)

I have tried printing the sensor support in black or blocking the green light on the sensor to have several measurement (on food coloring RYB and white paper), not improving the data quality (regarding making the readings more distinguishable for different colors )if I remember correctly. (not sure if I've added these to the [previous issue](https://github.com/AccelerationConsortium/ac-training-lab/issues/87).

I have not tried a black enclosure, but I think it is very likely the LED light would overpower signal on all eight channels. I can still print one to have a try anyway

sgbaird, 2025-01-21:
Ah, got it. Thanks for checking that, and don't worry about it then. If we're able to resolve the differences in the current setup with reasonable accuracy, then great 👍. For future reference, we can adjust the gain to prevent hitting the upper limits of the sensor, but like you said, it might overpower it in the sense that the differences are hard to resolve.

> Currently I have a MongoDB to log-in the data of well-plate status (empty/used) so I think I can also mark down the student_id for the budget consumption purpose, but we may need students sign up for their ID first.

I forgot you already had the MongoDB integration. Ignore what I said then, sounds good on a student_id for budget consumption. Subtracting from that number on the database as credits are "consumed" sounds good 👍 I think it would be ok with we generated IDs on our side and passed them out one by one to students, but I'm ok with either way.

Neil-YL, 2025-01-21:
Some updates:

- I finished the MQTT setup on OT-2. Currently I run an ipynb file on OT-2 notebook page and sent a RYB and well command via MQTT, the OT-2 will conduct the mixing and call the color sensor to read the color. 

To do:
- I still need to debug a bit of using the notebook to trigger the command it seems the calibration is not loaded and have some issue on picking up the first tip.
- changing the topic of the sensor side with the pico_id and maybe some tidy-up things like the formate of the payload between devices (I think I would finish this tonight)
- Update app.py with the new triggering of OT-2 
- Notebook page for the practical 

Sterling could we have a in-person meeting Thursday afternoon to discuss all of these above? 

- Bad news: I am not sure if it is because the color is left overnight but the Acrylic paint water also has the color layering things after mix on the well, like the other paint, may look into this tomorrow. I will test with some fresh mixed paint tomorrow.

sgbaird, 2025-01-21:
Sounds good! I'll be in tomorrow if you have time. Could you send an invite? 

Stirring sounds good, and we might want to do `touch_tip` as part of the workflow anyhow (maybe for both inside the well and inside the vials). We can also have caps that we can lightly set on top of the vials using a fake tip mechanism to avoid evaporation.

Which broker is the MQTT happening through?

Neil-YL, 2025-01-21:
> Sounds good! I'll be in tomorrow if you have time. Could you send an invite?
I sent one via teams (tomorrow 4:00-4:30pm) I think I would at least finish the hugging face app (and btw could you add me to the AC team member on hugging face tomorrow, my username is the same as Github, Neil-YL )

> Stirring sounds good, and we might want to do `touch_tip` as part of the workflow anyhow (maybe for both inside the well and inside the vials). We can also have caps that we can lightly set on top of the vials using a fake tip mechanism to avoid evaporation.

or maybe just moving around within the vial

> Which broker is the MQTT happening through?
I am using the same credentials for the Light public demo on hugging face.

sgbaird, 2025-01-22:
Current implementation is blue. Switching to green. Eventually go to green dotted line with low level commands

[image]

Neil-YL, 2025-01-23:
I changed the behavior of OT-2 and AS7341 to green one and deployed the HF space on my HF account, the HF app still has many things to  improve and I might sum up here later. (some behaviors are different when I tested locally and credentials protect setup)

If you want you can try it now I have removed all the wet stuffs so you are safe to have a dry run and get a set of 8-ch data back. 

https://huggingface.co/spaces/Neil-YL/OT-2-LCM

sgbaird, 2025-01-23:
Nice! Just tried, but maybe I messed up. Could you transfer ownership to AccelerationConsortium so I can see logs? I can give you admin access if you need

sgbaird, 2025-01-23:
Also, maybe we adjust so that it just has to be less than 300?

Neil-YL, 2025-01-23:
- currently the output feedback will load last run once you click submit so it looks a bit messy, I am working on it now to separate users and global queue.

300 things are one thing I would love to improve: 

- if we restrict it as 300, students should not be able to press submit(I have tried locally)
- less than 300, I should also adjust code run on the OT-2 but I think it is fine?
- Some more restrictions by the device (like I think for P300 minimum dispense would be 10 or 20uL?)

> Sensor: Data received.
{'sensor_data': {'ch583': 116, 'ch670': 56, 'ch510': 89, 'ch410': 0, 'ch620': 113, 'ch470': 20, 'ch550': 114, 'ch440': 17}, 'session_id': 'new test', 'command': {'Y': 100, 'B': 100, 'well': 'A10', 'R': 100}, 'experiment_id': 'dc94fa51'}
Sending sensor to charging position.
OT-2: Sensor returned to charging position.

Last log I have, I think you are good? We are safe it is dry run

I think I will have many trials-and-errors tonight so if you don't mind I think I could move it to AC's account tomorrow?

sgbaird, 2025-01-23:
Ok, nice! Could you record a video? (No worries if it's tomorrow). Also, could you give me admin access to your HF repo? (So I can see logs)

Switching tomorrow is np

sgbaird, 2025-01-23:
Bare minimum is to display an error and return before doing any MQTT communication. I think there's some gradio built in component for this related to error display.

We can probably get it to dynamically display / show an error in real-time as the sliders get adjusted, but it might be more effort than it's worth to try to get it to **stop** the user from selecting an invalid option and clicking submit. I think someone may have run into a similar issue with making a submit button unclickable in certain scenarios.

I'd need to see the OT-2 code (maybe open a PR with your latest implementation?) but I think a "less than 300" check should be ok.

We could either just put a note saying what the minimum resolution of the pipette is, or we could display a warning or stop the user. Technically we could also probably reduce the allowable selections so it's zero or 5+, etc.

Anyway, it would help to see both the HF code and the OT-2 code. Exciting to see this!

Neil-YL, 2025-01-23:
> I forgot you already had the MongoDB integration. Ignore what I said then, sounds good on a student_id for budget consumption. Subtracting from that number on the database as credits are "consumed" [...]

I did not implement this part yet. Having many thoughts on giving students credits to try whenever they like or everyone have an one hour play time.

Some concerns for using budget credits:
Says if a student started his practical in the morning, have the first two or three tries and leave. Then want to continue days after, in the evening: 
- The base colors are not precisely identical for a long period of time. Let alone the evaporation, every time I hand-mix the color  it could not be the same. (in future we may need to use other liquid-base color with precise volume to volume measurement or premix liquid color)
- Lighting, similar to the case of Bambu practice. Day time with the room light on/off and night time with the room light off, although not sure how the difference would be if the OT-2 built-in light is on.
- Not affecting the experiment performance, but if multiple students are working at a same time for a full optimazation it would be a painful experience.

sgbaird, 2025-01-23:
All valid points to be addressed. I still prefer the budget and queue approach. I think this enhances the real-world aspect of it, as long as we do our best to provide the right metadata and reasonably address some of the concerns with our procedures.

In advance, maybe we mix a large quantity of solution in a large container, keep it usually sealed, and then refill from there as needed? Then, we wouldn't need to have as high accuracy or mix each time.

Let's keep ot-2 light on always. We can add background or blank capture as part of workflow and data we send back. This is what I do for the light-mixing demo, but it's not an urgent priority. Also related to AS7341 LED discussion #87 

The equipment usage sharing is also real-world, but not part of the learning outcome. Instead we might just ask people to organize amongst themselves for when they plan to use it in bulk and not restrict it explicitly. I think we can put this one on Jae to have the students work it out. Google sheets signup sheet or a ping on a slack channel (similar to shared usage of Bambu X1E). On our side, we'll work on getting a large number of iterations without intervention, which could either be rinsing and reusing wells or adding more well plates. Further down the line, we could use plate stacking #143. If needed, eventually we could move the workflow to the Flex (e.g., if OT-2 is getting used significantly for other projects).

---

Thoughts? (We can also chat tomorrow)

Neil-YL, 2025-01-24:
OT2MQTT code: https://github.com/AccelerationConsortium/ac-training-lab/pull/170/files

Neil-YL, 2025-01-31:
Notebook:https://colab.research.google.com/drive/1PwVGB7eguByIx5Rx0rEn0hMs1x076rdP?usp=sharing 

Hugging face space: https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM 

YouTube Stream:  https://www.youtube.com/watch?v=J06AwzMeOEc&ab_channel=ACHardwareStreams

Neil-YL, 2025-02-07:
> Let's keep ot-2 light on always. We can add background or blank capture as part of workflow and data we send back. This is what I do for the light-mixing demo, but it's not an urgent priority. Also [...]

I have been testing with the OT-2 light on and the room light off, but the lighting conditions for each well vary significantly, causing fluctuations in the results. I printed a black enclosure and turned on the LED on AS7341. However, this made some channels less distinguishable of some similar colors. I think either incorporating background/location calibration data or applying weighted calculations to certain channels could help improve the experiment’s performance. Additionally, I am saving the well location for each measurement to Mongodb, so we can explore how to integrate this data later.

Neil-YL, 2025-02-12:
After discussion with Runze, yesterday we test some white color calibration, because he lean towards the data without LED on AS7341. I have measured a white color to have a full-reflection value for each wells location and use it as the denominator for other color measure value. The relative standard deviation (RSD) can be reduced from 6-7% to 1.2-2.3%.

[image]

[image]

Neil-YL, 2025-02-18:
@sgbaird 
Teams meeting with Jae this afternoon. He has proposed a new idea to use OT-2 with similar set up for the MO, with the data for the infrared (I believe it would be Near Infra-red light, NIR in AS7341 smux setting).

 He posted some written-ups on the Slack group. I will try to have some measurement first.

sgbaird, 2025-02-19:
@Neil-YL thanks for the update on the color calibration. As long as students have enough information to act on, I'm ok with it. We just need to keep track of it. With the light mixing demos, I typically take a background with all the LEDs off. I suppose the equivalent in this case is to have a step at the beginning of the workflow that captures the spectrum of the empty well before dispensing liquid into the well (i.e., as if you put 0,0,0 for the RYB liquid amounts).

For multi-objective with IR, can you post some of the details here? I'd still like to prioritize getting the Bambu lab setup ready by the 23rd.

Neil-YL, 2025-02-19:
I will keep you informed if I have any progress on the NIR (on a new issue post). 

For the background measurement, I think I can simply call the function for measurement before the color mixing and put two results together (save it temporarily on HF and wait for a normal color measurement, then combine two measurements and return as one output into current output)

Sissi will try to downgrade to 1.02 tomorrow.

sgbaird, 2025-02-26:
Moving the discussion specific to the light sensor measurements to https://github.com/AccelerationConsortium/ac-training-lab/issues/87#issuecomment-2685941026.

Neil-YL, 2025-03-04:
A crack on the fake-tip was observed after the practical:
https://github.com/AccelerationConsortium/ac-training-lab/issues/64#issuecomment-2698571324

sgbaird, 2025-04-11:
@Neil-YL thanks for refreshing the expired certificate on the Pico W!

sgbaird, 2025-04-22:
Hugging Face Space stuck in "building". This has happened before I think. I updated to the latest gradio version via README.md but it's still stuck. I've tried reset and factory rebuild. It may be an issue on HF side, as it seemed to be the last time I ran into this and based on various forum posts. Hopefully goes away on its own.

[image]

Neil-YL, 2025-04-22:
> Hugging Face Space stuck in "building". This has happened before I think. I updated to the latest gradio version via README.md but it's still stuck. I've tried reset and factory rebuild. It may be [...]

Seems good now.