AccelerationConsortium/ac-dev-lab #156: Test out a few parameter combinations with A1 mini and grid of squares/logos etc.

sgbaird, 2025-01-20:
@Neil-YL could you try printing a grid pattern of squares, each square ~10 mm x 10 mm, with 5 mm spacing between each square? I think that will get us ~60 samples per plate.

184 mm *184 mm / (10+2.5 * 10+2.5) = 56

**Let's switch to the 0.2 mm nozzle for the A1 mini**

What are the default parameters for 0.2 mm nozzle and PLA on the A1 mini?

Let's try printing three plates:
1. recommended nozzle temperature
2. 10C more than recommended nozzle temperature
3. 20C less than recommended nozzle temperature

Neil-YL, 2025-01-20:
The recommended temperature would be between 190 °C and 240 °C. I think the default setting would be 220.
For the printed square, how about the thickness? Should it be a single layer test or multi-layers?

sgbaird, 2025-01-20:
Nice! Single layer I think. For thickness.. up to you, just lmk what you decide.

Neil-YL, 2025-01-20:
A "standard" layer thickness in Bambu App is 0.2mm so maybe let's use this thickness?

sgbaird, 2025-01-20:
Sounds good!

Neil-YL, 2025-01-20:
For the nozzle temperature, should I go aggressively 220/250/170 °C or 220/230/200 °C

sgbaird, 2025-01-20:
I think still the 220/230/200 and we'll evaluate from there.

sgbaird, 2025-01-20:
You're welcome to try the other ones, too, though maybe don't go over 240 C unless you think it's safe.

Neil-YL, 2025-01-20:
First print: 121 10mmx10mm square. Nozzle Temperature: 230 °C 

[image]

https://github.com/user-attachments/assets/7b189b68-526b-4ffb-a7a2-8296e7114a52

I will use white filament for next print(200  °C ).

Neil-YL, 2025-01-21:
Nozzle temperature: 200  °C 

[image]

sgbaird, 2025-01-21:
This is promising! It feels a bit odd to say, but this is great that it looks poor 😅 it's interesting that not every print exhibits the same level of poor quality. However, it's very nice that the prints didn't seem to interfere with each other.

The edges / corners certainly seem worse. I'll give this some more thought.

Neil-YL, 2025-01-21:
And this is the default nozzle temp: 220  °C 

[image]

sgbaird, 2025-01-23:
We should track and expose a few variables:
- number of days since opening the filament
- distance from the center for a given iteration

We could also track the humidity of the room, but I think that's more likely to stay constant over the course of weeks.

Maybe we should restrict the printing to avoid some of the edges where prints are more likely to fail than in the middle.

We will also likely want to have a step where we extrude filament into a trash for a set period of time, and allow things to reach a reasonable steady state (nozzle temperature and bed temperature) prior to doing the iteration.

sgbaird, 2025-02-25:
Taking a bunch of content from https://github.com/AccelerationConsortium/ac-training-lab/issues/168 and porting here.

https://github.com/AccelerationConsortium/ac-training-lab/issues/168#issuecomment-2648961688

> Current progress and choke points：
> 
> https://utoronto-my.sharepoint.com/:w:/g/personal/sissi_feng_utoronto_ca/Efo8kI05SrRIsMQshBLWC8sBrom8XhHEND6Th2kAihnGfQ

----

> * to prevent stringing ruining later iterations, constrain the movement so the extruder is never passing over unused slots (if it strings onto old squares that already had the picture taken, not a big deal)
> * I'm not as worried about 10 min per iteration, better than not being implemented. Let's stick with single GCode file per square probably.
> * please implement a purge / reset step at the end of each iteration. Choose some standard set of parameters (maybe the default recommended ones), go to a position off the build plate, extrude for some amount of time (e.g., 1-2 min) onto the desk
> * at the _beginning_ of each step, do the same: move to waste extrusion position, change to the new settings, let everything get into a steady state for some amount of time (e.g., 5 minutes), extrude for a minute or two, then carry out the iteration. Again, make sure that unused squares aren't "trespassed" until it's time to actually use them
> * typically, movement should be far up in the z-direction before moving in XY in-between iterations. XY movement into other squares with low clearance between build plate and extruder is more likely to cause issues. I.e., lift the extruder up all the way before moving it in XY when switching slots, etc.
> * material build-up on nozzle. Thanks for the picture - that helped me understand. Maybe requires a removal stand. Something really simple. See video below:
> 
>  PXL_20250212_042129384.3.mp4 
> Otherwise, we can try to position the livestream camera so that we can see if there is material build up on the extruder nozzle. If there are certain conditions that are more likely to produce that build up, we can try to narrow what's allowed (e.g., not letting nozzle get too hot)
> 
> Keep the other parameters as possibilities for people to adjust (fan speed, print speed, etc.). Something to be left to the students. They get to figure out what's important or not.

----

> > Nozzle buildup needs to be heated to 220-240 for easy removal, so the stand needs to consider that temperature.
> > Another scenario to consider: Let's say the current parameter settings are 210/70 (the recommended default optimal parameter is 220/65). Instead of spitting out the normal amount and sticking to the bed, the filament is carried by the nozzle head with the spit out portion on the plate to complete the whole movement, and finally this time the task camera has no square to capture. This part of the filament stuck to the nozzle head is not sure if it will be completely removed in the reset step at the end.
> 
> Could you draw/include a picture to help me understand? Would having a small "trash" area to the side of each square, where the reset procedure begins, help with this? If we can get a reasonable reset procedure going, maybe this issue can be tested and kept in mind. Is this something that you think would happen frequently?

----

(For video) https://github.com/AccelerationConsortium/ac-training-lab/issues/168#issuecomment-2659763406

> 1215_1739550328.mp4 
> (The video uses a continuous printout for illustrative purposes, not the single square printout that will be used for the experiment.)
> 
> 1. Reset and “Trash” Wipe at the Start of Each Print
>    At the beginning of every print task, there is already a reset procedure where any leftover filament from the previous job is wiped off onto the lower edge of the build plate (as seen around the 1:35 mark in the video). A similar wiping action occurs again around the 3:15 mark in the video, removing any excess filament from the nozzle head.
> 2. Non-Optimal Parameter Settings Can Cause Poor Adhesion
>    We experimented with parameters of 210°C nozzle temperature and 70°C bed temperature (instead of the recommended 220°C/65°C). In the video at about 4:24, you can see a situation where the filament does not properly adhere to the bed. As a result, part—or sometimes all—of the printed square remains stuck to the nozzle head rather than the build plate. If the entire piece is lifted and carried away by the nozzle, then by the time the print finishes, there is no square left on the build plate for the camera to capture.
> 
> How frequency: We have conducted multiple print trials using the 210°C/70°C settings. The print quality has been inconsistent: sometimes the square prints normally. sometimes a corner is missing. sometimes the entire piece is gone. We have not yet recorded exact statistics on how frequently each issue occurs.
> 
> 3. Temperature Drift
>    Another observation is temperature fluctuation. At around 3:28 in the video, the set temperature is 220°C, but the actual nozzle temperature drifts among 222°C, 221°C, 220°C, and 219°C. This slight fluctuation could also play a role in inconsistent print adhesion.

----

> I think it's ok if the piece gets lifted off as long as it doesn't interfere with later prints. This can essentially be considered a failure condition (I.e., in a real print, if that happens it would probably be unrecoverable). We need to make sure the algorithm that computes metrics can handle this. Maybe we need some assessment of color of the filament. But we can worry about that a bit later. It's ok if there's some inconsistency, as long as we've done our best to make sure a proper reset has taken place. Position on the plate will affect quality, but as long as we tell people the next position available on the plate / which position had been printed on, and how long it's been since the filament has been opened, I think we're ok.

sgbaird, 2025-02-25:
Ideally, I'd like to start exploring more parameters combinations and getting a sense of potential failure conditions of the system. Much of the functionality is tested, but this is waiting on #160 to be able to start testing the full workflow I believe.

sgbaird, 2025-02-28:
Ok, I think I'm seeing the nuance now, this clarification helps a lot. This is reminiscent of something I think Jae talked about a while ago. I understand the class-wide student collaboration / batch optimization as a potential assignment setup; however, the priority is still to get the workflow set up in the same way as with the OT-2 demo and start collecting real data (likely just via Sobol sampling as implemented in Ax, plus repeats) as soon as is feasible.

SissiFeng, 2025-02-28:
Ok then, I'll spend all my time today debugging pi 4b. 
Jae's TA is coming in next Monday to ask about data collection, and also ideally we'd like to have a baseline for the trained model as their evaluation metrics, in order to discuss it with Jae next Tuesday.

sgbaird, 2025-02-28:
Ok, thank you! I'll ping Jae about it.

SissiFeng, 2025-03-06:
[image]

Neil-YL, 2025-03-10:
> Ok, thanks! Let's leave the choice of running repeats to the users (i.e., HF should still be a single parameter + print), but we can adjust the notebook to use three repeats.

I fixed this last Friday. I believe it was caused by our customized G-code at the beginning of each print.

sgbaird, 2025-03-12:
I think fan setting was also adjusted, so it's not on full blast all the time when parameter is 100%

Neil-YL, 2025-03-13:
> [@Neil-YL](https://github.com/Neil-YL) removed the dynamic flow calibration step from the GCode, so it's not printing over itself repeatedly. I think the file is still local though (not on git).

I updated the gcode template file in AC directory on Pi4

sgbaird, 2025-03-14:
@Neil-YL could you provide an update on the current status of the G-code workflow?

sgbaird, 2025-03-14:
The partial bed leveling has default points that are maybe ~15-20 mm away from the contour of the square, and there are 6 touch points in this case.

https://github.com/user-attachments/assets/1f9d1326-7123-4535-8b67-80c3901403cb

Neil-YL, 2025-03-14:
To post on the Bambu community:

Hello everyone! We are developing a self-driving lab teaching demo using the A1-Mini. As part of this, I would like to print a series of squares (10mm by 10mm each, 9 columns and 9 rows, as shown in the figure below) on a single plate. Ideally, each square would have its own G-code file, allowing us to modify different printing parameters (e.g., nozzle temperature, bed temperature) for each square. The G-code files would be sent to the printer sequentially, without removing the previously printed squares, until all available slots are used.

[image]

For each square, I would like to retain the local bed leveling setup by Bambu Studio to study how other parameters affect print quality. However, from my understanding, the calibration points are preset as an 18×18 mesh. To avoid the current print job’s bed leveling process from touching previously printed squares, I would like to know: the exact locations of these calibration points, or if it is possible to manually control or customize the bed leveling calibration points with the G29 command.

One example of the G29 command in our current G-code is: `G29 A1 X30.7939 Y124.851 I10 J10`, defined by:  `G29 A1 X{first_layer_print_min[0]} Y{first_layer_print_min[1]} I{first_layer_print_size[0]} J{first_layer_print_size[1]}`

Any insights or documentation on this would be greatly appreciated.

Thanks, everyone!

sgbaird, 2025-03-17:
Looks like we haven't heard back on the forum: https://forum.bambulab.com/t/user-specified-partial-bed-leveling-calibration-points/152631. @Neil-YL could you provide an update by the end of the day on the status of the g-code workflow?

sgbaird, 2025-03-18:
Ok, sounds good!

sgbaird, 2025-03-19:
@Neil-YL did you already upload these files to GitHub? If so, which branch / PR?

Neil-YL, 2025-03-19:
> [@Neil-YL](https://github.com/Neil-YL) did you already upload these files to GitHub? If so, which branch / PR?
> 
> EDIT: I just merged `a1-mini` into `a1-mini-yh` and then merged a PR for `a1-mini-yh`. Are these the right files now? https://github.com/AccelerationConsortium/ac-training-lab/tree/main/src/ac_training_lab/bambu_a1_mini/gcode
> 
> (lmk if I somehow removed your latest edits)

They are update to date

Neil-YL, 2025-03-19:
We finished the data collecting for the students today.
50 sets of parameters(seeding parameters generated by Runze)

An example photo:
[image]

sgbaird, 2025-03-19:
Nice! Yeah, I guess I couldn't get the autonomous workflow implemented in time, though I'm making steady progress still.

Neil-YL, 2025-03-20:
> [@Neil-YL](https://github.com/Neil-YL) did you already upload these files to GitHub? If so, which branch / PR?
>
> EDIT: I just merged `a1-mini` into `a1-mini-yh` and then merged a PR for `a1-mini-yh`. [...]

I believe I should modify the G-code processing section in the MQTT.py. Which branch should I be working in to avoid conflicts?

sgbaird, 2025-03-20:
@Neil-YL you can use `a1-mini` or a new branch. The file should be called `device.py` now. Also, we might switch to OS Lite (bookworm) from the desktop version even for the A1 mini RPi 4B. Open to thoughts.

sgbaird, 2025-03-21:
https://github.com/mchrisgm/bambulabs_api/issues/113?notification_referrer_id=NT_kwDOArXQBbQxNTQyNDg4NTc0MDo0NTQ2OTcwMQ#issuecomment-2744338470

A few functions to implement:

`park_toolhead()` (M17 R requires above workaround)
`move_x`
`move_y`
`move_z` (I think this is already implemented in the API)
`move_xy`
`move_xyz`

Each encoding the logic specific to our experiments:

If moving xy, +y should always come 2nd, -y should always come first.

If moving xyz, raise z should always be first, lower z should always be last.

Each function should have lower and upper bounds imposed specific to the A1 mini, with safety in mind.

Thoughts?

Neil-YL, 2025-03-24:
> [mchrisgm/bambulabs_api#113](https://github.com/mchrisgm/bambulabs_api/issues/113)
> 
> A few functions to implement:
> 
> `park_toolhead()` (M17 R requires above workaround) `move_x` `move_y` `move_z` (I think this is already implemented in the API) `move_xy` `move_xyz`
> 
> Each encoding the logic specific to our experiments:
> 
> If moving xy, +y should always come 2nd, -y should always come first.
> 
> If moving xyz, raise z should always be first, lower z should always be last.
> 
> Each function should have lower and upper bounds imposed specific to the A1 mini, with safety in mind.
> 
> Thoughts?

If the moving command is used only after the print finishes/idle situation, I think we will need "M17" not "M17 R" ?
since M18 is executed in the ending code for Bambu A1mini

```
;=====printer finish  sound=========
M400 S1
M18 X Y Z
M73 P100 R0
; EXECUTABLE_BLOCK_END
```

And we need to decide whether to use M91 or M90.

sgbaird, 2025-03-24:
I pulled `M17 R` (reset motor currents I think) from one of the recent gcode files. Not familiar with M91 / M90 though. Good point about M18 being in ending code for gcode.

Neil-YL, 2025-03-26:
stp file and bambu studio file of the square

[square 10x10.zip](https://github.com/user-attachments/files/19472168/square.10x10.zip)