AccelerationConsortium/ac-dev-lab #157: Dehumidifier for powder dispensing

sgbaird, 2025-01-20:
I.e., try to avoid clumping and other factors that humidity affects when dispensing powder. Any suggestions on devices? We could probably just use something without any programmatic control (i.e., set it and forget it), and use a humidity sensor to keep track of humidity.

sgbaird, 2025-03-01:
@evelynnesher pinging you since this relates to powder dispensing (and possibly RS232) - more of an FYI

evelynnesher, 2025-03-01:
We can place a silica gel canister with rechargeable desiccants in the powder area to prevent clumping caused by humidity. A humidity sensor would monitor the environment and alert us (possibly via MQTT) if levels get too high, so we can replace or recharge the canister monthly or as needed. Does that sound like what you had in mind, or were you thinking of something more like an actual dehumidifier?

sgbaird, 2025-03-01:
Great point! I can imagine a custom 3D printed hopper lid that holds some rechargeable desiccants or similar and having them scattered at the corners of the chamber (not touching the platen of course). I like the idea of a humidity sensor with an alert, too. This would be a great first implementation. Once @Luthiraa wraps up https://github.com/AccelerationConsortium/ac-training-lab/pull/127, we could put the sensor somewhere nearby (maybe just thrown in with the desiccants in the hopper via a long Stemma-QT cable 😅)

Could you find a link or two for a silica gel canister / rechargeable desiccants? My first instinct would be amazon.ca.

Later, I'd still want to have an actual dehumidifier, but this is down the road (probably 4-6+ months). Could you look briefly and see if there are any reasonable options with an RS232 port or similar?

evelynnesher, 2025-03-01:
Sure! I’ll look into both desiccant/dehumidifier options and different ways to build the sensors. I’ll provide different options and try to come up with a comprehensive plan.

sgbaird, 2025-03-01:
Thanks! Don't worry about the sensor bit too much. I bought many types of sensors already, and we'd either use the BME680 (see https://github.com/AccelerationConsortium/ac-training-lab/issues/13, preferably after Luthira has completed this), another similar-sized board, or a [Shelly H&T Gen3](https://www.aartech.ca/shelly-ht-gen3-wh) (I have one of these at home and really like it - can be used with one's own MQTT server, too [[1](https://support.ilevia.com/shelly/how-to-set-up-the-shelly-ht-inside-the-home-automation-configuration-software-eve-manager-pro/)][[2](https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Mqtt/)]). If the BME680, we'd use a long stemma-QT cable and have it inside the hopper. If the Shelly H&T Gen3, it could probably fit within the chamber (or be mounted to the side with the adhesive mount it comes with).

evelynnesher, 2025-03-06:
Here are a few desiccant options to consider:

The first four links are desiccants that change color when they need to be dried or replaced, which can be useful for monitoring humidity levels visually. However, we should still rely on a sensor for accurate tracking.

https://www.amazon.ca/Desiccant-Packets-Humidity-Indicator-Moisture/dp/B01MPYB16J/ref=sr_1_2_sspa?crid=1X95D9QHQBF6T&dib=eyJ2IjoiMSJ9.o_burpNltloLtCOcMdE6kZDEHO4Q06g3BWaYlWzCxlzVzOJGen_qrCCUr3Fc9j6w9qAuPg6kdwIBTltnBdO1wwNQBY0niK0TQHDdbr93kwVvI8VoiBOxf5I9XNYZ4QV8DNC60Z5TEjZcrO7BKf57-S8PHnnHI-Cd_MqJlXZPKec9PAhhcYtsV6QGXZp048Y_U821WtL8wN_-H9X24kspUQLYBGLO6N_bEFtSd58hYCGNbAnGqCfr5lgtKqU3BIZyTrioZyHHXs1WPoe2oNYzIAGIdEH_PSUxvdvuoT-WfzU.ovD4TSW_cwLmDjFeTz-Uz_1OM8DfEqEK6_b4X9bB_zA&dib_tag=se&keywords=rechargeable%2Bdesiccant&qid=1741222709&sprefix=rechargeable%2Bdesiccan%2Caps%2C131&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

These bulk options could work well. We can 3D print custom shapes to fit these desiccants for both large and small spaces. This would allow us to tailor them per project.
https://www.amazon.ca/Fonday-Moisture-Indicating-Desiccant-Industry/dp/B0BXJNJ73M/ref=sr_1_5?crid=1X95D9QHQBF6T&dib=eyJ2IjoiMSJ9.o_burpNltloLtCOcMdE6kZDEHO4Q06g3BWaYlWzCxlzVzOJGen_qrCCUr3Fc9j6w9qAuPg6kdwIBTltnBdO1wwNQBY0niK0TQHDdbr93kwVvI8VoiBOxf5I9XNYZ4QV8DNC60Z5TEjZcrO7BKf57-S8PHnnHI-Cd_MqJlXZPKec9PAhhcYtsV6QGXZp048Y_U821WtL8wN_-H9X24kspUQLYBGLO6N_bEFtSd58hYCGNbAnGqCfr5lgtKqU3BIZyTrioZyHHXs1WPoe2oNYzIJ9S64pY2EabO1c5tPAb_cNCfRY2xkHLhN4urnZoIQ_1OVzMgdyjrhj4qFuaV-5LHeS3huIV9HF-jizBWDMCZOkmvm5fFVqkCBX-Bk1BUn4_7Wle8saN26ZoylWIbN72LrLY-KSAtHw2zJUn1n_ZjZStfI22w9Kl9E5KKp2bfO9F.u903ZvSVz7-V_bTgt66kSyRNhftbjaSFFZBAatzQvus&dib_tag=se&keywords=rechargeable%2Bdesiccant&qid=1741222709&sprefix=rechargeable%2Bdesiccan%2Caps%2C131&sr=8-5&th=1

https://www.amazon.ca/Wisesorb-Indicating-Reusable-Desiccant-Dehumidifier/dp/B09QFJQQ7Q/ref=sxin_15_pa_sp_search_thematic_sspa?content-id=amzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24%3Aamzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24&crid=1X95D9QHQBF6T&cv_ct_cx=rechargeable%2Bdesiccant&keywords=rechargeable%2Bdesiccant&pd_rd_i=B09QFJQQ7Q&pd_rd_r=0ed6f3a4-443b-4747-a300-ddca66fe3001&pd_rd_w=pQ9H7&pd_rd_wg=PMOGN&pf_rd_p=46621be6-fabe-4126-8501-d32c96c42a24&pf_rd_r=B5DA74MER0YXTRZ2CQNZ&qid=1741222709&sbo=RZvfv%2F%2FHxDF%2BO5021pAnSA%3D%3D&sprefix=rechargeable%2Bdesiccan%2Caps%2C131&sr=1-2-acb80629-ce74-4cc5-9423-11e8801573fb-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9zZWFyY2hfdGhlbWF0aWM&th=1

https://www.amazon.ca/Reusable-Indicating-Rechargeable-Desiccant-Drawstring/dp/B07QRRHJV3/ref=sr_1_25?crid=1X95D9QHQBF6T&dib=eyJ2IjoiMSJ9.o_burpNltloLtCOcMdE6kZDEHO4Q06g3BWaYlWzCxlzVzOJGen_qrCCUr3Fc9j6w9qAuPg6kdwIBTltnBdO1wwNQBY0niK0TQHDdbr93kwVvI8VoiBOxf5I9XNYZ4QV8DNC60Z5TEjZcrO7BKf57-S8PHnnHI-Cd_MqJlXZPKec9PAhhcYtsV6QGXZp048Y_U821WtL8wN_-H9X24kspUQLYBGLO6N_bEFtSd58hYCGNbAnGqCfr5lgtKqU3BIZyTrioZyHHXs1WPoe2oNYzIJ9S64pY2EabO1c5tPAb_cNCfRY2xkHLhN4urnZoIQ_1OVzMgdyjrhj4qFuaV-5LHeS3huIV9HF-jizBWDMCZOkmvm5fFVqkCBX-Bk1BUn4_7Wle8saN26ZoylWIbN72LrLY-KSAtHw2zJUn1n_ZjZStfI22w9Kl9E5KKp2bfO9F.u903ZvSVz7-V_bTgt66kSyRNhftbjaSFFZBAatzQvus&dib_tag=se&keywords=rechargeable+desiccant&qid=1741222709&sprefix=rechargeable+desiccan%2Caps%2C131&sr=8-25

We also have the option of packaged ones and we can rely solely on the sensor to know when to change them
https://www.amazon.ca/Wisesorb-Desiccant-Indicating-Rechargeable-Electronics/dp/B0BP1QMWR7/ref=sr_1_1_sspa?crid=1X95D9QHQBF6T&dib=eyJ2IjoiMSJ9.o_burpNltloLtCOcMdE6kZDEHO4Q06g3BWaYlWzCxlzVzOJGen_qrCCUr3Fc9j6w9qAuPg6kdwIBTltnBdO1wwNQBY0niK0TQHDdbr93kwVvI8VoiBOxf5I9XNYZ4QV8DNC60Z5TEjZcrO7BKf57-S8PHnnHI-Cd_MqJlXZPKec9PAhhcYtsV6QGXZp048Y_U821WtL8wN_-H9X24kspUQLYBGLO6N_bEFtSd58hYCGNbAnGqCfr5lgtKqU3BIZyTrioZyHHXs1WPoe2oNYzIAGIdEH_PSUxvdvuoT-WfzU.ovD4TSW_cwLmDjFeTz-Uz_1OM8DfEqEK6_b4X9bB_zA&dib_tag=se&keywords=rechargeable%2Bdesiccant&qid=1741222709&sprefix=rechargeable%2Bdesiccan%2Caps%2C131&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

https://www.amazon.ca/Packets-Dry-Premium-Desiccant-Dehumidifiers/dp/B01F4EYU20/ref=sxin_15_pa_sp_search_thematic_sspa?content-id=amzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24%3Aamzn1.sym.46621be6-fabe-4126-8501-d32c96c42a24&crid=1X95D9QHQBF6T&cv_ct_cx=rechargeable+desiccant&keywords=rechargeable+desiccant&pd_rd_i=B01F4EYU20&pd_rd_r=0ed6f3a4-443b-4747-a300-ddca66fe3001&pd_rd_w=pQ9H7&pd_rd_wg=PMOGN&pf_rd_p=46621be6-fabe-4126-8501-d32c96c42a24&pf_rd_r=B5DA74MER0YXTRZ2CQNZ&qid=1741222709&sbo=RZvfv%2F%2FHxDF%2BO5021pAnSA%3D%3D&sprefix=rechargeable+desiccan%2Caps%2C131&sr=1-3-acb80629-ce74-4cc5-9423-11e8801573fb-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9zZWFyY2hfdGhlbWF0aWM&psc=1

Let me know if you prefer the color-changing or plain option, and whether you’d like to go with the packaged or loose/bulk version. I will narrow it down based on your review!

sgbaird, 2025-03-10:
Thanks! Probably the gel beads, maybe just https://www.amazon.ca/Wisesorb-Indicating-Reusable-Desiccant-Dehumidifier/dp/B09QFJQQ7Q/ with a custom autotrickler lid.

And for the inner chamber (not the hopper), one of the packaged ones to stick in the corner. Dimension-wise, maybe https://www.amazon.ca/dp/B01MPYB16J/